ZerBea/hcxdumptool #183: build for android fails

sinihabe, 2021-08-08:
OS: clean and fresh ubuntu 20.04 desktop.
Android SDK installed, ndk-build (22.1.7171670) in path
Repo cloned with --recursive flag
Linux binary compiles just fine.

Building for android fails.

```
user@ubuntu:~/hcxdumptool$ ndk-build
[arm64-v8a] Compile        : hcxdumptool <= hcxdumptool.c
In file included from ./hcxdumptool.c:29:
./include/android-ifaddrs/ifaddrs.c:89:9: warning: unused variable 'l_result' [-Wunused-variable]
    int l_result;
        ^
./include/android-ifaddrs/ifaddrs.c:291:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./hcxdumptool.c:41:10: fatal error: 'openssl/conf.h' file not found
#include <openssl/conf.h>
         ^~~~~~~~~~~~~~~~
```

```
ls -l /usr/include/openssl/conf.h
-rw-r--r-- 1 root root 5601 Apr 27 17:37 /usr/include/openssl/conf.h
```
Do i need android specific header files ?

[image]

JuniorJPDJ, 2021-08-08:
Did you follow this instruction: https://github.com/ZerBea/hcxdumptool/pull/16#commitcomment-30006431
?

JuniorJPDJ, 2021-08-08:
You are probably missing openssl-dev package on your system

sinihabe, 2021-08-08:
> You are probably missing openssl-dev package on your system

libssl-dev is installed, and /usr/include/openssl/conf.h file actually exists in the system. Giving its location to ndk-build is ignored. Building linux binary succeeds.

JuniorJPDJ, 2021-08-11:
> Using this repo: JuniorJPDJ/hcxdumptool-android ndk-build succeeds.

This is very-very-very-very old and it was merged to official repository.

I'll try playing with this yesterday, probably something big changed here and changes need to be done in ndk support.

JuniorJPDJ, 2021-08-11:
Yup, openssl dependency was added in 2019 and mine fork is from 2018:

```
17.09.2019
==========
hcxdumptool: added weak password check
             high experimental, only on WPA2 QOS
             new dependency: lib crypto (openssl)
```

ZerBea, 2021-08-12:
Please notice that OpenSSL 3.0 will be released soon:
https://wiki.openssl.org/index.php/OpenSSL_3.0
If it find its way into Arch Core repository
https://archlinux.org/packages/core/x86_64/openssl/
I'm going to use the new EVP-HMAC API (which is better to handle and faster).
That will make OpenSSL 3.0 mandatory.