ZerBea/hcxdumptool #184: -b, --bssid arg

soxrok2212, 2021-08-11:
Hi! Great work on hcxtools!

I'm wondering if it's possible to get similar functionality to aircrack-ng's `--bssid` arg to specify a specific AP, and without having to create a file and pass it in with `--filterlist_ap`, etc. Would be a lot quiter and less tedious.

Hopefully this feature isn't already here and I'm just oblivious...

Thank you!

ZerBea, 2021-08-11:
How should the BSSID option work:
as display filter?
as capture filter (in that case, useful packets will get lost)?
as attack filter?
as protect filter?
 
By the current  hcxdumptool options you can set, what ever you want, from totally silent to aggressive transmission, from display filter to  to capture filter, from attack (target) filter to protect (own devices) filter. In addition to that, you can reload the filter list while hcxdumptool is running.
Therefore we need at least two option switches: target filter list and kind of filter.

hcxdumptool's filtering is totally different to aircrack-ng (or better explained: the whole engine is totally different to aircrack-ng). On aircrack-ng suite you have different tools:
airodump-ng to capture (receive only)
aireplay-ng to attack (transmit only)

Here is an example: filtered airodump-ng cap file:
```
$ hcxpcapngtool 6.2.0-42-gcb7bf91 reading from 5a2ce49f548c0c44f532859236232f01.cap...

summary capture file
--------------------
file name................................: 5a2ce49f548c0c44f532859236232f01.cap
version (pcap/cap).......................: 2.4 (very basic format without any additional information)
timestamp minimum (GMT)..................: 03.08.2021 19:50:48
timestamp maximum (GMT)..................: 03.08.2021 21:03:05
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11 (105)
endianess (capture system)...............: little endian
packets inside...........................: 17976
BEACON (total)...........................: 1
ACTION (total)...........................: 41
PROBERESPONSE............................: 241
DEAUTHENTICATION (total).................: 12879
AUTHENTICATION (total)...................: 20
AUTHENTICATION (OPEN SYSTEM).............: 20
ASSOCIATIONREQUEST (total)...............: 3
ASSOCIATIONREQUEST (PSK).................: 3
WPA encrypted............................: 353
EAPOL messages (total)...................: 25
EAPOL RSN messages.......................: 25
ESSID (total unique).....................: 1
EAPOLTIME gap (measured maximum usec)....: 9752
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (recommended NC).........: 8
EAPOL M1 messages (total)................: 14
EAPOL M2 messages (total)................: 1
EAPOL M3 messages (total)................: 10
EAPOL pairs (total)......................: 1
EAPOL pairs (best).......................: 1
EAPOL M12E2 (challenge)..................: 1
PMKID (useless)..........................: 14

Warning: out of sequence timestamps!
This dump file contains frames with out of sequence timestamps.
That is a bug of the capturing tool.

Warning: excessive number of deauthentication/disassociation frames detected!
That can cause that an ACCESS POINT change channel, reset EAPOL TIMER,
renew ANONCE and set PMKID to zero.
This could prevent to calculate a valid EAPOL MESSAGE PAIR
or to get a valid PMKID.

Warning: missing frames!
This dump file does not contain undirected proberequest frames.
An undirected proberequest may contain information about the PSK.
It always happens if the capture file was cleaned or
it could happen if filter options are used during capturing.
That makes it hard to recover the PSK.
```

The last two warnings will explain what happened while doing the aircrack-ng way:

Warning: missing frames!
Especially if the CLIENT use MAC randomization, all information of it will get lost if you set a BSSID filter like the aircrack-ng way.

Warning: excessive number of deauthentication/disassociation frames detected!
aireplay-ng and airodump-ng don't talk with each other and the deauthentication went on and on and on, even if an EAPOL 4way handshake was received.

ZerBea, 2021-08-11:
compared to a hcxdumptool/hcxlabtool pcapng file:

```
hcxpcapngtool 6.2.0-42-gcb7bf91 reading from 389797df3a6041522dd7d321faf33c59.pcapng...

summary capture file
--------------------
file name................................: 389797df3a6041522dd7d321faf33c59.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.10.36-2-ARCH
application..............................: hcxlabtool 0.1.0
interface name...........................: wlan0
interface vendor.........................: 74da38
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: a468bc027b0d (incremented on every new client)
MAC CLIENT...............................: fcc233e9f145
REPLAYCOUNT..............................: 64550
ANONCE...................................: 389797df3a6041522dd7d321faf33c59e20246fae16baada9653747eac25e55c
SNONCE...................................: ee8e5600006bcf37d77bc1933736f3f370f9e100daa2bd58b2e975eef1687e48
timestamp minimum (GMT)..................: 19.05.2021 03:29:13
timestamp maximum (GMT)..................: 19.05.2021 09:24:19
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 99338
WIRELESS DISTRIBUTION SYSTEM.............: 92
BEACON (total)...........................: 3211
PROBEREQUEST.............................: 4357
PROBEREQUEST (directed)..................: 18
PROBERESPONSE............................: 1839
AUTHENTICATION (total)...................: 2146
AUTHENTICATION (OPEN SYSTEM).............: 2146
ASSOCIATIONREQUEST (total)...............: 1248
ASSOCIATIONREQUEST (PSK).................: 1155
REASSOCIATIONREQUEST (total).............: 74
REASSOCIATIONREQUEST (PSK)...............: 40
IDENTITIES...............................: 9
EAP (total)..............................: 1118
EAP CODE request.........................: 764
EAP CODE response........................: 24
EAP ID...................................: 409
EAP-SIM..................................: 34
EAP-AKA..................................: 345
EAPOL messages (total)...................: 84728
EAPOL RSN messages.......................: 84666
EAPOL WPA messages.......................: 62
ESSID (total unique).....................: 5574
ESSID changes (detected maximum).........: 1 (information: option --max-essids=2 and --all recommended)
EAPOLTIME gap (measured maximum usec)....: 78780448
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (suggested NC)...........: 14
EAPOL M1 messages (total)................: 26850
EAPOL M2 messages (total)................: 57012
EAPOL M3 messages (total)................: 727
EAPOL M4 messages (total)................: 139
EAPOL pairs (total)......................: 403107
EAPOL pairs (best).......................: 951
EAPOL ROGUE pairs........................: 893
EAPOL M12E2 (challenge)..................: 918
EAPOL M32E2 (authorized).................: 32
EAPOL M34E4 (authorized).................: 1
PMKID (useless)..........................: 205
PMKID (total)............................: 510
PMKID (best).............................: 75
PMKID ROGUE..............................: 53
malformed packets (total)................: 2
ESSID error (malformed packets)..........: 2

session summary
---------------
processed pcapng files................: 1
```

...big difference.

ZerBea, 2021-08-11:
Please also notice:
target of aircrack-ng = AP (mostly hardened) 
target of hcxdumptool/hcxlabtool = CLIENT (mostly weak)

ZerBea, 2021-08-12:
"Would be a lot quieter and less tedious"
No.

From --help:
```
--filtermode=<digit> : user space filter mode for filter list
                                 ...
                                  notice: this filter option will let hcxdumptool protect or attack a target - it is neither a capture nor a display filter
```
What you see in real time display is not what hcxdumptool transmit

ZerBea, 2021-08-15:
I think I found a solution.

By this commit:
https://github.com/ZerBea/hcxdumptool/commit/e9099ce43dcde9d75b228f992817e4789d932d1a
you can either use a filter list or a single MAC

example using a filter list:
```
$ sudo hcxdumptool -i wlp39s0f3u1u5 --filterlist_ap=filterlistap --filtermode=1
initialization of hcxdumptool 6.2.0-25-g8f86712...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u5
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE HARDWARE MAC....: 1cbfce3b5bdf (not used for the attack)
INTERFACE VIRTUAL MAC.....: 1cbfce3b5bdf (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.13.10-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 128
FILTERLIST ACCESS POINT...: 4 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: protect
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 0022f160ec99 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 0022f160ec9a (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 0022f160ec9b (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: e0cb1dc49825
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63530
ANONCE....................: 6d452ea4854a0d5e586fc78acee42b0cd1a83248f725e341e78cec3b42ecb908
SNONCE....................: b78fe1c7a96ed0bce4d0bc0b2cb01a99d07f4d37120583bb3030963c9f6e780b
```

example using a single MAC instead of a filter list:
```
$ sudo hcxdumptool -i wlp39s0f3u1u5 --filterlist_ap=11:22:33:44:55:66 --filtermode=1
initialization of hcxdumptool 6.2.0-25-g8f86712...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u5
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE HARDWARE MAC....: 1cbfce3b5bdf (not used for the attack)
INTERFACE VIRTUAL MAC.....: 1cbfce3b5bdf (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.13.10-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 128
FILTERLIST ACCESS POINT...: 1 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: protect
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 1000aa5c9b49 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 1000aa5c9b4a (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 1000aa5c9b4b (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: e00db913fb69
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64013
ANONCE....................: 456300291ca7689ee9d583cea5983188aa3b10b9f102714f22456379336d0759
SNONCE....................: 3545a38e8ed68f7481e9bd531f65f5f56c2be3abe0d9a0daa426148891e3f531

or

$ sudo hcxdumptool -i wlp39s0f3u1u5 --filterlist_ap=112233445566 --filtermode=1
initialization of hcxdumptool 6.2.0-25-g8f86712...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u5
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE HARDWARE MAC....: 1cbfce3b5bdf (not used for the attack)
INTERFACE VIRTUAL MAC.....: 1cbfce3b5bdf (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.13.10-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 128
FILTERLIST ACCESS POINT...: 1 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: protect
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 205b2ac978eb (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 205b2ac978ec (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 205b2ac978ed (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b025aa8496a1
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62841
ANONCE....................: 061bd252325382f439f440b15542656a57626bbeb6e76821835962ecee6f9cc0
SNONCE....................: e977e41581cad5672638cba863b7ed6b862ab51d9239dda195c40c37ae55d9f4
```

Please notice:
This is a filter within the transmit branch. It is neither a capture filter, nor a display filter. You can't compare this filtering with the method used by aircrack-ng.

soxrok2212, 2021-08-15:
I think this is what I was looking for. Not necessarily a packet filter, but in the sense of "okay, my org has access points with SSID 'X', I found one access point and have the MAC address, let's target that quickly, aka do not need to make a separate file with the MAC. Just pass it in as an arg.

Thank you!

closed 2021-08-15 by soxrok2212

ZerBea, 2021-08-16:
It's a fine improvement and I'm glad that I could help. 

Cheers
Mike