ZerBea/hcxdumptool #185: M3M4ZEROED => valid EAPOL MESSAGE PAIR

scorpius, 2021-08-12:
I execute this:
_sudo hcxdumptool -i wlan0 -o vector1.pcapng --stop_ap_attacks=6000 --resume_ap_attacks=12000 --active_beacon --tot=30 --enable_status=1_

and the real time display shows just a single entry as follows:
_[EAPOL:**M3M4ZEROED** EAPOLTIME:2622 RC:2 KDV:1]_

Then hcxpcapngtool outputs a hash, but is it valid? I usually look for M1M2 or M2M3 to indicate a valid hanshake, and believe **M3M4ZEROED** means that the key won't be found.

ZerBea, 2021-08-12:
If you see this message it means that hcxdumptool received an M3 and a zeroed M4 message.
M1 (PMKID), M2 are still missing and you have to wait until hcxdumptool will receive it.
Don't worry, hcxpcapngtool will not convert a M3M4ZEROED message pair to a hc22000 hash file, because it is useless
I recommend to increase the stay time (-t 15) to give hcxdumtool the chance to receive weak signals or signals from crowded channels.

ZerBea, 2021-08-12:
You can verify that a M3M4ZEROED message pair is not converted by running hcxhashtool:
`$ hcxhashtool -i your_hashfile.22000 --info=stdout`

You should see only challenges or authorized message pairs.
If this is not the case, we have a bug. In that case, please attach the pcapng file and the converted hash file.
Also make sure, you're running the latest git head of all tools.

ZerBea, 2021-08-12:
Plesa also notice that hcxdumptool is more (much more) strict than hcxpcapngtool regarding the EAPOL TIME OUT (time between two EAPOL message pairs:

```
hcxdumptool:
--eapoltimeout=<digit>             : set EAPOL TIMEOUT (microseconds)
                                     default: 20000 usec

vs

hcxpcapngtool:
--eapoltimeout=<digit>             : set EAPOL TIMEOUT (milliseconds)
                                   : default: 5000 ms
```

That means:
If hcxdumptool receive a MESSAGE PAIR outside its default EAPOL time, it will store it to pcapng, but it will not notify you via status, because it try to get a better one within its EAPOLTIMEOUT.

It is necessary to use different EAPOL TIMEOUTS, because hcxpcapngtool should work on Wireshark, tshark, tcpdump, kismet and aircrack-ng dump files, too.

ZerBea, 2021-08-12:
If your target is far away, the signal is weak or the channel is crowed, you can increase hcxdumptools EAPOLTIMEOUT value --eapoltimeout=50000. Unfortunately this will increase the chance to get a non valid message pair, too.

ZerBea, 2021-08-12:
Read more about the EAPOL timers here:
https://community.cisco.com/t5/wireless-mobility-documents/eap-timers-on-wireless-lan-controllers/ta-p/3122477
EAPOL-Key Timeout:
"The available values in 6.0 are between 200 and  5000 milliseconds, while codes prior to 6.0 allow for values between 1  and 5 seconds"

As you can see, both (hcxdumptool and hcxpcapngtool) are inside this range.

scorpius, 2021-08-12:
Thanks for all the valuable info!

hcxhashtool (6.0.5) reports:
**_RC INFO....: replycount checked
RC INFO....: AP-LESS attack / nc not reqired
MP M1M2 E2.: not authorized_**

and the latest reports:
**_RC INFO....: ROGUE attack / NC not required
MP M1M2 E2.: challenge_**

and no mention of M3M4, also the hashline is the same.

So I was using a very old version of the tools, I have now updated to the latest and will start over.

ZerBea, 2021-08-12:
Glad to hear that it works, now.
Beside the default values, hcxdumptool and hcxtools have lots of options to control their behavior to get exactly the best MESSAGE PAIR (the one with matching REPLAYCOUNT and the lowest EAPOLTIME between M1-M2, M2-M3, M3-M4 not zeroed, M1-M4 not zeroed).
In addition to that, hcxdumptool will request missing frames from every target (regardless if it is an ACCESS POINT or a CLIENT) to avoid a packet loss. Additional values are stored in pcapng comment fields to inform hcxpcapng tool how to convert the recorded traffic. The same applies to hcxpcapngtool which use the MESSAGE PAIR field to tell hashcat how to handle the converted hash.

Closed this issue report. Please reopen, if it doesn't work as expected.

closed 2021-08-12 by ZerBea