ZerBea/hcxdumptool #186: Intel AX201 driver errors

iyanmv, 2021-08-30:
Hi,

I think when I bought the new Lenovo Thinkpad X1 Yoga (Gen 6) I was able to use `hcxdumptool` without any issues. Now, after a few months using the laptop, when I tried again, I got many driver errors and it seems like injection is not working as good as before. Of course, in the meantime there have been BIOS updates, kernel updates, firmware updates... so it's hard for me to debug this.

Maybe here someone can help me to debug and try to find the real issue here, and open a bug in the proper project upstream (Lenovo support, linux kernel, iwlwifi...).

This laptop has an [Intel AX201](https://www.intel.com/content/dam/www/public/us/en/documents/product-briefs/wi-fi-6-ax201-module-brief.pdf) chipset. Please let me know what logs or information would be useful to share here and I will try to share as soon as possible. For the moment I copy-paste what I see in `dmesg` error messages when I call `hcxdumptool`.

```
[ 1566.917386] RIP: 0033:0x7f6620e3b59b
[ 1566.917392] Code: ff ff ff 85 c0 79 9b 49 c7 c4 ff ff ff ff 5b 5d 4c 89 e0 41 5c c3 66 0f 1f 84 00 00 00 00 00 f3 0f 1e fa b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d a5 a8 0c 00 f7 d8 64 89 01 48
[ 1566.917395] RSP: 002b:00007ffc67e86bb8 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[ 1566.917401] RAX: ffffffffffffffda RBX: 00000000000000a9 RCX: 00007f6620e3b59b
[ 1566.917404] RDX: 0000564320407da0 RSI: 0000000000008b04 RDI: 0000000000000003
[ 1566.917406] RBP: 0000000000000025 R08: 0000000000000002 R09: 006e6f6d33663032
[ 1566.917408] R10: 000000000000001a R11: 0000000000000246 R12: 0000564320407da0
[ 1566.917410] R13: 0000000000000003 R14: 00005643203f4040 R15: 00007ffc67e87de7
[ 1566.917417] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1566.917420] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1566.917447] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1566.917450] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1566.917456] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1566.917458] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1566.917464] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1566.917466] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1574.321266] iwlwifi 0000:00:14.3: Microcode SW error detected. Restarting 0x0.
[ 1574.322399] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[ 1574.322401] iwlwifi 0000:00:14.3: Status: 0x00000040, count: 6
[ 1574.322404] iwlwifi 0000:00:14.3: Loaded firmware version: 62.49eeb572.0 QuZ-a0-hr-b0-62.ucode
[ 1574.322407] iwlwifi 0000:00:14.3: 0x0000125F | ADVANCED_SYSASSERT
[ 1574.322411] iwlwifi 0000:00:14.3: 0x0080A210 | trm_hw_status0
[ 1574.322413] iwlwifi 0000:00:14.3: 0x00000000 | trm_hw_status1
[ 1574.322415] iwlwifi 0000:00:14.3: 0x004CAEFE | branchlink2
[ 1574.322417] iwlwifi 0000:00:14.3: 0x000015E2 | interruptlink1
[ 1574.322418] iwlwifi 0000:00:14.3: 0x000015E2 | interruptlink2
[ 1574.322420] iwlwifi 0000:00:14.3: 0x00000001 | data1
[ 1574.322422] iwlwifi 0000:00:14.3: 0xDEADBEEF | data2
[ 1574.322424] iwlwifi 0000:00:14.3: 0xDEADBEEF | data3
[ 1574.322426] iwlwifi 0000:00:14.3: 0x00000000 | beacon time
[ 1574.322428] iwlwifi 0000:00:14.3: 0x002095BC | tsf low
[ 1574.322429] iwlwifi 0000:00:14.3: 0x00000000 | tsf hi
[ 1574.322431] iwlwifi 0000:00:14.3: 0x00000000 | time gp1
[ 1574.322433] iwlwifi 0000:00:14.3: 0x0020FC23 | time gp2
[ 1574.322435] iwlwifi 0000:00:14.3: 0x00000001 | uCode revision type
[ 1574.322436] iwlwifi 0000:00:14.3: 0x0000003E | uCode version major
[ 1574.322438] iwlwifi 0000:00:14.3: 0x49EEB572 | uCode version minor
[ 1574.322440] iwlwifi 0000:00:14.3: 0x00000351 | hw version
[ 1574.322442] iwlwifi 0000:00:14.3: 0x00489004 | board version
[ 1574.322444] iwlwifi 0000:00:14.3: 0x801DFC27 | hcmd
[ 1574.322446] iwlwifi 0000:00:14.3: 0x24020000 | isr0
[ 1574.322447] iwlwifi 0000:00:14.3: 0x01000000 | isr1
[ 1574.322449] iwlwifi 0000:00:14.3: 0x00B00002 | isr2
[ 1574.322451] iwlwifi 0000:00:14.3: 0x00C0000D | isr3
[ 1574.322452] iwlwifi 0000:00:14.3: 0x00000000 | isr4
[ 1574.322454] iwlwifi 0000:00:14.3: 0x0106001C | last cmd Id
[ 1574.322456] iwlwifi 0000:00:14.3: 0x0000AD62 | wait_event
[ 1574.322458] iwlwifi 0000:00:14.3: 0x00000080 | l2p_control
[ 1574.322459] iwlwifi 0000:00:14.3: 0x00010034 | l2p_duration
[ 1574.322461] iwlwifi 0000:00:14.3: 0x0000003F | l2p_mhvalid
[ 1574.322463] iwlwifi 0000:00:14.3: 0x00008000 | l2p_addr_match
[ 1574.322465] iwlwifi 0000:00:14.3: 0x0000000B | lmpm_pmg_sel
[ 1574.322467] iwlwifi 0000:00:14.3: 0x00000000 | timestamp
[ 1574.322469] iwlwifi 0000:00:14.3: 0x00D0D020 | flow_handler
[ 1574.322586] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[ 1574.322588] iwlwifi 0000:00:14.3: Status: 0x00000040, count: 7
[ 1574.322590] iwlwifi 0000:00:14.3: 0x20000070 | NMI_INTERRUPT_LMAC_FATAL
[ 1574.322592] iwlwifi 0000:00:14.3: 0x00000000 | umac branchlink1
[ 1574.322594] iwlwifi 0000:00:14.3: 0x80454C8A | umac branchlink2
[ 1574.322595] iwlwifi 0000:00:14.3: 0x80473A88 | umac interruptlink1
[ 1574.322597] iwlwifi 0000:00:14.3: 0x80466C8A | umac interruptlink2
[... 200 lines cut ...]
[ 1580.583058] iwlwifi 0000:00:14.3: Failed sending remove station
[ 1580.583062] iwlwifi 0000:00:14.3: Failed to send binding (action:3): -5
[ 1580.583076] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583085] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583087] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583116] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583118] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583123] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583125] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583130] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583132] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583138] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583140] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583145] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583147] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.583152] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.583154] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1580.959388] iwlwifi 0000:00:14.3: PHY ctxt cmd error. ret=-5
[ 1580.959390] iwlwifi 0000:00:14.3: Failed to add PHY context
[ 1585.691559] device wlp0s20f3mon left promiscuous mode
```

By the way, I am running Arch Linux, and I tried with official pacakges `linux` (5.13.13) and `linux-lts` (5.10.61), as well as compiled linux-mainline (5.14.0). With all I get similar error messages.

I also realized that initialization phase takes too long and it puts one core to work 100% for a few seconds. I don't think that was the case in the past, and using `airmon-ng` to enable monitor mode seems to work better for me now.

ZerBea, 2021-08-30:
As you noticed correctly this a firmware or driver issue.
It is either caused by firmware (I assume this)
"Microcode SW error detected."
"FW error in SYNC CMD REMOVE_STA"

or by driver
"PHY ctxt cmd error."

but not by hcxdumptool
"hcxdumptool Not tainted 5.13.13-arch1-1 #1" 

There is nothing I can do, except to suggest to report this issue on 
https://bugzilla.kernel.org/

There have been several several problems running Intel chipsets in the past.
So I decided to print a warning in README.md Adapter section
https://github.com/ZerBea/hcxdumptool#readme
```
Not recommended WiFi chipsets:
    Intel PRO/Wireless
    Broadcom
    Realtek RTL8811AU, RTL8812AU, RTL 8814AU (due to NETLINK dependency)
```

ZerBea, 2021-08-30:
BTW:
Running airmon-ng is not a good idea due to NETLINK dependency.
airmon-ng is running iw and iw is running NETLINK.
You can see this if you enable iw debugging:
```
$ sudo iw --debug dev wlp0s20f3 set type monitor
```

which will show you that NETLINK is in use.

Here is an example of the debugging output of iw:
```
$ sudo iw --debug dev wlp39s0f3u1u1u2 set type monitor
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 33 <0x21>
    .flags = 5 <REQUEST,ACK>
    .seq = 1630327873
    .port = -1778364009
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 6
    .version = 0
    .unused = 0
  [PAYLOAD] 16 octets
    08 00 03 00 03 00 00 00 08 00 05 00 06 00 00 00 ................
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1630327873
    .port = -1778364009
  [ERRORMSG] 20 octets
    .error = 0 "Erfolg"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 33 <0x21>
    .flags = 5 <REQUEST,ACK>
    .seq = 1630327873
    .port = -1778364009
---------------------------  END NETLINK MESSAGE   ---------------------------
```

versus hcxdumptool monitor mode via ioctl() system calls:
```
$ sudo hcxdumptool -m wlp39s0f3u1u1u2
setting interface wlp39s0f3u1u1u2 to monitor mode
```

hcxdumptool driver test:
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --check_driver
initialization of hcxdumptool 6.2.0-33-g5297097...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```
and hcxdumptool full packet injection test:
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --check_injection
initialization of hcxdumptool 6.2.0-33-g5297097...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14
packet injection is working on 2.4GHz!
injection ratio: 44% (BEACON: 204 PROBERESPONSE: 90)
your injection ratio is average, but there is still room for improvement
antenna ratio: 91% (NETWORK: 34 PROBERESPONSE: 31)
your antenna ratio is huge - say kids what time is it?

terminating...
```

ZerBea, 2021-08-30:
You can run all three commands and check dmesg after each command to figure out what happens.
Do not use airmon-ng or iw before the test.

iyanmv, 2021-08-31:
Thanks for the help! I guess I will open first a bug report in Arch, as recommended in https://bugzilla.kernel.org/, and if they tell me to open a bug directly at the kernel bugzilla, then I will do that.

Just out of curiosity, why is so bad to use NETLINK instead of ioctl calls?

ZerBea, 2021-08-31:
hcxdumptool/hcxlabtool is running high performance attacks. To control the device, fast system calls are mandatory.

Here is a good explanation:
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls?share=1

ZerBea, 2021-08-31:
In addition to that, we have to create and evaluate additional NETLINK headers. That produce overhead and will cost CPU cycles which we don't have on a Raspberry Pi.

ZerBea, 2021-08-31:
You mentioned airmon-ng. That let me assume you are familiar with the aircrack-ng suite.

To figure out if the driver has a problem, running NETLINK to init the device you can do the following test:
Get aircrack-ng source.
Configure aircrack-ng without libnl dependency by passing --disable-libnl to configure
Put interface  wlp0s20f3 to monitor mode.
Then run an injection test  by aireplay-ng -9 wlp0s20f3
https://www.aircrack-ng.org/doku.php?id=injection_test
See aireplay-ng messages and dmesg, what happens.

iyanmv, 2021-08-31:
First of all, thank you very much for keep writing here even though it is clear not a `hcxdumptool` issue. I will open a bug in Arch when I have a bit more of time. But for the moment I did what you just told me, and I copy here the outputs:

dmesg
```
[  469.638086] device wlp0s20f3mon entered promiscuous mode
[  471.490675] iwlwifi 0000:00:14.3: Microcode SW error detected. Restarting 0x0.
[  471.491702] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[  471.491703] iwlwifi 0000:00:14.3: Status: 0x00000040, count: 6
[  471.491704] iwlwifi 0000:00:14.3: Loaded firmware version: 62.49eeb572.0 QuZ-a0-hr-b0-62.ucode
[  471.491705] iwlwifi 0000:00:14.3: 0x00004217 | ADVANCED_SYSASSERT          
[  471.491706] iwlwifi 0000:00:14.3: 0x00A022F0 | trm_hw_status0
[  471.491707] iwlwifi 0000:00:14.3: 0x00000000 | trm_hw_status1
[  471.491707] iwlwifi 0000:00:14.3: 0x004CAEFE | branchlink2
[  471.491708] iwlwifi 0000:00:14.3: 0x000015E2 | interruptlink1
[  471.491709] iwlwifi 0000:00:14.3: 0x000015E2 | interruptlink2
[  471.491709] iwlwifi 0000:00:14.3: 0x044E00B4 | data1
[  471.491710] iwlwifi 0000:00:14.3: 0x00000010 | data2
[  471.491710] iwlwifi 0000:00:14.3: 0xDEADBEEF | data3
[  471.491711] iwlwifi 0000:00:14.3: 0x00000000 | beacon time
[  471.491711] iwlwifi 0000:00:14.3: 0x013E8CC1 | tsf low
[  471.491712] iwlwifi 0000:00:14.3: 0x00000000 | tsf hi
[  471.491712] iwlwifi 0000:00:14.3: 0x00000000 | time gp1
[  471.491713] iwlwifi 0000:00:14.3: 0x013EF3EE | time gp2
[  471.491713] iwlwifi 0000:00:14.3: 0x00000001 | uCode revision type
[  471.491714] iwlwifi 0000:00:14.3: 0x0000003E | uCode version major
[  471.491714] iwlwifi 0000:00:14.3: 0x49EEB572 | uCode version minor
[  471.491715] iwlwifi 0000:00:14.3: 0x00000351 | hw version
[  471.491716] iwlwifi 0000:00:14.3: 0x00489004 | board version
[  471.491716] iwlwifi 0000:00:14.3: 0x0103001C | hcmd
[  471.491717] iwlwifi 0000:00:14.3: 0x00020000 | isr0
[  471.491717] iwlwifi 0000:00:14.3: 0x00000000 | isr1
[  471.491718] iwlwifi 0000:00:14.3: 0x08F00002 | isr2
[  471.491718] iwlwifi 0000:00:14.3: 0x00C20008 | isr3
[  471.491719] iwlwifi 0000:00:14.3: 0x00000000 | isr4
[  471.491719] iwlwifi 0000:00:14.3: 0x0103001C | last cmd Id
[  471.491720] iwlwifi 0000:00:14.3: 0x000146F8 | wait_event
[  471.491720] iwlwifi 0000:00:14.3: 0x00000080 | l2p_control
[  471.491721] iwlwifi 0000:00:14.3: 0x00000020 | l2p_duration
[  471.491721] iwlwifi 0000:00:14.3: 0x0000003F | l2p_mhvalid
[  471.491722] iwlwifi 0000:00:14.3: 0x00008000 | l2p_addr_match
[  471.491722] iwlwifi 0000:00:14.3: 0x00000009 | lmpm_pmg_sel
[  471.491723] iwlwifi 0000:00:14.3: 0x00000000 | timestamp
[  471.491723] iwlwifi 0000:00:14.3: 0x00002064 | flow_handler
[  471.492304] iwlwifi 0000:00:14.3: Start IWL Error Log Dump:
[  471.492305] iwlwifi 0000:00:14.3: Status: 0x00000040, count: 7
[  471.492306] iwlwifi 0000:00:14.3: 0x20000070 | NMI_INTERRUPT_LMAC_FATAL
[  471.492307] iwlwifi 0000:00:14.3: 0x00000000 | umac branchlink1
[  471.492307] iwlwifi 0000:00:14.3: 0x80454C8A | umac branchlink2
[  471.492308] iwlwifi 0000:00:14.3: 0x80473A88 | umac interruptlink1
[  471.492308] iwlwifi 0000:00:14.3: 0x80473A88 | umac interruptlink2
[  471.492309] iwlwifi 0000:00:14.3: 0x00000400 | umac data1
[  471.492310] iwlwifi 0000:00:14.3: 0x80473A88 | umac data2
[  471.492310] iwlwifi 0000:00:14.3: 0x00000000 | umac data3
[  471.492311] iwlwifi 0000:00:14.3: 0x0000003E | umac major
[  471.492311] iwlwifi 0000:00:14.3: 0x49EEB572 | umac minor
[  471.492312] iwlwifi 0000:00:14.3: 0x013EF40B | frame pointer
[  471.492312] iwlwifi 0000:00:14.3: 0xC0886270 | stack pointer
[  471.492313] iwlwifi 0000:00:14.3: 0x00250108 | last host cmd
[  471.492313] iwlwifi 0000:00:14.3: 0x00000000 | isr status reg
[  471.492433] iwlwifi 0000:00:14.3: IML/ROM dump:
[  471.492433] iwlwifi 0000:00:14.3: 0x00000003 | IML/ROM error/state
[  471.492520] iwlwifi 0000:00:14.3: 0x000060F0 | IML/ROM data1
[  471.492609] iwlwifi 0000:00:14.3: 0x00000080 | IML/ROM WFPM_AUTH_KEY_0
[  471.492652] iwlwifi 0000:00:14.3: Fseq Registers:
[  471.492675] iwlwifi 0000:00:14.3: 0x20000000 | FSEQ_ERROR_CODE
[  471.492698] iwlwifi 0000:00:14.3: 0x80290033 | FSEQ_TOP_INIT_VERSION
[  471.492719] iwlwifi 0000:00:14.3: 0x00090006 | FSEQ_CNVIO_INIT_VERSION
[  471.492741] iwlwifi 0000:00:14.3: 0x0000A482 | FSEQ_OTP_VERSION
[  471.492763] iwlwifi 0000:00:14.3: 0x00000003 | FSEQ_TOP_CONTENT_VERSION
[  471.492785] iwlwifi 0000:00:14.3: 0x4552414E | FSEQ_ALIVE_TOKEN
[... 70 lines cut ...]
[  477.920121] iwlwifi 0000:00:14.3: 0x0000A482 | FSEQ_OTP_VERSION
[  477.920147] iwlwifi 0000:00:14.3: 0x00000003 | FSEQ_TOP_CONTENT_VERSION
[  477.920172] iwlwifi 0000:00:14.3: 0x4552414E | FSEQ_ALIVE_TOKEN
[  477.920198] iwlwifi 0000:00:14.3: 0x20000302 | FSEQ_CNVI_ID
[  477.920225] iwlwifi 0000:00:14.3: 0x01300504 | FSEQ_CNVR_ID
[  477.920251] iwlwifi 0000:00:14.3: 0x20000302 | CNVI_AUX_MISC_CHIP
[  477.920279] iwlwifi 0000:00:14.3: 0x01300504 | CNVR_AUX_MISC_CHIP
[  477.920308] iwlwifi 0000:00:14.3: 0x05B0905B | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[  477.920364] iwlwifi 0000:00:14.3: 0x0000025B | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[  477.920520] iwlwifi 0000:00:14.3: WRT: Collecting data: ini trigger 4 fired (delay=0ms).
[  477.920529] ieee80211 phy0: Hardware restart was requested
```
aireplay-ng -9
```
18:35:57  Trying broadcast probe requests...
18:35:59  No Answer...
18:35:59  Found 1 AP 

18:35:59  Trying directed probe requests...
18:35:59  FC:4A:E9:70:A9:32 - channel: 10 - 'Kablonet Netmaster-A92D-G'
18:36:05   0/30:   0%
```
Do you see anything relevant in the dmesg that I should highlight in the bug reports in Arch/kernel?

By the way, I remember doing the same test when I got the laptop and I was getting 100% almost all the times. Also, `aireplay-ng --deauth` was working perfectly and now it doesn't, so I would say that currently injection is broken for AX201.

ZerBea, 2021-08-31:
Thanks for sharing your test results.
As you can see, if aircrack-ng is compiled without libnl, packet injection is not working running AX201 driver :
```
18:35:59  No Answer...
and
18:36:05   0/30:   0%
```
Looks like init of the driver via NETLINK isn't completely done.

I've seen this problem on Realtek drivers, too:
https://github.com/aircrack-ng/rtl8812au/issues/376#issuecomment-526120499

iyanmv, 2021-08-31:
Now I was able to totally crashed the laptop by running `hcxdumptool -i wlp0s20f3 -s 1 --do_rcascan`. The caps lock key led started to blink (I had to figure out what this means from Lenovo docs) and the system was totally unresponsive so I had to force a reboot by keep pressing the power button.

ZerBea, 2021-08-31:
That sounds not good. 
The iwlwifi driver need to be fixed.
There are lots of similar reports:
https://ask.fedoraproject.org/t/intel-r-wi-fi-6-ax201-driver-always-crashing-iwlwifi/10728
https://ask.fedoraproject.org/t/iwlwifi-driver-randomly-crashes-inoperable-until-reboot/12672
https://community.intel.com/t5/Wireless/iwlwifi-driver-randomly-crashes-in-Linux/m-p/643389
https://askubuntu.com/questions/1219893/iwlwifi-driver-crashing

ZerBea, 2021-09-01:
I think, we can close this issue, because it is related to the broken driver.

closed 2021-09-01 by ZerBea

iyanmv, 2021-09-02:
Just for future reference, if someone is reading this closed issue. I open a bug report at the kernel bugzilla. Here is the link:
https://bugzilla.kernel.org/show_bug.cgi?id=214291

ZerBea, 2021-09-03:
Good idea to report this issue on bugzilla.kernel.org.
I suspect, the issue is related to INTEL MICROCODE.

iyanmv, 2021-11-30:
@ZerBea thanks for keeping the discussion upstream, I thought I was not going to hear from Intel at this point.

ZerBea, 2021-12-01:
Can you please comment output of
$ hcxdumptool -i interface -C
to get a frequency list reported from the device as I did here
https://bugzilla.kernel.org/show_bug.cgi?id=214291#c7

ZerBea, 2021-12-02:
Thanks. It looks like ioctl(SIOCSIWFREQ) failed.

To find out which ioctl() is working and which is damaged, please use iw to set monitor mode:
$ sudo ip link set dev INTERFACE down
$ sudo iw dev INTERFACE set type monitor
$ sudo ip link set dev INTERFACE up

Than do a driver test:
$ sudo hcxdumptool -i INTERFACE --check_driver

Followed by an injection test
$ sudo hcxdumptool -i INTERFACE --check_injection

iyanmv, 2021-12-03:
```
[root@thinkpad ~]# ip link set dev wlp0s20f3 down
[root@thinkpad ~]# iw dev wlp0s20f3 set type monitor
[root@thinkpad ~]# ip link set dev wlp0s20f3 up
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 --check_driver
initialization of hcxdumptool 6.2.4...
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 --check_injection
initialization of hcxdumptool 6.2.4...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
channel 1 not available
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 2,3,4,5,6,7,8,9,10,11,12,13,40,44,48,52,56,60,64,112,116,120,124,128,132,136,140,144,149,153,157,161,165
packet injection is working on 2.4GHz!
injection ratio: 94% (BEACON: 138 PROBERESPONSE: 130)
your injection ratio is huge - say kids what time is it?
antenna ratio: 28% (NETWORK: 21 PROBERESPONSE: 6)
your antenna ratio is average, but there is still room for improvement

terminating...
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 -C
initialization of hcxdumptool 6.2.4...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
available channels:

terminating...
```

iyanmv, 2021-12-03:
I rebooted the laptop before doing this, just to be sure that NETLINK is not used.
```
[root@thinkpad ~]# hcxdumptool -m wlp0s20f3
setting interface wlp0s20f3 to monitor mode
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 --check_driver
initialization of hcxdumptool 6.2.4...
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 --check_injection
initialization of hcxdumptool 6.2.4...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,60,64
packet injection is working on 2.4GHz!
injection ratio: 100% (BEACON: 56 PROBERESPONSE: 614)
your injection ratio is huge - say kids what time is it?
antenna ratio: 83% (NETWORK: 12 PROBERESPONSE: 10)
your antenna ratio is excellent, let's ride!

terminating...
[root@thinkpad ~]# hcxdumptool -i wlp0s20f3 -C
initialization of hcxdumptool 6.2.4...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
available channels:

terminating...
```

ZerBea, 2021-12-04:
Thanks for the test.
It looks like the driver need NETLINK to set monitor mode and to bring the interface up - otherwise the interface will not be init completely
available channels by NETLINK init: 2,3,4,5,6,7,8,9,10,11,12,13,40,44,48,52,56,60,64,112,116,120,124,128,132,136,140,144,149,153,157,161,165
versus ioctl() init:
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,60,64 ... than the driver crashed

I pushed an update. Please do a git pull, compile and comment output of 
$ hcxdumptool -i wlp0s20f3 -C

BTW:
Can I send you a mail to the addr mentioned here:
https://github.com/iyanmv

iyanmv, 2021-12-04:
Here it is. Ignore the warning about NetworkManager, I added the wireless device on `/etc/NetworkManager/conf.d/unmanaged.conf`
```
# ./hcxdumptool -i wlp0s20f3 -C
initialization of hcxdumptool 6.2.5-6-g5739c87...
warning possible interfere: NetworkManager is running with pid 20254

wlp0s20f3 available frequencies, channels and tx power reported by driver:
2412MHz   1 (-2147483648 dBm)
2417MHz   2 (-2147483648 dBm)
2422MHz   3 (-2147483648 dBm)
2427MHz   4 (-2147483648 dBm)
2432MHz   5 (-2147483648 dBm)
2437MHz   6 (-2147483648 dBm)
2442MHz   7 (-2147483648 dBm)
2447MHz   8 (-2147483648 dBm)
2452MHz   9 (-2147483648 dBm)
2457MHz  10 (-2147483648 dBm)
2462MHz  11 (-2147483648 dBm)
2467MHz  12 (-2147483648 dBm)
2472MHz  13 (-2147483648 dBm)
5200MHz  40 (-2147483648 dBm)
5220MHz  44 (-2147483648 dBm)
5240MHz  48 (-2147483648 dBm)
5260MHz  52 (-2147483648 dBm)
5280MHz  56 (-2147483648 dBm)
5300MHz  60 (-2147483648 dBm)
5320MHz  64 (-2147483648 dBm)
5600MHz 120 (-2147483648 dBm)
5620MHz 124 (-2147483648 dBm)
5640MHz 128 (-2147483648 dBm)
5660MHz 132 (-2147483648 dBm)
5680MHz 136 (-2147483648 dBm)
5700MHz 140 (-2147483648 dBm)
5720MHz 144 (-2147483648 dBm)
5745MHz 149 (-2147483648 dBm)
5765MHz 153 (-2147483648 dBm)
5785MHz 157 (-2147483648 dBm)
5805MHz 161 (-2147483648 dBm)
5825MHz 165 (-2147483648 dBm)

terminating...
```
> BTW:
> Can I send you a mail to the addr mentioned here:
> https://github.com/iyanmv

Sure!

ZerBea, 2021-12-05:
Thanks.
Many ioctl() system calls give incorrect values or no values.
More and more it looks like the driver need NETLINK to work as expected.

iyanmv, 2021-12-05:
With NETLINK I see that a few more channels appear now (e.g. 36, 100, 104), but some are still missing (e.g. 38, 54). Can this be caused by some regdomain issues?
```
# ip link set wlp0s20f3 down
# iw wlp0s20f3 set type monitor
# ip link set wlp0s20f3 up
# ./hcxdumptool -i wlp0s20f3 -C
initialization of hcxdumptool 6.2.5-6-g5739c87...
warning possible interfere: NetworkManager is running with pid 634

interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
wlp0s20f3 available frequencies, channels and tx power reported by driver:
2412MHz   1 (-2147483648 dBm)
2417MHz   2 (-2147483648 dBm)
2422MHz   3 (-2147483648 dBm)
2427MHz   4 (-2147483648 dBm)
2432MHz   5 (-2147483648 dBm)
2437MHz   6 (-2147483648 dBm)
2442MHz   7 (-2147483648 dBm)
2447MHz   8 (-2147483648 dBm)
2452MHz   9 (-2147483648 dBm)
2457MHz  10 (-2147483648 dBm)
2462MHz  11 (-2147483648 dBm)
2467MHz  12 (-2147483648 dBm)
2472MHz  13 (-2147483648 dBm)
5180MHz  36 (-2147483648 dBm)
5200MHz  40 (-2147483648 dBm)
5220MHz  44 (-2147483648 dBm)
5240MHz  48 (-2147483648 dBm)
5260MHz  52 (-2147483648 dBm)
5280MHz  56 (-2147483648 dBm)
5300MHz  60 (-2147483648 dBm)
5320MHz  64 (-2147483648 dBm)
5500MHz 100 (-2147483648 dBm)
5520MHz 104 (-2147483648 dBm)
5540MHz 108 (-2147483648 dBm)
5560MHz 112 (-2147483648 dBm)
5580MHz 116 (-2147483648 dBm)
5600MHz 120 (-2147483648 dBm)
5620MHz 124 (-2147483648 dBm)
5640MHz 128 (-2147483648 dBm)
5660MHz 132 (-2147483648 dBm)
5680MHz 136 (-2147483648 dBm)
5700MHz 140 (-2147483648 dBm)
5720MHz 144 (-2147483648 dBm)
5745MHz 149 (-2147483648 dBm)
5765MHz 153 (-2147483648 dBm)
5785MHz 157 (-2147483648 dBm)
5805MHz 161 (-2147483648 dBm)
5825MHz 165 (-2147483648 dBm)

terminating...
```

ZerBea, 2021-12-05:
No, I don't think that it is related to the regulatory domain.
It should not happen that the driver reports different channels/frequencies depending on how the monitor mode was set:
less channels if monitor mode set by ioctl()
```
	iwr.u.mode = IW_MODE_MONITOR;
	if(ioctl(fd_socket, SIOCSIWMODE, &iwr) < 0)
```
versus set monitor mode by iw using NETLINK.

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.

iyanmv, 2023-02-26:
```
$ ./hcxlabtool -I wlp0s20f3

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   2 0456e5e5c112 0456e5e5c112 + wlp0s20f3        iwlwifi (NETLINK & WIRELESS EXTENSIONS)

+ monitor mode available
- no monitor mode available

available frequencies: frequency [channel] tx-power
  2412 [  1] 22.0 dBm     2417 [  2] 22.0 dBm     2422 [  3] 22.0 dBm     2427 [  4] 22.0 dBm
  2432 [  5] 22.0 dBm     2437 [  6] 22.0 dBm     2442 [  7] 22.0 dBm     2447 [  8] 22.0 dBm
  2452 [  9] 22.0 dBm     2457 [ 10] 22.0 dBm     2462 [ 11] 22.0 dBm     2467 [ 12] 22.0 dBm
  2472 [ 13] 22.0 dBm     2484 [ 14] disabled     5180 [ 36] 22.0 dBm     5200 [ 40] 22.0 dBm
  5220 [ 44] 22.0 dBm     5240 [ 48] 22.0 dBm     5260 [ 52] 22.0 dBm     5280 [ 56] 22.0 dBm
  5300 [ 60] 22.0 dBm     5320 [ 64] 22.0 dBm     5340 [ 68] disabled     5360 [ 72] disabled
  5380 [ 76] disabled     5400 [ 80] disabled     5420 [ 84] disabled     5440 [ 88] disabled
  5460 [ 92] disabled     5480 [ 96] disabled     5500 [100] 22.0 dBm     5520 [104] 22.0 dBm
  5540 [108] 22.0 dBm     5560 [112] 22.0 dBm     5580 [116] 22.0 dBm     5600 [120] 22.0 dBm
  5620 [124] 22.0 dBm     5640 [128] 22.0 dBm     5660 [132] 22.0 dBm     5680 [136] 22.0 dBm
  5700 [140] 22.0 dBm     5720 [144] 22.0 dBm     5745 [149] 22.0 dBm     5765 [153] 22.0 dBm
  5785 [157] 22.0 dBm     5805 [161] 22.0 dBm     5825 [165] 22.0 dBm     5845 [169] disabled
  5865 [173] disabled     5885 [177] disabled     5905 [181] disabled

bye-bye
```

ZerBea, 2023-02-26:
Great, thanks. Now hcxlabtool should work on this driver.

BTW:
This "5905 [181] disabled" is related to your wireless regulatory domain settings.
https://github.com/aircrack-ng/aircrack-ng/discussions/2430#discussioncomment-5023204

iyanmv, 2023-02-26:
Great! And what about hcxdumptool?

ZerBea, 2023-02-26:
hcxdumptool is a dinosaur (due to WIRLESS EXTENSION dependency) and it is acting like a toothless tiger (too many features that slow it down).
hcxlabtool will be the successor due to full NETLINK/RTNETLINK support. With hcxlabtool I'm going back to the roots and set focus on ultra fast and effective layer 2 attack vectors.

Simply run
$ hcxlabtool
and see what will happen.

iyanmv, 2023-06-30:
Hi @ZerBea

I missed [your last comment](https://bugzilla.kernel.org/show_bug.cgi?id=214291#c34) on the kernel bugzilla but I have just tried version 6.3.0 (which I think it includes the changes to NL80211) and I can still crash my laptop.

What else do you suggest me to try? Should I continue the conversation in the kernel bug now that you have removed WEXT completely?

ZerBea, 2023-07-01:
SInce v6.3.0 WEXT is completely removed and replaced by RTNETLINK and NL80211 (as suggested by the iwlwifi driver maintainer).
According to this
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi
packet injection is not mentioned (only sniffer mode).

or this
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi#about_the_monitorsniffer_mode
"This will put lots of pressure on the memory subsystem, but it will allow you to hear 12K long packets. You may see firmware crashes in case you didn't set that module parameter. "

and this
https://www.intel.com/content/www/us/en/support/articles/000058933/wireless/intel-wireless-ac-products.html

it looks like this chipset/driver is not the best choice to be used as a penetration testing device.

BTW:
There are a lot of problems with this chipset:
https://duckduckgo.com/?q=ax201+crash&t=ffab&ia=web

I think it is the best to do the conversion on bugzilla, because there is nothing I can do.
MediaTek, Ralink, Realtek and Atheros drivers are working fine, so I still suspect an iwlwifi driver issue.

ZerBea, 2023-07-02:
https://github.com/ZerBea/hcxdumptool/commit/4ba9d5bad5dd48e77532d22243d2e57d02a4d6fb

ZerBea, 2023-07-02:
Bugzilla report:
https://bugzilla.kernel.org/show_bug.cgi?id=217051