ZerBea/hcxdumptool #188: STUCK

Gouki-b, 2021-09-14:
After a minute of doing its work it just stops:

hcxdumptool -o ALL.pcapng -i wlan0 --enable_status=1 

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: e84e0603272c
DRIVER....................: rtl8187
DRIVER VERSION............: 5.10.0-kali9-amd64
DRIVER FIRMWARE VERSION...: N/A
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ROGUE (ACCESS POINT)......: 706f81790ceb (BROADCAST HIDDEN)
ROGUE (ACCESS POINT)......: 706f81000cec (BROADCAST OPEN)
ROGUE (ACCESS POINT)......: 706f81790ced (incremented on every new client)
ROGUE (CLIENT)............: b0ece1d47f28
EAPOLTIMEOUT..............: 20000 usec
REPLAYCOUNT...............: 62559
ANONCE....................: 18839914e0bd892b04a1d153d0736500b0e6f2d2a1bb7a6535d735668fabf972
SNONCE....................: 90e9ef09a668eabace592ddee3a69ffc834a7b53102b667e0808ab855a742b2a

19:11:24   3 b24e26c7b931 183d5ea1adc0 vip_409879447 [EAPOL:M1M2 EAPOLTIME:10441 RC:28 KDV:2]
19:11:24   3 b24e26c7b931 183d5ea1adc0 vip_409879447 [EAPOL:M2M3 EAPOLTIME:9056 RC:29 KDV:2]
19:11:24   3 20f4785618aa 10a3b815c7b1 Optima-15C7B0 [EAPOL:M1M2ROGUE EAPOLTIME:15243 RC:62559 KDV:2]
19:11:24   3 20f4785618aa 10a3b815c7b1 Optima-15C7B0 [EAPOL:M1M2 EAPOLTIME:3603 RC:1 KDV:2]
19:11:24   3 20f4785618aa 10a3b815c7b1 Optima-15C7B0 [EAPOL:M2M3 EAPOLTIME:3186 RC:2 KDV:2]
19:11:24   3 20f4785618aa 10a3b815c7b1 Optima-15C7B0 [EAPOL:M3M4ZEROED EAPOLTIME:5498 RC:2 KDV:2]
19:11:27   3 4e24b8e2bc2a b04e26c7b931 admin [EAPOL:M1M2ROGUE EAPOLTIME:2371 RC:62559 KDV:2]
19:11:27   3 4e24b8e2bc2a b04e26c7b931 admin [EAPOL:M1M2 EAPOLTIME:3761 RC:1 KDV:2]
19:11:27   3 4e24b8e2bc2a b04e26c7b931 admin [EAPOL:M2M3 EAPOLTIME:3392 RC:2 KDV:2]
19:11:27   3 4e24b8e2bc2a b04e26c7b931 admin [EAPOL:M3M4ZEROED EAPOLTIME:4830 RC:2 KDV:2]
19:11:27   3 b24e26c7b931 183d5ea1adc0 vip_409879447 [EAPOL:M1M2 EAPOLTIME:3646 RC:31 KDV:2]
19:11:27   3 b24e26c7b931 183d5ea1adc0 vip_409879447 [EAPOL:M2M3 EAPOLTIME:8903 RC:32 KDV:2]
19:11:29   5 b24e26c7b931 183d5ea1adc0 vip_409879447 [EAPOL:M1M2 EAPOLTIME:3334 RC:34 KDV:2]
19:11:30   5 b24e26c7b930 183d5ea1adc0 vip_409879447 [EAPOL:M1M2 EAPOLTIME:15969 RC:20 KDV:2]
19:11:30   5 b24e26c7b930 183d5ea1adc0 vip_409879447 [EAPOL:M2M3 EAPOLTIME:8442 RC:21 KDV:2]
19:11:30   5 b24e26e2bc2a 183d5ea1adc0 vip_409879447 [EAPOL:M2M3 EAPOLTIME:9702 RC:8 KDV:2]
19:11:44   2 1203ef254fc3 70c7f281ad50 A1_718705142 [EAPOL:M3M4ZEROED EAPOLTIME:2295 RC:5 KDV:2]
19:11:45   2 103d0a7df2f1 70c7f281ad50 A1_718705142 [EAPOL:M1M2ROGUE EAPOLTIME:1093 RC:62559 KDV:2]

All interfering processes are killed beforehand.

ZerBea, 2021-09-14:
Are you sure that hcxdumptool stopped?
You selected --enable_status=1
That means if there are no more EAPOL messages (4-way handshakes or PMKIDs) to request there will be no more status output as long as no new APs/CLIENTs ere in range.

That behavior is explained in --help:
```
--enable_status=<digit>            : enable real-time display (waterfall)
                                     only incomming traffic
                                     only once at the first occurrence due to MAC randomization of CLIENTs
```
If you would like to see more status messages choose --enable_status=31

Also make sure that you're running latest version (6.2.4):
```
$ sudo hcxdumptool -o ALL.pcapng -i wlp39s0f3u1u1u2 --enable_status=31
initialization of hcxdumptool 6.2.4-1-ge6c73d2...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u1u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca3124eb (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca3124eb (not used for the attack)
DRIVER....................: rtl8187
DRIVER VERSION............: 5.14.2-arch1-2
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 0023f7da9583 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 0023f7da9584 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 0023f7da9585 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: c8aacc1404cf
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63814
ANONCE....................: 88952c66fc8c0350c8ec213cb741ce5b2489f66715f200200932defb92bf5cb8
SNONCE....................: 84b18517a47f8b8746e95251a3a81fd77050f0737398c2488c2b6e173de09a80
```

after pressing ctrl+c
the latest version will print an information about ERRORs, detected during runtime:

```
^C
terminating...
1 driver error encountered
2 radiotap errors encountered
```

ZerBea, 2021-09-14:
Using --enable_status=95 will give you and additional "alive message" containing additional information:
```
21:42:00   2 ERROR:0 INCOMING:103 AGE:1 OUTGOING:76 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
21:43:00   3 ERROR:0 INCOMING:1059 AGE:5 OUTGOING:640 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
```
Now, you'll see that incoming and outgoing packet counter will increase.

Also I recommend to activate BEACONING.
Examples:
```
print EAPOL messages and alive status:
$ hcxdumptool -o ALL.pcapng -i wlan0  --active_beacon --enable_status=65
print EAPOL, AUTHENTICATION, ASSOCIATION, BEACON, ROGUE AP messages and alive status:
$ hcxdumptool -o ALL.pcapng -i wlan0  --active_beacon --enable_status=95

```

ZerBea, 2021-09-15:
I tested hcxdumptool and the driver without having problems:
```
$ lsusb
Bus 005 Device 006: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter
$ uname -r
5.14.3-arch1-1
```

$ sudo hcxdumptool -o ALL.pcapng -i wlp39s0f3u1u1u2 --enable_status=64
```
initialization of hcxdumptool 6.2.4-1-ge6c73d2...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u1u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca3124eb (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca3124eb (not used for the attack)
DRIVER....................: rtl8187
DRIVER VERSION............: 5.14.3-arch1-1
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 706f8116e81c (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 706f8116e81d (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 706f8116e81e (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: f04f7c7bb42c
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 61623
ANONCE....................: b1e88230aa3e2ef5c1e45591c70beb8e64f5b7b38cc2a568a8b0fdb7b87535c2
SNONCE....................: f1d8c0cce560332ef0c9b1b877aef29df6e6a5e2723a38e6ccfdc1c542240419

08:07:00   7 ERROR:2 INCOMING:314 AGE:1 OUTGOING:47 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:08:00   8 ERROR:2 INCOMING:1891 AGE:1 OUTGOING:321 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:09:00   9 ERROR:2 INCOMING:5516 AGE:1 OUTGOING:451 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:10:00  10 ERROR:2 INCOMING:6780 AGE:1 OUTGOING:538 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:11:00  11 ERROR:2 INCOMING:7821 AGE:5 OUTGOING:614 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:12:00  12 ERROR:2 INCOMING:8858 AGE:5 OUTGOING:653 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:13:00  13 ERROR:2 INCOMING:9842 AGE:9 OUTGOING:669 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:14:00  14 ERROR:2 INCOMING:12140 AGE:9 OUTGOING:741 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:15:00   1 ERROR:2 INCOMING:13245 AGE:17 OUTGOING:764 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:16:00   2 ERROR:2 INCOMING:14379 AGE:1 OUTGOING:888 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:17:00   3 ERROR:2 INCOMING:15276 AGE:3 OUTGOING:997 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:18:00   4 ERROR:2 INCOMING:16802 AGE:7 OUTGOING:1068 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:19:00   5 ERROR:2 INCOMING:17664 AGE:2 OUTGOING:1125 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:20:00   6 ERROR:2 INCOMING:18967 AGE:1 OUTGOING:1316 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:21:00   7 ERROR:2 INCOMING:19899 AGE:1 OUTGOING:1386 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:22:00   8 ERROR:2 INCOMING:21027 AGE:1 OUTGOING:1417 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:23:00   9 ERROR:2 INCOMING:22308 AGE:1 OUTGOING:1424 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:24:00  10 ERROR:2 INCOMING:25849 AGE:1 OUTGOING:1518 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:25:00  11 ERROR:2 INCOMING:27114 AGE:1 OUTGOING:1570 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:26:00  12 ERROR:2 INCOMING:27854 AGE:1 OUTGOING:1596 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:27:00  13 ERROR:2 INCOMING:28312 AGE:1 OUTGOING:1597 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:28:00  14 ERROR:2 INCOMING:29142 AGE:1 OUTGOING:1602 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:29:00   1 ERROR:2 INCOMING:30028 AGE:5 OUTGOING:1698 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:2 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:30:00   2 ERROR:2 INCOMING:30797 AGE:1 OUTGOING:1719 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:3 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
08:31:00   3 ERROR:2 INCOMING:31504 AGE:5 OUTGOING:1826 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:3 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0

^C
terminating...
1 driver error encountered
2 radiotap errors encountered
```
Everything is working as expected. Elapsed time (AGE) between received packets (1..17 seconds)
The driver ERROR is related to pselect() function, because it doesn't like to be interrupted by ctrl+c
The radiotap ERROR is related to the driver during its initialization.
The attack against the test target is working (PMKID and MESSAGE PAIR).

Gouki-b, 2021-09-16:
tnx for the help, ill let you know later if everything works ok

closed 2021-10-04 by ZerBea