ZerBea/hcxdumptool #189: Wrong mac address of wifi points

LowSkillDeveloper, 2021-09-16:
I have known for a long time that sometimes the captured points do not have the correct mac address. The name is correct and the password is also correct, but the mac of wifi station is not correct. It's randomly generated or something?
And sometimes it is not clear whether the real mac is or not, until you manually check it. Maybe you can add some parameter --zero_mac that you write and then such points, instead of an incomprehensible mac that looks like a real one, in hccapx file wrote 00:00:00:00:00:00 or something like that so that it was immediately clear that the mac address is not correct. Or to do it with hcxpcaptool, because this is very confusing.

LowSkillDeveloper, 2021-09-16:
When converting to cap, I use -O, that is raw. Could it be because of this?

ZerBea, 2021-09-16:
Most of state of the art CLIENTs doing MAC randomization to prevent tracking.
hcxdumptool/hcxlabtool doing the same MAC randomization to prevent counter measures against it.
In addition to that, on directed PROBEREQUESTs, hcxdumpttol/hcxlabtool will use the probed MAC from the CLIENT. 
Zeroed MACs are out of the protocol and most of the CLIENTs will not respond to them. So it is not a good idea to use them for an attack and it doesn't make sense to add such an option to hcxdumptto/hcxlabtool.

Please notice that calculating a PMKID or a PKE (from EAPOL) is based on the MAC addresses of AP and CLIENT used during the AUTHENTICATION. We can't change them by the conversion tool because that will lead to uncrackable hashes!
There is no chance to add such an option to the conversion tool.

If you don't want to feed hashcat with EAPOL MESSAGEPAIRs from hcxdumptool ROGUE attacks filter them out and get only authenticated M2M3 MESSAGEPAIRs as described here or use the PMKID: 
https://hashcat.net/forum/thread-10253-post-53580.html#pid53580

BTW:
hcxpcaptool is deprecated and not longer maintained. You should use hcxpcapngtool instead.
hash mode 2500 (hccapx) and 16800 are deprecated, too and replaced by hash mode 22000:
https://hashcat.net/forum/thread-10253-post-53580.html#pid53580

LowSkillDeveloper, 2021-09-16:
I mean, after the AP-LESS Attack, during the records of the information to the HCCAPX file about the AP, write its Mac like 00:00:00:00:00:00 or something like that. Something like a label for the future.
When viewing an APs in a CAP file, immediately know what wifi points with the wrong MAC address.

ZerBea, 2021-09-16:
Partly this information is stored in the pcapng header file and hcxpcapngtool will show you the basic start MAC:
```
$ hcxpcapngtool test.pcapng
hcxpcapngtool 6.2.4-1-g34a4170 reading from test.pcapng...

summary capture file
--------------------
file name................................: test.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.13.13-arch1-1
application..............................: hcxdumptool 6.2.0-40-gab5deb3
interface name...........................: wlp5s0f4u2
interface vendor.........................: 784476
openSSL version..........................: 1.1
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 00269f581765 (incremented on every new client)
MAC CLIENT...............................: e0cb1d72ccd8
```
This basic MAC_AP (00269f581765) is incremented on every new ESSID (00269f581766, 00269f581767, 00269f5817658,...).
If a CLIENT probe via undirected BROADCAST MAC (ffffffffffff), we use this replacement.
If a client use a directed MAC, we use this one instead.

After converting it to a hash line this information get lost, except of the value of the MESSAGE PAIR field bit 4.
If bit 4 is set, we know that the attack was ap-less and the MAC is possible a random one.
You can use hcxhashtool to identify them:
```
$ hcxhashtool -i test.hc22000 --apless -o apless.hc22000

OUI information file..........: /home/zerobeat/.hcxtools/oui.txt
OUI entires...................: 30555
total lines read..............: 97388
valid hash lines..............: 97388
PMKID hash lines..............: 13171
EAPOL hash lines..............: 84217
filter by M2..................: requested from client (AP-LESS)
EAPOL written.................: 36490
```
hashlines containing hashes from AP-LESS attacks will have message pair field *10 and you can assume that the used MAC_AP is mostly a random one.

closed 2021-09-16 by LowSkillDeveloper

ZerBea, 2021-09-16:
The same can be done by bash tools to get all AP-LESS hashes:
`$ cat hash.hc22000 | grep WPA.02 | grep 10$
` 
or to get all authorizde hashes:
`$ cat hash.hc22000 | grep WPA.02 | grep 02$
` 

There is no other way to do this.

ZerBea, 2021-09-16:
Please notice, if we are able to identify a randomized MAC, an AP or a CLIENT are able to do identify our attack, too. In that case they will initiate counter measures against us.

ZerBea, 2021-09-16:
If you don't want that hcxdumptool use randomized MAC_APs, you have to disable AP-LESS attacks.
In that case hcxdumptool will not respond to requests from CLIENTs.

LowSkillDeveloper, 2021-09-16:
> If you don't want that hcxdumptool use randomized MAC_APs, you have to disable AP-LESS attacks.
> In that case hcxdumptool will not respond to requests from CLIENTs.

I think I will do so, this is the simplest solution in my case.