ZerBea/hcxdumptool #196: Network is down

soxrok2212, 2021-11-15:
Hi! Let me begin with this: I'm not sure that this is a hcxdumptool issue, but probably something else with my setup, but hopefully someone here might have some hints.

My setup is as follows:
Host OS: Proxmox 7, `Linux proxmox 5.11.22-5-pve #1 SMP PVE 5.11.22-10 (Tue, 28 Sep 2021 08:15:41 +0200) x86_64 GNU/Linux`
Guest OS: Kali Rolling, `Linux kali 5.14.0-kali2-amd64 #1 SMP Debian 5.14.9-2kali1 (2021-10-04) x86_64 GNU/Linux`

When running hcxdumptool, I get frequent `failed to read packet: Network is down` messages on both USB passthrough devices and PCIe passthrough devices. The USB chips I've tried are AWUS036H (RTL8187), AWUS051NHv1 (RT2770/RT2750), TL-WN722Nv1 (AR9271), and the PCIe chip I've tried uses the RTL8822BE chip. All support injection just fine (exception on 5ghz on the AWUS051NH).

There are no other services interrupting the devices. wpa_supplicant is killed. I've watched dmesg while these errors pop up, but there don't seem to be any driver/device errors.

```
[154838.733778] device wlan0 entered promiscuous mode # start hcxdumptool
[165978.289700] device wlan0 left promiscuous mode # stop hcxdumptool
```

Here's a redacted example of what I see, behavior is the same across all 4 devices (USB and PCIe):
```
17:07:52 2447/8  xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxx [AUTHENTICATION]
17:08:04 2462/11  ffffffffffff xxxxxxxxxxxx xxxxxxxxxx [BEACON]
17:08:04 2462/11  ffffffffffff xxxxxxxxxxxx [HIDDEN BEACON]
17:08:36 5280/56  xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxxxx [PROBERESPONSE]

failed to read packet: Network is down
17:09:59 2412/1   xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxxx [PROBERESPONSE]
17:10:52 5200/40  xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxxxxx [PMKIDROGUE:abcdef0123456789abcdef0123456789 KDV:2]

failed to read packet: Network is down
17:19:53 5785/157 xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxxxxxxxxxx [AUTHENTICATION]
17:19:53 5785/157 xxxxxxxxxxxx xxxxxxxxxxxx xxxxxxxxxxxxxxxx [REASSOCIATION]
```

Other USB devices run fine. I have an RTLSDR on 24/7 running an ADS-B Receiver, so I don't think it's the actual USB bus.

If there's nothing that comes to mind here, I'll try shooting over to Proxmox forums. Just figured I'd start here since I don't see the same issue with other USB devices.

ZerBea, 2021-11-15:
First let's fix the simple thing: packet injection on 5GHz 
hcxdumptool respect the wireless regulatory domain settings.
To allow e,g, channel 13 or 14 or packet injection on 5GHz, regulatory domain must not be unset, which is the default on several distributions:
```
$ iw reg get
global
country 98: DFS-UNSET
	(2402 - 2472 @ 40), (N/A, 20), (N/A)
	(2457 - 2482 @ 25), (N/A, 20), (N/A), PASSIVE-SCAN
	(2474 - 2483 @ 9), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
	(5170 - 5250 @ 80), (N/A, 20), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW, PASSIVE-SCAN
	(5490 - 5725 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
	(5725 - 5730 @ 5), (N/A, 13), (0 ms), DFS, PASSIVE-SCAN
	(5735 - 5835 @ 80), (N/A, 13), (N/A), PASSIVE-SCAN
	(57240 - 63720 @ 2160), (N/A, 0), (N/A)
```
Just try regulatory setting which allow transmitting on 5GHz:
```
$ sudo iw reg set IN
$ iw reg get
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (N/A)
	(5470 - 5725 @ 160), (N/A, 24), (N/A)
	(5725 - 5875 @ 80), (N/A, 30), (N/A)
```
It is mandatory to disconnect all WiFi devices before setting the regulatory domain. Otherwise settings will be ignored.

Now the big issue: "Network is down"
This error message located in the main packet loop here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5827
and printed here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5842
hcxdumptool use perror() tho show the kind of the problem reported by the the OS
```
if(packetlen < 0)
	{
	perror("\nfailed to read packet");
	errorcount++;
	return;
	}
```
 after reading a packet:
`packetlen = recvfrom(fd_socket, epb +EPB_SIZE, PCAPNG_MAXSNAPLEN, 0, NULL, NULL);
`
https://linux.die.net/man/2/recvfrom
```

These call return the number of bytes received, or -1 if an error occurred.
```
In that case (interface is down)  we retrieve the kind of the error reported by the OS (ENETDOWN):
Here are some explainations:
https://www.man7.org/linux/man-pages/man3/errno.3.html
```
ENETDOWN
              Network is down (POSIX.1-2001).
```

https://man.openbsd.org/recv.2
`[ENETDOWN]     A socket operation encountered a dead network.`

https://www.ibm.com/docs/en/zvse/6.2?topic=SSB27H_6.2.0/fa2ti_errno_values.html
`ENETDOWN 1117 | The local interface to use or reach the destination is down.`

For me, it looks like a kind of power save function (or overheating protection - running hcxdumptool many devices get very hot), which disable the interface, if it assume it isn't in use.

BTW;
During init process, hcxdumptool try to disable power save, here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7195

But it looks like your OS enabled it again.

Unfortunately, there is nothing I can do (except of printing a warning). The received packet is lost.

ZerBea, 2021-11-15:
BTW:
Make sure to get access to USB hardware directly from the guest:
https://pve.proxmox.com/wiki/USB_Devices_in_Virtual_Machines

ZerBea, 2021-11-15:
BTW2:
hcxdumptool is in coexistence with tshark and Wireshark. You can run both of them in combination with hcxdumptool.
First start hcxdumptool than start tshark or Wireshark. Now stay tuned until hcxdumptool receive ENETDOWN from the OS and check what tshark/Wireshark report.
You can limit output to BEACONs and PROBEREQUESTs:
`$ tshark -i your_interface "wlan type mgt and (subtype beacon or subtype probe-req)"`

ZerBea, 2021-11-15:
Some injection test results on USB and PCIe.

ALFA AWUS036ACM:
```
$ lsusb
Bus 003 Device 003: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
$ sudo hcxdumptool -i wlp5s0f4u2 --check_injection
initialization of hcxdumptool 6.2.4-29-g4f944a2...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136,140,144,149,153,157,161,165,169,173
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 72% (BEACON: 99 PROBERESPONSE: 72)
your injection ratio is good
antenna ratio: 35% (NETWORK: 34 PROBERESPONSE: 12)
your antenna ratio is average, but there is still room for improvement

terminating...
```

build in PCIe WiFi card:
```
$ lspci
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
$ sudo hcxdumptool -i wlp4s0 --check_injection
initialization of hcxdumptool 6.2.4-29-g4f944a2...
starting antenna test and packet injection test (that can take up to two minutes)...
available channels: 1,2,3,4,5,6,7,8,9,10,11,12,13,14,36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136,140,144,149,153,157,161,165
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 100% (BEACON: 251 PROBERESPONSE: 420)
your injection ratio is huge - say kids what time is it?
antenna ratio: 51% (NETWORK: 58 PROBERESPONSE: 30)
your antenna ratio is good

terminating...
```

soxrok2212, 2021-11-15:
I initially thought it was power save too, but I did check to be sure it's not (or shouldn't be).
```
# cat /sys/bus/usb/devices/1-6/power/control
on
```
kernel.org says `on` means it will not go into power save mode: https://www.kernel.org/doc/html/v4.14/driver-api/usb/power-management.html#the-user-interface-for-dynamic-pm

I also checked in the qemu conf:
```
$ cat /sys/bus/usb/devices/2-1/power/control                                                                 1 ⨯
on
```

The USB device is passed through as suggested, USB 3.0 is not enabled since it is a USB 2.0 only device (AWUS036H):
```
usb0: host=0bda:8187
```

PCIe card is also passed through as recommended:
```
hostpci0: 0000:02:00.0
```

I checked the AWUS036H (which I have quite literally burnt 4 or 5 of them) felt warm to the touch, but not "hot", so I don't think it's a thermal issue. I can't check the PCIe chip though.

I'll probably have to take this up with Proxmox/qemu. Appreciate the help!

Side note: what exactly is the antenna ratio shown with `--check_injection`?

ZerBea, 2021-11-15:
I noticed overheating only in one case (Atheros device):
https://bugzilla.kernel.org/show_bug.cgi?id=61111

I have no idea what happens between the HOST and the GUEST. The GUEST OS report that recvfrom() failed (ENETDOWN), because it network adapter failed to get the packet.
None of my machines (Raspberry Pi, Notebooks (32bit and 64bit), Desktops) showing this issue on kernel 5.10.78-2 (RPI) and 5.14.16

I haven't updated to kernel 5.15.2, because of a radiotap header issue (unfixed):
https://lore.kernel.org/all/20211109100203.c61007433ed6.I1dade57aba7de9c4f48d68249adbae62636fd98c@changeid/
that caused that the RSSI is always 0.

BTW:
Have you read this:
https://forum.proxmox.com/threads/connection-lost-when-high-traffic.65261/
could be related to WiFI devices, too (hcxdumptool produce high traffic).

injection ratio is calculated based on how often a single AP respond:
injectionratio = (injectionhit *100) /injectioncount;

antenna ratio is calculated how often all APs together respond:
networkratio = (networkhit *100) /networkcount;

Let us say, you have an AP nearby and 2 APs far away.
Your injection ratio will be good, because the AP nearby will respond on every request - packet injection is working.
Antenna ratio may not be good, if the other APs do not respond.

If all 3 APs respond - packet injection is working and the antenna ratio is good, because you reach all of them.

ZerBea, 2021-11-18:
After some research I come more and more to the conclusion that the issue is related to the communication between HOST and GUEST.
I don't think that the control tool (SDR) run an AF_PACKET socket(?) and check lost packets like hcxdumptool does.

BTW:
THis is a good explanation of ENETDOWN:
```
Network is down.
A socket operation encountered a dead network. This could indicate a serious failure of the network system (i.e. the protocol stack that the socket runs over), the network interface, or the local network itself. 
```
Do you got an answer on Proxmox forum?

ZerBea, 2021-11-18:
But anyway, I added this to --help because full access to the WiFi hardware is mandatory as well as a working and stable AF_PACKET socket:
```
do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
do not run hcxdumtool on virtual machines or emulators
do not run hcxdumptool in combination with tools (channel hopper), that take access to the interface (except: tshark, wireshark, tcpdump)
do not use tools like machcanger, because hcxdumptool run its own MAC space and will ignore this changes
stop all this services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
```

soxrok2212, 2021-11-22:
Hi! I haven't had a chance to ask on the Proxmox forums, but I'm confident it's their issue and not one in hcxdumptool. I haven't been able to replicate it in other VMs (ESXi, VMware, or Virtualbox) so if it persists, I'll ask there.

Again, I appreciate your time and effort in helping with this!

closed 2021-11-22 by soxrok2212