ZerBea/hcxdumptool #197: Passive Mode for Monitor Mode-only Devices

s00se, 2021-12-20:
Is it possible to include an override to only allow passive mode collection for devices that do not have packet injection but do support RFMON, like many Android devices running qcacld-3.0

s00se, 2021-12-20:
After compiling my kernel (based on LineageOS 18.1) with WEXT and CRDA enabled, airmon-ng simply crashes my device and a hard reboot occurs, whereas before without the extensions it actually put the chipset into monitor mode and was usable with airodump-ng and other tools.

Thanks for clarifying.

ZerBea, 2021-12-20:
That is very interesting. airmon-ng use iw to set monitor mode and iw use NETLINK.
It looks like the wireless extensions (WEXT) inc combination with qcacld-3.0 are not completely functional and caused the kernel to crash.

What happens (kernel compiled without WEXT) if you set monitor mode as mentioned here:
https://github.com/kimocoder/qualcomm_android_monitor_mode
Than run hcxdumptool with options --force_interface --silent --enable_status=95
Now hcxdumptool will show you if ioctl() system calls are supported or not.

ZerBea, 2021-12-20:
The  output of iw will give us more information. Please try:
$ iw dev
It should look like this:
```
phy#0
	Interface wlp39s0f3u1u6
		ifindex 3
		wdev 0x1
		addr 74:da:38:f2:03:8e
		type managed
		txpower 0.00 dBm
```
Now set monitor by iw
$ sudo iw dev wlp39s0f3u1u6 set type monitor
and run
$ iw dev
again
output should be
```
phy#0
	Interface wlp39s0f3u1u6
		ifindex 3
		wdev 0x1
		addr 74:da:38:f2:03:8e
		type monitor
		txpower 0.00 dBm
```

Next step is to restart the device and set monitor mode as mentioned here:
https://github.com/kimocoder/qualcomm_android_monitor_mode
Than run iw dev to make sure monitor mode (type monitor) is really set on the physical device.

strasharo, 2021-12-20:
@s00se what device are you using?

s00se, 2021-12-20:
@strasharo Currently testing this with a Pixel 5 on persona build of LineageOS 18.1

ZerBea, 2021-12-21:
Presence of Wireless Extensions is mandatory.
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/cc2b1d8f7dbbcbad36f0caf9e1bcb3696ac12704
hcxdumptool detect presence of Wireless extensions like "Wireless Tools" do:
https://github.com/HewlettPackard/wireless-tools/blob/master/wireless_tools/iwlib.c#L673
If no wireless name : no wireless extensions
In that case hcxdumptool will not work as expected.

s00se, 2021-12-29:
I have removed identifying information:

`iw dev`:
```
phy#0
        Interface wifi-aware0
                ifindex 33
                wdev 0xe
                addr
                type NAN
        Interface p2p0
                ifindex 32
                wdev 0xd
                addr
                type managed
                txpower 0.00 dBm
        Interface wlan1
                ifindex 31
                wdev 0xc
                addr
                type managed
        Interface wlan0
                ifindex 30
                wdev 0xb
                addr
                ssid
                type managed
                txpower 37.00 dBm
```
After issuing `airmon-ng start wlan1` with **WEXT disabled in kernel**:
```
Your kernel supports rfkill but you don't have rfkill installed.
To ensure devices are unblocked you must install rfkill.

Found 1 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

  PID Name
27758 wpa_supplicant

PHY     Interface       Driver          Chipset

null    ??????          non-mac80211 device? (report this!)
null    p2p0            ??????          non-mac80211 device? (report this!)
null    wifi-aware0     ??????          non-mac80211 device? (report this!)
null    wlan0           ??????          non-mac80211 device? (report this!)
phy0    wlan1           icnss           Not pci, usb, or sdio

Failed to set wlan1 up using ip
                (monitor mode enabled)

Failed to set wlan1 up using ip

Error setting channel: command failed: No such device (-19)
```
The interface is actually put in RFMON, as I can see packets just fine using `airodump-ng wlan0` and `iw dev` does indeed show the device is out of managed mode, and now in monitor mode:
```
phy#0
        Interface wlan0
                ifindex 34
                wdev 0xf
                addr
                type monitor
```

s00se, 2021-12-29:
Do note, I am calling `airmon-ng` on **wlan1**, but it is actually **wlan0** that I need to use as an interface for `airodump-ng` to see packets. I don't know if there is a virtual interface created by Android subsystem or what.

ZerBea, 2022-01-02:
There is no mac80211 stack hcxdumptool can work on:
null    wifi-aware0     ??????          non-mac80211 device? (report this!)

Please also notice that airodump-ng and iw are running NETLINK and hcxdumptoll not.
To test if airodump-ng is still working if NETLINK is disabled just comple it with --disable-libnl and see what happens.
"Linux: LibNetlink 1 or 3. It can be disabled by passing --disable-libnl to configure."

ZerBea, 2022-01-02:
To clarify it:
iw use NETLINK
airodump-ng use NETLINK (if not compiled with --disable-libnl)
hcxdumptool use WEXT.