ZerBea/hcxdumptool #199: Channel or frequency switch not working on RTL8812AU

Czechball, 2021-12-24:
Hello.
When trying to use the latest version of hcxdumptool (https://github.com/ZerBea/hcxdumptool/commit/dc9e9721f4a21fd26ec002e4a5ae9ede8db7f7af) with the Alfa AWUS036ACH using RTL8812AU (latest version from https://github.com/aircrack-ng/rtl8812au - https://github.com/aircrack-ng/rtl8812au/commit/3a6402e9e79802891f1531b435be54f4d8b71f0b), it's not possible to set a channel or a frequency. When running hcxdumptool with the switch `-c` followed by either a channel number or a frequency, it will always result in a crash with the error message `no frequencies available`.
The channel switch was working fine until this update: `hcxdumptool: replaced channel scan engine by frequency scan engine`.
The systems I have tested this on were a Raspberry Pi 3B+ running Raspberry Pi OS `5.10.63-v7+` and Manjaro `5.10.84-1-MANJARO` on x86_64. I have also verified that the channel/frequency switch works on my other interfaces (MT7610U).

Are there any possible solutions to this other than using an older version of hcxdumptool?

Czechball, 2021-12-24:
To clarify: The latest version that supported channel switching on RTL8812AU was 6.2.4

ZerBea, 2021-12-24:
By latest commit, I added some additional information.
If channel setting failed, hcxdumptool will print the reported message from the driver.
E.g. if we try to set a channel (17) that isn't supported by the driver:
```
$ sudo hcxdumptool -i wlp5s0f3u2 -c 17
initialization of hcxdumptool 6.2.5-19-gdc9e972...
frequency/channel 17 not accepted by driver
no frequencies available

terminating...
1 driver error encountered
```
Please try and comment the report.

ZerBea, 2021-12-24:
Unfortunately rtl8812au and rtl8814au drivers are difficult to handle due to their NETLINK dependency.

ZerBea, 2021-12-25:
According to struct wireless.h
https://github.com/torvalds/linux/blob/master/include/uapi/linux/wireless.h#L910
the frequency or the channel is calculated :
struct iw_freq	freq;		/* frequency or channel :
				 * 0-1000 = channel
				 * > 1000 = frequency in Hz */

On hcxdumptool init we do a check to get this values in this function:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7387
Please add one line of code after this line
ptrfscanlist->frequency = pwrq.u.freq.m;
here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7419
`printf("debug freq/ch: %d, exponent: %d\n", pwrq.u.freq.m, pwrq.u.freq.e);`
compile and run hcxdumptool twice
time on the mt76 device
second time on the rtl8812au device

and comment the debug values for both drivers.
I don't have a rtl8812au device, but the mt76 debug should
look like this:
```
$ sudo ./hcxdumptool -i wlp5s0f3u3 -c 1
initialization of hcxdumptool 6.2.5-20-g08842f2...

debug freq/ch: 2412, exponent: 6
```
Explanation:
The reported frequency is 2412
It is greater than 1000, so we have to expect a frequeney.
The exponent is 6 and we have to expect a frequency of 2412000000 Hertz = 2412 MHz

At the moment, the exponent is hard coded to 6 because it will work on all
tested drivers - but unfortunately not on the rtl8812au one.
So wee need this exponent value.

Czechball, 2021-12-25:
Hey, thank you for your quick and helpful responses.
> If channel setting failed, hcxdumptool will print the reported message from the driver. 

So first, trying to just run supported vs unsupported channel on the RTL8812AU:
* Supported channel
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlan0 -c 1
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

unexpected frequency/channel!
wanted 1, reported from driver 241200000 (exponent 1)
no frequencies available

terminating...
1 driver error encountered

```
* Unsupported channel
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlan0 -c 15
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

unexpected frequency/channel!
wanted 15, reported from driver 507500000 (exponent 1)
no frequencies available

terminating...
1 driver error encountered

```
I think I can see the issue here, it's very weird.
Just for completion, here's the output of the former commands on Asus USB AC-51 (MT7610U):
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlp0s20u4 -c 1
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

start capturing (stop with ctrl+c)
```
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlp0s20u4 -c 15
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

frequency/channel 15 not accepted by driver
no frequencies available

terminating...
1 driver error encountered
```
> Please add one line of code after this line 
After adding the debug lines:
* Asus USB AC-51:
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlp0s20u4 -c 1
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

debug freq/ch: 2412, exponent: 6
```
* Alfa AWUS036ACH (RTL8812AU)
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlan0 -c 1
initialization of hcxdumptool 6.2.5-20-g08842f2...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
debug freq/ch: 241200000, exponent: 1
unexpected frequency/channel!
wanted 1, reported from driver 241200000 (exponent 1)
no frequencies available

terminating...
1 driver error encountered
```

ZerBea, 2021-12-25:
Thanks.
The rtl8812au driver is reporting the frequency a little bit different:
241200000 with exponent 1  = 2412000000 = 2412 MHz.

I'll move from hard code exponent 6 to dynamic exponent. Please give me some time to handle it. Than I'm going to push an update.

Merry Christmas

Czechball, 2021-12-25:
Thanks a lot, you're the best. Merry Christmas and a happy New Year, enjoy the holidays.

ZerBea, 2021-12-27:
Thanks.
Handling of frequencies "ioctl() SIOCGIWFREQ" reported by driver should be dynamic by this commit:
https://github.com/ZerBea/hcxdumptool/commit/5ae77c75db194182dd23d86994742f8835b3993f
Please test.

Handling of setting a frequency "ioctl() SIOCSIWFREQ" is still fixed on frequency in Hertz (by exponent 6).
Now we have to test if the driver accept it.

Czechball, 2021-12-27:
Seems like now both channel and frequency switches are working:
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlan0 -c 44 --enable_status 95 --silent
initialization of hcxdumptool 6.2.5-22-g82927ca...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11b
INTERFACE TX POWER........: 0 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: - (not used for the attack)
INTERFACE VIRTUAL MAC.....: - (not used for the attack)
DRIVER....................: rtl88XXau
DRIVER VERSION............: 5.10.84-1-MANJARO
DRIVER FIRMWARE VERSION...: 
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 000dc26d8dd5 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 000dc26d8dd6 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 000dc26d8dd7 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: dc70141e68fd
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62465
ANONCE....................: 1e098e85d60e5c832874395fa1ce7fd30664641157c27fd00e3082f121aff63e
SNONCE....................: cf4688fb6c86df9d4809bc2a910c369d5c15e93c9eb3c5a746c710a3c020037d

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
11:37:38 5220/44  ffffffffffff - ssid1 [BEACON]
11:37:38 5220/44  ffffffffffff - ssid2 [BEACON]
11:37:42 5220/44  - - ssid1 [PROBERESPONSE]
11:37:42 5220/44  - - ssid2 [PROBERESPONSE]

```
```
[czechball@altitude hcxdumptool]$ sudo ./hcxdumptool -i wlan0 -c 5220 --enable_status 95 --silent
initialization of hcxdumptool 6.2.5-22-g82927ca...
warning possible interfere: NetworkManager is running with pid 394

warning possible interfere: wpa_supplicant is running with pid 399

interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11b
INTERFACE TX POWER........: 0 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: - (not used for the attack)
INTERFACE VIRTUAL MAC.....: - (not used for the attack)
DRIVER....................: rtl88XXau
DRIVER VERSION............: 5.10.84-1-MANJARO
DRIVER FIRMWARE VERSION...: 
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
[... 79 lines cut ...]
DRIVER FIRMWARE VERSION...: 
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 4876041c4fa8 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 4876041c4fa9 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 4876041c4faa (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: f0a2255a97b8
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63937
ANONCE....................: 3c34026f004ac838a14ad69a934c0db00c131b0b1225c52e7701134cc6c4411c
SNONCE....................: 6349b57c1009f0d29a191b7d2a1e29c8648d021b55dbdce7784e9c26185375cd

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
11:41:19 2412/1   ffffffffffff - ssid1 [BEACON]
11:41:19 2412/1   ffffffffffff - ssid2 [BEACON]
11:41:19 2412/1   ffffffffffff - ssid3 [BEACON]
11:41:19 2412/1   ffffffffffff - ssid4 [BEACON]
```

ZerBea, 2021-12-27:
I'm glad to hear this. Thanks for the test.
Now hcxdumptool and hcxlabtool are able to handle all exponents (0, 1, 2, 3, 4, 5 and 6) in a dynamic way, which is a nice improvement.

closed 2021-12-27 by ZerBea

ZerBea, 2022-08-15:
@Czechball do you still have an ALFA AWUS036ACH. If yes, is latest hcxdumptool still working or do you have similar problems:
https://github.com/ZerBea/hcxdumptool/issues/227

Czechball, 2022-08-15:
Yeah, I'm still using that one, along with ~1 week old versions of aircrack-ng rtl8812au and hcxdumptool, haven't noticed any problems so far... Maybe a little bit of decreased performance, which could be related to some hardware issues. I'll check the exact versions when I'm home.

Czechball, 2022-08-19:
Hello. I've finally had some time to check my system.
I'm running on Raspberry Pi OS 5.15.56-v7l+ on a Raspberry Pi 4B 2GB.
My hcxdumptool version is 6.2.7-1-g7174fbc, rtl8812au (from aircrack-ng) version v5.6.4.2, commit 4ab079f. Driver was installed using make install, not dkms. The interface seems to be working fine, hcxdumptool is not complaining about anything and the log is full of traffic (I always do the manual monitor mode switch with ip and iw before starting hcxdumptool). There might be a problem with packet injection, its performance is bad, but I still need to investigate that.

ZerBea, 2022-08-21:
Great, thanks for the test. Looks like running ip and iw is still mandatory to set monitor mode by NETLINK. Otherwise the driver doesn't init the device completely.

Czechball, 2022-08-22:
Update: There aren't any issues with packet injection either, it's working pretty good - same as before. I just wasn't in an ideal environment when testing.

ZerBea, 2022-08-23:
Thanks for the update.
Can you please comment out put of available channels (-C)?

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.