ZerBea/hcxdumptool #201: Error 0

okcf10, 2022-01-08:
Hi So when I run the following 
`sudo hcxdumptool -i wlan0mon -enable_status=95 -o test.pcap`
I keep getting error 0

Live Ubuntu 20.04
Ralink Technology, Corp. RT2870/RT3070

Thanks

ZerBea, 2022-01-08:
There is no error 0 message.
Please comment complete output after running your command line.
Please also comment complete output of
$ sudo hcxdumptool -i your_interface --check-driver
and
$ sudo hcxdumptool -i your_interface --check_injection

Also please notice that hcxdumptool doesn't run on virtual monitor interfaces like wlan0mon.

From --help:
```
$ hcxdumptool -h
hcxdumptool 6.2.5-25-ga62a745  (C) 2021 ZeroBeat
usage  : hcxdumptool <options>
         press ctrl+c to terminate hcxdumptool
         press GPIO button to terminate hcxdumptool
         hardware modification is necessary, read more:
         https://github.com/ZerBea/hcxdumptool/tree/master/docs
         do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
         do not run hcxdumtool on virtual machines or emulators
         do not run hcxdumptool in combination with tools (channel hopper), that take access to the interface (except: tshark, wireshark, tcpdump)
         do not use tools like machcanger, because hcxdumptool run its own MAC space and will ignore this changes
         stop all this services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
```

ZerBea, 2022-01-08:
An example how to use hcxdumptool is here (section: Capture example
https://hashcat.net/wiki/doku.php?id=cracking_wpawpa2

ZerBea, 2022-01-09:
I'm not able to reproduce your reported issue, if I follow the instructions of --help:
```
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service
$ sudo hcxdumptool -i wlp39s0f3u1u1u1 --enable_status=95 -o test.pcapng
initialization of hcxdumptool 6.2.5-26-g7dc1cc4...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp39s0f3u1u1u1
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: c83a35cb08e3 (not used for the attack)
INTERFACE VIRTUAL MAC.....: c83a35cb08e3 (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.15.13-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 10ae60095ea6 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 10ae60095ea7 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 10ae60095ea8 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: e80410e5347f
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63693
ANONCE....................: f9c0c97df100fa5ca3a6c54647b9669a2cc225a958ceca592f7e209004d8acc9
SNONCE....................: cae149b498382006840fa09a3f70e1605b754017a9d32b6e00f4c4684ae97358

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
09:10:42 2412/1   ffffffffffff 112233445566 TEST_NET [BEACON]
09:10:43 2412/1   aabbccddeeff 112233445566 WLAN-V7ME3M [PMKIDROGUE:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx KDV:2]

09:13:00 2467/12    ERROR:0 INCOMING:751 AGE:5 OUTGOING:435 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
^C
terminating...
1 driver error encountered
ERRORs < 10 are related to a slow initialization and can be ignored
```

okcf10, 2022-01-10:
Thanks for the quick reply and sorry for the less information. So, I am not working on a virtual machine. I ran the commands and here is the output (in the screenshot)
[ The adapter name was wlan0mon because I ran airmon-ng before and it changed the name as it was too long]

[For this test I didn't use airmon-ng]
[image]

Here Is the error 

`sudo hcxdumptool -i wlx20f41b209080 --filterlist_ap=filter --filtermode=2 --enable_status=95 -o 1.pcapng

initialization of hcxdumptool 6.2.5-20-g08842f2... interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls start capturing (stop with ctrl+c) 
NMEA 0183 SENTENCE........: N/A INTERFACE NAME............:wlx20f41b209080 INTERFACE PROTOCOL........: IEEE 802.11 INTERFACE TX POWER........: 20 dBm (lowest value reported by the device) INTERFACE HARDWARE MAC....: 20f41b209080 (not used for the attack) INTERFACE VIRTUAL MAC.....: 20f41b209080 (not used for the attack) DRIVER....................: rt2800usb DRIVER VERSION............: 5.11.0-27-generic DRIVER FIRMWARE VERSION...: 0.36 openSSL version...........: 1.1 ERRORMAX..................: 100 errors BPF code blocks...........: 0 FILTERLIST ACCESS POINT...: 1 entries FILTERLIST CLIENT.........: 0 entries FILTERMODE................: attack WEAK CANDIDATE............: 12345678 ESSID list................: 0 entries ACCESS POINT (ROGUE)......: 000df2dd6174 (BROADCAST HIDDEN used for the attack) ACCESS POINT (ROGUE)......: 000df2dd6175 (BROADCAST OPEN used for the attack) ACCESS POINT (ROGUE)......: 000df2dd6176 (used for the attack and incremented on every new client) CLIENT (ROGUE)............: e0cb1d84809f EAPOLTIMEOUT..............: 20000 usec EAPOLEAPTIMEOUT...........: 2500000 usec REPLAYCOUNT...............: 62847 ANONCE....................: f11f20dcd065fc41cee71035020946ee3fbdb6ef44e1dd61cab3c77373e831f1 SNONCE....................: 8a06291129d908a0a807fd190b9e35b70e5f1b085541a2acc95ae8ca7c3a5770 TIME FREQ/CH MAC_DEST MAC_SOURCE ESSID [FRAME TYPE] 09:58:09 2412/1 ffffffffffff 0495e6466cd1 Network1 [BEACON] 09:58:11 2412/1 007263fa4f86 000df2dd6177 xxxxxxx [ROGUE PROBERESPONSE] 09:58:17 2412/1 3c846a830ec6 0495e6466cd1 Network1 [PROBERESPONSE] 09:59:00 2412/1 ERROR:0 INCOMING:4452 AGE:1 OUTGOING:423 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0 10:00:00 2412/1 ERROR:0 INCOMING:9152 AGE:1 OUTGOING:715 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0 10:01:00 2412/1 ERROR:0 INCOMING:13015 AGE:1 OUTGOING:726 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0 ^C terminating... 1 driver error encountered `

ZerBea, 2022-01-10:
The line starting with ERROR:...... is an internal status which is shown every minute. You requested that by adding 64 to --enable status.
I can't see any ERRORs  in your comments. Everything is working as expected.
ERROR = 0
65617 incoming packets
646 otugoing packets
It is activated by adding 64 to --enable_status
You can disable this status by --enable_status=31
Also all driver and injection test passsed fine.

BTW:
I'm not talking about a VIRTUAL MACHINE (VM). I'm talking about a VIRTUAL INTERFACE (VIF) (e.g.: wlan0mon) that airmon-ng and iw create. Hcxdumptool will not work as expected on this VIRTUAL MONITOR INTERFACEs.
Don't worry about the long interface names. Hcxdumptool will handle this.

Closed, because output looks like expected and every test passed as expected.
But you can still ask your questions here.

closed 2022-01-10 by ZerBea

ZerBea, 2022-01-10:
To test if targets are in range, you should use --do_rcascan:
sudo hcxdumptool -i wlx20f41b209080 --do_rcascan
No HITs means, no target is in your transmit range.

ZerBea, 2022-01-10:
The --enable_status bitmask explained:
```
--enable_status=<digit>            : enable real-time display (waterfall)
                                     only incoming traffic
                                     each message is displayed only once at the first occurrence to avoid spamming the real-time display
                                     bitmask:
                                         0: no status (default)
                                         1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
                                        32: GPS (once a minute)
                                        64: internal status (once a minute)
                                       128: run as server
                                       256: run as client
                                       512: EAP
                                      1024: EAP NAK
                                     characters < 0x20 && > 0x7e are replaced by .
                                     example: show everything but don't run as server or client (1+2+4+8+16 = 31)
                                              show only EAPOL and ASSOCIATION and REASSOCIATION (1+2 = 3)
```

You set --enable_status=95 which is: 1+2+4+8+16+64 = 95
 ```
                                        1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
                                          64: internal status (once a minute)
```
That include the internal status (ERROR: .... line) every minute

This --enable_status=31will exclude the internal status:  1+2+4+8+16 = 31
 ```
                                        1: EAPOL
                                         2: ASSOCIATION and REASSOCIATION
                                         4: AUTHENTICATION
                                         8: BEACON and PROBERESPONSE
                                        16: ROGUE AP
```
Using a bitmask as option value will makes us more flexible and keep the command line small.

okcf10, 2022-01-11:
Hey Thanks So much for the reply I had some questions
1.When do you know it's successful?
2.How Long can it take
3.How does the packet injection ratio effect the results 
[will having a better ratio increase changes of success]

ZerBea, 2022-01-11:
1.When do you know it's successful?
$ sudo hcxdumptool -i -i wlx20f41b209080 -o test.pcapng --active_beacon --enable_status=1
Now only EAPOL messages (that include PMKIDs) are shown.
If you got them, your attack was successful.
We are packet oriented. If one packet reach the target, it is enough.

2.How Long can it take
PMKID within 10 minutes
EAPOL message (M1-M4) within 10 minutes, but only if CLIENTs are connected
EAPOL message M2 within 10 minutes, but only if CLIENTs are present

3.How does the packet injection ratio effect the results
No effect. This is only useful to verify that injection and antenna is working as expected
$ sudo hcxdumptool -i wlx20f41b209080 --do_rcascan
will show you if your target is in range.

okcf10, 2022-01-28:
Hey Thanks, again so the network which I tried above doesn't support the pmkid attack?

ZerBea, 2022-01-28:
If you don't receive a PMKID from the target within the first 5 minutes, the AP doesn't transmit one, because it is disabled, by default.