ZerBea/hcxdumptool #203: Driver is busy: failed to transmit internal beacon

Niklas-debug, 2022-02-26:
Hello,

when I run the following command: `sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15`, I get the following output:

`initialization of hcxdumptool 6.2.5...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: cc6b1e528c17 (not used for the attack)
INTERFACE VIRTUAL MAC.....: cc6b1e528c17 (not used for the attack)
DRIVER....................: rtw_8822ce
DRIVER VERSION............: 5.16.0-kali1-amd64
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 005079d80cb9 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: 005079d80cba (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 005079d80cbb (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0febd685599
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64718
ANONCE....................: 38f36c4102b64c75d7d08b0fd5c8de7d89c09511ca4b09d384f6c03610dd9385
SNONCE....................: 742aa41b4833c03dc4169b2509479417f0f5039847174060a51006967eeeda1d

18:27:11 2412/1   ffffffffffff b44bd626ba02 WifiAccessPoint [BEACON]
18:27:11 2412/1   ffffffffffff 9eb94eb8e3c1 [HIDDEN BEACON]
18:27:11 2412/1   ffffffffffff eca81f2e2660WifiAccessPoint [BEACON]
18:27:11 2412/1   ffffffffffff eea81f2e2662 WifiAccessPoint [BEACON]
18:27:11 2412/1   ffffffffffff eea81f2e2664 WifiAccessPoint [BEACON]
18:27:11 2412/1   ffffffffffff 3e71bf3ebbb4 WifiAccessPoint [BEACON]
18:27:11 2412/1   ffffffffffff bce92f000af9 WifiAccessPoint [BEACON]
18:27:12 2417/2   0009b09d0e33 eca81f2e2660 WifiAccessPoint [PROBERESPONSE]
18:27:12 2417/2   0009b09d0e33 eea81f2e2662 WifiAccessPoint [PROBERESPONSE]
18:27:12 2417/2 driver is busy: failed to transmit internal beacon
18:27:13 2417/2 driver is busy: failed to transmit proberesponse
18:27:13 2417/2 driver is busy: failed to transmit proberesponse
18:27:14 2417/2 driver is busy: failed to transmit proberesponse
18:27:14 2417/2 driver is busy: failed to transmit internal beacon
18:27:15 2417/2 driver is busy: failed to transmit proberesponse
18:27:15 2417/2 driver is busy: failed to transmit proberesponse
18:27:16 2422/3 driver is busy: failed to transmit internal beacon
18:27:16 2422/3 driver is busy: failed to transmit internal beacon
18:27:16 2422/3 driver is busy: failed to transmit internal beacon
18:27:17 2422/3 driver is busy: failed to transmit proberesponse
18:27:18 2422/3 driver is busy: failed to transmit proberesponse
18:27:18 2422/3 driver is busy: failed to transmit proberesponse
18:27:19 2422/3 driver is busy: failed to transmit proberesponse
18:27:19 2422/3 driver is busy: failed to transmit proberesponse
18:27:20 2427/4 driver is busy: failed to transmit internal beacon
18:27:20 2427/4 driver is busy: failed to transmit internal beacon
18:27:20 2427/4 driver is busy: failed to transmit internal beacon
18:27:20 2427/4 driver is busy: failed to transmit proberesponse
18:27:22 2427/4 driver is busy: failed to transmit proberesponse
18:27:22 2427/4 driver is busy: failed to transmit proberesponse
18:27:23 2427/4 driver is busy: failed to transmit proberesponse
18:27:23 2427/4 driver is busy: failed to transmit proberesponse
18:27:24 2432/5 driver is busy: failed to transmit internal beacon
18:27:24 2432/5 driver is busy: failed to transmit internal beacon
18:27:24 2432/5 driver is busy: failed to transmit internal beacon
18:27:25 2432/5 driver is busy: failed to transmit proberesponse
18:27:26 2432/5 driver is busy: failed to transmit proberesponse
18:27:26 2432/5 driver is busy: failed to transmit proberesponse
18:27:27 2432/5 driver is busy: failed to transmit proberesponse
18:27:27 2432/5 driver is busy: failed to transmit proberesponse
18:27:28 2437/6 driver is busy: failed to transmit internal beacon
18:27:28 2437/6 driver is busy: failed to transmit internal beacon
18:27:28 2437/6 driver is busy: failed to transmit internal beacon
18:27:29 2437/6 driver is busy: failed to transmit authenticationrequest
18:27:29 2437/6 driver is busy: failed to transmit deauthentication
18:27:29 2437/6   ffffffffffff b44bd626ba00 WifiAccessPoint [BEACON]
^C18:28:00 2484/14 driver is busy: failed to transmit proberesponse

terminating...
1 driver error encountered`

Before I run this, I stop the following services:
`sudo systemctl stop wpa_supplicant.service`
`sudo systemctl stop NetworkManager.service`

I tried my internal Wi-Fi card and an ALFA Networks AWUS036EAC.

Is there a solution for this problem or other way to do this?

ZerBea, 2022-02-27:
hcxdumptool will show this warnings if the driver is not able to transmit any longer.

rtw_8821ce:
Tested a notebook (internal) device and it looks like the rtw_8822ce driver is broken.
I got a similar result on latest driver:
$ sudo hcxdumptool -i wlp4s0 -o dumpfile.pcapng --enable_status=15 --active_beacon
initialization of hcxdumptool 6.2.5-44-g83ead87 (this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlp4s0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 70665577c423 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 70665577c423 (not used for the attack)
DRIVER....................: rtw_8821ce
DRIVER VERSION............: 5.16.10-arch1-1
DRIVER FIRMWARE VERSION...: N/A

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
15:53:13 2417/2 driver is busy: failed to transmit internal beacon
15:53:14 2417/2 driver is busy: failed to transmit internal beacon
15:53:14 2417/2 driver is busy: failed to transmit internal beacon

By this commit:
https://github.com/ZerBea/hcxdumptool/commit/4bc7a29791a132e6b05a43d9ea350e1df39d0363
I removed this device from the known as working list until the driver will receive a fix.

AWUS036EAC
ALFA's Linux driver for this device is an RTL8812AU driver set from Realtek
https://deviwiki.com/wiki/ALFA_Network_AWUS036EAC
This driver doesn't work well in combination with hcxdumptool due to its NETLINK dependency as mentioned in README.md:
```
Not recommended WiFi chipsets:
* Broadcom (neither monitor mode nor frame injection)
* Intel PRO/Wireless (due to several driver issues and NETLINK dependency)
* Realtek RTL8811AU, RTL8812AU, RTL8814AU (due to NETLINK dependency)
```
https://github.com/aircrack-ng/rtl8812au/issues/862
https://github.com/aircrack-ng/rtl8812au/issues/476#issuecomment-558523331

You can try to use ip and iw to set monitor mode by NETLINK:
$ sudo systemctl stop wpa_supplicant.service
$ sudo systemctl stop NetworkManager.service
$ sudo ip link set wlan0 down
$ sudo iw dev wlan set type monitor
$ sudo ip link set wlan0 down

Than run hcxdumptool:
$ sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15

Bot issues are related to the Realtek drivers. Unfortunately there is nothing I can do to fix this by hcxdumptool.
rtl2800usb and mt76 drivers are working as expected running the same command line as mentioned above.

Niklas-debug, 2022-02-27:
Thank you for your answer. I was able to run hcxdumptool with the following commands:
$ sudo systemctl stop wpa_supplicant.service
$ sudo systemctl stop NetworkManager.service
$ sudo ip link set wlan1 down
$ sudo iw dev wlan1 set type monitor
$ sudo ip link set wlan1 up

Than run hcxdumptool:
$ sudo hcxdumptool -i wlan1 -o dumpfile.pcapng --active_beacon --enable_status=15

ZerBea, 2022-02-27:
NETLINK dependency on Realtek drivers is really a problem. It looks like the driver need some NETLINK commands to fully init the device. This init can be done by iw only, because hcxdumptool doesn't use NETLINK stuff.
Regarding the rtw_8821ce, we have to wait for an official fix.

closed 2022-02-27 by ZerBea

ZerBea, 2022-02-28:
I changed the headline if other users are running into the same issue.
Unfortunately it looks like that the use of "new" Realtek SOCs is not the best choice this days, if you want to inject packets, due to driver issues and/or NETLINK dependency. The same applies to Intel SOCs and iwlwifi driver as well as Broadcom SOCs (absolutely no monitor mode on stock drivers).
Luckily other SOCs (vendor: Ralink/MediaTek) are still working fine. The drivers (e.g.: rt2800usb and mt76) are well maintained, working out of the box (full monitor mode and packet injection by default) and they are part of the Linux stock kernel.
Here is a nice overview:
https://openwrt.org/docs/techref/hardware/soc/soc.mediatek

pjchacong, 2022-03-01:
I experienced this issue with my old Alfa card, so I bought the one down below on Amazon.

Happy to report that it works flawlessly out of the box. Just note that the blue LED in the adapter that indicates activity, doesn't work in Kali. I thought it was a HW issue, but it works on Windows.

Alfa Long-Range Dual-Band AC1200 USB 3.0 Wi-Fi Adapter w/ 2X 5dBi External Antennas - 2.4GHz 300Mbps/5GHz 867Mbps - 802.11ac & A, B, G, N - Windows, MacOS & Kali Linux Supported - (AWUS036ACM) https://www.amazon.com/dp/B08BJS8FXD/ref=cm_sw_r_apan_glt_i_J12WKQ57KJ2D7W4M2NFX?_encoding=UTF8&psc=1

ZerBea, 2022-03-01:
@pjchacong I fully agree. This device is fantastic in combination with a dual band panel antenna (e.g. DELOCK 88806):
https://duckduckgo.com/?q=DELOCK+88806&t=h_&iax=images&ia=images

Some additional information about the device (SOC, driver):
https://deviwiki.com/wiki/ALFA_Network_AWUS036ACM

The driver (mt76) is well (very well) maintained and if you have problems, you can report them to upstream, here:
https://github.com/openwrt/mt76
After a fix, the update will find its way directly to the Linux stock kernel (Lorenzo is doing a great job!):
https://patchwork.kernel.org/project/linux-wireless/list/
example of a similar issue on mt7615 device: MT7615 LEDs not working (even after adding -DCONFIG_MT76_LEDS)
https://github.com/openwrt/mt76/issues/355

BTW:
Regarding 5GHz packet injection, it is important to set the correct wireless regulatory domain, e.g.:
```
$ iw reg get
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (N/A)
	(5470 - 5725 @ 160), (N/A, 24), (N/A)
	(5725 - 5875 @ 80), (N/A, 30), (N/A)
```

ZerBea, 2022-03-01:
I forgot to mention:
The modified rtl8812au driver
https://github.com/aircrack-ng/rtl8812au
is not(!) part of the stock Linux kernel;

ZerBea, 2022-03-01:
Looks like there are several regressions on rtw_8822ce after kernel update from 5.15 to 5.16:
https://www.spinics.net/lists/linux-wireless/msg221075.html

HighSteam, 2022-07-14:
I am having the same issue with my driver. Anyone found a solution for this issue?

Niklas-debug, 2022-07-14:
> I am having the same issue with my driver. Anyone found a solution for this issue?

It's working for me now with the following driver: https://github.com/aircrack-ng/rtl8812au

ZerBea, 2022-07-18:
You need root permission to install kernel modules:
$ make && sudo make install
should do it.

The  rtl8812au driver will not work on your device (rtw8821ce).
The rtw8821ce driver is currently broken (regarding packet injection).
You kernel 5.17.15 is outdated: https://www.kernel.org/

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.