ZerBea/hcxdumptool #204: Question about choosing wifi card

ericsia, 2022-03-04:
I wanted to choose a new wifi card as I am very unfamiliar with mpcie etc. I will only use if for a while so a budget friendly one is my choice.
I confirmed I can choose 3 x 3
So I have option to get Intel 8260 vs 8265.
Upon googling I found hcxdumptool can't support 8260, I wonder if it is because of mimo reason.
Then checking 8265 seem like there isn't any issue. Does anyone know whether  8265 can capture pmkid using hcxdumptool?

ZerBea, 2022-03-04:
From README.md:
```
Not recommended WiFi chipsets:
* Broadcom (neither monitor mode nor frame injection)
* Intel PRO/Wireless (due to several driver issues and NETLINK dependency)
* Realtek RTL8811AU, RTL8812AU, RTL8814AU (due to NETLINK dependency)
```
As I write this comment, I only recommend Ralink and MediaTek chipsets. Drivers for both of them are part of the stock Linux kernel, well maintained and support full monitor mode and packet injection,

That is not correct:
"Upon googling I found hcxdumptool can't support 8260"
It should be:
"The driver doesn't support full monitor mode and packet injection and it depend on NETLINK"
https://bugzilla.kernel.org/show_bug.cgi?id=214291

ZerBea, 2022-03-04:
If you want to buy a WiFi adapter please notice this 4 points:

Is the chipset supported by Linux native driver:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/wireless?h=v5.17-rc6

Are there known issues of the driver:
https://bugzilla.kernel.org/

Does the driver support full monitor mode and packet injection:
https://deviwiki.com/

Are there known problems running hcxdumptool:
https://github.com/ZerBea/hcxdumptool/blob/master/README.md

ericsia, 2022-04-08:
@ZerBea just curious, have you ever run a benchmark hashcat windows vs hashcat linux non gpu which one is more faster?

ZerBea, 2022-04-08:
No, because I'm not a friend of benchmarks and I use Linux only.

ZerBea, 2022-04-11:
https://github.com/ZerBea/hcxdumptool/issues/211