ZerBea/hcxdumptool #205: Understanding info

hbith, 2022-03-09:
Could you help me understanding the relationship between info on hcxdumptool and hcxpcapngtool and .22000 output file?

for example I' ve made a capture, and  obtained 174 items on .22000 export...
hcxdumptool ends with:

13:04:00   4 ERROR:0 INCOMING:7806363 AGE:1 OUTGOING:156381 PMKIDROGUE:1 PMKID:2 M1M2ROGUE:326 M1M2:135 M2M3:27 M3M4:0 M3M4ZEROED:24 GPS:0

hcxpcapngtool:
summary capture file
--------------------
file name................................: road.202203081046.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.4.0-99-generic
application..............................: hcxdumptool 6.2.0-15-ga42fc84
interface name...........................: wlp3s0mon
interface vendor.........................: 2016d8
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 28ef0130d9aa (incremented on every new client)
MAC CLIENT...............................: e80410f2429c
REPLAYCOUNT..............................: 62324
ANONCE...................................: 7292c6c66a99ca14d08f281fc31483048d975a9c2f1a0a47bf79c95d3bc6a5b4
SNONCE...................................: d922c3c8cfd553cc111ff275a2b8b37517d35fc4b67aadda57e7980ef03f4ab2
timestamp minimum (GMT)..................: 08.03.2022 10:46:30
timestamp maximum (GMT)..................: 09.03.2022 13:03:51
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 45131
frames with correct FCS..................: 38073
BEACON (total)...........................: 116
PROBEREQUEST.............................: 1653
PROBEREQUEST (directed)..................: 30
PROBERESONSE.............................: 84
AUTHENTICATION (total)...................: 283
AUTHENTICATION (OPEN SYSTEM).............: 283
ASSOCIATIONREQUEST (total)...............: 154
ASSOCIATIONREQUEST (PSK).................: 151
REASSOCIATIONREQUEST (total).............: 50
REASSOCIATIONREQUEST (PSK)...............: 49
EAPOL messages (total)...................: 42680
EAPOL RSN messages.......................: 42013
EAPOL WPA messages.......................: 667
ESSID (total unique).....................: 1654
EAPOLTIME gap (measured maximum usec)....: 1621422706
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (suggested NC)...........: 5
EAPOL M1 messages........................: 13657
EAPOL M2 messages........................: 28684
EAPOL M3 messages........................: 240
EAPOL M4 messages........................: 99
EAPOL pairs (total)......................: 164654
EAPOL pairs (best).......................: 157
EAPOL ROGUE pairs........................: 127
EAPOL pairs written to combi hash file...: 157 (RC checked)
EAPOL M12E2..............................: 141
EAPOL M32E2..............................: 16
PMKID (total)............................: 471
PMKID (useless)..........................: 943
PMKID (best).............................: 17
PMKID ROGUE..............................: 1
PMKID written to combi hash file.........: 17
malformed packets (total)................: 3
ESSID error (malformed packets)..........: 3

ZerBea, 2022-03-09:
This is the status of RAW date, received by hcxdumptool:
`13:04:00 4 ERROR:0 INCOMING:7806363 AGE:1 OUTGOING:156381 PMKIDROGUE:1 PMKID:2 M1M2ROGUE:326 M1M2:135 M2M3:27 M3M4:0 M3M4ZEROED:24 GPS:0`
We are online and triggered by the traffic. The EAPOL TIMEOUT is much lower because we have not the time to do a deep analysis over a long time period. There is no time to remove duplicates or zeroed values from the counters. We stupid increase them on an event inside the EAPOL TIMEOUT-
```
--eapoltimeout=<digit>             : set EAPOL TIMEOUT (microseconds)
                                     default: 20000 usec
```

Now we are offline and have all the time we need to do a deeper analysis which include clean up, removal of duplicates and detection of zeroed values. The EAPOL TIMEOUT is much greater so that we have more stuff to analyze.
```
--eapoltimeout=<digit>             : set EAPOL TIMEOUT (milliseconds)
                                   : default: 5000 ms
```

hcxdumptool -> only approximate values (quick and dirty)
hcxpcapngtool -> deep analysis

Just play around with the EAPOL TIMEOUT values and you'll see that the results vary.

closed 2022-03-10 by hbith

ZerBea, 2022-03-10:
You're welcome.

Please notice that each option has a price tag. That include the status waterfall display, too.
All actions are triggered by the traffic on the channel and not by the capabilities of the system that is running hcxdumptool.
To take care about EAP AUTHENTICATIONs we additional have to slow down all attacks.

If you are an advanced user, I recommend to use hcxlabtool series:
https://github.com/ZerBea/wifi_laboratory
Instead of using options, everything is compiled directly into the binary (tailored by mission), depending on the compiler flags:
DGETM1 -> hunt for PMKIDs
DGETM2 -> hunt for EAPOL M2 frames of CLIENTs
DGETM2PR -> hunt for EAPOL M2 frames of CLIENTs - respond 10(!) APs simultaneously an a single CLIENT PROBEREQUEST (essidlist is mandatory)
DGETM1234 -> hunt for 4way handshakes
DBEACONUNSET -> set wildcard BEACON - force a CLIENT to send a BROADCAST WILDCARD PROBEREQUEST
DSTATUSOUT -> show limited waterfall status (not necessary on headless systems)  

all options can be cascaded (example see Makefile)

This is mostly to be used on headless systems like a Raspberry Pi:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-2