ZerBea/hcxdumptool #206: Build 32bit.exe from source?

Guest9595, 2022-03-20:
Hi!

Usually I prefer Linux, but i would need 32 bit win executable version for my tablet. I tried to build it from source code and I get that error with minGw-w64. So I want to make executable version for Windows. Linux version is working good, except some driver related issues.(NHA using ath9).

Terminal line:

sudo i686-w64-mingw32-gcc hcxdumptool.c -o hcxdumptool32_win

fatal error: linux/ethtool.h: No such file or directory 19 # include <linux/ethtool.h>

I tried:

   - tried to build it on Fedora with minGw-w64
    - tried to build it on Debian based Parrot Os with some extra dependencies which could fix it but did not.
    - Install apt-get install build-essential and similar
    - installed dependencies as in hcxdumptool manual.
    - cygwin on Windows with similar error.

I dont know what is missing is it possible to compile get it working for win32 bit? I think I installed all what is in the manual and even more? Any help appreciated or even better if you had the time to compile it.

Thank you!

RealEnder, 2022-03-21:
hcxdumptool heavily depends on linux kernel interfaces to talk to wifi adapters. This means it can't be just cross compiled for win, since win lacks this kind of APIs.
Some people have luck using hcxdumptool under Android using supported external USB WiFi adapter.

ZerBea, 2022-03-21:
As RealEnder mentioned, neither Windows nor MAC OS provide drivers that are capable to do full packet injection.
Android is partly capable to do this if the Linux kernel is compiled with CRDA and WEXT (which is not the case on most of the devices).
```
$ zcat /proc/config.gz | grep CFG80211
CONFIG_CFG80211=m
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
CONFIG_CFG80211_DEBUGFS=y
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
```
To perform this high speed, high density attack vector, hcxdumptool is directly talking to the Linux kernel API via ioctl() system calls and there is absolutely no way to do this on the proprietary operating systems mentioned above.

The developer of cygwin know about this and they don't supply this Linux only stuff:
https://cygwin.com/cgi-bin2/package-grep.cgi?grep=ethtool.h&arch=x86

And even if hcxdumptool is cross compiled successfully, it will not work running this operating systems or an emulator/wrapper.

ZerBea, 2022-03-21:
Please also notice that, in several cases, running hcxdumptool within a VM will not work, too: 
https://github.com/ZerBea/hcxdumptool/issues/196
hcxdumptool need full access to the device and doesn't like it to be interrupted by the VM.

ZerBea, 2022-03-21:
BTW:
hcxdumptool highly depend on the quality of the driver. Unfortunately it looks like the ath9k driver have some serious issues at the moment:
https://bugzilla.kernel.org/buglist.cgi?bug_status=NEW&bug_status=ASSIGNED&bug_status=NEEDINFO&bug_status=REOPENED&field0-0-0=product&field0-0-1=component&field0-0-2=alias&field0-0-3=short_desc&field0-0-4=status_whiteboard&field0-0-5=content&no_redirect=1&order=changeddate%20DESC%2Cbug_status%2Cpriority%2Cassigned_to%2Cbug_id&query_format=advanced&type0-0-0=substring&type0-0-1=substring&type0-0-2=substring&type0-0-3=substring&type0-0-4=substring&type0-0-5=matches&value0-0-0=ath9k&value0-0-1=ath9k&value0-0-2=ath9k&value0-0-3=ath9k&value0-0-4=ath9k&value0-0-5=%22ath9k%22

closed 2022-03-21 by ZerBea