ZerBea/hcxdumptool #207: Alfa AWUS036ACM not working correctly - 1 driver error

pjchacong, 2022-03-22:
I'm using this Alfa adapter. It worked fin until about two weeks ago (today is March 22, 2022), but since then, when using hcxdumptool, it retrieves a considerable smaller amount of responses, and runs"  very slowly--about 30 entries (responses) per hour, and often only one response is an EPOL. When cutting the process, I get the error "1 driver error?" 

Please advice. 

Thank you.

ZerBea, 2022-03-23:
To give an advice, I need more information:
type of device ALFA .....
the exact command line us use
complete output of
$ hcxdumptool -v
$ hcxdumptool -I
$ hcxdumptool -i interfacename --check_driver
$ hcxdumptool -i interfacename --check_injection

Please notice that the ALFA decvie use different chipsets. Some of them are hit by driver issues.

One driver ERROR is normal, because this ERROR is related to  pselect() after you terminated hcxdumptool by pressing ctrl+c which causes an interrupt that increase the ERROR counter.
That message was fixed by this commit
https://github.com/ZerBea/hcxdumptool/commit/81818db376467d5175d827219ef18f94a9fc01e3
To get rid of this single ERROR, please make sure, you're running the latest version.

Please also note that the realtime display depend on --enable_status. A value of 95 will give you more information as a value of 1:
```
--enable_status=95
 1: EAPOL
 2: ASSOCIATION and REASSOCIATION
 4: AUTHENTICATION
 8: BEACON and PROBERESPONSE
16: ROGUE AP
64: internal status (once a minute)
```

ZerBea, 2022-03-23:
A added a frequency statistic to latest hcxpcapngtool.
It will give you an information about received BEACONS. We take the channel information ELEMENT of an ACCESS POINT to count this.
And it will give you an information about the detect frequencies on which are traffic. We take the value directly from the device:
```
BEACON (detected on 2.4 GHz channel).....: 6 7 8 9 10 11 
...
frequency statistics (frequency: received packets)
--------------------------------------------------
 2437: 4221	 2442: 6007	 2447: 1831	 2452: 4575	
 2457: 4134	 2462: 4905	
```

I'll say, if you you many less/not crowded channels in your scan list. the status display will show you less information.
If you choose crowded channels, you'll get more information. 
Running -c 1,6,11 you'll get more traffic than running -c 36,4,44.  

Here is a nice video that show an ALFA in action:
https://forum.hashkiller.io/index.php?threads/hcxdumptool-hcxtools-modern-tool-to-analyze-your-own-wifi-network-part-1-hcxdumptool.55607/#post-306550

BTW:
I'm on kernel:
$ uname -r
5.16.16-arch1-1

and (Raspberry Pi ARMv6)
$ uname -r
5.10.103+

ZerBea, 2022-03-23:
Additional --do_rcascan will give you an information how man ACCESS POINTs are in range.
Therefore hcxdumptool transmit PROBEREQUESTs and count the BEACONs and (PROBE)RESPONSEs from the ACCESS POINTs. That is a good indicator to see if an AP is in range.

ZerBea, 2022-03-23:
To make sure, we have no driver regression I ran a test on an ALFA AWUS036ACM:
$ lsusb
Bus 001 Device 003: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
```
$ sudo hcxdumptool -i wlp3s0f0u2 --enable_status=64 -c 11 -o test16111.pcapng --bpfc=targets.bpfc
initialization of hcxdumptool 6.2.5-54-g38ea25c (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy1
INTERFACE NAME............: wlp3s0f0u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca1d743a (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca1d743a (not used for the attack)
DRIVER....................: mt76x2u
DRIVER VERSION............: 5.16.16-arch1-1
DRIVER FIRMWARE VERSION...: 0.0.00-b1
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 79
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 980ee4ff53eb (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 980ee4ff53ec (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 980ee4ff53ed (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: e80410403f66
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64111
ANONCE....................: 9a20c03c8da537622ceab9b76b745c2dda7b480b01b206cc6031abdf4ef7d921
SNONCE....................: d4057befc85dd414ca7ef245e46106f6cc1bb8d5b0b816b1b79a6817d3aa2534

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
09:49:00 2462/11    ERROR:0 INCOMING:1218 AGE:1 OUTGOING:601 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
09:50:00 2462/11    ERROR:0 INCOMING:5609 AGE:1 OUTGOING:2305 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
09:51:00 2462/11    ERROR:0 INCOMING:10101 AGE:1 OUTGOING:3711 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
09:52:00 2462/11    ERROR:0 INCOMING:14466 AGE:1 OUTGOING:5293 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
09:53:00 2462/11    ERROR:0 INCOMING:18800 AGE:1 OUTGOING:6979 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
```
Everything looks like expected:

There are no ERRORs during runtime.
We can see incoming packets (age between them <= 1 second) and outgoing packets.
All targets (2 APs and one CLIENT as stored in the Berkeley Packet Filter) are attacked successful.

pjchacong, 2022-03-24:
Thank you so much, @ZerBea for these detailed responses. I will first update my hcxdunmtool to make sure I have your latest version, and then I will make some more testing with the options that you shared.

For visibility, the full command that I am running is sudo hcxdumptool -i wlan1 -o dumpfile.pcapng --active_beacon --enable_status=15

ZerBea, 2022-03-24:
Ok. Adding the results of my tests above, allow you to compare it to your results. If they are similar, we can rule out a hardware issue.
Please notice that the realtime display is limited to 
```
only incoming traffic
each message is displayed only once at the first occurrence to avoid spamming the real-time dis
```
Let us say there are 10 targets in range and hcxdumptool attacked them successfully and it received from each target a message matching to the requested enable_status option, there will be no longer this type of messages displayed.
e..g. if we got a BEACON, there will be no more BEACONs of this target displayed

ZerBea, 2022-04-06:
Latest driver test is here:
https://github.com/ZerBea/hcxdumptool/issues/211#issuecomment-1090123208

closed 2022-04-06 by ZerBea

aurhack, 2022-04-06:
zerbea this outputs when i check the driver injection of AWUS036ACM : 

initialization of hcxdumptool 6.2.5-61-g7fedbd0 (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 5865/173 proberesponse 171   
packet injection is working on 2.4GHz!
injection ratio: 7% (BEACON: 2367 PROBERESPONSE: 171)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 16% (NETWORK: 118 PROBERESPONSE: 20)
your incection ratio is poor - improve your antenna and get closer to the target

terminating...

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.