ZerBea/hcxdumptool #209: driver doesn't report frequency

hklsb, 2022-03-30:
Hi I was using aircrack-ng + airodump-ng with atheros wireless adapter, successful.
But I realize hcxdumptool is modern way especially with new wpa2 thing.
So I tried but get error, anyone have this issue?

`uname -r`
5.13.0-39-generic

`hcxdumptool -v`
hcxdumptool 6.2.5-59-gf138052 (C) 2021 ZeroBeat

`hcxdumptool -I`
wlan interfaces:
phy2	c01c300da43d	wlxc01c300da43d	(driver:ath9k_htc)

`sudo hcxdumptool -i wlxc01c300da43d --check_driver`
initialization of hcxdumptool 6.2.5-59-gf138052 (depending on the capabilities of the device, this may take some time)...
starting driver test...
driver doesn't report frequency
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored

Note: I've been reading and looking for these issues for hours but couldn't find a solution.
Is my adapter not supported by hcxdumptool?

ZerBea, 2022-03-30:
$ hcxdumptool -I
wlan interfaces:
phy12	f81a67d974b9	wlp5s0f4u2	(driver:ath9k_htc)

$ sudo hcxdumptool -i wlp5s0f4u2 --check_injection
initialization of hcxdumptool 6.2.5-59-gf138052 (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2484/14 proberesponse 294   
packet injection is working on 2.4GHz!
injection ratio: 64% (BEACON: 454 PROBERESPONSE: 294)
your injection ratio is good
antenna ratio: 60% (NETWORK: 10 PROBERESPONSE: 6)
your antenna ratio is good

terminating...

aurhack, 2022-03-30:
@ZerBea I'm running Kali Linux on bare metal and i'm having an error doing this command : 

sudo hcxdumptool -i wlan1 -o dumpfile.pcapng --active_beacon --enable_status=1

After getting some beacons probes, pmkidrogues and that type of things this message prints a lot : 

23:04:00 2427/4 driver is busy: failed to transmit internal beacon
23:04:01 2427/4 driver is busy: failed to transmit internal beacon
23:04:01 2427/4 driver is busy: failed to transmit internal beacon
23:04:02 2427/4 driver is busy: failed to transmit internal beacon
23:04:03 2427/4 driver is busy: failed to transmit internal beacon
23:04:03 2427/4 driver is busy: failed to transmit internal beacon
23:04:05 2432/5 driver is busy: failed to transmit internal beacon
23:04:05 2432/5 driver is busy: failed to transmit internal beacon
23:04:06 2432/5 driver is busy: failed to transmit internal beacon
23:04:06 2432/5 driver is busy: failed to transmit internal beacon

when i run dmesg this sentence says this : 

device wlan1 entered promiscuous mode
[ 1523.275065] ath: phy3: Unable to remove monitor interface at idx: 0
[ 1525.291232] ath: phy3: Unable to remove station entry for monitor mode
[ 1542.884096] usb 1-8: USB disconnect, device number 8
[ 1543.181747] device wlan1 left promiscuous mode

Kali Linux on Virtual Machine works good but this happens on bare metal you know why?

I'm running an AWUS036NHA to this.

hklsb, 2022-03-31:
Thanks for your response, I'll try immediately and let you know the results!

hklsb, 2022-03-31:
Now it's getting better. passed driver test but couldn't capture anything. 
The first try, it shows a couple of APs and then prints "driver is busy" something like that.
Subsequent try didn't show it anymore just quiet.

```
sudo hcxdumptool -i wlxc01c300da43d --check_driver
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

sudo hcxdumptool -i wlxc01c300da43d --check_injection
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2484/14 proberesponse 0   
warning: no PROBERESPONSE received - packet injection is probably not working!

terminating...

sudo hcxdumptool -i wlxc01c300da43d -o dumpfile.pcapng --active_beacon --enable_status=15
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy1
INTERFACE NAME............: wlxc01c300da43d
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: c01c300da43d (not used for the attack)
INTERFACE VIRTUAL MAC.....: c01c300da43d (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.13.0-39-generic
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 00057899a3cb (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 00057899a3cc (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 00057899a3cd (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0ece17ce2d2
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62400
ANONCE....................: 9f58b612b1cc34b12f581986d01699efde884da5178c290df16be91a80b917d3
SNONCE....................: 774e8590fc0d36413413058e0f7e6739c7151a7ef1cbbd58e48a713baf1b29f5

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
```

hklsb, 2022-03-31:
Here is the injection result I ran after rebooting...
```
sudo hcxdumptool -i wlxc01c300da43d --check_injection
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2484/14 proberesponse 369   
packet injection is working on 2.4GHz!
injection ratio: 32% (BEACON: 1142 PROBERESPONSE: 369)
your injection ratio is average, but there is still room for improvement
antenna ratio: 57% (NETWORK: 28 PROBERESPONSE: 16)
your antenna ratio is good

terminating...
```

ZerBea, 2022-03-31:
I got another issue report on ath9k driver (closed, because it is similar to yours):
https://github.com/ZerBea/hcxdumptool/issues/210

Running enable_status=64 (if you want to see other messages, too, set it to 95), you'll get some more information (ERROR counter state, and state of incoming and outgoing packets.

Some times the driver is working as expected (INCOMING and OUTGOING packet counter increase.
```
$ sudo hcxdumptool -i wlp5s0f4u2 --active_beacon --enable_status=64
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy2
INTERFACE NAME............: wlp5s0f4u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.16.16-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 14070883a5af (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 14070883a5b0 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 14070883a5b1 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: acde483de17f
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 61655
ANONCE....................: 4074b480ae07b8adf026a3de64d629fd36a066586fb72832f34f442970ab0a04
SNONCE....................: abb6f3ec4ea249890eb56c12c2828ae9f16aa015de28e4995817ad29001a2abf

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
09:12:00 2422/3     ERROR:0 INCOMING:22 AGE:1 OUTGOING:81 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
09:13:00 2427/4     ERROR:0 INCOMING:2097 AGE:5 OUTGOING:1567 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:0 GPS:0
09:14:00 2432/5     ERROR:0 INCOMING:3563 AGE:9 OUTGOING:2700 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:0 GPS:0
09:15:00 2437/6     ERROR:0 INCOMING:4641 AGE:2 OUTGOING:3775 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:0 GPS:0
09:16:00 2442/7     ERROR:0 INCOMING:6307 AGE:1 OUTGOING:4824 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:0 GPS:0
09:17:00 2447/8     ERROR:0 INCOMING:7678 AGE:1 OUTGOING:6038 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:6 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:1 GPS:0
^C
terminating...
```
But sometimes the driver doesn't respond (INCOMING:0 AGE:1 OUTGOING:0):
```
$ sudo hcxdumptool -i wlp5s0f4u2 --enable_status=64
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy1
INTERFACE NAME............: wlp5s0f4u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.16.16-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 10b7135493c7 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 10b7135493c8 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 10b7135493c9 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0febd97efde
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63758
ANONCE....................: ac658fc213545ae66cba45072ded8ba02587492a611d7f86b90135dea580096a
SNONCE....................: e84ec3f4174180ca351574601d9dbc398535c95c7ebb83ebc07e08e1cc31a0ae

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
09:04:00 2417/2     ERROR:0 INCOMING:0 AGE:1 OUTGOING:0 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
^C
terminating...
```
After some investigation on bugzilla.kernel.org that there are some related issues on this driver:
https://bugzilla.kernel.org/show_bug.cgi?id=215703
https://bugzilla.kernel.org/show_bug.cgi?id=42877
https://bugzilla.kernel.org/show_bug.cgi?id=215698

Like wpa_supplican, hcxdumptool noticed that, too, as mentioned here:
https://bugzilla.kernel.org/show_bug.cgi?id=215703
and print "driver doesn't respond" or ""driver is busy: failed to transmit internal beacon".

Now we have to figure out, why the driver is broken.
I'm running
$ uname -r
5.16.16-arch1-1
and can confirm that it is broken.

BTW:
By this commit, I removed ath9k devices from the know as working list, until this driver issue will receive a fix:
https://github.com/ZerBea/hcxdumptool/commit/7fedbd0685046f75b9873af36c866bb3c35b75a6

hklsb, 2022-03-31:
Thanks for caring for this driver.
I don't know if this helps but after all of these error thing,
when I start NetworkManager.service and wpa_supplicant.service
my driver couldn't find a single wireless network.
I had to unplug the adapter physically and replug it to have wireless network.

ZerBea, 2022-03-31:
Unfortunately, you are right. After after the error occurred, the driver is dead.

hklsb, 2022-03-31:
Actually I don't think roll back kernal is not a solution because up before that all your programs are running smoothly and you tackled the kernal and the specific program might be working but what might happen to the rest of programs? I am not a pro but advise me.

aurhack, 2022-03-31:
You can rollback to a version all programs worked well anyways, like on a  version all programs and drivers worked well you know

ZerBea, 2022-03-31:
I'm still not sure how to deal with this issue. Also I'm not sure which kernel version is affected:
https://lkml.org/lkml/2022/3/23/72
Now I'm going to check kernel 5.10.103.

aurhack, 2022-03-31:
@ZerBea I can say that on my vm i didn't updated or upgraded anything

it works on 
┌──(kali㉿kali)-[~]
└─$ uname -r
5.15.0-kali3-amd64

so that's why i said we could rollback, in my case so i don't know the other case you know how i can downgrade my version of the bare metal kali from 5.16.0 to 5.15.0 ?

aurhack, 2022-03-31:
└─$ sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=95
initialization of hcxdumptool 6.2.5-55-gbb8353c (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca994771 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca994771 (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.15.0-kali3-amd64
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 001761f0cc7b (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 001761f0cc7c (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 001761f0cc7d (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0ece1e50a32
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62597
ANONCE....................: 657258163dc23b198d14cded899608a732c46af9d57f7c695d3e0b2eca7cbac1
SNONCE....................: bbd6cfeefa5740d745340e021f68f66e990bdaa73c12988afc2cab144453a36d

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
09:32:32 2417/2   ffffffffffff 4829523a1ae6 vodafone1AE0 [BEACON]
09:32:32 2417/2   ffffffffffff c8b4228309ff MOVISTAR_09F0 [BEACON]
09:32:32 2417/2   ffffffffffff 88832251f788 Orlando Araujo  [BEACON]
09:32:32 2417/2   b8b7f1ca195c 001761f0cc7e DefaultSSID [ROGUE PROBERESPONSE]
09:32:32 2417/2   ffffffffffff 34576093519e MOVISTAR_519D [BEACON]
09:32:32 2417/2   ffffffffffff 047153eed478 pepephone_ADSL4JEF [BEACON]
09:32:32 2417/2   ffffffffffff f85b3bfe532f SANTOANDRE [BEACON]
09:32:32 2417/2   66eecc8ad282 3c846afaf048 TP-Link_F048 [PROBERESPONSE]
09:32:32 2417/2   66eecc8ad282 e4c32a6f1ebe Monitos2_2.4GHz [PROBERESPONSE]
09:32:32 2417/2   ffffffffffff e81b696fd558 vodafoneBA1321 [BEACON]
09:32:32 2417/2   ffffffffffff e4c32a6f1ebe Monitos2_2.4GHz [BEACON]
09:32:32 2417/2   ffffffffffff 704f576ba787 MiFibra-9474 [BEACON]
09:32:32 2417/2   ffffffffffff 608d26ec22d3 Livebox6-22D4 [BEACON]
09:32:32 2417/2   ffffffffffff a4ceda5d21a6 MiFibra-4730 [BEACON]
09:32:33 2417/2   ffffffffffff e475dc1fb63e MiFibra-4730 [BEACON]
09:32:33 2417/2   001dc9a18259 001761f0cc7f MOVISTAR_7C9E [ROGUE PROBERESPONSE]
09:32:33 2417/2   ffffffffffff 3c846afaf048 TP-Link_F048 [BEACON]
09:32:33 2417/2   ffffffffffff a02d130f39cc vodafone9D9C [BEACON]
09:32:33 2417/2   ffffffffffff 44fb5ace5bbb MIWIFI_xz54 [BEACON]
09:32:34 2417/2   ffffffffffff e475dca65706 PAQUITA [BEACON]
09:32:34 2417/2   b8b7f1b0dbe3 001761f0cc7e DefaultSSID [AUTHENTICATION]
09:32:34 2417/2   b8b7f1b0dbe3 001761f0cc7e DefaultSSID [ASSOCIATION]
09:32:34 2417/2   209148c72d20 704d7b604460 ASUS_ED44 [PROBERESPONSE]
09:32:35 2417/2   20e2040cdb20 001761f0cc80 vodafoneBA8120 [ROGUE PROBERESPONSE]
09:32:35 2417/2   ffffffffffff 8cfdde8fc0ee vodafoneC0E8 [BEACON]
09:32:35 2417/2   3280cccbcac7 704f576ba787 MiFibra-9474 [PROBERESPONSE]
09:32:37 2422/3   209148c72d20 94917f8fb6af MOVISTAR_B6A0 [PROBERESPONSE]
09:32:37 2422/3   ffffffffffff 94917f8fb6af MOVISTAR_B6A0 [BEACON]
09:32:37 2422/3   20e2040cdb20 001761f0cc80 vodafoneBA8120 [AUTHENTICATION]
09:32:37 2422/3   20e2040cdb20 001761f0cc80 vodafoneBA8120 [ASSOCIATION]
09:32:41 2427/4   ffffffffffff e4ca12c3012e MIWIFI_2G_juZr [BEACON]
09:32:41 2427/4   b0ece1e50a32 e4ca12c3012e MIWIFI_2G_juZr [PMKIDROGUE:cf46329f3add924ea69e9266337da202 KDV:2]
09:32:41 2427/4   ffffffffffff 80787100f04f MOVISTAR_F040 [BEACON]
09:32:41 2427/4   68474935476b 80787100f04f MOVISTAR_F040 [PROBERESPONSE]
09:32:41 2427/4   ffffffffffff a02d13102f65 Vodafone_4960_74FL [BEACON]
09:32:43 2427/4   2c9ffbc570b1 001761f0cc7e DefaultSSID [AUTHENTICATION]
09:32:43 2427/4   2c9ffbc570b1 001761f0cc7e DefaultSSID [ASSOCIATION]
[... 6 lines cut ...]
09:32:48 2437/6   ffffffffffff b246fc601148 MOVISTAR_1148 [BEACON]
09:32:48 2437/6   ffffffffffff cceddc936e51 MOVISTAR_6E50 [BEACON]
09:32:48 2437/6   b0ece1e50a32 cceddc129841 MOVISTAR_9840 [PMKIDROGUE:8720c3ee80601528697af03ec6f37b5d KDV:2]
09:32:48 2437/6   ffffffffffff 946ab05e09ae MiFibra-09AC [BEACON]
09:32:48 2437/6   ffffffffffff 841ea37ea4fe sagemcomA4F8 [BEACON]
09:32:48 2437/6   ffffffffffff 209a7db84066 MOVISTAR_F819 [BEACON]
09:32:48 2437/6   ffffffffffff f4239c181fd0 vodafoneBA2052 [BEACON]
09:32:48 2437/6   ffffffffffff 7829ed9fe80c MOVISTAR_E80B [BEACON]
09:32:49 2437/6   4401bbbad4b6 209a7db84066 MOVISTAR_F819 [PROBERESPONSE]
09:32:49 2437/6   ffffffffffff a0648faf76d8 MOVISTAR_76D7 [BEACON]
09:32:49 2437/6   ffffffffffff b4fbe497c83d WIFI_FAUS [BEACON]
09:32:49 2437/6   b0ece1e50a32 cceddc936e51 MOVISTAR_6E50 [PMKIDROGUE:2e6b0951b8289659ae86c5306ede95a1 KDV:2]
09:32:49 2437/6   b0ece1e50a32 841ea37ea4fe sagemcomA4F8 [PMKIDROGUE:75ff0eb962aa5e87481830277a7395ec KDV:2]
09:32:49 2437/6   b0ece1e50a32 209a7db84066 MOVISTAR_F819 [PMKIDROGUE:9206e109d8b30b304426ea1f2ff0a56f KDV:2]
[... cut ...]
09:32:50 2437/6   b0ece1e50a32 a0648faf76d8 MOVISTAR_76D7 [PMKIDROGUE:986efcb903cdc6b7fb1de9847a7ea11f KDV:2]
09:32:51 2437/6   b0ece1e50a32 345760b7d457 MOVISTAR_D456 [PMKIDROGUE:3d9db04403b64301b483ff2e2223c0f4 KDV:2]
09:32:51 2437/6   ffffffffffff c8b422d8c34f MOVISTAR_C340 [BEACON]
09:32:51 2437/6   68474935476b 4829523a1ae6 vodafone1AE0 [PROBERESPONSE]
09:33:00 2452/9     ERROR:0 INCOMING:1606 AGE:8 OUTGOING:537 PMKIDROGUE:8 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0

Maybe it still doesn't works pretty well on vm tho.. hmm

aurhack, 2022-03-31:
Anyways zerbea you know if it's better to downgrade or use vm instead? you know when this issue will be fixed? thanks for everything

hklsb, 2022-03-31:
> 

Talking about downgrade, what am I suppose to do? Mine is 5.13.0. Upgrading to 5.15 ? Let me search online how to upgrade to a specific kernel version.

ZerBea, 2022-03-31:
Finished the tests on kernel 5.10.103+ and 5.16.16 running 6 devices in parallel:
$ hcxdumptool -I
wlan interfaces:
phy0	70665578cdab	wlp4s0	(driver:rtw_8821ce)
phy1	f81a67077d0e	wlp5s0f3u3	(driver:ath9k_htc)
phy2	fc34973278be	wlp5s0f3u3	(driver:mt76x0u)
phy3	74da38eb4600	wlp5s0f3u3	(driver:mt7601u)
phy4	dc4ef4086e71	wlp5s0f3u3	(driver:rt2800usb)
phy5	00c0caad0e49	wlp5s0f3u3	(driver:mt76x2u)

driver:rtw_8821ce -> broken
driver:ath9k_htc -> broken
driver:mt76x0u -> working as expected
driver:mt7601u -> working as expected
driver:rt2800usb -> working as expected
driver:mt76x2u -> working as expected

ZerBea, 2022-03-31:
Also I'm not sure how the HOST handle the GUEST if running hcxdumptool in a VM.
Please take a look at this issue report:
https://github.com/ZerBea/hcxdumptool/issues/196
After a while, the HOST disconnected the GUEST and the network went down.
Are you running the same kernel within the VM and on bare Linux?

hklsb, 2022-03-31:
Forn me, bare metal. X86-64, Ubuntu

ZerBea, 2022-03-31:
Thanks.
Updated to
$ uname -r
5.17.1-arch1-1
and it looks better:
```
$ sudo hcxdumptool -i wlp5s0f4u2 --enable_status=64 --active_beacon -t 10 --bpfc=test.bpfc -c 1,6,11,2,3,4,5,6,7,8,9,10,11,12,13
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy1
INTERFACE NAME............: wlp5s0f4u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.17.1-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 79
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 8c840156a7ff (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 8c840156a800 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 8c840156a801 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0ece1bfef6e
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63069
ANONCE....................: 6ede111e970b0c46dd84a6c457e516fd729d482fbf97647f16c36a5bbc3d7557
SNONCE....................: 3e19eaab033729b3b701e61fb3fa34055d04bb603b4bddc204bc88424152ba68

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
16:33:00 2427/4     ERROR:0 INCOMING:1936 AGE:14 OUTGOING:734 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:2 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
16:34:00 2457/10    ERROR:0 INCOMING:2964 AGE:1 OUTGOING:1770 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:2 M1M2:6 M2M3:3 M3M4:0 M3M4ZEROED:4 GPS:0
16:35:00 2462/11    ERROR:0 INCOMING:5099 AGE:1 OUTGOING:2527 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:2 M1M2:7 M2M3:4 M3M4:0 M3M4ZEROED:5 GPS:0
16:36:00 2442/7     ERROR:0 INCOMING:6604 AGE:1 OUTGOING:3661 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:2 M1M2:10 M2M3:7 M3M4:0 M3M4ZEROED:6 GPS:0
16:37:00 2472/13    ERROR:0 INCOMING:8932 AGE:1 OUTGOING:4482 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:11 M2M3:8 M3M4:0 M3M4ZEROED:7 GPS:0
16:38:00 2427/4     ERROR:0 INCOMING:13210 AGE:14 OUTGOING:5393 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
16:39:00 2457/10    ERROR:0 INCOMING:13814 AGE:3 OUTGOING:6360 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
16:40:00 2462/11    ERROR:0 INCOMING:16180 AGE:1 OUTGOING:7021 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
16:41:00 2442/7     ERROR:0 INCOMING:18150 AGE:1 OUTGOING:7801 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
16:42:00 2472/13    ERROR:0 INCOMING:20085 AGE:1 OUTGOING:8552 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
16:43:00 2427/4     ERROR:0 INCOMING:21851 AGE:14 OUTGOING:9375 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
^C
terminating...

```

ZerBea, 2022-03-31:
There are several changes from 5,16 to 5.17:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/diff/?id=v5.17&id2=v5.16&dt=2
```
-rw-r--r-- | drivers/net/wireless/ath/ath9k/ar9002_mac.c | 2 |   |   |   |  
-- | -- | -- | -- | -- | -- | --
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/ar9003_calib.c | 14 |   |   |   |  
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/hif_usb.c | 7 |   |   |   |  
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/htc.h | 2 |   |   |   |  
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/htc_drv_txrx.c | 13 |   |   |   |  
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/wmi.c | 4 |   |   |   |  
  |   |  
-rw-r--r-- | drivers/net/wireless/ath/ath9k/xmit.c
```

hklsb, 2022-03-31:
Just upgrade to 5.17.1
Here are the results but I'm not sure if it's working!
Please advise.

```
uname -r
5.17.1-051701-generic

hcxdumptool -I
wlan interfaces:
phy0	c01c300da43d	wlxc01c300da43d	(driver:ath9k_htc)

sudo hcxdumptool -i wlxc01c300da43d --check_driver
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

sudo hcxdumptool -i wlxc01c300da43d --check_injection
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2484/14 proberesponse 297   
packet injection is working on 2.4GHz!
injection ratio: 31% (BEACON: 955 PROBERESPONSE: 297)
your injection ratio is average, but there is still room for improvement
antenna ratio: 51% (NETWORK: 29 PROBERESPONSE: 15)
your antenna ratio is good

terminating...

sudo hcxdumptool -i wlxc01c300da43d -o dumpfile.pcapng --active_beacon --enable_status=64
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlxc01c300da43d
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: c01c300da43d (not used for the attack)
INTERFACE VIRTUAL MAC.....: c01c300da43d (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.17.1-051701-generic
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 000e17c9e7f7 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 000e17c9e7f8 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 000e17c9e7f9 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: acde488e0857
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 65242
ANONCE....................: e8056b6e018e0f5e460cf75e3c083b291ec386d9592c0992abc4946bd60b2a36
SNONCE....................: 6720d8cfa72520aaa174f44733b9b5242e89b18768688a4f9fa4a2507391b348

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
23:55:00 2427/4     ERROR:0 INCOMING:226 AGE:8 OUTGOING:282 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
23:56:00 2432/5     ERROR:0 INCOMING:226 AGE:68 OUTGOING:1494 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
23:57:00 2437/6     ERROR:0 INCOMING:226 AGE:128 OUTGOING:2707 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
23:58:00 2442/7     ERROR:0 INCOMING:226 AGE:188 OUTGOING:3919 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
23:59:00 2447/8     ERROR:0 INCOMING:226 AGE:248 OUTGOING:5132 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0

```

ZerBea, 2022-03-31:
I don't think it is working, because INCOMING packet count is too less and stopped at 266
compared ti mine
```
16:33:00 2427/4     ERROR:0 INCOMING:1936 AGE:14 OUTGOING:734 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:2 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
...
16:43:00 2427/4     ERROR:0 INCOMING:21851 AGE:14 OUTGOING:9375 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:12 M2M3:9 M3M4:0 M3M4ZEROED:8 GPS:0
```

ZerBea, 2022-03-31:
Same chipset, same firmware and same driver, so we should expect the same behavior - but unfortunately that isn't the case.
Let's take a look at the kernel config. Maybe there are som differences:
```
$ zcat /proc/config.gz | grep 80211
CONFIG_CFG80211=m
# CONFIG_NL80211_TESTMODE is not set
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
CONFIG_CFG80211_DEBUGFS=y
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
CONFIG_LIB80211=m
CONFIG_LIB80211_CRYPT_WEP=m
CONFIG_LIB80211_CRYPT_CCMP=m
CONFIG_LIB80211_CRYPT_TKIP=m
# CONFIG_LIB80211_DEBUG is not set
CONFIG_MAC80211=m
CONFIG_MAC80211_HAS_RC=y
CONFIG_MAC80211_RC_MINSTREL=y
CONFIG_MAC80211_RC_DEFAULT_MINSTREL=y
CONFIG_MAC80211_RC_DEFAULT="minstrel_ht"
CONFIG_MAC80211_MESH=y
CONFIG_MAC80211_LEDS=y
CONFIG_MAC80211_DEBUGFS=y
# CONFIG_MAC80211_MESSAGE_TRACING is not set
# CONFIG_MAC80211_DEBUG_MENU is not set
CONFIG_MAC80211_STA_HASH_MAX_SIZE=0
CONFIG_MAC80211_HWSIM=m
```

ZerBea, 2022-03-31:
device capabilities:
```
$ iw phy phy6 info | grep supports
	Device supports RSN-IBSS.
	Device supports T-DLS.
	Device supports TX status socket option.
	Device supports HT-IBSS.
	Device supports SAE with AUTHENTICATE command
	Device supports low priority scan.
	Device supports scan flush.
	Device supports AP scan.
	Device supports per-vif TX power setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Device supports ACK timeout estimation.
	Device supports configuring vdev MAC-addr on create.
```

ZerBea, 2022-03-31:
Conclusion:
TP-Link TL-WN722N V1 is working on kernel 5.17.1
CHANEVE AR9271 still issues

@aurhack how about AWUS036NHA ?

hklsb, 2022-03-31:
Both commands didn't work for me. How do I do that? @ZerBea

hklsb, 2022-03-31:
OK I got the first one:- here
```
cat /boot/config-5.17.1-051701-generic | grep 80211
CONFIG_CFG80211=m
# CONFIG_NL80211_TESTMODE is not set
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
# CONFIG_CFG80211_CERTIFICATION_ONUS is not set
CONFIG_CFG80211_REQUIRE_SIGNED_REGDB=y
CONFIG_CFG80211_USE_KERNEL_REGDB_KEYS=y
CONFIG_CFG80211_DEFAULT_PS=y
CONFIG_CFG80211_DEBUGFS=y
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_CFG80211_WEXT_EXPORT=y
CONFIG_LIB80211=m
CONFIG_LIB80211_CRYPT_WEP=m
CONFIG_LIB80211_CRYPT_CCMP=m
CONFIG_LIB80211_CRYPT_TKIP=m
# CONFIG_LIB80211_DEBUG is not set
CONFIG_MAC80211=m
CONFIG_MAC80211_HAS_RC=y
CONFIG_MAC80211_RC_MINSTREL=y
CONFIG_MAC80211_RC_DEFAULT_MINSTREL=y
CONFIG_MAC80211_RC_DEFAULT="minstrel_ht"
CONFIG_MAC80211_MESH=y
CONFIG_MAC80211_LEDS=y
CONFIG_MAC80211_DEBUGFS=y
CONFIG_MAC80211_MESSAGE_TRACING=y
# CONFIG_MAC80211_DEBUG_MENU is not set
CONFIG_MAC80211_STA_HASH_MAX_SIZE=0
CONFIG_MAC80211_HWSIM=m

```

ZerBea, 2022-03-31:
Ok, thanks. That is identical to mine.
For the other command you need iw:
https://zoomadmin.com/HowToInstall/UbuntuPackage/iw

hklsb, 2022-03-31:
Maybe this is the one?
I already have iw, it's just I don't know the commands.
```
iw phy phy4 info | grep supports
	Device supports RSN-IBSS.
	Device supports T-DLS.
	Device supports TX status socket option.
	Device supports HT-IBSS.
	Device supports SAE with AUTHENTICATE command
	Device supports low priority scan.
	Device supports scan flush.
	Device supports AP scan.
	Device supports per-vif TX power setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Device supports configuring vdev MAC-addr on create.

```

ZerBea, 2022-03-31:
Thanks again. Here is something different. TL-WN722N support ACK timeout estimation:
Device supports ACK timeout estimation but I assume that this doesn't cause the issue, because active monitor mode is not implemented in hcxdumptool.
Currently I'm testing active monitoring on hcxlabtool starting with this commit:
https://github.com/ZerBea/wifi_laboratory/commit/a475e885e9985d0ee43ea89f2a0af936933835d4

aurhack, 2022-03-31:
> Conclusion: TP-Link TL-WN722N V1 is working on kernel 5.17.1 CHANEVE AR9271 still issues
> 
> @aurhack how about AWUS036NHA ?

Let me try rn! Sorry i was studying i study on afternoons.

aurhack, 2022-03-31:
@ZerBea how i can update my kali linux to version 5.17.1?

ZerBea, 2022-04-01:
Maybe purchasing an AWUS036AC is not a good idea.
The stock driver supplied by the kernel dos not(!) support monitor mode an packet injection.
https://deviwiki.com/wiki/ALFA_Network_AWUS036AC
You need a third party driver to be installed:
https://github.com/aircrack-ng/rtl8812au

Take a look at the video made by David:
https://davidbombal.com/best-wifi-hacking-adapters-in-2021-kali-linux-parrot-os/
​Realtek RTL8812AU: 6:39​  -> hard to install on KALI

Also please take a look at the issue reports regarding this driver:
https://github.com/aircrack-ng/rtl8812au/issues

From hcxdumptool README.md adapter section:
```
Not recommended WiFi chipsets:
* Broadcom (neither monitor mode nor frame injection)
* Intel PRO/Wireless (due to several driver issues and NETLINK dependency)
* Realtek RTL8811AU, RTL8812AU, RTL8814AU (due to NETLINK dependency)
```

ZerBea, 2022-04-01:
Result of the long term test an AR9271:
```
$ sudo hcxdumptool -i wlp5s0f3u3 --enable_status=64 -o test.pcapng --active_beacon --bpfc=test.bpfc -c 1,6,11 --stop_client_m2_attacks=2
initialization of hcxdumptool 6.2.5-60-gd42d16d (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy8
INTERFACE NAME............: wlp5s0f3u3
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.17.1-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 79
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 000eefe34597 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 000eefe34598 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 000eefe34599 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: fcc233ed11c9
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 65292
ANONCE....................: 91db5865f64eb12897c11c30cb218bb950d759a42bc09a698ab80aa3871e95b0
SNONCE....................: fe3b437cedef288c22075e7c1ec7511d9628bd5172dce832ab27e9d644135766

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
22:42:00 2437/6     ERROR:0 INCOMING:2448 AGE:1 OUTGOING:602 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:6 M2M3:4 M3M4:0 M3M4ZEROED:3 GPS:0
22:43:00 2437/6     ERROR:0 INCOMING:5211 AGE:1 OUTGOING:1303 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:8 M2M3:6 M3M4:0 M3M4ZEROED:4 GPS:0
22:44:00 2437/6     ERROR:0 INCOMING:8333 AGE:3 OUTGOING:2049 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:8 M2M3:6 M3M4:0 M3M4ZEROED:4 GPS:0
22:45:00 2437/6     ERROR:0 INCOMING:10917 AGE:1 OUTGOING:2844 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:8 M2M3:6 M3M4:0 M3M4ZEROED:4 GPS:0
22:46:00 2437/6     ERROR:0 INCOMING:13739 AGE:1 OUTGOING:3466 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
...
07:32:00 2437/6     ERROR:0 INCOMING:1489944 AGE:5 OUTGOING:351250 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
07:33:00 2437/6     ERROR:0 INCOMING:1494321 AGE:5 OUTGOING:351843 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
07:34:00 2437/6     ERROR:0 INCOMING:1498866 AGE:5 OUTGOING:352424 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
07:35:00 2437/6     ERROR:0 INCOMING:1503482 AGE:1 OUTGOING:352997 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
07:36:00 2437/6     ERROR:0 INCOMING:1508511 AGE:1 OUTGOING:353934 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:9 M2M3:7 M3M4:0 M3M4ZEROED:5 GPS:0
^C
terminating...
```
INCOMING and OUTGOING packet count increased
no ERROR occurred 
All test targets are attacked successful

ZerBea, 2022-04-01:
Please take a look at this newer video made by David:
https://davidbombal.com/wifi-wpa-wpa2-cracking-with-hashcat-and-hcxdumptool/

hklsb, 2022-04-01:
Hey @ZerBea 
Today I'm fortunate. I did it again and it works for 2-4 minutes.
It managed to capture 8 handshakes but as usual after that
`driver is busy:...`

Here is the result with hashcat:-
```
Session..........: hashcat
Status...........: Running
Hash.Mode........: 22000 (WPA-PBKDF2-PMKID+EAPOL)
Hash.Target......: capture.hc22000
Time.Started.....: Fri Apr  1 15:30:17 2022 (2 mins, 25 secs)
Time.Estimated...: Fri Apr  1 16:31:52 2022 (59 mins, 10 secs)
Kernel.Feature...: Pure Kernel
Guess.Base.......: File (/usr/share/wordlists/wpa/step1.txt)
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:    57136 H/s (8.77ms) @ Accel:16 Loops:64 Thr:256 Vec:1
Recovered.Total..: 0/8 (0.00%) Digests, 0/6 (0.00%) Salts
Progress.........: 10508862/213347952 (4.93%)
Rejected.........: 2251326/10508862 (21.42%)
Restore.Point....: 1732579/35557992 (4.87%)
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:2624-2688
Candidate.Engine.: Device Generator
Candidates.#1....: 15089413 -> 15m06a27o07d
Hardware.Mon.#1..: Temp: 83c Fan: 52%
```

ZerBea, 2022-04-01:
Here is another interesting issue report on that driver:
https://github.com/qca/open-ath9k-htc-firmware/issues/173
It confirm the behavior (working after re-connect the device) that you mentioned this comment:
https://github.com/ZerBea/hcxdumptool/issues/209#issuecomment-1084218309

hklsb, 2022-04-01:
> That and my own tests confirm a driver issue which is possible fixed by kernel 5.17.1-arch1-1 (the driver no longer crashes). By this commit [d42d16d](https://github.com/ZerBea/hcxdumptool/commit/d42d16d7172e7df903e846d3de039232b0d0c0a5) I removed the detection of a (not responding driver) buggy driver. That allow hcxdumptool to run as long as possible until the driver crashes completely. Unfortunately there is nothing more I can do.

Maybe that's why it reconnects wireless network automatically
after starting wpa_supplicant and NetworkManager services.
It's an improvement.
Thanks

hklsb, 2022-04-01:
But I'm still puzzled by aircrack-ng and airodump-ng working perfectly
through all these though!

ZerBea, 2022-04-01:
In contrast to wpa_supplicatn, hcxdumptool doesn't restart a "broken" device.
Instead, especially on headless operation (Raspberry Pi), there is an option to reboot the entire system:
```
--error_max=<digit>                : terminate hcxdumptool if error maximum reached
                                     default: 100 errors
--reboot                           : once hcxdumptool terminated, reboot system
```

varoudis, 2022-04-01:
@ZerBea just to cross-report here too. With me bare metal was ok (24h test) and VM was not! This was replicated to Kali simply joining a wifi AP (no monitor mode or something fancy..)

ZerBea, 2022-04-01:
@varoudis , thanks for that information, which is very much appreciated.
I'm going to collect some more reported problems here:
https://forums.kali.org/showthread.php?56804-AR9271-crashes-Linux

Also I'm still not sure if the issue is related to the firmware or the driver. 
In 2019 I noticed an issue on several drivers. At least we located the real cause, which was related to the xhci subsystem:
https://bugzilla.kernel.org/show_bug.cgi?id=202541

ZerBea, 2022-04-01:
@hklsb there is a big difference between airodump-ng and hcxdumptool.
airodump-ng does not transmit while hcxdumptool transmit several 802.11 frames like management frames and data frames, e.g.:
BEACON frames
AUTHENTICATION frames
PROBEREQUEST and PROBERESPONSE frames
ASSOCIATION and REASSOCIATION frames
ACTION frames
NULL frames
EAPOL frames
ACK frames

Doin this, hcxdumptool is acting as ACCESS POINT, CLIENT and DEAUTHENTICATOR at the same time. That can produce a huge workload on the driver, because 512 APs, 1024 ROGUE APs and 1024 CLIENTs can be handled simultaneously.   

BTW:
If the driver is broken, you can run aireplay-ng injectiontest to determine that the driver is working or not.
`--test              : tests injection and quality (-9)
`
No response means driver is dead.

hklsb, 2022-04-01:
Maybe my usb adapter is not powerful enough for hcxdumptool?
Can I limit the size of APs with hcxdumptool?
It would be nice if it's possible to work with the speed and power of the adapter used.

ZerBea, 2022-04-01:
No, the power of an adapter doesn't matter, but it is mandatory that the driver support full monitor mode, full packet injection and is flawless.
An external antenna connector is fine, because a good antenna is the best amplifier and transmission branch as well as on reception branch. This increases the range enormously (much more than a high power adapter).
Please take a look at this picture showing a minimal working hardware configuration:
https://github.com/ZerBea/hcxdumptool/wiki/Penetration-testing-system-1

ZerBea, 2022-04-02:
Here is an interesting comment about an ALFA AWUS036ACH (rtl8812au driver):
https://forum.aircrack-ng.org/index.php/topic,9087.msg16330.html#msg16330
I fully agree to MisterX.

buyukbang, 2022-05-06:
Hello,

Today I decided to test my TL-WN722N V1.1 for the very first time with hcxdumptool and I experienced same issue as #80. After reading that topic I was about to buy a new usb adapter, Then I noticed #209 and read all posts. Then downloaded today's daily kali iso (5.16.0-kali7-amd64) found here: https://archive.kali.org/kali-daily-images/ , intalled it to VMware. Now in this version of Kali TL-WN722N V1.1 works like a charm without any problem :)

ZerBea, 2022-05-06:
Thanks for this information. In most of the cases the problem is related to the driver.
A good idea is to watch the wireless mailing list:
https://patchwork.kernel.org/project/linux-wireless/list/
and bugzilla:
https://bugzilla.kernel.org/
and there are still some unfixed:
https://bugzilla.kernel.org/show_bug.cgi?id=215698
https://bugzilla.kernel.org/show_bug.cgi?id=42877

ZerBea, 2022-05-16:
Closed. We have to wait for a driver fix.

closed 2022-05-16 by ZerBea