ZerBea/hcxdumptool #212: PACKET_IGNORE_OUTGOING should be PACKET_OUTGOING

HascEasy, 2022-04-12:
https://github.com/ZerBea/hcxdumptool/blob/7fedbd0685046f75b9873af36c866bb3c35b75a6/hcxdumptool.c#L7308

On Ubuntu 18.04 PACKET_IGNORE_OUTGOING should be PACKET_OUTGOING

ZerBea, 2022-04-12:
No, because PACKET_IGNORE_OUTGOING and PACKET_OUTGOING have different values:
```
$ grep -r PACKET_IGNORE_OUTGOING
netpacket/packet.h:#define	PACKET_IGNORE_OUTGOING		23
linux/if_packet.h:#define PACKET_IGNORE_OUTGOING		23

$ grep -r PACKET_OUTGOING
netpacket/packet.h:#define PACKET_OUTGOING		4		/* Originated by us . */
linux/if_packet.h:#define PACKET_OUTGOING		4		/* Outgoing of any type */

```

ZerBea, 2022-04-12:
By this commit
https://github.com/ZerBea/hcxdumptool/commit/875986ae77465b56b06663eed1b63fc6706350b1
we do not set this socket options on old kernel versions.
Be warned that this increase CPU usage.

closed 2022-04-12 by ZerBea

ZerBea, 2022-04-12:
By the second commit
https://github.com/ZerBea/hcxdumptool/commit/0e4158f431ed1a7717d08b3eb23624d28bc15724
I removed PACKET_OUTGOING from bind, too.
No outgoing packets should be ignored on all versions.

ZerBea, 2022-04-13:
By this commit
https://github.com/ZerBea/hcxdumptool/commit/76fb696b597ceac4d0465ee8ea75dca1f76834f8
it should be clear that we are not distribution dependent.

ZerBea, 2022-04-13:
Commits tested on $ uname -r
Arch: 5.17.1-arch1-1
Debian bullseye: 5.15.32+
without issues.

ZerBea, 2022-04-14:
Is there any reason, why we should monitor outgoing packets, too on UBUNTU 18.04?
What kernel version is UBUNTU 18.04 running? $ uname -r
Also please comment the output of:
$ grep PACKET_OUTGOING -r /usr/include
$ grep PACKET_IGNORE_OUTGOING -r /usr/include