ZerBea/hcxdumptool #214: driver errors - kali 2022.1 - asus awus036ach

yv35404, 2022-04-22:
hi there
i'm quite new to hcxdumptool and am enjoying the heck out of it ;-)
i noticed, that while doing i.e.
sudo hcxdumptool -o test.pcapng -i wlan0 --enable_status=15 
that it will be showing lots of responses etc. at first. then, after like 8 - 9 minutes, this rate will go down drastically. it will still run, but with a much slower rate. whenerver i ctrl+c quit it, it will show 1 driver error encountered.
after i noticed this, i ran:
hcxdumptool -i wlan0 --check_injection with the following result:

initialization of hcxdumptool 6.2.5...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 5825/165 proberesponse 1057   
packet injection is working on 2.4GHz!
injection ratio: 100% (BEACON: 401 PROBERESPONSE: 1057)
your injection ratio is huge - say kids what time is it?
antenna ratio: 37% (NETWORK: 16 PROBERESPONSE: 6)
your antenna ratio is average, but there is still room for improvement

i also ran:
hcxdumptool -i wlan0 --check_driver with the following outcome:

initialization of hcxdumptool 6.2.5...
starting driver test...
driver use unsupported frequency format
expected frequency: 2412 - reported frequency: 241200000
expected exponent;  6    - reported exponent:  1
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

so i see there's something with the driver and the unsupported frequency format. i'm using the driver found on https://github.com/aircrack-ng/rtl8812au

i later noticed, that when running 
sudo hcxdumptool -i wlan0 --do_rcascan 
that it will quite after like 1 minute stating the max number of errors (100 as per default) is reached. 

i'm googling and googling but i can't figure out how and where to search for the problem so i figured maybe you could help me with some hints? :-)

i don't know what infos you need, so here goes:

kali is installed in dualboot (no vm) on an asus zenbook duo 14, details:

lsb_release -a    
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2022.1
Codename:       kali-rolling

hcxdumptool -v  
hcxdumptool 6.2.5 (C) 2021 ZeroBeat

yeah so let me know whatever you need to know in case you got a minute to help me :-)

have a great weekend, keep up your awesome work and cheers from switzerland
:-)                             

oh and btw: the alfa antenna doesn't get hot or anything (just i case one wants to know hehe)

ZerBea, 2022-04-22:
This issue was fixed by this commit:
https://github.com/ZerBea/hcxdumptool/commit/5ae77c75db194182dd23d86994742f8835b3993f
please use latest git head.
$ hcxdumptool -v
hcxdumptool 6.2.5-63-g0e4158f (C) 2021 ZeroBeat

yv35404, 2022-04-22:
thanks a ton for your quick answer! 
okay, now i feel stupid, how exactly do i update to the latest version? i'm really sorry, i'm still a bit unused to using git :-/

closed 2022-04-22 by yv35404

ZerBea, 2022-04-22:
No problem.
Your issue report reminded me to release v 6.2.6 - which is done some minutes ago.

yv35404, 2022-04-22:
yeah by the time i found out how to update it, you released it already haha
awesome, thanks a ton! have a great weekend

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.