ZerBea/hcxdumptool #216: --disable_deauthentication doesn't seem to work

mtnsec, 2022-05-18:
I need to run hcxdumptool without injecting deauth frames.  I'm using the --disable_deauthentication option, but devices around me are getting kicked off the wireless network, even with that option set.

Here are some of the things I've tried:

Version:  hcxdumptool 6.2.6-6-ga3fcc5c (C) 2022 ZeroBeat

This command causes my wireless devices to disconnect every minute or so (as long as the command is running)

`hcxdumptool -i wlan0 --enable_status 69 -o testing --disable_deauthentication`

If I use --essidlist=<file> my wireless devices still disconnect. Even if the network they're on is not in that list:

`hcxdumptool --disable_deauthentication -i wlan0 --enable_status 69 -o testing --essidlist=<file>`

Is there something wrong in my command line?  Thanks!

ZerBea, 2022-05-19:
There are more ways to disconnect a CLIENT than running stupid DEAUTHENTICATIONs.
A wireless dumper (kismet) is able to detect DEAUTHENTICATION attacks, so hcxdumptool has an option to deactivate this attack. You have disabled DEAUTHENTICATIONsand if you run Wireshark or tshark  in parallel you'll see that hcxdumptool doesn't transmit this kind of frames. The remaining DISASSOCIATION frames come from the APs and the CLIENTs.
If you are interested to read more:
https://hashcat.net/forum/thread-6661-post-55011.html#pid55011

BTW:
Stupid DEAUTHENTICATION attacks will not work if MFP (management frame protection) is enabled by an AP.
Just take a look at the RSN IE-TAG. You can find it in ASSOCIATION/REASSOCIATION REQUEST, EAPOL M2, PROBERESPONSE, BEACON frames:
```
Tag: RSN Information
    Tag Number: RSN Information (48)
    Tag length: 20
    RSN Version: 1
    Group Cipher Suite: 00:0f:ac (Ieee 802.11) AES (CCM)
    Pairwise Cipher Suite Count: 1
    Pairwise Cipher Suite List 00:0f:ac (Ieee 802.11) AES (CCM)
    Auth Key Management (AKM) Suite Count: 1
    Auth Key Management (AKM) List 00:0f:ac (Ieee 802.11) SAE (SHA256)
    RSN Capabilities: 0x00c0
        .... .... .... ...0 = RSN Pre-Auth capabilities: Transmitter does not support pre-authentication
        .... .... .... ..0. = RSN No Pairwise capabilities: Transmitter can support WEP default key 0 simultaneously with Pairwise key
        .... .... .... 00.. = RSN PTKSA Replay Counter capabilities: 1 replay counter per PTKSA/GTKSA/STAKeySA (0x0)
        .... .... ..00 .... = RSN GTKSA Replay Counter capabilities: 1 replay counter per PTKSA/GTKSA/STAKeySA (0x0)
        .... .... .1.. .... = Management Frame Protection Required: True
        .... .... 1... .... = Management Frame Protection Capable: True
        .... ...0 .... .... = Joint Multi-band RSNA: False
        .... ..0. .... .... = PeerKey Enabled: False
        ..0. .... .... .... = Extended Key ID for Individually Addressed Frames: Not supported
```

If you do not want that your devices are disconnect you have to use either the protection filter mode in combination with a  MAC list or (much better) the Berkeley Packet Filter as described in --help

In addition you can stop stop attack against APs or CLIENTs after x tries:
--stop_ap_attacks=<digit>          : stop attacks against ACCESS POINTs if <n> BEACONs received
--stop_client_m2_attacks=<digit>   : stop attacks against CLIENTS after 10 M2 frames received
                                     affected: ap-less (EAPOL 2/4 - M2) attack
                                     require hcxpcangtool --all option
                                     warning: that can prevent that a CLIENT can establish a connection to an assigned ACCESS POINT

BTW:
`--essidlist is an attack mode. Running it (or better in combination with --active_beacon) all CLIENTs will try to connect to hcxdumptool (and they will disconnect from their ACCESS POINT) it the target ESSID is inside this list..
`
Conclusion:
`Is there something wrong in my command line? Thanks!
`
The answer is yes. You're welcome.

Closed this because it is not an issue, but you can still ask your questions here.

closed 2022-05-19 by ZerBea

ZerBea, 2022-05-19:
To get an overall picture of what types of attacks are executed by hcxdumptool and how the target (ACCESS POINT and/or CLIENT act), I recommend to run Wireshark (or thark) in parallel (hcxdumptool is 100% compatible to Wireshark an tshark).
This is also very helpful to design a Berkeley Packet filter (hcxdumptool behavior can be controlled by BPF code).