ZerBea/hcxdumptool #217: System hangs

RaduNico, 2022-05-20:
Upon running the command `sudo ./hcxdumptool -i wlp0s20f3 --do_rcascan` the program runs for a a few seconds and then the syste completely hangs/becomes unresponsive (the mouse does not move). I've stopped all the interfering programs (wpa_supplicant and NetworkManager), but the issue persists.

The environment is the following:
```
$ uname -a 
Linux worky 5.4.0-110-generic #124-Ubuntu SMP Thu Apr 14 19:46:19 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
$ ./hcxdumptool -v
hcxdumptool 6.2.6-6-ga3fcc5c (C) 2022 ZeroBeat
```

I am not sure how could I further debug this issue, but I can provide any requested information.

ZerBea, 2022-05-20:
Usually this happens if the driver crashed. Potential driver candidates are iwlwifi and ath9k as well as rtw88, rtl8812au and rtl8814au (why I can't recommend them).

Please comment output of
```
$ hcxdumptool -I
$ sudo hcxdumptool -i wlp0s20f3 --check_driver
$ sudo hcxdumptool -i wlp0s20f3 --check_injection
```

A good candidate to figure out what happened is dmesg log:
`$ sudo dmesg
`
Output should look like this:
Here I connected the wlan-adapter
```
[  461.120472] mt7601u 5-1.1.1:1.0: ASIC revision: 76010001 MAC revision: 76010500
[  461.133504] mt7601u 5-1.1.1:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
[  461.571801] mt7601u 5-1.1.1:1.0: EEPROM ver:0d fae:00
[  461.572299] mt7601u 5-1.1.1:1.0: EEPROM country region 01 (channels 1-13)
[  461.805737] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[  461.806422] usbcore: registered new interface driver mt7601u
[  461.816139] mt7601u 5-1.1.1:1.0 wlp39s0f3u1u1u1: renamed from wlan0
```
Than running hcxdumptool:
`[  580.016333] device wlp39s0f3u1u1u1 entered promiscuous mode
`
After terminating hcxdumptool:
`[  585.808319] device wlp39s0f3u1u1u1 left promiscuous mode
`

ZerBea, 2022-05-20:
BTW:
Your kernel "Linux worky 5.4.0-110-generic" is a little bit outdated.
Longterm is on: 5.4.195 as of 2022-05-18
https://www.kernel.org/

RaduNico, 2022-05-20:
Here is the requested data:

```
$ sudo ./hcxdumptool -I
wlan interfaces:
phy0	40ec99a9061e	wlp0s20f3	(driver:iwlwifi)
$ sudo ./hcxdumptool  -i wlp0s20f3 --check_driver
initialization of hcxdumptool 6.2.6-6-ga3fcc5c (depending on the capabilities of the device, this may take some time)...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

The command `sudo hcxdumptool -i wlp0s20f3 --check_injection` results in the system hanging almost immediatlely (it does display some output).

I've started a rcascan and closed it before it the system freeze and it appears that the driver is indeen crashing. I'm attaching the entire dmesg output as a file.

```
$ sudo ./hcxdumptool -i wlp0s20f3 --do_rcascan
[...]
 BSSID        FREQ   CH RSSI BEACON RESPONSE ESSID  SCAN-FREQ: 2437 INJECTION-RATIO:   0% [15:11:25]
-----------------------------------------------------------------------------------------------------
 e067b379edd1 2432    5  -86      1        0 NEXGOEDD0-POS
^C
terminating...
3 driver errors encountered
ERRORs < 10 are related to a slow initialization and can be ignored
```

[dmesg.txt](https://github.com/ZerBea/hcxdumptool/files/8740834/dmesg.txt)

I am not sure if it is any help, but I've tried testing for injection using the aircrack suite (after conducting the above testing and a complete system reboot)  and it appears to be working:
```
$ sudo aireplay-ng --test wlp0s20f3mon
15:02:14  Trying broadcast probe requests...
15:02:16  No Answer...
15:02:16  Found 2 APs

15:02:16  Trying directed probe requests...
15:02:16  2C:4D:54:76:50:60 - channel: 11 - 'RORX'
15:02:21  Ping (min/avg/max): 26.168ms/26.168ms/26.168ms Power: -84.00
15:02:21   1/30:   3%

15:02:21  Injection is working!

15:02:22  18:31:BF:65:24:B8 - channel: 10 - 'OFFICE'
15:02:28   0/30:   0%
```

ZerBea, 2022-05-20:
Thanks for the information. The iwlwifi issue is know and reported to bugzilla here:
https://bugzilla.kernel.org/show_bug.cgi?id=214291
so we can close this issue and wait for an upstream fix.

I really love rt2800usb (Ralink) and mt76 (MediaTek (former Ralink) devices, because they are working out of the box and the drivers are part of the stock kernel. They are well maintained.

BTW:
You can't compare aireplay-ng with hcxdumptool because the "engine behind the scenes" is completely different.
While aireplay-ng and airodump-ng running NETLINK (libnl on virtual monitor interface wlp0s20f3mon), hcxdumptool doesn't do this.
Similar to aireplay-ng injection test, you can run hcxdumptool -i interface --check_injection.

closed 2022-05-20 by ZerBea

RaduNico, 2022-05-21:
Thanks for the quick and detailed response!

ZerBea, 2022-05-21:
Some useful links regarding iwlwifi driver:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi
It should support monitor mode, but no indications about packet injection in the documentation. 

Debugging the driver is explained here:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi/debugging

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.

CravateRouge, 2023-09-14:
@ZerBea , I still have the same issue with hcxdumptool_6.3.1-1_amd64.deb using iwlwifi

ZerBea, 2023-09-14:
Looks like nobody have the time (linux developer) to fix that
https://bugzilla.kernel.org/show_bug.cgi?id=214291#c36

or is interested in (provide information)
https://www.intel.com/content/www/us/en/support/articles/000058933/wireless/intel-wireless-ac-products.html

JavaScythe, 2024-06-26:
@ZerBea 

Had similar issue (no reported error but 0 packets captured) with iwlwifi
I have iwd instead of wpa_supplicant, but I was unable to capture anything with iwd stopped, so I left it on.

Built hcxlabtool and tried that and it (probably) worked.

10 ERROR(s) during runtime
4170 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm

hcxpcapngtool converted to mode22000 without complaining

ZerBea, 2024-06-26:
@JavaScythe 
hcxlabtool will be the successor of hcxdumptool. It is WiFi 7 ready (as well as hcxdumptool >= 6.3.0).
Basically the engine of both tools is the same, but hcxlabtool is less sensitive to driver errors. As a result, the result will not be as expected.
But anyway, I do not recommend to use an Intel chipset in combination with hcxdumptool/hcxlabtool.

Absolutely no comment about packet injection here:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi#about_the_monitorsniffer_mode

But many problems on other tools too, e.g.:
https://community.intel.com/t5/Wireless/Problem-injecting-unicast-packets-using-pcap-library/m-p/1560256

JavaScythe, 2024-06-26:
Thank you.
Not to hijack the thread, but PMKID and EAPOL pair capture for WPA cracking should only be passive, right, so I should be fine?

ZerBea, 2024-06-26:
You're welcome. There are better (passive) tools to capture wifi traffic than hcxdumptool (e.g. tshark or wireshark are made for this).

JavaScythe, 2024-06-27:
Better in what aspect? Times, performance?

ZerBea, 2024-06-27:
hcxdumptool/hcxlabtool sends 802.11 packets via a RAW packet socket to the driver and it sends nl80211 messages to the driver to control its behavior (e.g. channel switch).
Due to a bug inside, the driver can't handle this and it is becoming increasingly unstable.

There is a function in Makefile to retrieve more information.
```
# uncomment to enable DEBUG log
#DEFS		+= -DHCXDEBUG
```

If hcxdumptool/hcxlabtool is compiled with enabled DEBUG mode, you'll get a log file (hcxdumptool.log) that include detailed information which frames are not handled by the driver.