ZerBea/hcxdumptool #219: atheros ar9271 compatibility issue?

vertexgamer, 2022-05-30:
Using both the latest versions of kali for vmware (64bit) and raspberry (32bit), using the latest hcxdumptool release from github (at this time) and using an alfa network awus036nha, after a few minutes from the start of the capture, hcxdumptool stop receiving data and the desktop envirorment starts freezing. I did kill both the network manager and wpa supplicant services before starting the capture. 
I know the atheros is not on the recommended list, but i wanted to ask anyway, as it's strange that such a popular adapter could have compatibility issues.
I'm a beginner, so sorry if i'm missing the obvious

vertexgamer, 2022-05-30:
EDIT: I tested on a 3rd computer using the same exact vm, and it worked as intended. I have 0 idea of what could have caused the issue before, tomorrow i will check again

ZerBea, 2022-05-30:
Unfortunately such problems are mostly related to a VIRTUAL Machine (VM).
In case of running hcxdumptool run within a VM and the host take access to the device you run into this issue:
https://github.com/ZerBea/hcxdumptool/issues/196
That is the reason why I do not recommend running a VM.

Or the problems are related to the driver:
https://github.com/ZerBea/hcxdumptool/issues/209
https://github.com/ZerBea/hcxdumptool/issues/132

This kind of issues are known and reported to upstream:
https://bugzilla.kernel.org/show_bug.cgi?id=215918
https://bugzilla.kernel.org/show_bug.cgi?id=215698
https://bugzilla.kernel.org/show_bug.cgi?id=202601
That is the reason why I do not recommend this devices (or remove them from the known as working list).

In both cases, mentioned above, there is nothing I can to to work around this issues and we have to wait for a kernel fix.
To be well informed, it is a good idea to monitor wireless mailing list:
https://patchwork.kernel.org/project/linux-wireless/list/
and bugzilla:
https://bugzilla.kernel.org/

In most of the cases, dmesg will show you what's going on.
The device is connected to the system:
```
[ 1476.615670] usb 5-1.1.1: new high-speed USB device number 8 using xhci_hcd
[ 1476.826051] usb 5-1.1.1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[ 1476.826062] usb 5-1.1.1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 1476.826066] usb 5-1.1.1: Product: USB2.0 WLAN
[ 1476.826068] usb 5-1.1.1: Manufacturer: ATHEROS
[ 1476.826071] usb 5-1.1.1: SerialNumber: 12345
[ 1476.972232] usb 5-1.1.1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 1476.972306] usbcore: registered new interface driver ath9k_htc
[ 1477.268275] usb 5-1.1.1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[ 1477.518724] ath9k_htc 5-1.1.1:1.0: ath9k_htc: HTC initialized with 33 credits
[ 1477.756097] ath9k_htc 5-1.1.1:1.0: ath9k_htc: FW Version: 1.4
[ 1477.756104] ath9k_htc 5-1.1.1:1.0: FW RMW support: On
[ 1477.756107] ath: EEPROM regdomain: 0x809c
[ 1477.756109] ath: EEPROM indicates we should expect a country code
[ 1477.756111] ath: doing EEPROM country->regdmn map search
[ 1477.756112] ath: country maps to regdmn code: 0x52
[ 1477.756113] ath: Country alpha2 being used: CN
[ 1477.756115] ath: Regpair used: 0x52
[ 1477.760195] ieee80211 phy2: Atheros AR9271 Rev:1[ 1477.768964] ath9k_htc 5-1.1.1:1.0 wlp39s0f3u1u1u1: renamed from wlan0
```
After starting hcxdumptool, the device entered monitor mode:
`[ 1483.822694] device wlp39s0f3u1u1u1 entered promiscuous mode
`

After hcxdumptool terminated, the device left monitor mode:
`[ 1639.234775] device wlp39s0f3u1u1u1 left promiscuous mode
`

If the kernel noticed a driver  problem (unfortunately not all problems are detected by the kernel itself), it will be reported between entered and left promiscuous mode).

ZerBea, 2022-05-30:
BTW:
You issue report reminded me to test  the Atheros device after kernel upgrade to
```
$ uname -r
5.18.0-arch1-1
```
because the Atheros driver received many updates:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/diff/?id=v5.18&id2=v5.17&dt=2
search for "drivers/net/wireless/ath/ath"

ZerBea, 2022-05-30:
Looking good after half an hour testing:
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u1 --enable_status=95 --active_beacon
...
08:30:00 2442/1     ERROR:0 INCOMING:242 AGE:1 OUTGOING:407 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
08:45:00 2447/8     ERROR:0 INCOMING:4844 AGE:1 OUTGOING:17471 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
09:00:00 2442/7     ERROR:0 INCOMING:7528 AGE:1 OUTGOING:32128 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
```

0 errors (ERROR:0) reported by hcxdumptool, incoming count (INCOMING:7528) increase outgoing count (OUTGOING:32128) increase and last packet was received less than one second ago (AGE:1).
Both test target APs (PMKIDROGUE:2 ) and the test single CLIENT (M1M2ROGUE:1) are attacked successfully.
dmesg showing now problems.

vertexgamer, 2022-05-30:
Hello, first of all thanks for your time, secondly today i noticed that my rpi3 was strangely constantly power limited during the capture, considering that my setup didn't change during multiple runs except for the usb port used to connect the wifi adapter, i believe it's an hardware related issue.
P.S. in case it wasn't a power issue, would you suggest me to try arch for arm? Or a guide to update the current drivers?

EDIT: nvm, still failed after 1h, this time giving "driver is busy: failed to transmit internal beacon".

ZerBea, 2022-05-31:
Power supply and power consumption are the key points. My Raspberry Pi models (A, B, Zero) are very sensitive to voltage fluctuations. It took me awhile to get a good power supply. The USB ports are limited, too:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#typical-power-requirements

I had to modify boot config.txt (head less operation mode)  this way:
```
# See /boot/overlays/README for all available options

gpu_mem=64
initramfs initramfs-linux.img followkernel
max_usb_current=1
hdmi_blanking=1
dtoverlay=pi3-disable-bt
dtoverlay=pi3-disable-wifi
dtparam=audio=off
dtparam=watchdog=on
disable_splash=1
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
```

My first choice is always Arch Linux (desktop, notebook and the RPI's). If you're firm with Linux, I recommend to use it.
Unfortunately Arch Linux Arm dropped support for the old RPI's ( < model 2) in 2022 and I have to use Raspbian testing (removed all unnecessary stuff, services, GUI, tools), instead.
https://archlinuxarm.org/about

If you get "driver is busy", the driver is dead.

evrial, 2022-06-07:
Confirming the same shit. Freeze in random period average 5-10 min and then termination doesn't work, when uplug the adapter then it terminates. And honestly stable work only with Ralink chips defeats the purpose of having such tool.

```
uname -r
5.16.0-kali7-amd64
```
check_driver
```
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

dmesg
[dmesg.txt](https://github.com/ZerBea/hcxdumptool/files/8854782/dmesg.txt)

evrial, 2022-06-07:
UPD
Super weird stuff, did a run for 1 hour without freeze, only change is no write to file. Even more weird is captured packets from my AP running on 5 GHz Mode: 802.11ac only
`sudo hcxdumptool -i wlan1 --enable_status=95 --active_beacon`

ZerBea, 2022-06-07:
These experiences  led me to add a warning:
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u1 --check_driver
initialization of hcxdumptool 6.2.6-34-g1a4c150 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: ath9k_htc (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```
https://bugzilla.kernel.org/show_bug.cgi?id=202601
https://bugzilla.kernel.org/show_bug.cgi?id=199709
https://bugzilla.kernel.org/show_bug.cgi?id=205813
https://bugzilla.kernel.org/show_bug.cgi?id=188201
https://bugzilla.kernel.org/show_bug.cgi?id=215698

As of Today: Only MediaTek and Ralink drivers working out of the box without problems..

ZerBea, 2022-06-07:
Also freezes and crashes on aircrack-ng suite in combination with this driver:
https://forums.kali.org/showthread.php?56804-AR9271-crashes-Linux

ZerBea, 2022-06-08:
I' testing the changes of 
$ uname -r
5.18.1-arch1-1
and
$ lsusb
Bus 005 Device 006: ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n
That is an old TP-TL-WN722N version 1

Running the test up to one hour, everything is looking fine.

```
$ sudo hcxdumptool -i wlp39s0f3u1u1u1 --enable_status=15 -c 1
initialization of hcxdumptool 6.2.6-34-g1a4c150 (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy2
INTERFACE NAME............: wlp39s0f3u1u1u1
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc (this driver is not recommended - expect driver errors)
DRIVER VERSION............: 5.18.1-arch1-1
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 0c81120dcc29 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 0c81120dcc2a (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 0c81120dcc2b (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: c8aacc610210
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 61902
ANONCE....................: 26adf186266281d44eabee0fa5c608faf1abcf9b2ab795845650b9ae1aba6ad0
SNONCE....................: 7d7c263877c197d84a6b8ae93ada742fa96332767e2b8ce4af45df5f76a1b55d

08:02:00 2412/1     ERROR:0 INCOMING:107 AGE:1 OUTGOING:27 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4
...
08:03:00 2412/1     ERROR:0 INCOMING:4895 AGE:1 OUTGOING:328 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
^C
terminating...
```
The first test AP was attacked immediately, the second one less than 2 minutes later.
hcxdumptool terminated without ERRORs.

ZerBea, 2022-06-08:
Kernel  5.15.34-1.0-lts seems to work,too. Same interface, Arch Linux 32 bit system, older hcxdumptool, test AP was attacked successful:
```
$ sudo hcxdumptool -i wlp0s29f7u3 --enable_status=64 -c 1
initialization of hcxdumptool 6.2.5...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlp0s29f7u3
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: f81a67077d0e (not used for the attack)
INTERFACE VIRTUAL MAC.....: f81a67077d0e (not used for the attack)
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.15.34-1.0-lts
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: a468bcf0cca6 (BROADCAST HIDDEN used for the attack)
ACCESS POINT (ROGUE)......: a468bcf0cca7 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: a468bcf0cca8 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: dc70144c421b
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62374
ANONCE....................: 496518797626e08b41d73af08ea75c66c346432b95ec4af9969d426bbbc2c0bd
SNONCE....................: 16a0f6ced5b3f7831e13711b2ef12e25cbdbbfb951f5f80b51d00ed09219a884

08:23:00 2412/1   ERROR:0 INCOMING:205 AGE:2 OUTGOING:257 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
08:34:00 2412/1   ERROR:0 INCOMING:6255 AGE:1 OUTGOING:6376 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
```

evrial, 2022-06-08:
Yea. Confirming Ralink 3070L and RTL8812au are stable, though success with PMKID only with Ralink, but RSSI is weaker.

ZerBea, 2022-06-08:
Unfortunately rtl8812au depend on NL80211 (NETLINK)  and the initialization must be done by NETLINK (iw INTERFACE set type monitor). 
ioctl() calls (CFG80211) to set monitor are not accepted by this driver.

evrial, 2022-06-08:
But if check_driver has success, that mean CFG80211 works and PMKID packets?

evrial, 2022-06-10:
Ok excuse for off-topic
Just confirmed self-attack using rogue AP on wlan1 (tp-link ar9271 2.4 GHz) to attack client wlan0 (Apple's BCM4360) connected on 5.8GHz to my home AP running single band 802.11ac. So stupid, any advice to fix that?

ZerBea, 2022-06-10:
Using a strong PSK (63 random characters) or WPA3 is a good idea. It doesn't prevent to be attacked, but it make life much harder for an attacker.

evrial, 2022-06-10:
So this case was documented that 5GHz AC client and AP are vulnerable to cheap 2 GHz stick?

evrial, 2022-06-10:
Man I got 5GHz AP in 2014 and thought frequency makes in safer.

ZerBea, 2022-06-11:
That is history since I published hcxdumptool/hcxtools.
https://hashcat.net/forum/thread-6661-post-55540.html#pid55540

evrial, 2022-06-11:
Thanks, I'll spend some time learning. But I was sure that must be a radio shutdown for 2.4GHz for clients, all I care is 5GHz, I run AP on lowest power and radiowaves don't radiate as far as 2.4

ZerBea, 2022-06-11:
One of my test target devices is an iPad. It is funny how fast hcxdumptool is able to "fool" it.

evrial, 2022-06-11:
Bonkers, you're a genius.

ZerBea, 2022-12-23:
Looks like the latest driver (kernel >= 5.15 is now working.

closed 2022-12-23 by ZerBea

vertexgamer, 2022-12-24:
> Looks like the latest driver (kernel >= 5.15 is now working.

yeeeeee