ZerBea/hcxdumptool #220: Option to not do channel management

PlasmaHH, 2022-05-31:
Sometimes its convenient to run the tool alongside other tools to passively capture with --silent and then it would be useful also not do the channel management (and not complain about wrong expected channel)

ZerBea, 2022-05-31:
Added this feature by commit:
https://github.com/ZerBea/hcxdumptool/commit/744c8fee5f043bc93e56cc34a291d08fc13d558f

example:
$ sudo hcxdumptool -i wlp39s0f3u1u1u2 --enable_status=63 --passive

Channel management is completely disabled and the initial channel must be set by a third party tool.
Please test.

closed 2022-05-31 by PlasmaHH

PlasmaHH, 2022-05-31:
Ah, the script thing makes sense. It is just almost impossible to read as a human ;) With my formatting I could for example see easily that a packet was the first ~60something bytes of a HTTP response... 

I am using the iwlwifi driver and I have lots of problems, tons of errors in dmesg when I run hcxdumptool in any way that causes it to send. It breaks the device sometimes so hard I have to unload and reload (thus basically hard resetting the wifi chip) the module. I still want to test if it has the same problems with aireplay doing deassociation floods....

The broken packets I receive are usually at the beginning, often so much that the default error limit of 100 is not enough. They seem to be a mix of different packets. Sometimes they seem to be valid packets, just the radiotap header is missing. Sometimes they are of that kind but cut short ( I haven't checked if the tool does something to request only packets of that kind) and sometimes they are like 16kb or so in size, with a valid wifi packet without the radiotap header at the beinning and the rest filled up with the same 64bit value repeating all the time... 

and then there are packets that seem to be with a valid radiotap header, but it_present is set to 0 ( haven't looked how much sense other values would make as a header yet fully, but the number of bytes are there )... these kinds of packets arrive also sometimes in between, being the main cause of annoyance for longer sessions as they will sooner or later reach any error limit....

if I have the time I plan to look into the packets I receive, maybe one can have some (optional) heuristics that tries to make sense... like figuring out if there are any valid bits/bytes for interpreting the packet as a radiotap less one, or maybe even checking if where a mac would be is an already seen one, and based on that not increase the error count or even make use of that packet...

ZerBea, 2022-06-01:
This driver is one of the "problem drivers" and the issues are well known, e.g.:
https://github.com/ZerBea/hcxdumptool/issues/217
and reported to bugzilla, e.g.:
https://bugzilla.kernel.org/show_bug.cgi?id=214291
but the priority to fix them are low:
https://bugzilla.kernel.org/show_bug.cgi?id=214291#c11

There is an option to increase the ERROR level, you can use:
```
--error_max=<digit>                : terminate hcxdumptool if error maximum reached
                                     default: 100 errors
```

Additional heuristics is not useful because every additional check will cost us time. hcxdumptool interact with the targets (ACCESS POINTs as well as CLIENTs). If we have received a packet, we must (mandatory) respond in time (that also depend on a free transmission slot on the channel which is done by the firmware). If we do not respond right in time the the target doesn't respond.

Please note that you can't compare the behavior of aireplay-ng with the behavior of hcdumptool. The entire aircrack-ng suite make use of NETLINK, while hcxdumptool doesn't.
I really have good reasons not to use the entire NETLINK stuff, coming up with an additional library (libnl) as well as the entire libpcap stuff.
A good description is here):
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls?share=1

If you like to compare the behavior, you must (mandatory) compile aircrack-ng without NETLINK dependency (--disable-libnl) as described here:
https://github.com/aircrack-ng/aircrack-ng/blob/master/INSTALLING#L13
And you have to use "ifconfig/iwconfig" instead of "ip link/iw" (due to its NETLINK dependency) to control the device.

BTW:
hcxdumptool/hcxlabtool is running as expected on the recommended system (RPI) and the recommended chipsets/drivers (Ralink/MEdiaTek). Everything we add to work around a buggy driver will(!) slow us down.
To finally fix these errors, I recommend to collect them and report it to bugzilla. That is much better than just fighting the symptoms by adding additional heuristics.

ZerBea, 2022-06-01:
To improve hcxlabtool/hcxdumptool, mostly I use this cheap adapter: ALLNET ALLWA0150 (mt7601):
https://www.reichelt.de/de/en/allnet-wireless-nano-usb-adapter-150-mbit-s-allnet-allwa0150-p149756.html?GROUPID=5839&START=0&OFFSET=16&SID=93b9cd6b8d4008444af143e4f205b9a3437e9e414e51fa6d55c5d&LANGUAGE=EN&&r=1

I'll say that there is absolutely no need to buy/use and expensive high power device. 802.11 is packet oriented. If we receive the target and the target receive one of our (rogue) packets, it is more than enough.

The drivers of the recommended chipsets are well maintained. I know both maintainers
Lorenzo (mt76)
https://github.com/openwrt/mt76
https://bugzilla.kernel.org/show_bug.cgi?id=205305
and Stanislaw (rt2800usb).
https://bugzilla.kernel.org/show_bug.cgi?id=202241

We can be absolutely sure that an error on this drivers will be fixed as soon as possible.
This is the major reason, why I recommend to use MediaTek and Ralink chipsets.

ZerBea, 2022-06-01:
Exactly for this purpose is the Berkeley Packet filter. If discover a pattern, you can filter this packet out, before they reach hcxdumptool. That is extreme fast.

ZerBea, 2022-06-01:
Added your suggestion by latest commit:
https://github.com/ZerBea/hcxdumptool/commits/master

PlasmaHH, 2022-06-01:
Might give bpf a try, could avoid it mostly so far ;)

For completeness sake because a coworker asked me why I suggested that change:

https://godbolt.org/z/4q7z8KnMr ( you can also select the arm compiler for the RPi there )

ZerBea, 2022-06-01:
What do you mean here:
"With my formatting I could for example see easily that a packet was the first ~60something bytes of a HTTP response..."
https://github.com/ZerBea/hcxdumptool/issues/220#issuecomment-1142540657

For sure, hcxdumptool is able to record open WiFi IP (v4 and v6) traffic:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5962
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5972

Depending on the option this frames are stored to the pcapng file:
```
-f <frames>    : frames to save
                 bitmask:
                  0: clear default values
                  1: MANAGEMENT frames (default)
                  2: EAP and EAPOL frames (default)
                  4: IPV4 frames
                  8: IPV6 frames
                 16: WEP encrypted frames
                 32: WPA encrypted frames
                 64: vendor defined frames (AWDL)
```
Bitmask 4 and 8 include eap md5 challenge, eapleap and tacacs-plus radius authentications which can be converted by hcxpcapngtool to hashcat format:
```
--eapmd5=<file>                    : output EAP MD5 CHALLENGE (hashcat -m 4800)
--eapmd5-john=<file>               : output EAP MD5 CHALLENGE (john chap)
--eapleap=<file>                   : output EAP LEAP and MSCHAPV2 CHALLENGE (hashcat -m 5500, john netntlm)
--tacacs-plus=<file>               : output TACACS PLUS (hashcat -m 16100, john tacacs-plus)
```
 
If this frames contain a valid radiotap header, a valid mac header and IPv4 or IPv6 payload hcxdumptool will not increase its error counter.

Examples are here:
https://github.com/joswr1ght/asleap/tree/master/samples
Use hcxpcapngtool to convert and hashcat to recover the PSK.

ZerBea, 2022-06-02:
You can always check if hcxdumptool is working as expected.

load 802.11 hwsim module:
`$ sudo modprobe mac80211_hwsim
`
create hostapd. config and the following lines:
```
interface=wlan1
driver=nl80211
hw_mode=g
channel=1
ssid=mac80211 test
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
wpa_passphrase=12345678
```
open terminal 1 and run hcxdumptool to get driver information:
```
$ hcxdumptool -I
wlan interfaces:
phy0	020000000000	wlan0	(driver:mac80211_hwsim)
phy1	020000000100	wlan1	(driver:mac80211_hwsim)
```

test driver:
```
$ sudo hcxdumptool -i wlan0 --check_driver
initialization of hcxdumptool 6.2.6-20-g255d638 (depending on the capabilities of the device, this may take some time)...
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
detected driver: mac80211_hwsim (hardware simulation)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

open terminal 2 and run hostapd on interface 2:
`$ sudo hostapd hostapd.conf
`

run hcxdumptool to attack hostapd:
```
$ sudo hcxdumptool -i wlan0 --enable_status=95 -c 1
initialization of hcxdumptool 6.2.6-21-g135a7ca (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 020000000000 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 020000000000 (not used for the attack)
DRIVER....................: mac80211_hwsim (hardware simulation)
DRIVER VERSION............: 5.18.1-arch1-1
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 00221c0283b5 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 00221c0283b6 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 00221c0283b7 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b025aa923929
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64774
ANONCE....................: f34f03abaa4bb2e4ccdaf2255e18b93e4a87c7144976b10976ae3a608ab879f1
SNONCE....................: c959ba479603d28c9b2f57f736a000ad8d4d33725f7cda7f63e8d39879c3f6c3

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
12:41:56 2412/1   ffffffffffff 020000000100 mac80211 test [BEACON]
12:42:00 2412/1     ERROR:0 INCOMING:39 AGE:1 OUTGOING:6 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
12:43:00 2412/1     ERROR:0 INCOMING:653 AGE:1 OUTGOING:48 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
```

You'll notice that INCOMING count increase, OUTGOING count increase (until hcxdumptool noticed that the attack is useless and stop it), age of last received packet is mostly 1 second and ERROR count will not increase.

That is the reference that hcxdumptool is really working as expected.

PlasmaHH, 2022-06-03:
By that comment I just meant htat reading the hex values takes some time, and by adding that text output one could easily see the HTTP header and from there go back and see that its a tcp packet with an ethernet frame before it...

hcxdumptool itself is working fine, its just the driver or maybe even the firmware making problems, but sometimes you have to deal with what you are given and can't get access to other hardware easily, so I am trying to make the best out of that ;)

ZerBea, 2022-06-03:
Ok, thanks for the explanation. Now it's clear.

I played around with that driver and noticed that every Intel chipset has a different behavior. That will make it hard to find a solution. Last but not least I gave up and recommend MediaTek and Ralink chipsets only,

BTW:
To monitor IP traffic I use tshark. All hcxxxxx tools are 100% compatible to tshark and Wireshark. That include the capture file format (default pcapng, compressd by gz or uncompressed) as well as the display filters and capture filtes.
```
$ tshark -r ipv4.cap -O http -x

Frame 7: 326 bytes on wire (2608 bits), 326 bytes captured (2608 bits)
IEEE 802.11 Data, Flags: ......F.
Logical-Link Control
Internet Protocol Version 4, Src: 192.168.0.1, Dst: 239.255.255.250
User Datagram Protocol, Src Port: 1900, Dst Port: 1900
Simple Service Discovery Protocol

0000  08 02 00 00 ff ff ff ff ff ff 00 11 95 39 06 ef   .............9..
0010  00 11 95 39 06 ef 30 c0 aa aa 03 00 00 00 08 00   ...9..0.........
0020  45 00 01 26 9f 14 00 00 7f 11 eb 0e c0 a8 00 01   E..&............
0030  ef ff ff fa 07 6c 07 6c 01 12 52 5f 4e 4f 54 49   .....l.l..R_NOTI
0040  46 59 20 2a 20 48 54 54 50 2f 31 2e 31 0d 0a 48   FY * HTTP/1.1..H
0050  4f 53 54 3a 32 33 39 2e 32 35 35 2e 32 35 35 2e   OST:239.255.255.
0060  32 35 30 3a 31 39 30 30 0d 0a 43 41 43 48 45 2d   250:1900..CACHE-
0070  43 4f 4e 54 52 4f 4c 3a 6d 61 78 2d 61 67 65 3d   CONTROL:max-age=
0080  31 32 30 0d 0a 4c 4f 43 41 54 49 4f 4e 3a 68 74   120..LOCATION:ht
0090  74 70 3a 2f 2f 31 39 32 2e 31 36 38 2e 30 2e 31   tp://192.168.0.1
00a0  3a 35 36 37 38 2f 69 67 64 2e 78 6d 6c 0d 0a 4e   :5678/igd.xml..N
00b0  54 3a 75 75 69 64 3a 75 70 6e 70 2d 57 41 4e 43   T:uuid:upnp-WANC
00c0  6f 6e 6e 65 63 74 69 6f 6e 44 65 76 69 63 65 2d   onnectionDevice-
00d0  31 5f 30 2d 31 32 33 34 35 36 37 38 39 30 30 30   1_0-123456789000
00e0  30 31 0d 0a 4e 54 53 3a 73 73 64 70 3a 61 6c 69   01..NTS:ssdp:ali
00f0  76 65 0d 0a 53 45 52 56 45 52 3a 45 6d 62 65 64   ve..SERVER:Embed
0100  64 65 64 20 55 50 6e 50 2f 31 2e 30 0d 0a 55 53   ded UPnP/1.0..US
0110  4e 3a 75 75 69 64 3a 75 70 6e 70 2d 57 41 4e 43   N:uuid:upnp-WANC
0120  6f 6e 6e 65 63 74 69 6f 6e 44 65 76 69 63 65 2d   onnectionDevice-
0130  31 5f 30 2d 31 32 33 34 35 36 37 38 39 30 30 30   1_0-123456789000
0140  30 31 0d 0a 0d 0a                                 01....
```

PlasmaHH, 2022-06-03:
I was at that point intrested in the packets that did cause the error counters to increase, and as far as I could tell they were not written to the pcap file

ZerBea, 2022-06-03:
This ERRORs ar not caused by a (damaged) packet. It is caused inside the firmware or inside the driver or both.