ZerBea/hcxdumptool #221: hcxdumptool does not work well wit TP-Link AC600 High Gain Wifi USB Adapter (Archer T2U Plus)

albe-jj, 2022-06-03:
Hi ZerBea, thank you for the useful repository!

I'm facing issues when running hcxudumptool with the Archer T2U Plus adapter (on a kali virtual machine).
Essentially if I run `$sudo airmon-ng start wlan0 & airmon-ng stop wlan0`   then I can get to hcxudumptool to somewhat work (see below).

```
$ sudo hcxdumptool -i wlan0 --check_driver
initialization of hcxdumptool 6.2.6-22-ga02a399 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: rtl88XXau (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

$ sudo hcxdumptool -i wlan0 --check_injection
12:52:02     ERROR: 86 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 2452, DETECTED: 245200000]
stage 1 of 2 probing frequency 2455/9 proberesponse 444   12:52:03     ERROR: 87 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 2452, DETECTED: 245200000]
12:52:04     ERROR: 88 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 2452, DETECTED: 245200000]
stage 1 of 2 probing frequency 2456/9 proberesponse 454   12:52:05     ERROR: 89 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 2452, DETECTED: 245200000]
^C
packet injection is working on 2.4GHz!
injection ratio: 22% (BEACON: 2061 PROBERESPONSE: 463)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 52% (NETWORK: 51 PROBERESPONSE: 27)
your antenna ratio is good
89 driver errors encountered during the test`
```

If I plug in the adapter and directly run hcxdumptool then I run into driver error:
```
$ sudo hcxdumptool -i wlan0 --check_driver                                          
initialization of hcxdumptool 6.2.6-22-ga02a399 (depending on the capabilities of the device, this may take some time)...
starting driver test...
failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver: Operation not permitted
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
failed to restore old SIOCSIWMODE: Operation not permitted

```

Some additional info:
```
$ uname -r  
5.16.0-kali7-amd64

$ hcxdumptool -I
wlan interfaces:
phy0    984827c7c1a8    wlan0   (driver:rtl88XXau)

$ sudo ip link set wlan0 down
$ sudo iw dev wlan0 set type monitor
$ sudo ip link set wlan0 up

$ sudo iw dev wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 98:48:27:c7:c1:a8
        type monitor
        wiphy 0
        channel 7 (2442 MHz), width: 20 MHz, center1: 2442 MHz
        txpower 20.00 dBm

$ lsusb
Bus 001 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0e0f:0008 VMware, Inc. Virtual Bluetooth Adapter
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

```

ZerBea, 2022-06-03:
Exactly this is the reason why a added this message
"detected driver: rtl88XXau (this driver is not recommended - expect driver errors)"

From your comment
`12:52:02     ERROR: 86 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 2452, DETECTED: 245200000]
`
I assume that driver report the frequency a little bit different to the rt2800 and mt76 drivers.
By this commit I tried a workaround:
https://github.com/ZerBea/hcxdumptool/commits/master
Please try doing the following procedure:
do a git update to leatest head and compile
than run this commands and comment output
```
$ sudo ip link set wlan0 down
$ sudo iw dev wlan0 set type monitor
$ sudo ip link set wlan0 up
$ sudo hcxdumptool -i wlan0 --check_driver
$ sudo hcxdumptool -i wlan0 --check_injection
```
Maybe we have to adjust the frequency handling again a little bit.
Do not use airmon-ng before!

ZerBea, 2022-06-03:
This driver depend on NETLINK. Activating monitor mode by airmon-ng which internally run iw) will do this by NETLINK messages. As a result we got a NETLINK interface (AF_NETLIN) while hcxdumptool use a raw packet socket (PF_PACKET)
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7277
From now on, we will run into problems.

albe-jj, 2022-06-03:
Thanks for the reply. Seems your commit fixed the error. Here the results.
```
$ sudo hcxdumptool -i wlan0 --check_driver                                              
initialization of hcxdumptool 6.2.6-24-g76216e1 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: rtl88XXau (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
                                                                                                                                  
$ sudo hcxdumptool -i wlan0 --check_injection                                           
initialization of hcxdumptool 6.2.6-24-g76216e1 (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 1 of 2 probing frequency 5248/49 proberesponse 1512   ^C
packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 34% (BEACON: 4336 PROBERESPONSE: 1512)
your injection ratio is average, but there is still room for improvement
antenna ratio: 97% (NETWORK: 40 PROBERESPONSE: 39)
your antenna ratio is huge - say kids what time is it?

terminating...
```

The frequency sweep is quite slow, takes like 2 second per MHz. So I aborted the --check-injection first run after 10 minutes.

I tried also to let it run for like 30 minutes and when it reached somewhere around 5.8GHz started giving error below. Maybe reached the adapter frequency limit?

```
$ sudo hcxdumptool -i wlan0 --check_injection                                           
initialization of hcxdumptool 6.2.6-24-g76216e1 (depending on the capabilities of the device, this may take some time)...
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
detected driver: rtl88XXau (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
initialization of hcxdumptool 6.2.6-24-g76216e1 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 1 of 2 probing frequency 5890/178 proberesponse 1405   09:20:23     ERROR: 1 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 0, DETECTED: 0]
09:20:24     ERROR: 2 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 0, DETECTED: 0]
stage 1 of 2 probing frequency 5891/178 proberesponse 1405   09:20:25     ERROR: 3 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 0, DETECTED: 0]
09:20:26     ERROR: 4 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 0, DETECTED: 0]
stage 1 of 2 probing frequency 5892/178 proberesponse 1405   09:20:27     ERROR: 5 [INTERFACE IS NOT ON EXPECTED CHANNEL, EXPECTED: 0, DETECTED: 0]
....

packet injection is working on 2.4GHz!
packet injection is working on 5GHz!
injection ratio: 32% (BEACON: 4362 PROBERESPONSE: 1405)
your injection ratio is average, but there is still room for improvement
antenna ratio: 70% (NETWORK: 84 PROBERESPONSE: 59)
your antenna ratio is good
19 driver errors encountered during the test

terminating...
19 driver errors encountered

```

Also what is the antenna ratio? and why is it so different in the two runs?

albe-jj, 2022-06-03:
Thanks for the explanation and for making the workarounds! Works well also with latest commit.
```

$ sudo systemctl stop NetworkManager.service; sudo systemctl stop wpa_supplicant.service
sudo ip link set wlan0 down
sudo iw dev wlan0 set type monitor
sudo ip link set wlan0 up
sudo hcxdumptool -i wlan0 --check_driver
sudo hcxdumptool -i wlan0 --check_injection
initialization of hcxdumptool 6.2.6-25-g60b1bb7 (depending on the capabilities of the device, this may take some time)...
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
detected driver: rtl88XXau (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
initialization of hcxdumptool 6.2.6-25-g60b1bb7 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
starting antenna test and packet injection test (that can take up to two minutes)...
stage 1 of 2 probing frequency 2481/13 proberesponse 874   ^C
packet injection is working on 2.4GHz!
injection ratio: 22% (BEACON: 3939 PROBERESPONSE: 888)
your injection ratio is poor - improve your equipment and/or get closer to the target
antenna ratio: 34% (NETWORK: 61 PROBERESPONSE: 21)
your antenna ratio is average, but there is still room for improvement

terminating...

```

ZerBea, 2022-06-03:
Fine. Than we can close this issue report.

closed 2022-06-03 by ZerBea

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.