ZerBea/hcxdumptool #223: failed to detect wlan interface, driver does not support CFG80211?

lexittr, 2022-06-24:
I am running bare metal kali (installed yesterday). I already ran apt update and upgrade. 
Before running this command i ran:
`$ sudo systemctl stop NetworkManager.service`
`$ sudo systemctl stop wpa_supplicant.service `
OUTPUT:
$ sudo hcxdumptool -i interface -o dumpfile.pcapng --active_beacon --enable_status=15
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
failed to detect wlan interface: interface possible not comected
driver doesn't support CFG80211
kernel possible complied without Wireless Extensions (expected: CONFIG_CFG80211_WEXT=y and CONFIG_CFG80211_WEXT_EXPORT=y): No such device
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device

installed it using:
`sudo apt install hcxdumptool` and went through the github installation (make)
ifconfig:
`$ ifconfig 
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e8:6a:64:13:b6:91  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 914  bytes 83340 (81.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 914  bytes 83340 (81.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.86.73  netmask 255.255.255.0  broadcast 192.168.86.255
        inet6 fe80::897d:6d65:85c7:d8d8  prefixlen 64  scopeid 0x20<link>
        ether 80:2b:f9:49:4c:1b  txqueuelen 1000  (Ethernet)
        RX packets 1251169  bytes 1870593777 (1.7 GiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 163441  bytes 18152771 (17.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`

ZerBea, 2022-06-24:
Your command line is wrong- exactly your interface name: "interface"
`$ sudo hcxdumptool -i interface -o dumpfile.pcapng --active_beacon --enable_status=15
`and hcxdumptool told you this:
`failed to get interface information: No such device
`
From hcxdumptool --help:
```
-i <interface> : interface (monitor mode will be enabled by hcxdumptool)
                 it is mandatory that the driver support ioctl() system calls, monitor mode and full packet injection!
```
where <interface> is the name of our interface to be used for the attack.

This command will give you an overview about the interfaces connected to your system:
`$ hcxdumptool -I`

example:
```
$ hcxdumptool -I
wlan interfaces:
phy0	70665578cdab	wlp4s0	(driver:rtw_8821ce)
```
You have to use the correct interface name:
```
$ sudo hcxdumptool -i wlp4s0 -o dumpfile.pcapng --active_beacon --enable_status=15
initialization of hcxdumptool 6.2.6-34-g1a4c150 (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlp4s0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 30 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 70665578cdab (not used for the attack)
INTERFACE VIRTUAL MAC.....: 70665578cdab (not used for the attack)
DRIVER....................: rtw_8821ce (this driver is not recommended - expect driver errors)
DRIVER VERSION............: 5.18.6-arch1-1
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 005047b53519 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 005047b5351a (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 005047b5351b (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: a4a6a9dca4a7
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 61690
ANONCE....................: e08678ded6f744c6442bd04c1d5ee71fd4914f8debdc0e28a2d981adfc723dad
SNONCE....................: a349aabf7a3688787f882e3957d3b1c8254d702685afa3d15374a51e8b649192

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
^C
terminating...
```

Closed, because it is not an issue, but you can still ask your questions here.

BTW:
Make sure you're running a recommended interface (devices with Intel are not recommended due to several driver issues,  or Broadcom chipsets do not support monitor mode).
More information in README.md.

closed 2022-06-24 by ZerBea

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.