ZerBea/hcxdumptool #224: Possible regression with commit 0b9aaad

rgerganov, 2022-07-15:
I am running `hcxdumptool` on RaspberryPi Zero 2 with Alfa AWUS036ACM. I built the tool from the latest source (commit 20031ae) and when I run it like this:

```
$ sudo hcxdumptool -i wlan1 -o all.pcapng --enable_status=31
```
once in a while I am getting these errors:
```
14:19:44 2427/4 driver is busy/broken: failed to transmit associationrequest
14:19:44 2427/4 driver is busy/broken: failed to transmit associationrequest
14:19:45 2427/4 driver is busy/broken: failed to transmit associationrequest
14:19:45 2427/4 driver is busy/broken: failed to transmit associationrequest
```

The tool appears to be working but I was curious to find out what is causing these errors and rolled back to version 6.2.6 (commit 2b711d6). With this version I didn't get any errors for very long period of time. So I did a binary search and I found these errors start to appear with commit 0b9aaad. Unfortunately I don't understand the code enough to troubleshoot further.

Are these errors false positive or indicate an actual problem?

This is the kernel version:
```
$ uname -a
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
```

If you need any other info, I'd be happy to provide it.

ZerBea, 2022-07-15:
This ERROR message
"14:19:44 2427/4 driver is busy/broken: failed to transmit associationrequest"
means that the driver doesn't accept packets sent via the raw socket by hcxdumptool any longer, because it is possilbe broken (too high tx power, over heating problems or instable power supply).

The commit you mentioned
https://github.com/ZerBea/hcxdumptool/commit/0b9aaad4456660a7b861372d243a90544353d050
added active monitor mode on transmission branch.That means if a packet transmitted by hcxdumptool was not acknowledged by the target, it will be transmitted again up to 7 times.
That is working fine on an ALFA AWUS036CM:
```
$ lsusb
Bus 003 Device 003: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

$ sudo hcxdumptool -i wlp5s0f4u2 -o all.pcapng --enable_status=64
initialization of hcxdumptool 6.2.6-40-g20031ae (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy2
INTERFACE NAME............: wlp5s0f4u2
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca021987 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca021987 (not used for the attack)
DRIVER....................: mt76x2u
DRIVER VERSION............: 5.18.11-arch1-1
DRIVER FIRMWARE VERSION...: 0.0.00-b1
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 48f317f2a124 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 48f317f2a125 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 48f317f2a126 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b0febdf91905
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63789
ANONCE....................: 04835ba809f4f3cba144b505de674db6303678674e949897d3b917360d19a784
SNONCE....................: 9a014ae981f6b24f815402a603da9393278c992ea227e427f4799b2dcda209a9

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
18:44:00 2447/8     ERROR:0 INCOMING:518 AGE:1 OUTGOING:106 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:45:00 5500/100   ERROR:0 INCOMING:1248 AGE:17 OUTGOING:130 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:46:00 5805/161   ERROR:0 INCOMING:1248 AGE:77 OUTGOING:130 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:47:00 2467/12    ERROR:0 INCOMING:2281 AGE:1 OUTGOING:169 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:48:00 5580/116   ERROR:0 INCOMING:2630 AGE:33 OUTGOING:177 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:49:00 2412/1     ERROR:0 INCOMING:2630 AGE:93 OUTGOING:177 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
18:50:00 5200/40    ERROR:0 INCOMING:3220 AGE:1 OUTGOING:197 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:51:00 5660/132   ERROR:0 INCOMING:3306 AGE:52 OUTGOING:202 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:52:00 2432/5     ERROR:0 INCOMING:3822 AGE:9 OUTGOING:218 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:53:00 5280/56    ERROR:0 INCOMING:4043 AGE:5 OUTGOING:221 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:54:00 5745/149   ERROR:0 INCOMING:4043 AGE:65 OUTGOING:221 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:55:00 2452/9     ERROR:0 INCOMING:5038 AGE:7 OUTGOING:237 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:56:00 5520/104   ERROR:0 INCOMING:5450 AGE:21 OUTGOING:256 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:57:00 5825/165   ERROR:0 INCOMING:5450 AGE:81 OUTGOING:256 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:58:00 2472/13    ERROR:0 INCOMING:5772 AGE:3 OUTGOING:262 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
18:59:00 5600/120   ERROR:0 INCOMING:5948 AGE:37 OUTGOING:268 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:1 M3M4:0 M3M4ZEROED:0 GPS:0
^C
terminating...
```
Running the test about 15 minutes without ERRORs. The test targets (AP and CLIENT) are attacked successfully.

If the target is far away, the device transmit up to 7 times more packets. A stable power supply is mandatory as well as reduced tx power (max 20dBm):
```
$ sudo hcxdumptool -i wlp5s0f4u2 -C
initialization of hcxdumptool 6.2.6-40-g20031ae (depending on the capabilities of the device, this may take some time)...
wlp5s0f4u2 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
...
5180MHz  36 (20 dBm)
...
5865MHz 173 (20 dBm)

terminating...
```

ZerBea, 2022-07-15:
Is the last version, you mentioned as working:
https://github.com/ZerBea/hcxdumptool/commit/2b711d68172a731be4cdd928b452fdedccf465f3
is completely ERROR free, or do you get this kind of ERRORs here too after a longer period?

rgerganov, 2022-07-16:
The last working version is commit 3f7d98d02 which doesn't produce any errors.
```
$ lsusb
Bus 001 Device 002: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
This is with the latest code:
```
$ sudo hcxdumptool -i wlan1 -o all.pcapng --enable_status=64
initialization of hcxdumptool 6.2.6-40-g20031ae (depending on the capabilities of the device, this may take some time)...
warning possible interfere: wpa_supplicant is running with pid 437 360

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlan1
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0cab062b7 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0cab062b7 (not used for the attack)
DRIVER....................: mt76x2u
DRIVER VERSION............: 5.15.32-v7+
DRIVER FIRMWARE VERSION...: 0.0.00-b1
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 0024fb4ccb84 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 0024fb4ccb85 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 0024fb4ccb86 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b025aa4357e2
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62862
ANONCE....................: fc874df08728000f3b991127f3472f975f163fdd4e1bd4580a2f4b71f3a1934f
SNONCE....................: 0369a031c6d44f8a601ed8d59d625bac9e2c874606cd2a11711bd5beb8a2b8f2

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
05:57:25 2417/2 driver is busy/broken: failed to transmit associationrequest
05:57:41 2437/6 driver is busy/broken: failed to transmit associationrequest
05:58:00 2462/11    ERROR:1 INCOMING:1721 AGE:1 OUTGOING:248 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:1 GPS:0
05:58:19 5180/36 driver is busy/broken: failed to transmit associationrequest
05:59:00 5560/112   ERROR:1 INCOMING:3354 AGE:29 OUTGOING:479 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:1 GPS:0
06:00:00 5865/173   ERROR:1 INCOMING:3354 AGE:89 OUTGOING:479 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:1 GPS:0
06:01:00 5180/36    ERROR:1 INCOMING:8343 AGE:5 OUTGOING:722 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
06:02:00 5640/128   ERROR:1 INCOMING:8733 AGE:44 OUTGOING:837 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
06:03:00 2427/4     ERROR:1 INCOMING:9611 AGE:1 OUTGOING:865 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
06:03:20 2452/9 driver is busy/broken: failed to transmit associationrequest
06:03:57 5240/48 driver is busy/broken: failed to transmit associationrequest
06:04:00 5260/52    ERROR:1 INCOMING:13457 AGE:1 OUTGOING:1244 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:4 GPS:0
06:05:00 5720/144   ERROR:1 INCOMING:13461 AGE:60 OUTGOING:1244 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:2 M2M3:2 M3M4:0 M3M4ZEROED:4 GPS:0
06:06:00 2447/8     ERROR:1 INCOMING:15341 AGE:1 OUTGOING:1321 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:06:31 5180/36 driver is busy/broken: failed to transmit associationrequest
06:07:00 5500/100   ERROR:1 INCOMING:17150 AGE:17 OUTGOING:1514 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:08:00 5805/161   ERROR:1 INCOMING:17150 AGE:77 OUTGOING:1514 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:09:00 2467/12    ERROR:1 INCOMING:20558 AGE:1 OUTGOING:1642 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:10:00 5580/116   ERROR:1 INCOMING:22121 AGE:32 OUTGOING:1832 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:11:00 2412/1     ERROR:1 INCOMING:22123 AGE:1 OUTGOING:1832 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
06:11:51 2472/13 driver is busy/broken: failed to transmit associationrequest
06:12:00 5200/40    ERROR:1 INCOMING:26847 AGE:1 OUTGOING:2067 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:3 M2M3:3 M3M4:0 M3M4ZEROED:5 GPS:0
^C
terminating...
1 radiotap error encountered
```
And this is with version 6.2.6-7-g3f7d98d:
```
[... 36 lines cut ...]
06:16:00 5825/165   ERROR:0 INCOMING:4368 AGE:80 OUTGOING:325 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
06:17:00 2472/13    ERROR:0 INCOMING:7165 AGE:1 OUTGOING:469 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:1 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
06:18:00 5600/120   ERROR:0 INCOMING:8821 AGE:36 OUTGOING:542 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
06:19:00 2417/2     ERROR:0 INCOMING:9260 AGE:1 OUTGOING:551 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:3 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
06:20:00 5220/44    ERROR:0 INCOMING:11893 AGE:5 OUTGOING:690 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
06:21:00 5680/136   ERROR:0 INCOMING:12184 AGE:41 OUTGOING:708 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:4 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:1 GPS:0
[... cut ...]
06:29:00 5620/124   ERROR:0 INCOMING:23237 AGE:34 OUTGOING:1180 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:5 M1M2:1 M2M3:1 M3M4:0 M3M4ZEROED:2 GPS:0
^C
terminating...
```
I am using the builtin wifi adapter (wlan0) for SSH and I have disabled `wpa_supplicant` for wlan1 which is AWUS036ACM:
```
$ ps xa | grep wpa_sup
  360 ?        Ss     0:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
  437 ?        Ss     0:00 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
```

I am using 5V 2A adapter to power the Raspberry.

ZerBea, 2022-07-16:
Ok, Thanks for the comment. It looks like the driver is only blocking ASSOCIATIONREQUEST packets in active monitor mode, and we have to find out why. That doesn't appear on my systems (everything disabled/deactivated), so I can't reproduce it.
Please deactivate build in WiFI, disable SSH and  stop wpa_supplicant completely.
Than retry latest git head again to find out if the ASSIOCIATIONREQUESTs are still blocked.

ZerBea, 2022-07-16:
I think I got it. Please try latest commit.

Running active monitor mode in transmission branch increased traffic 7 times.
Now we are running a TX timeout of 1 second, which should be enough for the device to transmit all packets before we're doing a ne request.

BTW:
By the commit before, I eliminated the warning about deprecated openssl API.

rgerganov, 2022-07-18:
I don't get any errors with the latest code. However, my Raspberry seems to crash after running
```
sudo ./hcxdumptool -i wlan1 -o all.pcapng --enable_status=64
```
for about an hour. I will investigate and open a new issue if needed. Thanks for addressing this one.

ZerBea, 2022-07-18:
No problem, you're welcome.
The issue you reported is caused by a timing problem.
I the channel is busy, the firmware doesn't allow the device to transmit and the firmware doesn't accept new packets from the driver. Now we are running a 5 second timeout which is more than enough to handle this and short enough to detect if driver is broken.

Please check the temperature of the RPI, the connection of the power cable and the connection of the USB device. All of them are know to shut down the RPI.

closed 2022-07-18 by ZerBea

ZerBea, 2022-07-19:
BTW:
"I am using the builtin wifi adapter (wlan0) for SSH"

I recommend to protect wlan0 and the associated CLIENT via BPF. That reduce traffic and avoid timing issues.

rgerganov, 2022-07-19:
Thanks for the advice. I am actually planning to try Bluetooth as control channel for hcxdumptool. Are you aware of any interference between BT and wlan?

ZerBea, 2022-07-19:
Yes. There are impacts related to pselect() system call on all file descriptor operations.
To avoid this the RPI should run completely headless.
Run each tool 10 minutes:
$ hcxlabgetmallpr -i wlan1 -c 1,2,3,4,5,6,7,8,9,10,11,12,13
$ hcxdumptool -i wlan1 -o dump.pcapng --enable_status=15 -c 1,2,3,4,5,6,7,8,9,10,11,12,13
than compare the dump file contents by hcxpcapngtool.

pampam133, 2022-10-31:
Hello ZerBea. Your work is fantastic.
I use my raspberry pi zero this way. SSH to wlan0 with phone. 
I experienced hcxdumptool catches more stuff with the same adapter(T2UH) on desktop kali, than ssh-d to raspberry. Hard to describe, but it is far more snappy, faster, working better on desktop. Especially on association captures. Unfortunatelly I want to use it with the pi. 
What do you think, the reason could be what you did mentioned above?
"There are impacts related to pselect() system call on all file descriptor operations."
And if it yes for this case, disabling wlan0 and using an ESP-01 serial bridge for SSH could lead to better results? I could solder a pi hat and a socket for the ESP in an afternoon and i have a couple around.
Power supply current cannot be a problem, I am using a 2A power bank.

ZerBea, 2022-10-31:
Connecting the RPI via WiFI (and) SSH while running hcxdumptool is a big problem.
The single core CPU is not able to handle multiple threads at the same time. That will result in a packet loss. The same applies to a SSH connection via BT.
Also it is not a good idea to connect the RPI via WiFi while running attacks in the same WiFi spectrum, because the WiFi channels are overlapped. If the attack WiFi device is close to the remote WiFi device more than 6 channel are overlapped.
https://electronics.stackexchange.com/questions/155643/why-do-wifi-channels-overlap
https://www.data-alliance.net/wifi-channels-guide
That will lead to a packet loss, too.

I tested this running 3 RPI (one on channel 1, the second on channel 6 and the third on channel 11). All WiFi devices are close together (< 50 cm) and they interfered with each other.

The best way is to set up a minimalist system (no GUI and no unneeded additional tasks) except the hardware modification (LED and push button) and run it completely headless. You will get best results if everything is disabled via config.txt:
```
gpu_mem=64
initramfs initramfs-linux.img followkernel
max_usb_current=1
hdmi_blanking=1
dtoverlay=pi3-disable-bt
dtoverlay=pi3-disable-wifi
dtparam=audio=off
dtparam=watchdog=on
disable_splash=1
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on
dtparam=pwr_led_trigger=none
dtparam=pwr_led_activelow=off
```

and all unneeded tasks are disabled via systemctl except this ones:
```
$ systemctl list-units --type=service --state=running
  UNIT                         LOAD   ACTIVE SUB     DESCRIPTION
  autologin@tty1.service       loaded active running Getty on tty1
  dbus.service                 loaded active running D-Bus System Message Bus
  dhcpcd.service               loaded active running DHCP Client Daemon
  serial-getty@ttyAMA0.service loaded active running Serial Getty on ttyAMA0
  systemd-journald.service     loaded active running Journal Service
  systemd-logind.service       loaded active running User Login Management
  systemd-timesyncd.service    loaded active running Network Time Synchronization
  systemd-udevd.service        loaded active running Rule-based Manager for Device Events an>
  user@0.service               loaded active running User Manager for UID 0
  watchdog.service             loaded active running watchdog daemon
```

pampam133, 2022-10-31:
Thanks for your rich answer and caring!
It seems you understood my thinking an concerns. 
My RPI has headless raspbian. There is no UI when i connect the hdmi. It was considered for more processing power at the beginning. I made a hat with 2 buttons and a status led, works as expected. When the ssh connection was killed before the BPF filter hcxdumptools needed to stopped by button press. Nowdays this is not a problem since I use the filter.
1-2 year ago I used a nethunter nexus 5. There was not a ssh connection but the wn722n. I got problems(i read here too) but it worked for some time, pretty good and I had also a custom antenna/dongle device(ath9k, unknown board/chipset) which signal weakened, captures got worse. Then I bought a T2UH after reading your experience. Perfect, one I say. I got a directional biquad with lips for it. Captured lot of stuff with it at the beginning(RPI, same build). Was lot of association capture. Then I updated and the effect got "weaker". I dont remember the working version. Maybe 1-1,5 years old, and maybe not even the hcxdumptool update caused it.
Unfortunately the nethunter doesent recognize the T2UH so I am using the pi. (At least the 3.x, newest kernel on the nexus 5). 

As for the interference: More or less I think there is not much interference behind a reflector. I placed my RPI behind the reflector, that also got lips, so i should not be a big problem, but I am not an Electromagnetics expert and I am curious about your opinion about that. There was a time with this config when thing worked fine. Maybe I shield the T2UH. The old antenna-donge got a shielded aluminum house(ath9k). Perhaps i should make one myself. Now the dongle is between the back of a reflector and the rpi which has a thick aluminum heatsink on it.

Maybe next time I try some usb terminal solution somehow or maybe I need to buy a Pi zero 2W for a second core and a plus processing power.

I am looking forward your opinion and thanks again!
(Sorry for my rusty english)

ZerBea, 2022-11-01:
Try to use hcxlabtool:
$ hcxlabgetmallpr --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --gpio_button=4 --gpio_statusled=17 --bpfc=own.bpfc --essidlist=topcommonessid --essidmax=50 --m2attempt=4 -c 13,1,6,11,2,1,6,11,3,1,6,11,4,1,6,11,5,1,6,11,7,1,6,11,8,1,6,11,9,1,6,11,10,1,6,1,12,1,6,11 &

ZerBea, 2022-11-01:
BTW:
Don't underestimate the backward radiation of a panel antenna
https://elboxrf.com/img/bookmarks/126/108.png

pampam133, 2022-11-01:
Alright, I will try this. I have to admit I am also in the group who likes lot of status output. Sometimes when I start, it gives very few status other times it gives me full page status text instant. In the first case my walking will be unfruiting because it is barely catches anything and it can cause  problems later in the pcapngtool. When i got start with a page full of status text(lots of APs info) the run will be success with moderately captures. Not as much Association and EAPOL as the old days, but considerably good.

I will try pimp my system a little more. First I make a shielded case for the adapter cuz it is a fun tinkering and never hurts. Second I buy a hub and try it without a second wifi. As I searched USB tethering is my solution for it. Third i am start to experiment with your wifi labtool, but i am a little afraid from it(for the difficulty at first) Or maybe the difference between the nexus5 and the RPI was just really the GHz difference and the cores.

BTW I took apart my Adapter from curiosity and i saw that the signal lane is completely at the surface layer.

https://wikidevi.wi-cat.ru/images/9/99/Pc-board.JPG

This can collect a lot of EMI this way.

As for the backward radiation:
https://buildyourownantenna.blogspot.com/2014/07/double-biquad-sector-antenna-for-2450-mhz-wifi.html

You are right. I saw this graph before, but I thought it is insignificant magnitude. Now I know I was wrong.
Thanks for your time and help!

ZerBea, 2022-11-01:
Unfortunately my my lessons learned list is long:
```
a beautiful status output make the attack tool slow and sluggish.
too many features make the attack tool slow and sluggish.
response time behavior becomes very bad on big ringbuffers.
transmitting too many packets make a channel busy.
a Raspberry Pi is not able to handle more than one interface at the same time.
pselect doesn't like to be interrupted by another timer.
active monitor mode (enabled by radiotap header) is mandatatory on AUTHENTICATION, ASSOCIATION and EAPOL frames!
it is mandatory to ACK NULL and ACTION frames within a 4way handshake!
setting a short preamble in radiotap header is ignored on tx.
entire AUTHENTICATION process should be done running a data rate of 1.0 Mb/s.
there are (much) better ways than injecting stupid DEAUTHENTICATION frames to disconnect a CLIENT.
the most useful frame is an EAPOL M2 frame!
```
from https://github.com/ZerBea/wifi_laboratory