ZerBea/hcxdumptool #225: Android issue

kimocoder, 2022-07-23:
Hi brother, we now have problems with all adapters in Android for some reason it will not work anymore.

Tried a lot of adapters, but all failing for some reason with later releases/commits of hcxdumptool, even kmown supported adapters with all nescessary/provided settings for hcxdumptool.

Instead of sending an email, I opened this thread for anyone to participate in it 🥇

RealEnder, 2022-07-23:
Hi Chris, do you have a working commit?

kimocoder, 2022-07-23:
hold on, I will provide more information and logs after dinner 💯

kimocoder, 2022-07-23:
[image]

kimocoder, 2022-07-23:
The issue above is shown is all adapters I try now, different phones, kernels, OOS, LOS, A11, A12. Anywhere.

kimocoder, 2022-07-23:
[image]

[image]

kimocoder, 2022-07-23:
Above is 6 months back, even doing on a VIF adapter which I know I should not be doing. Same driver "rtl
2800usb"

kimocoder, 2022-07-23:
If you need me to do some tests or anything debug, just let me know what I can do, because I'm doing better integration of hcx tools in wifite2 these days, and we really need this working 🥇

ZerBea, 2022-07-23:
Ok, let's hunt for the problem.
Main change was adding active monitor mode in transmit branch which is working as expected on Linux systems.
I'll add some more information to opensocket() to find out what exactly went wrong.

kimocoder, 2022-07-23:
Awesome. Im available with anything for testing purposes, just let me know what to do or provide 👍

ZerBea, 2022-07-23:
I think I got it. It is located in opensocket
here:
```
#ifdef PACKET_IGNORE_OUTGOING
if(setsockopt(fd_socket, SOL_PACKET, PACKET_IGNORE_OUTGOING, &enable, sizeof(int)) < 0) return false;
#endif
```

or here:
```
if(set_channel_test(2462) == false) return false;
if(set_channel_test(2412) == false) return false;
```

Pushed an update to identify the problem exactly. Please test latest git head.

kimocoder, 2022-07-23:
hold on, tests coming up 👍

ZerBea, 2022-07-23:
Ok, stand by.

kimocoder, 2022-07-23:
[image]

kimocoder, 2022-07-23:
We're back in business sir 🥇 
Thanks for participating 🥇

ZerBea, 2022-07-23:
No problem, you're always welcome.

The price tag on the latest changes is huge:

Active monitor produce 7 times more traffic in case of a crappy connection (target far away).
Ignoring outgoing packets by hcxdumptool increase CPU cycles.

But it still works fine on small devices.
Some commits before I set the time out to 5 seconds - that should be more than enough.

closed 2022-07-23 by ZerBea

kimocoder, 2022-07-23:
[image]

Confirmation in tool 👍  Open source community will always be the best

ZerBea, 2022-07-23:
BTW:
I wonder about Android.
They define PACKET_IGNORE_OUTGOING
but the Protocol is not available.

kimocoder, 2022-07-23:
I can look for the string tonight 👍  I'll notify you

ZerBea, 2022-07-23:
BTW:
hcxlabgetmallpr is the evil little sister of hcxdumptool.

kimocoder, 2022-07-23:
I will investigate the usages.
Right now, my plan is to get rid of airmon-ng usage in wifite2, using a VIF interface will cause interferrance with adapter and rather use 'ip' and 'iw' and better integration/usage of hcx tools

kimocoder, 2022-07-23:
So, it looks like PACKET_IGNORE_OUTGOING was added in kernel v4.20. Most Android devices at this point haven't even got to kernel v4.20. Latest devices have, but majority not.

add PACKET_IGNORE_OUTGOING sockopt from linux v4.20

new in linux commit fa788d986a3aac5069378ed04697bd06f83d3488

kimocoder, 2022-07-23:
Correction, it was added in kernel v4.19-rc2 but the same problem, majority of devices isn't on kernel v4.19 at this point too.

RealEnder, 2022-07-23:
@ZerBea , this can go as compile-time option of hcxdumptool with 4.20+ default.
@kimocoder You can use hcxdumptool to set monitor mode directly on the interface. There is a simple CPU cracker in hcxtools, but in their current version can't read wordlist from file. john may be better alternative, from aircrack-ng/pyrit for sure. wifite2 uses some other old tools, which can be also stripped - it will be easier to maintain with less dependencies.

kimocoder, 2022-07-23:
Yes, I need to find some replacements soon, crackers used isn't really maintained anymore, that includes John. hashcat is the one which is active maintained at this point, and aircrack-ng of cource. But dont really know about replacement tools atm, we'll see. but moving away from VIF interfaces would be a good start.

ZerBea, 2022-07-24:
By latest commit, we test for kernel >= 4.20 as compile option. Thanks @RealEnder 
I thought it was enough to test if PACKET_IGNORE_OUTGOING is defined:
https://github.com/ZerBea/hcxdumptool/issues/212
Obviously that wasn't enough.
More information is here:
https://thomasw.dev/post/packet_ignore_outgoing/

@kimocoder JtR (single) and hashcat (a9) have a nice new mode if the PSK is calculated from BSSID or ESSID. That is working like a charm and can be added to wifite2 in an easy way.

RealEnder, 2022-07-24:
JtR is maintained and functionality added constantly: https://github.com/openwall/john
If compilation is made on the phone (maybe this is the case more often) it will work, but if using crossbuild environment kernel version may be different.

ZerBea, 2022-07-24:
Just for information:
https://source.android.com/devices/architecture/kernel/android-common

kimocoder, 2022-07-24:
I'm pulling AKC (common) and CAF (CodeLinaro) to my kernels, vendors certainly never touch it sfter release 🥇  will look at the crackers soon, but the handling of monitor mode is a priority

ZerBea, 2022-07-24:
For sure, monitor mode and packet injection is high priority.
Unfortunately the airmon-ng way is not the best choice due to iw (and libnl) dependency and the reason why I added to set monitor mode to hcxdumptool/hcxlabtool series.
https://github.com/aircrack-ng/aircrack-ng/issues/1965

ZerBea, 2022-07-28:
BTW:
Just received this issue report:
https://github.com/ZerBea/hcxtools/issues/218
A good reason to run hcxdumptool instead of a stupid DEAUTHENTICATOR.

kimocoder, 2022-08-07:
Possible to tag a new release and push soon? We want changes to Kali :1st_place_medal:

ZerBea, 2022-08-07:
Done: -> 6.2.7
https://github.com/ZerBea/hcxdumptool/releases

and flagged out
https://archlinux.org/packages/community/x86_64/hcxdumptool/