ZerBea/hcxdumptool #226: driver doesn't report frequency

Strapicarus-Rush, 2022-07-27:
Hi, hcxdumptool not working for me. first i try in kali and now on debian and the result is the same. i follow the readme and a video, also have missing zlib and follow instructions on an issue on hcxtools #35 to install the lib.

commands: 
hcxdumptool -i wlan0 --do_rcascan
hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15

reports errors: 
**driver doesn't report frequency
channel test failed
warning: failed to init socket
terminating...
1 driver error encountered**

Tested on: 
kali kernel 5.18.0 (maybe not accurate)
debian 11 5.18.14-1 sid 

Dependencies:
libcurl4-openssl-dev 7.84.0-2
libssl-dev 3.0.5-1
pkg-config 0.29.2-1

Compiled with:
gcc version 12.1.0

hardware lsusb reports:
ID 148f:3070 Ralink RT2870/RT3070 Wireless Adapter (alfa AWUS036NH).

If there is anymore info that i should post please let me know. 
Thanks.

ZerBea, 2022-07-27:
I can't reproduce that issue running an AWUS036NH:
$ lsusb
`Bus 005 Device 015: ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
`

Are you running latest git head?
$ hcxdumptool -v
`hcxdumptool 6.2.6-48-gd60c2d3 (C) 2022 ZeroBeat
`

$ hcxdumptool -I
```
wlan interfaces:
phy0	00c0ca367a0d	wlp39s0f3u1u4	(driver:rt2800usb)
```

$ sudo hcxdumptool -i wlp39s0f3u1u4 -o dumpfile.pcapng --active_beacon --enable_status=15
```
initialization of hcxdumptool 6.2.6-48-gd60c2d3 (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlp39s0f3u1u4
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca357b0d (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca357b0d (not used for the attack)
DRIVER....................: rt2800usb
DRIVER VERSION............: 5.18.14-arch1-1
DRIVER FIRMWARE VERSION...: 0.36
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 24bf748c3596 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 24bf748c3597 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 24bf748c3598 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: f04f7c63dd43
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64474
ANONCE....................: e097e7d0868faa83f3b022e31a0a92b87a06535dbb62ce4192ac8cf40b19ce92
SNONCE....................: 2a91827be409625aace81bc0faf2cf74cf6257d9ece5e5c3713410a67e9e5149

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
10:06:00 2412/1   ffffffffffff dc34c8cecde1 TEST_NET [BEACON]
^C
terminating...
```

Strapicarus-Rush, 2022-07-27:
Sorry, yeah works well after disconnecting and connecting with the networkmanager and wpa_suplicant services stopped. Maybe my mistake was stopping the services while still connected to the network. Thanks for that fast response, made me think to disconect and reconnect, sorry for wasting your time.

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.