ZerBea/hcxdumptool #227: Stuck at "initialization of hcxdumptool" Alfa Awus036ACH

OranGe-66, 2022-08-15:
I am using Kali and I was able to install the drivers of the adapter and set the adapter to monitor mode successfully.

ZerBea, 2022-08-15:
The AWUS036ACH USB adapter incorporates RTL8812AU WLAN chipset. The driver depend on NETLINK and is not recommended.

Please remove the adapter. Than plug-in the adapter and run:
$ sudo hcxdumptool -iwlan0 --check-driver
and comment the output.
Do not run iw to set monitor mode before!

Output on a kernel stock driver should look like this:
```
$ sudo hcxdumptool -i wlp39s0f3u1u6 --check_driver
initialization of hcxdumptool 6.2.7-1-g7174fbc (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: mt76x0u
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

ZerBea, 2022-08-15:
Next step is to set monitor mode by iw /which ruse NETLINK):
$ sudo ip link set wlan0 down
$ sudo iw wlan0 set monitor none
$ sudo ip link set wlan0 up
$ sudo hcxdumptool -i wlan0 --check-driver
and comment the output.

ZerBea, 2022-08-15:
Last step is to set monitor mode by iw (which ruse NETLINK):
$ sudo ip link set wlan0 down
$ sudo iw wlan0 set monitor none
$ sudo ip link set wlan0 up
$ sudo hcxdumptool -i wlan0 -C
to request frequencies supported by the driver.

ZerBea, 2022-08-15:
More information about `the` driver is here:
https://github.com/ZerBea/hcxdumptool/issues/214

OranGe-66, 2022-08-15:
Sorry for the late reply.
So I installed the latest version of Kali (2022.3) alongside with the latest drivers I was able to find for RTL8812au and then I tried out the commands you provided and here is the result :

`$ sudo hcxdumptool -i wlan0 --check_driver`
[image]

```
$ sudo ip link set wlan0 down
$ sudo iw wlan0 set monitor none
$ sudo ip link set wlan0 up
$ sudo hcxdumptool -i wlan0 --check_driver
```
[image]

```
$ sudo ip link set wlan0 down
$ sudo iw wlan0 set monitor none
$ sudo ip link set wlan0 up
$ sudo hcxdumptool -i wlan0 -C
```
[image]

The error message "driver doesn't support frequency scan" started appearing after 10 minutes from the moment I entered the command.

ZerBea, 2022-08-15:
Thanks for the comments.

Looks like hcxdumptool doesn't have full access to the device, because NetworkManager and wpa_supplicant are still running.

Please stop both tools:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service
Than run hcxdumptool again.

To get Networkaccess back, start them if hcxdumptool terminated:
$ sudo systemctl start NetworkManager.service
$ sudo systemctl start wpa_supplicant.service

ZerBea, 2022-08-15:
This is the driver supplied by KALI:
https://github.com/aircrack-ng/rtl8812au

OranGe-66, 2022-08-15:
Yes that's the one I installed. 
So I stopped both NetworkManager and wpa_supplicant, however unfortunately I still get the same error.

[image]

This time it took about 6 minutes from the moment I entered the command for it to appear.

ZerBea, 2022-08-15:
10 minutes is way too long until the driver respond to hcxdumptool.

Some examples:
ASUS AC51
```
$ hcxdumptool -I
wlan interfaces:
phy2	0c9d92b486ca	wlp39s0f3u1u1u4	(driver:mt76x0u)

$ time sudo hcxdumptool -i wlp39s0f3u1u1u4 -C
initialization of hcxdumptool 6.2.7-1-g7174fbc (depending on the capabilities of the device, this may take some time)...
wlp39s0f3u1u1u4 available frequencies, channels and tx power reported by driver:
2412MHz   1 (16 dBm)
2417MHz   2 (16 dBm)
2422MHz   3 (16 dBm)
2427MHz   4 (16 dBm)
2432MHz   5 (16 dBm)
2437MHz   6 (16 dBm)
2442MHz   7 (16 dBm)
2447MHz   8 (16 dBm)
2452MHz   9 (16 dBm)
2457MHz  10 (16 dBm)
2462MHz  11 (16 dBm)
2467MHz  12 (16 dBm)
2472MHz  13 (16 dBm)
2484MHz  14 ( 0 dBm)
5180MHz  36 (19 dBm)
5200MHz  40 (19 dBm)
5220MHz  44 (19 dBm)
5240MHz  48 (19 dBm)
5260MHz  52 (19 dBm)
5280MHz  56 (19 dBm)
5300MHz  60 (19 dBm)
5320MHz  64 (19 dBm)
5500MHz 100 ( 0 dBm)
5520MHz 104 ( 0 dBm)
5540MHz 108 ( 0 dBm)
5560MHz 112 ( 0 dBm)
5580MHz 116 ( 0 dBm)
5600MHz 120 ( 0 dBm)
5620MHz 124 ( 0 dBm)
5640MHz 128 ( 0 dBm)
5660MHz 132 ( 0 dBm)
5680MHz 136 ( 0 dBm)
5700MHz 140 ( 0 dBm)
5720MHz 144 ( 0 dBm)
5745MHz 149 (19 dBm)
5765MHz 153 (19 dBm)
5785MHz 157 (19 dBm)
5805MHz 161 (19 dBm)
5825MHz 165 (19 dBm)
5845MHz 169 ( 0 dBm)
5865MHz 173 ( 0 dBm)

terminating...

real	0m2,814s
user	0m0,062s
sys	0m0,136s
```

TP-LINK TL-WN722N v1
```
$ hcxdumptool -I
wlan interfaces:
phy3	f81a67077d0e	wlp39s0f3u1u1u4	(driver:ath9k_htc)

$ time sudo hcxdumptool -i wlp39s0f3u1u1u4 -C
initialization of hcxdumptool 6.2.7-1-g7174fbc (depending on the capabilities of the device, this may take some time)...
wlp39s0f3u1u1u4 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 (20 dBm)

terminating...

real	0m1,151s
user	0m0,065s
sys	0m0,105s
```

ALFA AWUS036ACM
```
$ hcxdumptool -I
wlan interfaces:
phy5	00c0caad0e49	wlp39s0f3u1u1u4	(driver:mt76x2u)

$ time sudo hcxdumptool -i wlp39s0f3u1u1u4 -C
initialization of hcxdumptool 6.2.7-1-g7174fbc (depending on the capabilities of the device, this may take some time)...
wlp39s0f3u1u1u4 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 ( 3 dBm)
5180MHz  36 (20 dBm)
5200MHz  40 (20 dBm)
5220MHz  44 (20 dBm)
5240MHz  48 (20 dBm)
5260MHz  52 (20 dBm)
5280MHz  56 (20 dBm)
5300MHz  60 (20 dBm)
5320MHz  64 (20 dBm)
5500MHz 100 ( 3 dBm)
5520MHz 104 ( 3 dBm)
5540MHz 108 ( 3 dBm)
5560MHz 112 ( 3 dBm)
5580MHz 116 ( 3 dBm)
5600MHz 120 ( 3 dBm)
5620MHz 124 ( 3 dBm)
5640MHz 128 ( 3 dBm)
5660MHz 132 ( 3 dBm)
5680MHz 136 ( 3 dBm)
5700MHz 140 ( 3 dBm)
5720MHz 144 ( 3 dBm)
5745MHz 149 (20 dBm)
5765MHz 153 (20 dBm)
5785MHz 157 (20 dBm)
5805MHz 161 (20 dBm)
5825MHz 165 (20 dBm)
5845MHz 169 ( 3 dBm)
5865MHz 173 ( 3 dBm)

terminating...

real	0m18,636s
user	0m0,038s
sys	0m0,113s
```

ZerBea, 2022-08-15:
5745MHz 149 (20 dBm) = full tx power allowed by wireless regulatory domain

5720MHz 144 ( 3 dBm) = full tx power not allowed by wireless regulatory domain

$ iw reg get
global
country 98: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 20), (0 ms), DFS
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), DFS
	(5725 - 5850 @ 80), (N/A, 30), (N/A)

ZerBea, 2022-08-15:
Something in your configuration prevent that hcxdumptool can change the frequency/channel and the ERROR message is shown after the timeout.

ZerBea, 2022-08-15:
By latest commit I added some more information so that we can see what values the driver respond.
Please try latest git hat and comment output of
sudo hcxdumptool -i wlan0 -C

OranGe-66, 2022-08-15:
Here is the result : 

[image]

Once again sorry for the late reply.

ZerBea, 2022-08-15:
Thanks. It looks like this driver neither report a channel (all values < 1000 are channel numbers) nor a frequency (all values >= 1000 are frequencies). I guess it is a driver issue, because all tested drivers (MediaTek, Ralink and Atheros) are working as expected.

Device 021: ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
```
$ hcxdumptool -I
wlan interfaces:
phy8	c83a35cc88c9	wlp39s0f3u1u1u4	(driver:rt2800usb)

$ time sudo hcxdumptool -i wlp39s0f3u1u1u4 -C
initialization of hcxdumptool 6.2.7-4-gfe0d024 (depending on the capabilities of the device, this may take some time)...
wlp39s0f3u1u1u4 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 ( 0 dBm)

terminating...

real	0m0,711s
user	0m0,027s
sys	0m0,074s
```

ZerBea, 2022-08-15:
The last hcxdumptool workaround  on this driver was late 2021:
https://github.com/ZerBea/hcxdumptool/issues/199

Unfortunately I'm running out of ideas.  
I'll ask the maintainer of the driver.
https://github.com/aircrack-ng/rtl8812au
Maybe he has an idea.

ZerBea, 2022-08-15:
BTW:
The man function to get the frequencies supported by the driver is here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7713
Afer running this ioctl() call
if(ioctl(fd_socket, SIOCGIWFREQ, &pwrq) < 0) continue;
we should get either an ERROR (in that case we do nothing) or a frequency or a channel in pwrq.u.freq.e.
This is not the case and pwrq.u.freq.e always is 0.

OranGe-66, 2022-08-15:
Alright thanks a lot for being concerned I'll wait for your response.

I have to mention though that the awus036ach adapter that I am using is the new version with the USB Type-C, I am not sure if it differs a lot from the previous version but this might help in debugging.

ZerBea, 2022-08-16:
I don't think the issue is related to hcxdumptool, because it looks like the aircrack-ng driver is still working. I got the first response here:
https://github.com/ZerBea/hcxdumptool/issues/199#issuecomment-1215713383
and it looks like everything is still working as expected.

Unfortunately there are many "screws" which can cause the problem:
ALFA AWUS036ACH TYP-C (new device - possible not completely covered by the driver)
VIRTUAL MACHINE (not recommended for hcxdumptool - the VM can prevent that hcxdumptool has full access to the device)
RTL8812AU driver that depend on NETLINK (hcxdumptool use ioctl() system calls instead of NETLINK messages

What happens if you try to the the frequency via iw (NETLINK)?
$ sudo ip link set wlan0 down
$ sudo iw wlan0 set monitor none
$ sudo ip link set wlan0 up
$ sudo iw dev wlan0 set freq 2412
$ sudo iw dev wlan0 set freq 2442
$ sudo iw dev wlan0 set freq 2462

ZerBea, 2022-08-16:
By latest commit I added some initial tests to improve init speed. If this test failed, hcxdumptool terminated immediately.

OranGe-66, 2022-08-16:
> ALFA AWUS036ACH TYP-C (new device - possible not completely covered by the driver)
I think this is the one. Do you know someone that has the exact same adapter to confirm whether he also encounters the same problem or not?

> What happens if you try to the the frequency via iw (NETLINK)?
Everything works fine : 
[image]

> By latest commit I added some initial tests to improve init speed. If this test failed, hcxdumptool terminated immediately.
Here is the result when I tried the latest Git :
[image]

ZerBea, 2022-08-16:
Thanks.
Now we know that iw is able to set the frequency by NETLINK message.

Let's try the same using iwconfig by ioctl() system call:
$ sudo hcxdumptool -m wlan0
$ sudo iwconfig wlan0 freq 2.412G
$ iwconfig
$ sudo iwconfig wlan0 freq 2.442G
$ iwconfig

ZerBea, 2022-08-16:
There is a difference in my iwconfig output:
```
$ sudo hcxdumptool -m wlp39s0f3u1u1u4
setting interface wlp39s0f3u1u1u4 to monitor mode

$ sudo iwconfig wlp39s0f3u1u1u4
wlp39s0f3u1u1u4  IEEE 802.11  Mode:Monitor  Frequency:2.412 GHz  Tx-Power=16 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
```
My interface is not shared (no unassociated ESSID and no Nickname).

OranGe-66, 2022-08-16:
Here is the result : 
[image]

Sorry but if you don't mind me asking, what does a shared interface mean? and how does it differ from a non-shared one?

OranGe-66, 2022-08-16:
BTW did you consider trying to fetch the channels supported by the driver instead of the frequencies? (if possible ofc)

ZerBea, 2022-08-16:
Shared Interface means that hcxdumptool doesn't have exclusive access to the interface.
Thanks for the iwconfig results.

Unfortunately, the result is incomplete, because I forgot to mention that you have to do the entire process directly after the device is plugged in. That way we make sure the adapter is not partly init by NETLINK.
$ sudo hcxdumptool -m wlan0
$ sudo iwconfig wlan0 freq 2.442G
$ iwconfig wlan0

There is no difference by setting the frequency or the channel:
```
struct iw_freq	freq;		/* frequency or channel :
					 * 0-1000 = channel
					 * > 1000 = frequency in Hz */
```
https://github.com/torvalds/linux/blob/master/include/uapi/linux/wireless.h#L910
Every driver understand this.

ZerBea, 2022-08-16:
Problem is that we request the actual channel/frequency from the driver and it report: 0
if(ioctl(fd_socket, SIOCGIWFREQ, &pwrq) < 0) continue;
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L7738

OranGe-66, 2022-08-16:
Thanks for the reply.
Here is the result after a fresh reboot then plugging in the adapter : 
[image]

However looks like I figured something out, I set the frequency to 2457hz (which is an arbitrary frequency I chose), then I added -c 10 which is the channel that has that frequency and it worked fine. 
[image]

So I tried specifying a frequency other than the one that I set my adapter to and it worked fine too.
[image]
[image]

However when I don't specify a frequency it doesn't work, for example when I run the following commands :
```
$ sudo hcxdumptool -i wlan0 -o file.pcapng --active_beacon --enable_status=15 
$ sudo hcxdumptool -i wlan0 -C

```

ZerBea, 2022-08-16:
Thanks again.
It looks like the problem is related to the driver. After you plugged in the device neither hcxdumptool, nor iw config is able to set the frequency.
You requested 2.442G via iwconfig wlan0 2.442G , but the adapter remains on 2.412G. That should not happen and point to a driver issue.

Please try
$ sudo hcxdumptool -i wlan 0 -c 2457 instead of -c 10

ZerBea, 2022-08-16:
Here, iwconfig was able to set the channel:
https://github.com/ZerBea/hcxdumptool/issues/227#issuecomment-1216326555
After a reboot and a fresh plug in, the settings by iwconfig are ignored completely (without an ERROR):
https://github.com/ZerBea/hcxdumptool/issues/227#issuecomment-1216452614

I don't think I'm able to code a workaround to this problem.

ZerBea, 2022-08-16:
Please report this issue directly on the driver site:
https://github.com/aircrack-ng/rtl8812au/issues
it could be related to this issue:
https://github.com/aircrack-ng/rtl8812au/issues/587
https://github.com/aircrack-ng/rtl8812au/issues/725

ZerBea, 2022-08-16:
Moved the issue to:
https://github.com/aircrack-ng/rtl8812au/issues/587
and closed it here.

closed 2022-08-16 by ZerBea

OranGe-66, 2022-08-16:
So I did another fresh reboot and then plugged in the adapter and gave it another try, it worked fine this time :
[image]
I did this once again to make sure and it worked fine too. 

I also tried `$ sudo hcxdumptool -i wlan 0 -c 2457` and it worked fine 
[image]

ZerBea, 2022-08-16:
Nice to hear that it works. But I still suspect a driver issue that caused this trouble. For me it looks like the driver sometimes failed to init the device completely.

OranGe-66, 2022-08-16:
Thanks for the reply.
I agree with you it might be a driver issue.

So at the end I think I'll be going with adding the frequencies manually like this : 
`sudo hcxdumptool -i wlan0 -o file.pcapng --active_beacon --enable_status=15 -c 1,2,3,4,5,...`

Its downside is that it takes a long time to start but it eventually works, however does it function the same as when not specifying the frequencies like this or is there a difference between the two commands?  
`sudo hcxdumptool -i wlan0 -o file.pcapng --active_beacon --enable_status=15`

ZerBea, 2022-08-16:
Difference is:
-c 1,2,3,4,5 -> hcxdumptool scan on this channels/frequencies as requested
if -c is absent, hcxdumtool is scanning all available channels from 1 .... last available channel/frequency.

BTW:
1,2,3,4,5 is not good -> 1,6,11,2,1,6,11,3,1,6,11,4 ..... 1,6,11,13 is much better.

ZerBea, 2022-08-16:
From other users I know that the hcxdumptool is working if the rtl8812au driver initialized the device successfully. Unfortunately that sometimes fail.

ZerBea, 2022-08-16:
Just to complete the device list. I got my hands on a notebook of a friend.
Looking good, running a rtw_8821ce device:
```
$ hcxdumptool -I
wlan interfaces:
phy0	7066557cb81a	wlp4s0	(driver:rtw_8821ce)

$ time sudo hcxdumptool -i wlp4s0 -C
initialization of hcxdumptool 6.2.7-9-g35929c3 (depending on the capabilities of the device, this may take some time)...
wlp4s0 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 ( 0 dBm)
5180MHz  36 (30 dBm)
5200MHz  40 (30 dBm)
5220MHz  44 (30 dBm)
5240MHz  48 (30 dBm)
5260MHz  52 (24 dBm)
5280MHz  56 (24 dBm)
5300MHz  60 (24 dBm)
5320MHz  64 (24 dBm)
5500MHz 100 (24 dBm)
5520MHz 104 (24 dBm)
5540MHz 108 (24 dBm)
5560MHz 112 (24 dBm)
5580MHz 116 (24 dBm)
5600MHz 120 (24 dBm)
5620MHz 124 (24 dBm)
5640MHz 128 (24 dBm)
5660MHz 132 (24 dBm)
5680MHz 136 (24 dBm)
5700MHz 140 (24 dBm)
5720MHz 144 (24 dBm)
5745MHz 149 (30 dBm)
5765MHz 153 (30 dBm)
5785MHz 157 (30 dBm)
5805MHz 161 (30 dBm)
5825MHz 165 (30 dBm)

terminating...

real	0m1,706s
user	0m0,059s
sys	0m0,157s
```

OranGe-66, 2022-08-17:
> BTW:
> 1,2,3,4,5 is not good -> 1,6,11,2,1,6,11,3,1,6,11,4 ..... 1,6,11,13 is much better.
Good to know, thanks for pointing it out.

> From other users I know that hcxdumptool is working if the rtl8812au driver initialized the device successfully. Unfortunately that sometimes fail.
Yes that's the problem with the rtl8812au driver. Although the adapter is really good but dealing with the driver issues is really frustrating.

BTW specifying multiple frequencies `-c 1,...` doesn't find a lot of networks when using `--do_rcascan `  

I used this command and left it to run for about 10 minutes and the result was 6 Networks on channel 1 and 2 only, although when I specified channel 1 only more than 10 Networks were found instantly. Am I missing something in the command?
`sudo hcxdumptool -i wlan0 --do_rcascan -c 1,2,3,4,5,6,7,8,9,10,11,12,13`

Same thing happens if I specified less channels such as `-c 1,2`

ZerBea, 2022-08-17:
That depend on the stay time on the channel and the BEACON interval of the APs.
A BEACON is transmitted ever 100ms = 10 BEACONs a second from a single AP
If you scan e.g. 13 channels the chance to get a BEACON of an AP is 1:13 (only).

OranGe-66, 2022-08-17:
Makes sense. 
I tried increasing the stay time but the result didn't change which makes sense since the default stay time which is 4 seconds is more than enough I guess.
In my case is there a method that allows for more BEACONs or I just have to wait?

ZerBea, 2022-08-17:
The stay time on rca_scan mode is 1 second. That is more than enough, because we are waiting for PROBERESPNSEs and BEACONs only.
Please notice that the intensity of an attack decreases (by automatic) the longer you stay on a channel. That prevent spamming a channel if the target is out of tx range or if the target ignore the attacks. That behavior can be controlled by:
```
--stop_client_m2_attacks=<digit>   : stop attacks against CLIENTS after 10 M2 frames received
                                     affected: ap-less (EAPOL 2/4 - M2) attack
                                     require hcxpcangtool --all option
                                     warning: that can prevent that a CLIENT can establish a connection to an assigned ACCESS POINT
--stop_ap_attacks=<digit>          : stop attacks against ACCESS POINTs if <n> BEACONs received
                                     default: stop after 600 BEACONs
--resume_ap_attacks=<digit>        : resume attacks against ACCESS POINTs after <n> BEACONs received
                                     default: 864000 BEACONs
```

Currently I'm testing a dynamic channel management (hcxlabtool). If that is working as expected, I'll add it to hcxdumptool.

ZerBea, 2023-01-25:
Latest git head should work now on rt88xxau driver (rtl8811au/rtl8812au/rtl8814au)
https://github.com/aircrack-ng/rtl8812au

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.