ZerBea/hcxdumptool #228: Wi-Fi Coconut.

nxfxcom, 2022-09-22:
Hello, has anybody managed yet to get the Hak5 coconut to work with the HCXdumptool? Would be such an amazing combo.

ZerBea, 2022-09-23:
According to this:
https://docs.hak5.org/wifi-coconut/software-and-drivers/native-drivers
the Coconut is running chipsets which are based on the rt2800usb driver. The driver support full monitor mode and full packet injection (active). 

Unfortunately it looks the hardware is not designed to be used by an interactive tool: 
"It is designed as a passive capture tool and the antennas are not optimized or connected for transmit mode, and excessive transmit attempts at higher powers may damage your device."
There is a clear warning:
"DO NOT use WiFi Coconut radios to transmit — doing so may damage the device and void the Hak5 limited warranty."

Closed this issue report, because it is not an hcxdumptool issue.
But if somebody is able to run hcxdumptool feel free to post this here.

BTW:
Running hcxdumptool (by default options) my devices are growing really hot!.

closed 2022-09-23 by ZerBea

RealEnder, 2022-09-23:
I don't have such device, but will be interesting to see the results. hcxdumptool has `--silent` option, which will not transmit at all. It's not much useful, but still capturing on all channels has its merits. With some hw mod this can be viable, but here I suspect will hit some issues.

nxfxcom, 2022-09-23:
Hello,

Happy to test anything and share output if it's helpful. I can't really use it so it's okay if I break it :)