ZerBea/hcxdumptool #229: cant run make

eliddell1, 2022-10-26:
I am on raspbian with openssl version 1.1.01
when i clone and run make i get the following error:

```
hcxdumptool.c:3900:9: warning: implicit declaration of function ‘EVP_PKEY_new_CMAC_key’ [-Wimplicit-function-declaration]
  pkey = EVP_PKEY_new_CMAC_key(NULL, testptk, 16, EVP_aes_128_cbc());
         ^~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:3900:7: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  pkey = EVP_PKEY_new_CMAC_key(NULL, testptk, 16, EVP_aes_128_cbc());
       ^
/tmp/ccE6sVM7.o: In function `detectweakwpa':
hcxdumptool.c:(.text.unlikely+0x1f00): undefined reference to `EVP_PKEY_new_CMAC_key'
collect2: error: ld returned 1 exit status
make: *** No rule to make target 'hcxdumptool', needed by 'build'.  Stop.
```
any ideas on how I can build the hcxdumptool

eliddell1, 2022-10-26:
libcurl4-openssl-dev is already the newest version (7.52.1-5+deb9u16).
libssl-dev is already the newest version (1.1.0l-1~deb9u6).
pkg-config is already the newest version (0.29-4).

ZerBea, 2022-10-27:
version 1.1.1 is required:
e.g. on Debian: libssl-dev (1.1.1n-0+deb10u3)
https://packages.debian.org/buster/libssl-dev
e.g. on Arch: openssl 1.1.1.q-1
https://archlinux.org/packages/core/x86_64/openssl/

Are you still on stretch? -> update to buster or better bookworm
https://buildd.raspbian.org/status/logs.php?pkg=openssl

BTW:
curl is outdated, too:
e.g. on Debian: libcurl4-openssl-dev (7.85.0-1~bpo11+1)
e.g. on Arch: curl 7.86.0-1
https://archlinux.org/packages/core/x86_64/curl/

ZerBea, 2022-10-27:
To retrieve the version of your distribution run
$ cat /etc/os-release

ZerBea, 2022-10-29:
I think we can close this issue, because it is related to an outdated and EOL Debian Version (stretch).
The package libssl-dev (1.1.0l-1~deb9u6 you mentioned belongs to Debian stretch:
https://packages.debian.org/en/stretch/libssl-dev

According to the release information
https://www.debian.org/releases/stretch/index.en.html
```
Debian stretch is EOL and will not receive updates any longer
Debian 9 has been superseded by Debian 10 ("buster"). Security updates have been discontinued as of July 6th, 2020.
However, stretch benefits from Long Term Support (LTS) until the end of June 2022. The LTS is limited to i386, amd64, armel, armhf and arm64. All other architectures are no longer supported in stretch. For more information, please refer to the [LTS section of the Debian Wiki](https://wiki.debian.org/LTS).
```

closed 2022-10-29 by ZerBea