ZerBea/hcxdumptool #230: hcxdumptool not working under TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]

abidmaria, 2022-11-18:
Hello ,
I am trying to use hcxdumptool , but i am getting the below errors :

sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15   

`
initialization of hcxdumptool 6.2.7-17-g87fcd83 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
driver doesn't support ioctl() SIOCSIWFREQ
frequency test failed
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored

`

Note that my wireless adaptor support monitor and injection  : 

`
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11bgn  ESSID:"Ooredoo_96"  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.422 GHz  Access Point: 1A:59:C0:3F:E9:39   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=1/100  Signal level=1/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

`

`

07:36:20  Trying broadcast probe requests...
07:36:20  Injection is working!
07:36:21  Found 3 APs

07:36:21  Trying directed probe requests...
07:36:21  1A:59:C0:3F:E9:39 - channel: 3 - 'Ooredoo_96'
07:36:22  Ping (min/avg/max): 4.238ms/13.395ms/28.225ms Power: -71.00
07:36:22  30/30: 100%

07:36:22  E8:9F:80:26:99:A2 - channel: 3 - 'Ooredoo_96_Ext'
07:36:22  Ping (min/avg/max): 0.451ms/6.033ms/16.040ms Power: -64.73
07:36:22  30/30: 100%

07:36:22  1E:59:C0:3F:E9:39 - channel: 3 - ''
07:36:22  Ping (min/avg/max): 4.385ms/13.173ms/30.151ms Power: -73.40
07:36:22  30/30: 100%

`I am using TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]

└─$ lsusb
Bus 001 Device 002: ID 2357:0109 TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Can you please help
Thanks

ZerBea, 2022-11-21:
It looks like this driver depend on NETLINK and doesn't support some (or all) ioctl() calls required by hcxdumptool.
Please connect the device. If it is already connected, please disconnect and reconnect it.
Now stop all services that take access to the device. Please do not use airmon-ng or iw.

Than run the following commands and post the output:
$ hcxdumptool -v
$ hcxdumptool -I
$ hcxdumptool -i wlan0 --check_driver

Difference betweeen aircrack-ng and hcxdumptool:
aircrack use NETLINK and ioctl() system calls
while
hcxdumptool uso only ioctl() system calls. The reason why I don't use NETLINK is described here:
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls?share=1

ZerBea, 2022-11-22:
Some more information about WiFi drivers is here:
https://github.com/ZerBea/hcxdumptool/issues/234

ZerBea, 2022-11-23:
To identify the driver, please comment the output of
$ hcxdumptool -I wlan0
I guess it's not a stock kernel driver.

abidmaria, 2022-11-23:
Thank you for your feedback 
as requested , below is the output  :

┌──(dimitri㉿kali)-[~]
└─$ hcxdumptool -I wlan0
**wlan interfaces:
phy0    1c61b41fb584    wlan0   (driver:rtl88XXau)**

ZerBea, 2022-11-24:
I think we can close this report, because other drivers are working as expected.

closed 2022-11-24 by ZerBea

abidmaria, 2022-11-24:
Agree , so it's better to include them in the future (if possible) , thank you for your excellent support.

ZerBea, 2022-11-24:
This is the git hub of rtl88XXau:
https://github.com/aircrack-ng/rtl8812au
You can issues related to the driver report there.

As mentioned before, the stock kernel driver
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtl8xxxu?h=v6.0.9
is working as expected:
 
```
$ hcxdumptool -I
wlan interfaces:
phy4	f81a670dc527	wlp39s0f3u1u4	(driver:rtl8192cu)
[zerobeat@tux1 ~]$ sudo hcxdumptool -i wlp39s0f3u1u4 -C
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
wlp39s0f3u1u4 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 ( 0 dBm)

terminating...
```
while the third party driver doesn't work.

ZerBea, 2022-11-24:
Another example of a working Realtek stock kernel driver. This time RTL8821CE PCIe card:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtw88?h=v6.0.9
```
$ lspci
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter

$ hcxdumptool -I
wlan interfaces:
phy0	70665578cdab	wlp4s0	(driver:rtw_8821ce)

$ sudo hcxdumptool -i wlp4s0 -C
initialization of hcxdumptool 6.2.7-13-gace520c (depending on the capabilities of the device, this may take some time)...
wlp4s0 available frequencies, channels and tx power reported by driver:
2412MHz   1 (20 dBm)
2417MHz   2 (20 dBm)
2422MHz   3 (20 dBm)
2427MHz   4 (20 dBm)
2432MHz   5 (20 dBm)
2437MHz   6 (20 dBm)
2442MHz   7 (20 dBm)
2447MHz   8 (20 dBm)
2452MHz   9 (20 dBm)
2457MHz  10 (20 dBm)
2462MHz  11 (20 dBm)
2467MHz  12 (20 dBm)
2472MHz  13 (20 dBm)
2484MHz  14 ( 0 dBm)
5180MHz  36 (30 dBm)
5200MHz  40 (30 dBm)
5220MHz  44 (30 dBm)
5240MHz  48 (30 dBm)
5260MHz  52 (24 dBm)
5280MHz  56 (24 dBm)
5300MHz  60 (24 dBm)
5320MHz  64 (24 dBm)
5500MHz 100 (24 dBm)
5520MHz 104 (24 dBm)
5540MHz 108 (24 dBm)
5560MHz 112 (24 dBm)
5580MHz 116 (24 dBm)
5600MHz 120 (24 dBm)
5620MHz 124 (24 dBm)
5640MHz 128 (24 dBm)
5660MHz 132 (24 dBm)
5680MHz 136 (24 dBm)
5700MHz 140 (24 dBm)
5720MHz 144 (24 dBm)
5745MHz 149 (30 dBm)
5765MHz 153 (30 dBm)
5785MHz 157 (30 dBm)
5805MHz 161 (30 dBm)
5825MHz 165 (30 dBm)

terminating...
```

ZerBea, 2023-01-25:
Latest git head hcxdumptool should work now running this driver:
https://github.com/kimocoder/realtek_rtwifi
https://github.com/kimocoder/realtek_rtwifi/issues/34

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.