ZerBea/hcxdumptool #231: zsh: segmentation fault  sudo hcxdumptool -i wlan0 -o output.pcapng --enable_status=15

abidmaria, 2022-11-19:
Hello 

I am using the below command and it works fine , i can get the beacon, ESSID ...but i am getting the below error 

zsh: segmentation fault  sudo hcxdumptool -i wlan0 -o output.pcapng --enable_status=15

Do you have any idea ?

Thanks

ZerBea, 2022-11-21:
I can't reproduce the segmentation fault, too, but I have an idea. It may related to the combination zsh & sudo.
Which version of sudo and zsh do you use?

abidmaria, 2022-11-21:
Hello , thank you for your feedback , below is the output :

Sudo version  :
└─$ sudo -V
Sudo version 1.9.11p3
Sudoers policy plugin version 1.9.11p3
Sudoers file grammar version 48
Sudoers I/O plugin version 1.9.11p3
Sudoers audit plugin version 1.9.11p3

zsh version  : 

└─$ zsh --version
**zsh 5.9 (x86_64-debian-linux-gnu)**

ZerBea, 2022-11-21:
Thanks.
Maybe your problem is related to this one:
https://github.com/nmap/nmap/issues/2482
because hcxdumptool/hcxtools using OpenSSL 3, too.
But I'm not sure.

ZerBea, 2022-11-22:
Started another test. This time hcxdumptool is compiled with debug options (Makefile: -ggdb -fsanitize=address):
```
CC		?= gcc
CFLAGS		?= -O3 -Wall -Wextra
CFLAGS		?= -ggdb -fsanitize=address
CFLAGS		+= -std=gnu99
```
still no segmentation fault.

ZerBea, 2022-11-22:
Please do a fresh git clone:
$ git clone https://github.com/ZerBea/hcxdumptool
Than replace the Makefile with the attached one (need to be decompressed):
[Makefile.zip](https://github.com/ZerBea/hcxdumptool/files/10067040/Makefile.zip)
than run
$ make
followed by
when finished run hcxdumptool:
$ sudo ./hcxdumptool -i wlan0 -o output.pcapng --enable_status=15
If it crashed, please comment the crash report.

ZerBea, 2022-11-22:
Ok, there is already a git clone on your system so you can proceed with the replacement.

ZerBea, 2022-11-22:
Just rename the old Makefile to Makefile.old
That will make it easier to go back.

ZerBea, 2022-11-23:
Ok,thanks.
There are no debug information from hcxdumptool so it could be either the zsh or sudo that caused the segmentation fault.
Please move to su and start hcxdumtool again
$ su
$ ./hcxdumptool -i wlan0 -o output.pcapng --enable_status=15

abidmaria, 2022-11-23:
Thank for your feedback and support : 
below is the output  :  

┌──(dimitri㉿kali)-[~]
└─$ su
Password: 
su: Authentication failure

i don't know the password's root ..so i have executed the command **sudo su**
                                                                                                                                                                      
┌──(dimitri㉿kali)-[~]
└─$ **sudo su**                                   

then : 
┌──(root㉿kali)-[/home/dimitri]
└─# **./hcxdumptool -i wlan0 -o output.pcapng --enable_status=15**
**zsh: permission denied: ./hcxdumptool**
                                                                                                                                                                      
┌──(root㉿kali)-[/home/dimitri]
└─# 

Thanks

ZerBea, 2022-11-23:
Ok, please try this:
$ sudo ls
Do you get a segmentation fault when trying to list the content of the directory?

ZerBea, 2022-11-23:
Now I'm sure your issue is related to sudo/su (not to hcxdumptool or zsh) like this issue, mentioned here:
https://bbs.archlinux.org/viewtopic.php?id=257278
That caused the segmentation fault due to insufficient permissions.

abidmaria, 2022-11-23:
Thank you , 
below is the output , i didn't get segmentation list when trying to list the content 

┌──(dimitri㉿kali)-[~]
└─$ sudo ls
[sudo] password for dimitri: 
Desktop  Documents  Downloads  hcxdumptool  Music  Pictures  Public  Templates  test  Videos
                                                                                                                                                                      
┌──(dimitri㉿kali)-[~]
└─$

poliang95, 2022-11-23:
Hi, mine also have the same problem, run for 2-5 minute and segmentation fault

┌──(root㉿kali)-[/home/kali]
└─# sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15 
initialization of hcxdumptool 6.2.7-17-g87fcd83 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls

start capturing (stop with ctrl+c)
NMEA 0183 PROTOCOL........: N/A
PHYSICAL INTERFACE........: phy0
INTERFACE NAME............: wlan0
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 00c0ca9897d7 (not used for the attack)
INTERFACE VIRTUAL MAC.....: 00c0ca9897d7 (not used for the attack)
DRIVER....................: ath9k_htc (this driver is not recommended - expect driver errors)
DRIVER VERSION............: 6.0.0-kali3-amd64
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 20d160093b17 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 20d160093b18 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 20d160093b19 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: b4e1ebf93309
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64663
ANONCE....................: 29f2cc327ab0545f03fd81dd340f0a4591e59052be4c9f01e42eec8a70d3622e
SNONCE....................: 4833d8f426411dedb90c6da3e43349b238dd190b0313aba27e670c21997b9873

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
12:52:54 2412/1   ffffffffffff a0a3f0992060 shafiq@eyra@unifi [BEACON]
12:52:54 2412/1   ffffffffffff d44f67d66c64 annac [BEACON]
12:52:54 2412/1   ffffffffffff ceb1825579c4 Nathan House_2.4 [BEACON]
...
...
...
12:54:06 2432/5   72209e44f974 98dac408bb93 liang ah peng-TIME2.4GHz [AUTHENTICATION]
12:54:06 2432/5   72209e44f974 98dac408bb93 liang ah peng-TIME2.4GHz [ASSOCIATION]
zsh: segmentation fault  sudo hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon

abidmaria, 2022-11-24:
> sudo dmesg > /tmp/dmesg.txt

Hello , thank you for feedback , below is the dmesg log

` [    0.000000] Linux version 6.0.0-kali3-amd64 (devel@kali.org) (gcc-12 (Debian 12.2.0-3) 12.2.0, GNU ld (GNU Binutils for Debian) 2.39) #1 SMP PREEMPT_DYNAMIC Debian 6.0.7-1kali1 (2022-11-07)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.0.0-kali3-amd64 root=UUID=569fc278-60cb-4708-83d2-2a9575bd56f1 ro quiet splash
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] signal: max sigframe size: 1776
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000023ebfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000003] kvm-clock: using sched offset of 12700398123 cycles
[    0.000006] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000011] tsc: Detected 2304.000 MHz processor
[    0.003369] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003377] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003385] last_pfn = 0x23ec00 max_arch_pfn = 0x400000000
[    0.003400] Disabled
[    0.003402] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.003405] CPU MTRRs all blank - virtualized system.
[    0.003409] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.003416] last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[    0.003479] found SMP MP-table at [mem 0x0009fff0-0x0009ffff]
[    0.004544] RAMDISK: [mem 0x2faa1000-0x33d47fff]
[    0.004552] ACPI: Early table checksum verification disabled
[    0.004556] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.004563] ACPI: XSDT 0x00000000DFFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.004571] ACPI: FACP 0x00000000DFFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.004579] ACPI: DSDT 0x00000000DFFF0610 002353 (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.004585] ACPI: FACS 0x00000000DFFF0200 000040
[    0.004590] ACPI: FACS 0x00000000DFFF0200 000040
[    0.004595] ACPI: APIC 0x00000000DFFF0240 000054 (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.004601] ACPI: SSDT 0x00000000DFFF02A0 00036C (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.004606] ACPI: Reserving FACP table memory at [mem 0xdfff00f0-0xdfff01e3]
[    0.004608] ACPI: Reserving DSDT table memory at [mem 0xdfff0610-0xdfff2962]
[    0.004610] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.004611] ACPI: Reserving FACS table memory at [mem 0xdfff0200-0xdfff023f]
[    0.004613] ACPI: Reserving APIC table memory at [mem 0xdfff0240-0xdfff0293]
[    0.004614] ACPI: Reserving SSDT table memory at [mem 0xdfff02a0-0xdfff060b]
[    0.004902] No NUMA configuration found
[    0.004904] Faking a node at [mem 0x0000000000000000-0x000000023ebfffff]
[    0.004916] NODE_DATA(0) allocated [mem 0x23ebd1000-0x23ebfbfff]
[    0.005441] Zone ranges:
[    0.005442]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.005446]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.005449]   Normal   [mem 0x0000000100000000-0x000000023ebfffff]
[    0.005451]   Device   empty
[    0.005453] Movable zone start for each node
[    0.005457] Early memory node ranges
[    0.005458]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.005460]   node   0: [mem 0x0000000000100000-0x00000000dffeffff]
[    0.005463]   node   0: [mem 0x0000000100000000-0x000000023ebfffff]
[    0.005466] Initmem setup node 0 [mem 0x0000000000001000-0x000000023ebfffff]
[    0.005552] On node 0, zone DMA: 1 pages in unavailable ranges
[... 591 lines cut ...]
[25232.671223] Monitor 0 (w,h)=(640,480) (x,y)=(0,0)
[25232.671986] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[25232.672404] RRScreenChangeNotify event received
[25232.811298] Monitor 0 (w,h)=(640,480) (x,y)=(0,0)
[25232.812226] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[25232.812759] RRScreenChangeNotify event received
[25232.857031] Monitor 0 (w,h)=(640,480) (x,y)=(0,0)
[25232.858236] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[25253.202241] New guest source window 0x1c007d1 ('Thunar')
[25260.535404] Processing message 102 failed with VERR_TIMEOUT
[... cut ...]
[25632.904899] Monitor 0 (w,h)=(1366,663) (x,y)=(0,0)
[25632.906095] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[25632.923203] RRScreenChangeNotify event received
[25632.947441] Monitor 0 (w,h)=(1366,663) (x,y)=(0,0)
[25632.948191] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
[25632.948666] RRScreenChangeNotify event received
[25633.094509] Monitor 0 (w,h)=(1366,663) (x,y)=(0,0)
[25633.095296] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
  `

ZerBea, 2022-11-24:
Thanks.
Unfortunately that is unreadable. Please attach the txt file

ZerBea, 2022-11-24:
@poliang95 thanks.
Same conditions: KALI + ZSH + sudo == segmentation fault.
Are you running KALI inside a VM?
To identify what caused it exactly (zsh or sudo), please run hcxdumptool in a bash shell.

Also please move to su and than run hcxdumptool without sudo.
$ su
$ hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=15

ZerBea, 2022-11-24:
I'm not able to reproduce that issue running this environment (no VM):
```
$ uname -r
6.0.9-arch1-1

$ bash --version
bash --version
GNU bash, Version 5.1.16(1)-release (x86_64-pc-linux-gnu)

$ sudo --version
sudo --version
Sudo version 1.9.12p1
Sudoers policy plugin version 1.9.12p1
Sudoers file grammar version 48
Sudoers I/O plugin version 1.9.12p1
Sudoers audit plugin version 1.9.12p1

```

running as expected 
```
$ sudo hcxdumptool -i wlp39s0f3u1u4 -o dumpfile.pcapng --active_beacon --enable_status=79
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...

start capturing (stop with ctrl+c)
NMEA 0183 PROTOCOL........: N/A
PHYSICAL INTERFACE........: phy1
INTERFACE NAME............: wlp39s0f3u1u4
INTERFACE PROTOCOL........: IEEE 802.11
INTERFACE TX POWER........: 20 dBm (lowest value reported by the device)
INTERFACE HARDWARE MAC....: 74da38f84e1b (not used for the attack)
INTERFACE VIRTUAL MAC.....: 74da38f84e1b (not used for the attack)
DRIVER....................: mt7601u
DRIVER VERSION............: 6.0.9-arch1-1
DRIVER FIRMWARE VERSION...: N/A
openSSL version...........: 1.0
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 00269f5915d4 (BROADCAST WILDCARD used for the attack)
ACCESS POINT (ROGUE)......: 00269f5915d5 (BROADCAST OPEN used for the attack)
ACCESS POINT (ROGUE)......: 00269f5915d6 (used for the attack and incremented on every new client)
CLIENT (ROGUE)............: c8aaccdffec1
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 63452
ANONCE....................: 03aefc3b31c004bdf6d6e50de95d73993eee9abe8970e5f679f5bd34ddc6b1c6
SNONCE....................: 1ef853835f24e5e9ca996f555a9fa076c73b2a4c7f0dae5cce5f82a791d0dfef

TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE]
07:02:20 2417/2   ffffffffffff a58094302cc2f TEST_NET [BEACON]
07:03:00 2467/12    ERROR:0 INCOMING:763 AGE:2 OUTGOING:758 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:04:00 2412/1     ERROR:0 INCOMING:3359 AGE:1 OUTGOING:1936 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:05:00 2422/3     ERROR:0 INCOMING:6205 AGE:5 OUTGOING:3060 PMKIDROGUE:1 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:06:00 2432/5     ERROR:0 INCOMING:7464 AGE:1 OUTGOING:4225 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:07:00 2442/7     ERROR:0 INCOMING:10011 AGE:1 OUTGOING:5372 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:08:00 2452/9     ERROR:0 INCOMING:11076 AGE:1 OUTGOING:6550 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:09:00 2462/11    ERROR:0 INCOMING:12260 AGE:17 OUTGOING:7737 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
...
07:10:00 2472/13    ERROR:0 INCOMING:13614 AGE:1 OUTGOING:8906 PMKIDROGUE:2 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
^C
terminating...
```

ZerBea, 2022-11-24:
Tried again to force the "segmentation error" - no luck:
```
$ sudo gdb --args ./hcxdumptool -i wlp39s0f3u1u4 -o /tmp/dumpfile.pcapng --active_beacon --enable_status=79
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./hcxdumptool...
(gdb) set logging on
(gdb) r
Starting program: /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool -i wlp39s0f3u1u4 -o /tmp/dumpfile.pcapng --active_beacon --enable_status=79
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
initialization of hcxdumptool 6.2.7-22-gc65aac5 (depending on the capabilities of the device, this may take some time)...
[Detaching after vfork from child process 6442]
[Detaching after vfork from child process 6443]
[Detaching after vfork from child process 6444]
[Detaching after vfork from child process 6445]
[Detaching after vfork from child process 6446]
[Detaching after vfork from child process 6447]

start capturing (stop with ctrl+c)
...
^C
Program received signal SIGINT, Interrupt.
0x00007ffff71169a0 in pselect () from /usr/lib/libc.so.6
(gdb) 

```

By latest commit, I added uncommented debug options.
Please to a fresh git clone
comment debug information in Makefile
#CFLAGS		+= -ggdb -fsanitize=address
to
CFLAGS		+= -ggdb -fsanitize=address

$ make clean
$ make
$ sudo gdb --args ./hcxdumptool -i wlp39s0f3u1u4 -o /tmp/dumpfile.pcapng --active_beacon --enable_status=79
on (gdb) prompt type "set logging on" and "r"
(gdb) set logging on
(gdb) r
now hcxdumptool will start and everything went in gdb.txt log file
wait for the segmentaion fault
quit gdb
(gdb) q
confirm with y
than attach gdb.txt here.

ZerBea, 2022-11-24:
If the error is caused by hcxdumptool, you'll see something similar to this (example code to force an error):
```
=================================================================
==10689==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7fffffffe5d0 at pc 0x555555555280 bp 0x7fffffffe570 sp 0x7fffffffe560
WRITE of size 1 at 0x7fffffffe5d0 thread T0
    #0 0x55555555527f in main /home/zerobeat/temp/sonstige/Simple.c:30
    #1 0x7ffff763c28f  (/usr/lib/libc.so.6+0x2328f)
    #2 0x7ffff763c349 in __libc_start_main (/usr/lib/libc.so.6+0x23349)
    #3 0x5555555550d4 in _start ../sysdeps/x86_64/start.S:115

Address 0x7fffffffe5d0 is located in stack of thread T0 at offset 64 in frame
    #0 0x5555555551b8 in main /home/zerobeat/temp/sonstige/Simple.c:24

  This frame has 1 object(s):
    [32, 64) 'essidlen' (line 26) <== Memory access at offset 64 overflows this variable
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/zerobeat/temp/sonstige/Simple.c:30 in main
Shadow bytes around the buggy address:
  0x10007fff7c60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7c70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7c80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7c90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7ca0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x10007fff7cb0: 00 00 f1 f1 f1 f1 00 00 00 00[f3]f3 f3 f3 00 00
  0x10007fff7cc0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7cd0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7ce0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7cf0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x10007fff7d00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==10689==ABORTING
[Inferior 1 (process 10689) exited with code 01]
(gdb) 
```

abidmaria, 2022-11-24:
> Thanks. Unfortunately that is unreadable. Please attach the txt file
Hello as requested 

[dmesg.txt](https://github.com/ZerBea/hcxdumptool/files/10083351/dmesg.txt)

ZerBea, 2022-11-24:
Thanks. Unfortunately nothing inside that can help us.
Her hcxdumptool entered promiscuous mode:
[ 6119.561985] device wlan0 entered promiscuous mode
...
[25633.095296] Sending monitor positions (8 of them)  to the host: VINF_SUCCESS
here ends the log.

Looks like we need to take a look at gdb.txt log to figure out whats going on.

abidmaria, 2022-11-24:
> 

Thank you for your feedback , below is my comment  :
I have changed root password since i don't know the default one  .
script : Test .sh
`  
#!/bin/bash
su root -c 'hcxdumptool -i wlan0 -o dumpfile.pcapng --active_beacon --enable_status=79'

`
Output of the Script :

[Script_output.txt](https://github.com/ZerBea/hcxdumptool/files/10084023/Script_output.txt)

i got "Segmentation fault" , please refer to attached file..

PS : i have changed the values of "MAC_SOURCE"  " ESSID"  in the file ..so please ignore this ..

ZerBea, 2022-11-24:
Thanks. We need the exact position where the segmentation error occur.
I still have no idea where this happens

BTW:
--enable_status=64 will show only status messages (every minute one) - that's enough to hunt for the issue.
` 07:04:00 2412/1     ERROR:0 INCOMING:3359 AGE:1 OUTGOING:1936 PMKIDROGUE:0 PMKID:0 M1M2ROGUE:0 M1M2:0 M2M3:0 M3M4:0 M3M4ZEROED:0 GPS:0
`

Now, as you changed the password, you can run gdb as described here:
https://github.com/ZerBea/hcxdumptool/issues/231#issuecomment-1326054533
Maybe the gdb.txt log is more useful for us.

ZerBea, 2022-11-24:
Just dived into KALI documentation and noticed that bash is still present.
To move to bash open a terminal than type 
$ bash
$ hcxdumptool -i wlan0-o dumpfile.pcapng --active_beacon --enable_status=15
now hcxdumptool should run inside a bash shell

abidmaria, 2022-11-24:
> git clone https://github.com/ZerBea/hcxdumptool

Thank you for your feedback ,
below is my comments :
1- git clone https://github.com/ZerBea/hcxdumptool
`
└─# git clone https://github.com/ZerBea/hcxdumptool
Cloning into 'hcxdumptool'...
remote: Enumerating objects: 4244, done.
remote: Counting objects: 100% (935/935), done.
remote: Compressing objects: 100% (352/352), done.
remote: Total 4244 (delta 613), reused 746 (delta 582), pack-reused 3309
Receiving objects: 100% (4244/4244), 1.40 MiB | 1000.00 KiB/s, done.
Resolving deltas: 100% (2875/2875), done.

`

2.Take backup of Makefile

`
┌──(root㉿kali)-[/home/dimitri/Lab/hcxdumptool]
└─# ls -ltra 
total 488
drwxr-xr-x 3 dimitri dimitri   4096 Nov 24 04:50 ..
-rw-r--r-- 1 root    root     12120 Nov 24 04:50 README.md
-rw-r--r-- 1 root    root      1670 Nov 24 04:50 Makefile
-rw-r--r-- 1 root    root       123 Nov 24 04:50 .gitmodules
-rw-r--r-- 1 root    root        21 Nov 24 04:50 .gitignore
drwxr-xr-x 2 root    root      4096 Nov 24 04:50 docs
-rw-r--r-- 1 root    root     71647 Nov 24 04:50 changelog
-rw-r--r-- 1 root    root       217 Nov 24 04:50 Android.mk
-rw-r--r-- 1 root    root       161 Nov 24 04:50 AndroidManifest.xml
drwxr-xr-x 2 root    root      4096 Nov 24 04:50 usefulscripts
drwxr-xr-x 2 root    root      4096 Nov 24 04:50 man
-rw-r--r-- 1 root    root      1081 Nov 24 04:50 license.txt
drwxr-xr-x 2 root    root      4096 Nov 24 04:50 jni
drwxr-xr-x 3 root    root      4096 Nov 24 04:50 include
-rw-r--r-- 1 root    root      8889 Nov 24 04:50 hcxpioff.c
-rw-r--r-- 1 root    root    342900 Nov 24 04:50 hcxdumptool.c
drwxr-xr-x 8 root    root      4096 Nov 24 04:50 .git
drwxr-xr-x 8 root    root      4096 Nov 24 04:50 .
                                                                                                                                                                      
┌──(root㉿kali)-[/home/dimitri/Lab/hcxdumptool]
└─# **cp Makefile Makefile.bk**

`

3. Change Makefile , uncomment debug information in Makefile  . 
4. 
$ make clean
$ make
$ sudo gdb --args ./hcxdumptool -i wlan0 -o /tmp/dumpfile.pcapng --active_beacon --enable_status=79

This is the output : 

`
└─$  sudo gdb --args ./hcxdumptool -i wlan0 -o /tmp/dumpfile.pcapng --active_beacon --enable_status=79
GNU gdb (Debian 12.1-4) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

[... 94 lines cut ...]
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==2009==ABORTING
**[Inferior 1 (process 2009) exited with code 01]**
(gdb) q

Please find attached the gdb.txt file 

`
[gdb.txt](https://github.com/ZerBea/hcxdumptool/files/10085489/gdb.txt)

ZerBea, 2022-11-24:
Great, thanks.

Now I'm able to reproduce the segmentation fault:

```
debug detektweak
=================================================================
==2572==ERROR: AddressSanitizer: global-buffer-overflow on address 0x5555555e8a80 at pc 0x7ffff785f427 bp 0x7fffffffe6a0 sp 0x7fffffffde48
WRITE of size 38656 at 0x5555555e8a80 thread T0
    #0 0x7ffff785f426 in __interceptor_memcpy /usr/src/debug/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:827
    #1 0x555555578716 in detectweakwpa /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:3677
    #2 0x5555555859db in printeapol /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:3775
    #3 0x55555559ca0d in process80211eapol_m3 /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:3967
    #4 0x55555559ca0d in process80211eapol /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:4270
    #5 0x5555555a4a7a in process80211eap /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:4289
    #6 0x5555555a4a7a in process_packet /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:5771
    #7 0x5555555abe8e in process_fd /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:6114
    #8 0x555555570b8d in main /home/zerobeat/Source/Programme/git_hcxdumptool/hcxdumptool.c:9518
    #9 0x7ffff703c28f  (/usr/lib/libc.so.6+0x2328f)
    #10 0x7ffff703c349 in __libc_start_main (/usr/lib/libc.so.6+0x23349)
    #11 0x5555555725a4 in _start ../sysdeps/x86_64/start.S:115

0x5555555e8a80 is located 32 bytes to the left of global variable 'keymic' defined in 'hcxdumptool.c:3660:16' (0x5555555e8aa0) of size 16
0x5555555e8a80 is located 0 bytes to the right of global variable 'eapoltmp' defined in 'hcxdumptool.c:3662:16' (0x5555555e8680) of size 1024
SUMMARY: AddressSanitizer: global-buffer-overflow /usr/src/debug/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:827 in __interceptor_memcpy
Shadow bytes around the buggy address:
  0x0aab2aab5100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aab5110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aab5120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aab5130: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0aab2aab5140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0aab2aab5150:[f9]f9 f9 f9 00 00 f9 f9 f9 f9 f9 f9 04 f9 f9 f9
  0x0aab2aab5160: f9 f9 f9 f9 00 00 f9 f9 f9 f9 f9 f9 00 f9 f9 f9
  0x0aab2aab5170: f9 f9 f9 f9 04 f9 f9 f9 f9 f9 f9 f9 04 f9 f9 f9
  0x0aab2aab5180: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 00 00 00 00
  0x0aab2aab5190: 00 00 00 00 01 f9 f9 f9 f9 f9 f9 f9 01 f9 f9 f9
  0x0aab2aab51a0: f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9 01 f9 f9 f9
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==2572==ABORTING
[Inferior 1 (process 2572) exited with code 01]

```

ZerBea, 2022-11-24:
Ok, pushed an update.
This commit should fix the segmentation fault:
https://github.com/ZerBea/hcxdumptool/commit/a64983fc50b7a29cb77e33d4b93fa12c0a6a28f1
Please try.

ZerBea, 2022-11-24:
Found some more problems when detecting weak EAPOL and fixed them.
Please test latest git head.
If everything is working as expected, we can close this issue report.

abidmaria, 2022-11-24:
Thank you , it works fine but now i am not able to found **dumpfile.pcapng** , file has not been generated ...

hcxdumptool -i wlan0 dumpfile.pcapng --active_beacon --enable_status=15

abidmaria, 2022-11-25:
Hello , also i would like to keep only one hcxdumptool directory , i have noticed that i have created 2 since i was doing some testing with Makefile and bash scripts ..., Please how to clean and keep only one ...or how to remove all then i can install the fresh one ...

directory : **/home/dimitri**

-rw-------  1 dimitri dimitri    20 Nov 23 09:04 .lesshst
drwxr-xr-x  3 root    root     4096 Nov 23 19:01 hs
-rw-------  1 dimitri dimitri    77 Nov 24 10:41 .bash_history
**drwxr-xr-x  9 dimitri dimitri  4096 Nov 24 10:49 hcxdumptool**
drwxr-xr-x  3 dimitri dimitri  4096 Nov 24 15:40 Lab
-rw-------  1 dimitri dimitri  4827 Nov 24 16:23 .zsh_history
-rw-------  1 dimitri dimitri  6056 Nov 24 16:23 .xsession-errors.old
-rw-------  1 dimitri dimitri    49 Nov 24 23:30 .Xauthority
-rw-r-----  1 dimitri dimitri     4 Nov 24 23:30 .vboxclient-clipboard.pid
-rw-r-----  1 dimitri dimitri     4 Nov 24 23:30 .vboxclient-seamless.pid
-rw-r-----  1 dimitri dimitri     4 Nov 24 23:30 .vboxclient-draganddrop.pid
-rw-r-----  1 dimitri dimitri     4 Nov 24 23:30 .vboxclient-display-svga-x11.pid
-rw-------  1 dimitri dimitri  5723 Nov 24 23:30 .xsession-errors
drwxr-xr-x 19 dimitri dimitri  4096 Nov 24 23:31 .

directory : **/home/dimitri/Lab**

-rw-r--r--  1 dimitri dimitri 1560 Nov 24 04:00 log.tmp
-rwx------  1 root    root     180 Nov 24 04:15 Test.sh
drwxr-xr-x  3 dimitri dimitri 4096 Nov 24 15:40 .
drwxr-xr-x  9 root    root    4096 Nov 24 15:40 **hcxdumptool**
drwxr-xr-x 19 dimitri dimitri 4096 Nov 24 23:31 ..

ZerBea, 2022-11-25:
Dump files are always stored int that directory from which you started hcxdumptool.
E.g. if you start hcxdumptool -o dumpfile.pcapng from /tmp your dump file is in /tmp/dumpfile.pcapng
I suggest to remove all hcxdumptool git clones.
Than get a fresh one
$ git clone https://github.com/ZerBea/hcxdumptool
$ cd hcxdumptool
$ make
$ make sudo make install
now create a working directory in $HOME (e.g. dumpfiles)
from now on, it is enough to tell hcxdumptool where this directory is:
$ hcxdumptool -i wlan0 -o $HOME/dumpfiles/dumpfile.pcapng --active_beacon --enable_status=15

This applies to all hcxtools.
E.g. hcxpcapngtool
create a hash directory in $HOME (e.g. hashfiles)
from now on, it is enough to tell hcxpcapngtool where this directory is:
$ hcxpcapngtool -o $HOME/hashfiles/hash.hc22000 $HOME/dumpfiles/*.*
or
create an additional wordlist folder to store weak candidates in $HOME (e.g. wordlists)
$ hcxpcapngtool -o $HOME/hashfiles/hash.hc22000 -R $HOME/wordlists/prlist $HOME/dumpfiles/*.*
That allow hashcat to test weak candidates:
$ hashcat -m 22000 $HOME/hashfiles/hash.hc22000 $HOME/wordlists/prlist

abidmaria, 2022-11-25:
Thank you , but how to remove  all hcxdumptool git clones ?

abidmaria, 2022-11-25:
> how to remove all hcxdumptool git clones ?

Please cofirm  : rm -r hcxdumptool , is it enough to remove all hcxdumptool git clones ?

ZerBea, 2022-11-25:
$ rm -r hcxdumptool 
will only remove the clone in the current directory
$ git clone https://github.com/ZerBea/hcxdumptool
will create a new one
$ make install
will overwrite hcxdumptool located in bin

abidmaria, 2022-11-25:
> $ rm -r hcxdumptool will only remove the clone in the current directory $ git clone https://github.com/ZerBea/hcxdumptool will create a new one $ make install will overwrite hcxdumptool located in bin

Thank you so much, for your excellent support.

- Please be informed that my home directory is **"/home/dimitri"**
- I have created 2 directories under home directory "**dumpfiles**" & "**hashfiles**"
- **hcxdumptool** directory is under my home directory 
- I am executing the command "hcxdumptool..." from my home directory 

So below are my observations related to **hcxdumptool**  command  :

**1)** -  I  got error  -> **failed to read packet: Network**   , please refer to the attached file (Network is down.txt)

[Network is down.txt](https://github.com/ZerBea/hcxdumptool/files/10093305/Network.is.down.txt)

**2)**- after this error , I have sent again the command without rebooting the system , i got error -->  **Operation not permitted** , please refer to the attached file (Operation not permitted.txt)

[Operation not permitted.txt](https://github.com/ZerBea/hcxdumptool/files/10093337/Operation.not.permitted.txt)

**3)**-after that , I have rebooted  the system with command " reboot " and now i am not able to receive data means NO (TIME     FREQ/CH  MAC_DEST     MAC_SOURCE   ESSID [FRAME TYPE] )  **,I run this command for more than 5  min** . please refer  to the attached file (No_TIME_FREQ_MAC_DEST _MAC_SOURCE_ESSID.txt)

[No_TIME_FREQ_MAC_DEST _MAC_SOURCE_ESSID.txt](https://github.com/ZerBea/hcxdumptool/files/10093350/No_TIME_FREQ_MAC_DEST._MAC_SOURCE_ESSID.txt)

4: finally , I shutdown the system and restart again VM , and start Kali ..and executed the same command , **it works fine** BUT the **dumpfile.pcapng file is not generated** , i run this command for more than 4 mins , please refer to the attached file (dumpfile.pcapng file is not generated.txt)

[dumpfile.pcapng file is not generated.txt](https://github.com/ZerBea/hcxdumptool/files/10093438/dumpfile.pcapng.file.is.not.generated.txt)

Thank you

ZerBea, 2022-11-25:
This is a system message:
`failed to read packet: Network is down
`
The device was disconnected either by the VM or a broken cable.

`failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver: Operation not permitted
`
Related to the first problem. The device is still disconnected.
You have to investigate why the VM disconnect the device.

`sudo hcxdumptool -i wlan0 /home/dimitri/dumpfiles/dumpfile.pcapng --active_beacon --enable_status=15
`
The dump file was not stored because you missed the option switch (-o). Correct command ist:
`sudo hcxdumptool -i wlan0 -o /home/dimitri/dumpfiles/dumpfile.pcapng --active_beacon --enable_status=15
`

abidmaria, 2022-11-25:
> This is a system message: `failed to read packet: Network is down ` The device was disconnected either by the VM or a broken cable.
>
> `failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by [...]

**Thank you so much for your excellent support and help**, it works fine , please you can close this thread.

ZerBea, 2022-11-25:
You're welcome.

closed 2022-11-25 by ZerBea