ZerBea/hcxdumptool #232: Project cannot compile

ShutdownRepo, 2022-11-20:
Hi, it seems the project doesn't compile anymore.
Checking out to 56d078de4d6f5cef07b378707ab478fde03168c0 works, meaning there were changes in the last 2 commits that induced those errors.

```
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto  -DVERSION_TAG=\"6.2.7-17-g87fcd83\" -DVERSION_YEAR=\"2022\"
hcxdumptool.c:60:8: error: unknown type name ‘EVP_MAC’
   60 | static EVP_MAC *hmac;
      |        ^~~~~~~
hcxdumptool.c:61:8: error: unknown type name ‘EVP_MAC’
   61 | static EVP_MAC *cmac;
      |        ^~~~~~~
hcxdumptool.c:62:8: error: unknown type name ‘EVP_MAC_CTX’
   62 | static EVP_MAC_CTX *ctxhmac;
      |        ^~~~~~~~~~~
hcxdumptool.c:63:8: error: unknown type name ‘EVP_MAC_CTX’
   63 | static EVP_MAC_CTX *ctxcmac;
      |        ^~~~~~~~~~~
hcxdumptool.c:64:8: error: unknown type name ‘OSSL_PARAM’
   64 | static OSSL_PARAM paramsmd5[3];
      |        ^~~~~~~~~~
hcxdumptool.c:65:8: error: unknown type name ‘OSSL_PARAM’
   65 | static OSSL_PARAM paramssha1[3];
      |        ^~~~~~~~~~
hcxdumptool.c:66:8: error: unknown type name ‘OSSL_PARAM’
   66 | static OSSL_PARAM paramssha256[3];
      |        ^~~~~~~~~~
hcxdumptool.c:67:8: error: unknown type name ‘OSSL_PARAM’
   67 | static OSSL_PARAM paramsaes128[3];
      |        ^~~~~~~~~~
hcxdumptool.c: In function ‘detectweakwpa’:
hcxdumptool.c:3697:6: warning: implicit declaration of function ‘EVP_MAC_init’; did you mean ‘EVP_MD_nid’? [-Wimplicit-function-declaration]
 3697 |  if(!EVP_MAC_init(ctxhmac, pmk, 32, paramssha1)) return false;
      |      ^~~~~~~~~~~~
      |      EVP_MD_nid
hcxdumptool.c:3698:6: warning: implicit declaration of function ‘EVP_MAC_update’; did you mean ‘HMAC_Update’? [-Wimplicit-function-declaration]
 3698 |  if(!EVP_MAC_update(ctxhmac, pkedata, 100)) return false;
      |      ^~~~~~~~~~~~~~
      |      HMAC_Update
hcxdumptool.c:3699:6: warning: implicit declaration of function ‘EVP_MAC_final’; did you mean ‘HMAC_Final’? [-Wimplicit-function-declaration]
 3699 |  if(!EVP_MAC_final(ctxhmac, pkedata, NULL, 100)) return false;
      |      ^~~~~~~~~~~~~
      |      HMAC_Final
hcxdumptool.c: In function ‘evpdeinitwpa’:
hcxdumptool.c:8067:2: warning: implicit declaration of function ‘EVP_MAC_CTX_free’; did you mean ‘EVP_MD_CTX_free’? [-Wimplicit-function-declaration]
 8067 |  EVP_MAC_CTX_free(ctxhmac);
      |  ^~~~~~~~~~~~~~~~
      |  EVP_MD_CTX_free
hcxdumptool.c:8068:2: warning: implicit declaration of function ‘EVP_MAC_free’; did you mean ‘EVP_PKEY_free’? [-Wimplicit-function-declaration]
 8068 |  EVP_MAC_free(hmac);
      |  ^~~~~~~~~~~~
      |  EVP_PKEY_free
hcxdumptool.c: In function ‘evpinitwpa’:
hcxdumptool.c:8098:8: warning: implicit declaration of function ‘EVP_MAC_fetch’ [-Wimplicit-function-declaration]
 8098 | hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
      |        ^~~~~~~~~~~~~
hcxdumptool.c:8098:6: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8098 | hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
      |      ^
hcxdumptool.c:8100:6: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8100 | cmac = EVP_MAC_fetch(NULL, "cmac", NULL);
      |      ^
hcxdumptool.c:8103:16: warning: implicit declaration of function ‘OSSL_PARAM_construct_utf8_string’ [-Wimplicit-function-declaration]
 8103 | paramsmd5[0] = OSSL_PARAM_construct_utf8_string("digest", "md5", 0);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:8104:16: warning: implicit declaration of function ‘OSSL_PARAM_construct_end’ [-Wimplicit-function-declaration]
 8104 | paramsmd5[1] = OSSL_PARAM_construct_end();
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:8115:11: warning: implicit declaration of function ‘EVP_MAC_CTX_new’; did you mean ‘EVP_MD_CTX_new’? [-Wimplicit-function-declaration]
 8115 | ctxhmac = EVP_MAC_CTX_new(hmac);
      |           ^~~~~~~~~~~~~~~
      |           EVP_MD_CTX_new
hcxdumptool.c:8115:9: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8115 | ctxhmac = EVP_MAC_CTX_new(hmac);
      |         ^
hcxdumptool.c:8117:9: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8117 | ctxcmac = EVP_MAC_CTX_new(cmac);
      |         ^
make: *** [Makefile:74: hcxdumptool] Error 1
```

ZerBea, 2022-11-21:
Closed, because it is solved. More information is here:
https://github.com/ZerBea/hcxtools/issues/233

closed 2022-11-21 by ZerBea

ShutdownRepo, 2022-11-22:
Thanks 😉 will update OpenSSL then, I thought it would be up to date from debian repos

ZerBea, 2022-11-22:
OpenSSL 3.0  is not backported to older versions, yet, but it is already present in bookworm.

ZerBea, 2022-11-22:
Compared to OpenSSL v1.1 the new version 3.0 is working like a charm.