ZerBea/hcxdumptool #233: failed to set channel

114h, 2022-11-21:
I don't know what is wrong. Changing channels with aircrack-ng was never a problem.

sudo hcxdumptool -i wlp4s0 -o Handshake.pcapng --disable_deauthentication --disable_client_attacks --enable_status=3
initialization...

start capturing (stop with ctrl+c)
INTERFACE................: wlp4s0
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries

failed to set channel

terminated...

116h, 2022-11-21:
I am using an ASUS AX3000 wlan card btw

ZerBea, 2022-11-21:
This is a way too less information.
Pleas stop all services that takes access to the device, Than run the following  commands and post the output:
$ hcxdumptool -v
$ hcxdumptool -I
$ hcxdumptool -i wlp4s0 --check_driver

Difference betweeen aircrack-ng and hcxdumptool:
aircrack use NETLINK and ioctl() system calls
while
hcxdumptool uso nly ioctl() system calls. The reason why I don't use NETLINK is described here:
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls?share=1

ZerBea, 2022-11-22:
BTW:
This looks like you're running an old version of hcxdumptooL.

```
$ sudo hcxdumptool -i wlp4s0 -o Handshake.pcapng --disable_deauthentication --disable_client_attacks --enable_status=3
initialization...

start capturing (stop with ctrl+c)
INTERFACE................: wlp4s0
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries

failed to set channel

terminated...
```

As of 08.11.2021, hcxdumptool moved from channel handling to frequency handling (required by some drivers):
```
08.11.2021
==========
hcxdumptool: replaced channel scan engine by frequency scan engine
this is high experimental
```

This message is removed "failed to set channel" completely.

ZerBea, 2022-11-22:
Some more information about WiFi drivers is here:
https://github.com/ZerBea/hcxdumptool/issues/234

ZerBea, 2022-11-23:
According to this:
https://wireless.wiki.kernel.org/en/users/drivers/iwlwifi?s[]=iwlwifi
It is an AX3000 Wifi 6 adapter that uses Intel's AX200 chipset.

There are a lot bugs reported on the iwlwifi driver:
https://bugzilla.kernel.org/buglist.cgi?quicksearch=AX200
especially this one:
https://bugzilla.kernel.org/show_bug.cgi?id=214291

As long as this bugs are not fixed, hcxdumptool will not work. There is nothing I can do.

That is noticed in README.md:
```
Not recommended WiFi chipsets (Broadcom, Intel, Realtek and Atheros), especially:
* Broadcom (neither monitor mode nor frame injection)
* Intel PRO/Wireless (due to several driver issues and NETLINK dependency)
* Realtek RTL8811AU, RTL8812AU, RTL8814AU (due to NETLINK dependency)
```

closed 2022-11-23 by ZerBea