ZerBea/hcxdumptool #234: driver doesn't support/allow frequency scan

abidmaria, 2022-11-22:
Hello ,
I am getting the below error , can you please help 

└─$ sudo hcxdumptool -i wlan0 -C
initialization of hcxdumptool 6.2.7-17-g87fcd83 (depending on the capabilities of the device, this may take some time)...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
wlan0 available frequencies, channels and tx power reported by driver:
**driver doesn't support/allow frequency scan (reported exponent: 1, reported frequency: 0)**

terminating...

I am using the below driver : 
Bus 001 Device 002: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]

Thanks

ZerBea, 2022-11-22:
Mostly latest Realtek drivers dropped ioctl() system call support in favor of NETLINK support.
That is pretty good, if you want to stream videos via WiFi, but it is not good if you decide to run high speed attacks.

How do you set monitor mode (iw?)
`interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
`

ZerBea, 2022-11-22:
To get more information, plese disconnect the device and reconnect it again.
Than stop all services that takes access to the device,
Now run the following commands and post the output:

```
$ hcxdumptool -I
$ hcxdumptool -i wlan0 --check_driver
```
hcxdumptool will check if the ioctl() system calls are supported by the driver and give us some additional information about the return values of the driver.

ZerBea, 2022-11-22:
If all ioctl() calls are supported output looks like this:
```
$ hcxdumptool -I
wlan interfaces:
phy0	74da384204a1	wlp39s0f3u1u1u1	(driver:mt7601u)
```
and this
```
$ sudo hcxdumptool -i wlp39s0f3u1u1u1 --check_driver
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: mt7601u
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

ZerBea, 2022-11-22:
Some additional information:
hcxdumptool use 
`ioctl(SIOCSIWMODE)
` 
to set monitor mode
while
iw use a huge (much overhead and libnl dependent NETLINK message to do the same thing:
```
$ sudo iw --debug dev wlp39s0f3u1u1u1 set type monitor
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 34 <0x22>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669104642
    .port = -1178586373
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 6
    .version = 0
    .unused = 0
  [PAYLOAD] 16 octets
    08 00 03 00 03 00 00 00 08 00 05 00 06 00 00 00 ................
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1669104642
    .port = -1178586373
  [ERRORMSG] 20 octets
    .error = 0 "Erfolg"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 34 <0x22>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669104642
    .port = -1178586373
---------------------------  END NETLINK MESSAGE   ---------------------------
``` 

The same applies to setting a frequency.
hcxdumptool:
`ioctl(SIOCSIWFREQ)
`
iw via NETLINK
```
$ sudo iw --debug dev wlp39s0f3u1u1u1 set freq 2412
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 60
    .type = 34 <0x22>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669104880
    .port = -532663233
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 2
    .version = 0
    .unused = 0
  [PAYLOAD] 40 octets
    08 00 03 00 04 00 00 00 08 00 26 00 6c 09 00 00 ..........&.l...
    08 00 9f 00 00 00 00 00 08 00 27 00 00 00 00 00 ..........'.....
    08 00 a0 00 6c 09 00 00                         ....l...
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1669104880
    .port = -532663233
  [ERRORMSG] 20 octets
    .error = 0 "Erfolg"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 34 <0x22>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669104880
    .port = -532663233
```

I will never support NETLINK, because that lead to a libnl dependency which produce huge overhead.

ZerBea, 2022-11-22:
The Ralink driver is working fine, too:
```
$ hcxdumptool -I
wlan interfaces:
phy2	0022c0a008f9	wlp39s0f3u1u4	(driver:rt2800usb)

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_driver 
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: rt2800usb
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
```

ZerBea, 2022-11-22:
Now we are doing the same running a Realtek device/driver.

$ hcxdumptool -I
wlan interfaces:
N/A	503eaad5e035	wlp39s0f3u1u4	(driver:r8188eu)

```
$ hcxdumptool -I
wlan interfaces:
N/A	503eaad5e035	wlp39s0f3u1u4	(driver:r8188eu)

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_driver
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
starting driver test...
failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver: Device or resource busy
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
failed to restore old SIOCSIWMODE: Device or resource busy
```
and failed because the driver doesn't respond to ioctl() system calls.

ZerBea, 2022-11-22:
And now we come to the part that really surprised me. Some older Realtek drivers/devices suport ioctl() system calls:
```
$ hcxdumptool -I
wlan interfaces:
phy5	f81a670dc527	wlp39s0f3u1u4	(driver:rtl8192cu)

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_driver
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: rtl8192cu (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_injection
initialization of hcxdumptool 6.2.7-16-g29c1743 (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2484/14 proberesponse 49   
packet injection is working on 2.4GHz!
injection ratio: 100% (BEACON: 38 PROBERESPONSE: 49)
your injection ratio is huge - say kids what time is it?
antenna ratio: 100% (NETWORK: 22 PROBERESPONSE: 22)
your antenna ratio is huge - say kids what time is it?
2 radiotap errors encountered during the test

terminating...
2 radiotap errors encountered
```

I really don't know why some of the latest Realtek drivers are compiled without ioctl() support. Due to their NETLINK dependency you can't use them running hcxdumptool any longer.

abidmaria, 2022-11-22:
> latest

Thank you  for your feedback ,
I am using the below commands to set the monitor mode .

sudo ifconfig wlan0 down
sudo airmon-ng check kill
sudo iwconfig wlan0 mode monitor

Thanks

ZerBea, 2022-11-22:
That's ok, but unfortunately it also prevent that hcxdumptool check the capabilities of the driver:
```
starting driver test...
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
```

ZerBea, 2022-11-22:
instead of airmon-g
`$ sudo airmon-ng check kill
`
you can use systemctl which does the same
stop services:
```
sudo systemctl stop NetworkManager.service
sudo systemctl stop wpa_supplicant.service

```
start services:
```
sudo systemctl start NetworkManager.service
sudo systemctl start wpa_supplicant.service
```

abidmaria, 2022-11-22:
> 

Thank you , after using these command to stop the service , now i am not able to connect to the network device , iwconfig , ip addr ... stuck ... no output ...

Can you please help  .

ZerBea, 2022-11-22:
After hcxduptool finished, you must akrivate it again:
```
sudo systemctl start NetworkManager.service
sudo systemctl start wpa_supplicant.service
```

If the interface doesn't respond you should reconnect it or restart your system.

abidmaria, 2022-11-22:
Thank you for your feedback , below is the output , please let me know your feedback. 
└─$ hcxdumptool -I
wlan interfaces:
phy0    1c61b41fb584    wlan0   (driver:rtl88XXau

==================================================

─# hcxdumptool -i wlan0 --check_driver
initialization of hcxdumptool 6.2.7-17-g87fcd83 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: rtl88XXau (this driver is not recommended - expect driver errors)
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

===================================================

└─# sudo iw --debug dev wlan0 set type monitor
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 31 <0x1f>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669121173
    .port = 1593838753
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 6
    .version = 0
    .unused = 0
  [PAYLOAD] 16 octets
    08 00 03 00 03 00 00 00 08 00 05 00 06 00 00 00 ................
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1669121173
    .port = 1593838753
  [ERRORMSG] 20 octets
    .error = 0 "Success"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 31 <0x1f>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669121173
    .port = 1593838753
---------------------------  END NETLINK MESSAGE   ---------------------------
                                                                                 
																				 
																				 
===================================================================================================

└─# sudo iw --debug dev wlan0 set freq 2412
-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 60
    .type = 31 <0x1f>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669121269
    .port = 1728056897
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 2
    .version = 0
    .unused = 0
  [PAYLOAD] 40 octets
    08 00 03 00 03 00 00 00 08 00 26 00 6c 09 00 00 ..........&.l...
    08 00 9f 00 00 00 00 00 08 00 27 00 00 00 00 00 ..........'.....
    08 00 a0 00 6c 09 00 00                         ....l...
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 36
    .type = 2 <ERROR>
    .flags = 256 <ROOT>
    .seq = 1669121269
    .port = 1728056897
  [ERRORMSG] 20 octets
    .error = 0 "Success"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 31 <0x1f>
    .flags = 5 <REQUEST,ACK>
    .seq = 1669121269
    .port = 1728056897
---------------------------  END NETLINK MESSAGE   -----------------------

==========================================================================================================

ZerBea, 2022-11-22:
Great, thanks. Looks like driver support both (ioctl and NETLINK).
Now please run:
$ hcxdumptool -i wlan0 --check_injection

ZerBea, 2022-11-22:
I have an idea why this driver is not working as expected.
This caused the problem:
```
sudo ifconfig wlan0 down
sudo airmon-ng check kill
sudo iwconfig wlan0 mode monitor
```
because the driver need to be initialized by NETLINK.
So this should solve the problem:
$ sudo ip link set wlan0 down
$ sudo iw dev wlan0 set type monitor
$ sudo ip link set wlan0 up
$ sudo iw dev wlan0 info --> to see if the interface is in monitor mode and init by NETLINK
$ sudo hcxdumptool -i wlan0 -C --> to see if we can get frequency information, now

abidmaria, 2022-11-22:
Thank you for your support and feedback ,

below is the output , same issue :

┌──(dimitri㉿kali)-[~]
└─$ sudo ip link set wlan0 down
[sudo] password for dimitri: 
                                                                                                                                                                       
┌──(dimitri㉿kali)-[~]
└─$ sudo iw dev wlan0 set type monitor
                                                                                                                                                                       
┌──(dimitri㉿kali)-[~]
└─$ sudo ip link set wlan0 up
                                                                                                                                                                       
┌──(dimitri㉿kali)-[~]
└─$ sudo iw dev wlan0 info
Interface wlan0
        ifindex 3
        wdev 0x1
        addr 96:2f:5b:63:2f:5f
        type monitor
        wiphy 0
        channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
        txpower 20.00 dBm
                                                                                                                                                                       
┌──(dimitri㉿kali)-[~]
└─$ sudo hcxdumptool -i wlan0 -C
initialization of hcxdumptool 6.2.7-22-gc65aac5 (depending on the capabilities of the device, this may take some time)...
warning possible interfere: NetworkManager is running with pid 517

warning possible interfere: wpa_supplicant is running with pid 661

interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
wlan0 available frequencies, channels and tx power reported by driver:
**driver doesn't support/allow frequency scan (reported exponent: 1, reported frequency: 0)**

terminating...

ZerBea, 2022-11-23:
hcxdumptool requested the frequency from the driver by ioctl() call and the driver respond with 0. It looks like this can be done via NETLINK, only. This is a No-Go (asynchronous and additional dependency) and hcxdumptool will not work on this driver. I think we can close this issue report, because there is nothing I can do.

Please notice, aircrack-ng is working because it use NETLINK (via libnl):
https://github.com/aircrack-ng/aircrack-ng/blob/master/INSTALLING#L13

BTW:
This driver is not a stock kernel driver. According KALI package tracker:
https://pkg.kali.org/pkg/realtek-rtl88xxau-dkms
the driver is from here:
https://github.com/aircrack-ng/rtl8812au
so it should work definitely on aircrack-ng

closed 2022-11-23 by ZerBea

abidmaria, 2022-11-23:
Thank you for your feedback ,
please 2 questions before closing the topic:
1-  since this driver is not a stock kernel driver , so i have to **uninstall**  [realtek-rtl88xxau-dkms] and **install** [rtl8812au].
2- please what is the best wireless adapter for hcxtool , and no errors are expecting .

**Thanks again for your excellent support.**

ZerBea, 2022-11-23:
`1- since this driver is not a stock kernel driver , so i have to uninstall [realtek-rtl88xxau-dkms] and install [rtl8812au].
`
No, because you're going to loose monitor mode and packet injection capabilities because the default Realtek driver doesn't support this features.
That is the reason why there are so many driver clones on git:
https://duckduckgo.com/?t=ffab&q=git+rtl88xxau&ia=web
https://duckduckgo.com/?q=git+rtl8xxxu&t=ffab&ia=web

`2- please what is the best wireless adapter for hcxtool , and no errors are expecting.
`
Every chipset from MediaTek (former Ralink) or Ralink is fine:
https://en.wikipedia.org/wiki/MediaTek
https://en.wikipedia.org/wiki/Ralink

Both drivers are part of the stock kernel and well maintained.
Here is a nice video, showing e.g. an ALFA AWUS036ACM (mt76 chipset) in action:
https://www.youtube.com/watch?v=Usw0IlGbkC4

abidmaria, 2022-11-23:
Thank you so much and thanks again for your excellent support.