ZerBea/hcxdumptool #236: make failing on hcxdumptool

64-nickel, 2022-12-06:
I had the same issue as described in https://githubplus.com/ZerBea/hcxdumptool/issues/232. My prereqs were already installed (apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev). That led me to https://githubplus.com/ZerBea/hcxdumptool/issues/233 that in turn led me to https://github.com/ZerBea/hcxtools/commit/5937d2ad9d021f3b5e2edd55d79439b8485d3222.

The solution appeared to be to install OpenSSL 3.0. So I installed newest OpenSSL from here: git clone https://github.com/openssl/openssl and ran:

./Configure
make
make test
sudo make install

All seemed to run without any issues.

I then rm -rf'd the entire ~/tmp/hcxdumptool folder and git clone'd it again from scratch. When I ran the make again, I got a different error for which I cannot find any references to remedy it. Below are the first few lines of the new error.

~/tmp/hcxdumptool$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99 -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto -DVERSION_TAG="6.2.7-28-g615b7b7" -DVERSION_YEAR="2022"
hcxdumptool.c: In function ‘process_fd’:
hcxdumptool.c:5963:36: warning: ‘
SNONCE....................:.’ directive output may be truncated writing 29 bytes into a region of size between 27 and 834 [-Wformat-truncation=]
5963 | snprintf(servermsg, SERVERMSG_MAX, "\e[?25l\nstart capturing (stop with ctrl+c)\n"
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:5989:159: note: format string is defined here
5989 | "ANONCE....................: %02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x\n"
| ^~~
5990 | "SNONCE....................: %02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x%02x\n"
| ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
5963 | snprintf(servermsg, SERVERMSG_MAX, "\e[?25l\nstart capturing (stop with ctrl+c)\n"
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
hcxdumptool.c:5963:36: note: directive argument in the range [0, 255]
...snipped here for brevity...

ZerBea, 2022-12-07:
Your commands are interesting:
```
./Configure
make
make test
sudo make install
```
Because make process neither use "./Configure" nor "make test" and you will get a warning:
```
$. /Configure
bash: ./configure: No such file or directory
$ make test
make: *** No rule to make target 'test'.  Stop.
```

I can't reproduce the issue, because hcxdumptool is compiling
```
$ gcc --version
gcc (GCC) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

 as expected:
```
$ git clone https://github.com/ZerBea/hcxdumptool
Cloning into 'hcxdumptool'...
remote: Enumerating objects: 4256, done.
remote: Counting objects: 100% (947/947), done.
remote: Compressing objects: 100% (360/360), done.
remote: Total 4256 (delta 621), reused 754 (delta 586), pack-reused 3309
Receiving objects: 100% (4256/4256), 1.40 MiB | 1.94 MiB/s, done.
Resolving deltas: 100% (2883/2883), done.

$ cd hcxdumptool

$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"

$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
install -D -m 0755 hcxpioff /usr/bin/hcxpioff
```

To verify this, I added "-Wformat-truncation" (that caused your warning) to CFLAGS and compiled it again:
```
$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -Wformat-truncation -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"
cc -O3 -Wall -Wextra -Wformat-truncation -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"
```

Explanation how the fields are sized and how many bytes are written into them:
ANONCE and SNONCE are fileds of 32 bytes:
```
static uint8_t myanonce[32];
static uint8_t mysnonce[32];
```
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L297

and we write exactly 32 bytes:
```
	myanonce[0], myanonce[1], myanonce[2], myanonce[3], myanonce[4], myanonce[5], myanonce[6], myanonce[7],
	myanonce[8], myanonce[9], myanonce[10], myanonce[11], myanonce[12], myanonce[13], myanonce[14], myanonce[15],
	myanonce[16], myanonce[17], myanonce[18], myanonce[19], myanonce[20], myanonce[21], myanonce[22], myanonce[23],
	myanonce[24], myanonce[25], myanonce[26], myanonce[27], myanonce[28], myanonce[29], myanonce[30], myanonce[31],

	mysnonce[0], mysnonce[1], mysnonce[2], mysnonce[3], mysnonce[4], mysnonce[5], mysnonce[6], mysnonce[7],
	mysnonce[8], mysnonce[9], mysnonce[10], mysnonce[11], mysnonce[12], mysnonce[13], mysnonce[14], mysnonce[15],
	mysnonce[16], mysnonce[17], mysnonce[18], mysnonce[19], mysnonce[20], mysnonce[21], mysnonce[22], mysnonce[23],
	mysnonce[24], mysnonce[25], mysnonce[26], mysnonce[27], mysnonce[28], mysnonce[29], mysnonce[30], mysnonce[31]);
```
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L6009

Which gcc version ($ gcc --version) do you use?

ZerBea, 2022-12-07:
I went back to gcc 11
```
$ gcc-11 --version
gcc-11 (GCC) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
Dies ist freie Software; die Kopierbedingungen stehen in den Quellen. Es
gibt KEINE Garantie; auch nicht für MARKTGÄNGIGKEIT oder FÜR SPEZIELLE ZWECKE.
```
Also no warnings when compiling hcxdumptool:
```
$ make
mkdir -p .deps
gcc-11 -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"
gcc-11 -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-28-g615b7b7\" -DVERSION_YEAR=\"2022\"

```

64-nickel, 2022-12-07:
"Because make process neither use "./Configure" nor "make test" and you will get a warning:" - This was for the OpenSSL build, not hcxdump.  
I looked at the gcc on the box with gcc --version.  It's reporting gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0

ZerBea, 2022-12-07:
Ok. Thanks for the build information.

Please notice that gcc 9 is outdated: This release series is no longer supported.
https://gcc.gnu.org/gcc-9/

ZerBea, 2022-12-23:
Closed, because requirements are not satisfied.

closed 2022-12-23 by ZerBea