ZerBea/hcxdumptool #239: Feature Request: hcxpcapngtool KML

PowerPress, 2022-12-13:
Could you please add an option to hcxpcapngtool that converts the GPS and cracked passwords with SSID to Google Earth KML

ZerBea, 2022-12-14:
There are hundreds of GPS formats and languages out in the wildness, e.g the one from Google:
```
$ gpsbabel --help
	kml                   Google Earth (Keyhole) Markup Language
	  deficon               Default icon name 
	  lines                 (0/1) Export linestrings for tracks and routes 
	  points                (0/1) Export placemarks for tracks and routes 
	  line_width            Width of lines, in pixels 
	  line_color            Line color, specified in hex AABBGGRR 
	  floating              (0/1) Altitudes are absolute and not clamped to ground 
	  extrude               (0/1) Draw extrusion line from trackpoint to ground 
	  track                 (0/1) Write KML track (default = 0) 
	  trackdata             (0/1) Include extended data for trackpoints (default = 1 
	  trackdirection        (0/1) Indicate direction of travel in track icons (defau 
	  units                 Units used when writing comments ('s'tatute, 'm'et 
	  labels                (0/1) Display labels on track and routepoints  (default  
	  max_position_point    Retain at most this number of position points  (0  
	  rotate_colors         Rotate colors for tracks and routes (default autom 
	  prec                  Precision of coordinates, number of decimals 
```
or e.g. wigle:
https://github.com/ZerBea/hcxdumptool/issues/157

Supporting this formats e.g. KML
https://developers.google.com/kml/documentation/kml_tut
goes far beyond the scope of my engagement. Therefore I decided to provide a general CSV option (beside the fast and short NMEA0183 option) that include all information.
```
--nmea=<file>                      : output GPS data in NMEA 0183 format
                                     format: NMEA 0183 $GPGGA, $GPRMC, $GPWPL
                                     to convert it to gpx, use GPSBabel:
                                     gpsbabel -i nmea -f hcxdumptool.nmea -o gpx,gpxver=1.1 -F hcxdumptool.gpx
                                     to display the track, open file.gpx with viking
--csv=<file>                       : output ACCESS POINT information in CSV format
                                     delimiter: tabulator (0x08)
                                     columns:
                                     YYYY-MM-DD HH:MM:SS MAC_AP ESSID ENC_TYPE CIPHER AKM COUNTRY_INFO CHANNEL RSSI GPS(DM.m) GPS(D.d) GPSFIX SATCOUNT HDOP ALTITUDE UNIT
                                     to convert it to other formats, use bash tools or scripting languages
                                     GPS FIX:
                                     0 = fix not available or invalid
                                     1 = fix valid (GPS SPS mode)
                                     2 = fix valid (differential GPS SPS Mode)
                                     3 = not supported
                                     4 = not supported
                                     5 = not supported
                                     6 = fix valid (Dead Reckoning Mode)
```
Now it is easy to write a (third party) script that take this fields and to convert them for whatever purpose.
A good scripting example (wigle) is here:
https://github.com/Czechball/hcx-to-wigle

BTW:
hcxpcapngtool is a conversion tool. It is not able to crack a NETWORK key that you can add to the GPS information.

ZerBea, 2023-04-06:
Refactored hcxdumptool store GPS data as NMEA 0183 standard messages:
```
--nmea_out=<outfile>           : write GPS information to a nmea-format file named <outfile>
                                  default outfile name: yyyyddmmhhmmss.nmea
                                  output: NMEA 0183 standard messages:
                                          $GPRMC: Position, velocity, time and date
                                          $GPGGA: Position, orthometric height, fix related data, time
                                          $GPWPL: Position and MAC AP
                                          $GPTXT: ESSID in HEX ASCII
                                  use gpsbabel to convert to other formats:
                                   gpsbabel -w -t -i nmea -f in_file.nmea -o gpx -F out_file_gpx
                                   gpsbabel -w -t -i nmea -f in_file.nmea -o kml -F out_file.kml
```
They can be converted via bash tools or gpsbabel.

closed 2023-04-06 by ZerBea