ZerBea/hcxdumptool #240: problem with hcxdumptool

eziocan, 2022-12-15:
hi, i'm trying to use "hcxdumptool", i would like to scan the wifi networks in my vicinity but i get this :

`$ sudo hcxdumptool --do_rcascan -i wlan0

initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time
interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls
driver doesn't report frequency
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored`

please someone can help me ?

ZerBea, 2022-12-16:
You're running an old version of hcxdumptool:
"initialization of hcxdumptool 6.2.6"

please update to latest version:
$ hcxdumptool -v
hcxdumptool 6.2.7-30-g362a817 (C) 2022 ZeroBeat

Your driver doesn't report a frequency if requested via ioctl() system call:
"driver doesn't report frequency"
This leads me to the assumption that you're running a driver that depend on NETLINK (libnl). This driver does not support ioctl() system calls, which are mandatory to perform hcxdumptools fast attacks.
Read more in Adapters section of README.md

Have you set monitor mode by iw or iwconfig, before you ran hcxdumptool?
Looks like you interface is already in monitor mode and hcxdumptool skipped some initial driver tests:
"interface is already in monitor mode, skipping ioctl(SIOCSIWMODE) and ioctl(SIOCSIFFLAGS) system calls"

Please comment the output of:
$ hcxdumptool -I
and
$ sudo hcxdumptool -i wlan0 --check_driver
and additional
$ sudo hcxdumptool -i wlan0 --check_injection
Do net set monitor mode by iw, before you start this commands!

If all tests are passed [driver support ioctl() system calls], output should look like this:
```
$ hcxdumptool -I
wlan interfaces:
phy0	74da381415c7	wlp39s0f3u1u4	(driver:mt7601u)

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_driver
initialization of hcxdumptool 6.2.7-30-g362a817 (depending on the capabilities of the device, this may take some time)...
starting driver test...
detected driver: mt7601u
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

$ sudo hcxdumptool -i wlp39s0f3u1u4 --check_injection
initialization of hcxdumptool 6.2.7-30-g362a817 (depending on the capabilities of the device, this may take some time)...
starting antenna test and packet injection test (that can take up to two minutes)...
stage 2 of 2 probing frequency 2472/13 proberesponse 103   
packet injection is working on 2.4GHz!
injection ratio: 68% (BEACON: 175 PROBERESPONSE: 119)
your injection ratio is good
antenna ratio: 37% (NETWORK: 117 PROBERESPONSE: 44)
your antenna ratio is average, but there is still room for improvement
```

eziocan, 2022-12-16:
first two commands ok , the last one , return this :
[image]

ZerBea, 2022-12-16:
"first two commands ok" is not enough.
I requested the output of the commands to identify the driver.
Please comment output of
$ hcxdumptool -I

ZerBea, 2022-12-16:
Please update to version 2.7, because this version has an improved driver detection:
```
02.06.2022
==========
hcxdumptool: added warning if a not recommended driver has been detected
```

ZerBea, 2022-12-23:
Possible driver problem and not hcxdumptool.

closed 2022-12-23 by ZerBea