ZerBea/hcxdumptool #241: known beacons attack

s0i37, 2022-12-23:
Hi.
I make the known beacons attacks like this:

`sudo hcxdumptool -i mon0 --enable_status=6 --essidlist=essids.txt`

However, hcxdumptool only announce  `WPA` network. Known beacons attack more interesting whith `OPN` networks

ZerBea, 2022-12-23:
No. Target is only WPA 1, WPA 2 and WPA 2 keyversion 3

BTW;
When running --essidlist=essids.txt, you should use --active_beacon in combination

ZerBea, 2022-12-23:
You can set BEACON params to what ever you want:
`--beaconparams=<TLVs>              : update or add Information Elements in all reactive and essidlist beacons
`

BTW:
This is more a discussion than an issue report. Closed it here and opened discussion

closed 2022-12-23 by ZerBea

s0i37, 2022-12-24:
`When running --essidlist=essids.txt, you should use --active_beacon in combination`
Not necessary.
I want only sends beacons from my dictionary, not previosly seen networks.

ZerBea, 2022-12-24:
Ok.