ZerBea/hcxdumptool #242: Enterprise attacks

s0i37, 2022-12-23:
Hello.

Can I attacks WPA Enterprise and obtain plaintext/netntlm1 credentials with hcxdumptool (like a eaphammer)?

`hcxdumptool -i mon0 --enable_status=1536 --wpaent --eapreq_follownak`

Thank you!

ZerBea, 2022-12-23:
No.

BTW:
Running hcxdumptool on a virtual monitor interface (mon0) is not a good idea.

ZerBea, 2022-12-23:
BTW:
This is more a discussion than an issue report. Closed it here and opened discussion

closed 2022-12-23 by ZerBea

ZerBea, 2022-12-23:
Please notice that hcxdumptool is designed to run attacks on 802.11 MAC layer, only.
@n0w1re added the EAP-TLS part, but I'm not really happy about it, because a protocol attack takes much more time than a MAC layer attack.
That is the major reason (time is a bottleneck) I coded hcxlabtool. Instead of using hundreds of options (every option hash a price tag), the attacks modes are directly hard coded.

ZerBea, 2022-12-23:
And as far as I know, eaphammer is running hcxdumptool to perform the MAC layer attacks and hcxtools to convert the results.

s0i37, 2022-12-24:
`Running hcxdumptool on a virtual monitor interface (mon0) is not a good idea.`
I know, but I dont understand that the problem. I prefer use many tools in the same time all of they use monitor mode.