ZerBea/hcxdumptool #245: Kernel panic

linsyking, 2023-01-01:
When I run this tool, it will always cause a kernel panic.

linsyking, 2023-01-01:
Here is the output of
```
journalctl -b-1 -p3
```
[log.log](https://github.com/ZerBea/hcxdumptool/files/10328689/log.log)

linsyking, 2023-01-01:
00:14.3 Network controller: Intel Corporation Comet Lake PCH CNVi WiFi
        Subsystem: Intel Corporation Wi-Fi 6 AX201 160MHz
        Kernel driver in use: iwlwifi
        Kernel modules: iwlwifi

ZerBea, 2023-01-02:
Due to a driver issue, this driver is not suitable to be used by hcxdumptool as mentioned in README.md driver section:
`* Intel PRO/Wireless (several driver issues and NETLINK dependency)`
Also you should have get a warning this driver is not recommended.
Pleasw take a look at the closed issues, e.g. this one.
https://github.com/ZerBea/hcxdumptool/issues/217

This problem is reported to bugzilla.kernel.org on 2021-09-02 15:32:47 UTC 
https://bugzilla.kernel.org/show_bug.cgi?id=214291
and it is still open.

BTW:
There is no entry that hcumptool caused the panic. I only see iwlwifi entries.

I think we can close this issue report (as the other issues reported before),

closed 2023-01-02 by ZerBea

ZerBea, 2023-01-05:
Looks like this is not an isolated case:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=459249;msg=7
and more
https://duckduckgo.com/?q=iwlwifi+kernel+panic&t=ffab&df=y&ia=web

ZerBea, 2023-01-05:
I'm testing hcxdumptool in combination with Realtek (rt2800usb) and MediaTek (mt76) drivers. This is the reference.
Both drivers are well maintained and they are working out of the box.
An example how fast issues are fixed is e.g. here
https://bugzilla.kernel.org/show_bug.cgi?id=205305
and regarding upstream, here:
https://github.com/openwrt/mt76/issues/216#issuecomment-500999516

ZerBea, 2023-01-06:
Please join our conversion on:
https://bugzilla.kernel.org/show_bug.cgi?id=214291
and provide your logs there, because I have no Intel device to do that.

ZerBea, 2023-02-26:
Can you please test hcxlabtool which use NL80211 and RTNETLINK instead of WIRELESS EXTENSIONS):
https://github.com/ZerBea/wifi_laboratory

$ make
$ ./hcxlabtool -I interfacename

Thanks.

linsyking, 2023-02-27:
```
available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 38fc9861cb4b 38fc9861cb4b + wlp0s20f3        iwlwifi (NETLINK & WIRELESS EXTENSIONS)

+ monitor mode available
- no monitor mode available

available frequencies: frequency [channel] tx-power
  2412 [  1] 22.0 dBm     2417 [  2] 22.0 dBm     2422 [  3] 22.0 dBm     2427 [  4] 22.0 dBm
  2432 [  5] 22.0 dBm     2437 [  6] 22.0 dBm     2442 [  7] 22.0 dBm     2447 [  8] 22.0 dBm
  2452 [  9] 22.0 dBm     2457 [ 10] 22.0 dBm     2462 [ 11] 22.0 dBm     2467 [ 12] 22.0 dBm
  2472 [ 13] 22.0 dBm     2484 [ 14] disabled     5180 [ 36] 22.0 dBm     5200 [ 40] 22.0 dBm
  5220 [ 44] 22.0 dBm     5240 [ 48] 22.0 dBm     5260 [ 52] 22.0 dBm     5280 [ 56] 22.0 dBm
  5300 [ 60] 22.0 dBm     5320 [ 64] 22.0 dBm     5340 [ 68] disabled     5360 [ 72] disabled
  5380 [ 76] disabled     5400 [ 80] disabled     5420 [ 84] disabled     5440 [ 88] disabled
  5460 [ 92] disabled     5480 [ 96] disabled     5500 [100] 22.0 dBm     5520 [104] 22.0 dBm
  5540 [108] 22.0 dBm     5560 [112] 22.0 dBm     5580 [116] 22.0 dBm     5600 [120] 22.0 dBm
  5620 [124] 22.0 dBm     5640 [128] 22.0 dBm     5660 [132] 22.0 dBm     5680 [136] 22.0 dBm
  5700 [140] 22.0 dBm     5720 [144] 22.0 dBm     5745 [149] 22.0 dBm     5765 [153] 22.0 dBm
  5785 [157] 22.0 dBm     5805 [161] 22.0 dBm     5825 [165] 22.0 dBm     5845 [169] disabled
  5865 [173] disabled     5885 [177] disabled     5905 [181] disabled

bye-bye
```

ZerBea, 2023-02-27:
Great, thanks - hcxlabtool should work on iwlwifi driver.
Please read more here:
Why are some frequencies disabled
https://github.com/ZerBea/hcxdumptool/issues/186#issuecomment-1445397147
Information about hcxdumptool and NETLINK
https://github.com/ZerBea/hcxdumptool/issues/186#issuecomment-1445404265