ZerBea/hcxdumptool #247: BUG Compile

vincenzogianfelice, 2023-01-19:
**Debian 11** (5.10)

```
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto  -DVERSION_TAG=\"6.2.7-45-g9bb7542\" -DVERSION_YEAR=\"2023\"
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-45-g9bb7542\" -DVERSION_YEAR=\"2023\"
hcxdumptool.c:61:8: error: unknown type name ‘EVP_MAC’
   61 | static EVP_MAC *hmac;
      |        ^~~~~~~
hcxdumptool.c:62:8: error: unknown type name ‘EVP_MAC’
   62 | static EVP_MAC *cmac;
      |        ^~~~~~~
hcxdumptool.c:63:8: error: unknown type name ‘EVP_MAC_CTX’
   63 | static EVP_MAC_CTX *ctxhmac;
      |        ^~~~~~~~~~~
hcxdumptool.c:64:8: error: unknown type name ‘EVP_MAC_CTX’
   64 | static EVP_MAC_CTX *ctxcmac;
      |        ^~~~~~~~~~~
hcxdumptool.c:65:8: error: unknown type name ‘OSSL_PARAM’
   65 | static OSSL_PARAM paramsmd5[3];
      |        ^~~~~~~~~~
hcxdumptool.c:66:8: error: unknown type name ‘OSSL_PARAM’
   66 | static OSSL_PARAM paramssha1[3];
      |        ^~~~~~~~~~
hcxdumptool.c:67:8: error: unknown type name ‘OSSL_PARAM’
   67 | static OSSL_PARAM paramssha256[3];
      |        ^~~~~~~~~~
hcxdumptool.c:68:8: error: unknown type name ‘OSSL_PARAM’
   68 | static OSSL_PARAM paramsaes128[3];
      |        ^~~~~~~~~~
hcxdumptool.c: In function ‘detectweakwpa’:
hcxdumptool.c:3703:6: warning: implicit declaration of function ‘EVP_MAC_init’; did you mean ‘EVP_MD_nid’? [-Wimplicit-function-declaration]
 3703 |  if(!EVP_MAC_init(ctxhmac, pmk, 32, paramssha1)) return false;
      |      ^~~~~~~~~~~~
      |      EVP_MD_nid
hcxdumptool.c:3704:6: warning: implicit declaration of function ‘EVP_MAC_update’; did you mean ‘HMAC_Update’? [-Wimplicit-function-declaration]
 3704 |  if(!EVP_MAC_update(ctxhmac, pkedata, 100)) return false;
      |      ^~~~~~~~~~~~~~
      |      HMAC_Update
hcxdumptool.c:3705:6: warning: implicit declaration of function ‘EVP_MAC_final’; did you mean ‘HMAC_Final’? [-Wimplicit-function-declaration]
 3705 |  if(!EVP_MAC_final(ctxhmac, testptk, NULL, 256)) return false;
      |      ^~~~~~~~~~~~~
      |      HMAC_Final
hcxdumptool.c: In function ‘evpdeinitwpa’:
hcxdumptool.c:8098:2: warning: implicit declaration of function ‘EVP_MAC_CTX_free’; did you mean ‘EVP_MD_CTX_free’? [-Wimplicit-function-declaration]
 8098 |  EVP_MAC_CTX_free(ctxhmac);
      |  ^~~~~~~~~~~~~~~~
      |  EVP_MD_CTX_free
hcxdumptool.c:8099:2: warning: implicit declaration of function ‘EVP_MAC_free’; did you mean ‘EVP_PKEY_free’? [-Wimplicit-function-declaration]
 8099 |  EVP_MAC_free(hmac);
      |  ^~~~~~~~~~~~
      |  EVP_PKEY_free
hcxdumptool.c: In function ‘evpinitwpa’:
hcxdumptool.c:8129:8: warning: implicit declaration of function ‘EVP_MAC_fetch’ [-Wimplicit-function-declaration]
 8129 | hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
      |        ^~~~~~~~~~~~~
hcxdumptool.c:8129:6: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8129 | hmac = EVP_MAC_fetch(NULL, "hmac", NULL);
      |      ^
hcxdumptool.c:8131:6: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8131 | cmac = EVP_MAC_fetch(NULL, "cmac", NULL);
      |      ^
hcxdumptool.c:8134:16: warning: implicit declaration of function ‘OSSL_PARAM_construct_utf8_string’ [-Wimplicit-function-declaration]
 8134 | paramsmd5[0] = OSSL_PARAM_construct_utf8_string("digest", "md5", 0);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:8135:16: warning: implicit declaration of function ‘OSSL_PARAM_construct_end’ [-Wimplicit-function-declaration]
 8135 | paramsmd5[1] = OSSL_PARAM_construct_end();
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
hcxdumptool.c:8146:11: warning: implicit declaration of function ‘EVP_MAC_CTX_new’; did you mean ‘EVP_MD_CTX_new’? [-Wimplicit-function-declaration]
 8146 | ctxhmac = EVP_MAC_CTX_new(hmac);
      |           ^~~~~~~~~~~~~~~
      |           EVP_MD_CTX_new
hcxdumptool.c:8146:9: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8146 | ctxhmac = EVP_MAC_CTX_new(hmac);
      |         ^
hcxdumptool.c:8148:9: warning: assignment to ‘int *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 8148 | ctxcmac = EVP_MAC_CTX_new(cmac);
      |         ^
make: *** [Makefile:75: hcxdumptool] Errore 1
```

ZerBea, 2023-01-19:
This is mentioned in README.md and changelog:
README.md:
```
Section Or install via packet manager
### Debian (e.g. Kali, Ubuntu) release requirements >= bookworm (testing/Debian 12)  
Section Requirements:
* libopenssl (>= 3.0) and openssl-dev installed
```

Changelog:
```
12.11.2022
==========
start moving to OpenSSL 3.0 EVP API
this is a huge step forward an will break backward compatibility
https://wiki.openssl.org/index.php/OpenSSL_3.0
```
It is not worth the effort to support the old version (1.1) because libssl3 is on Arch Linux (default)
https://archlinux.org/packages/core/x86_64/openssl/
and Debian:
https://packages.debian.org/bookworm/libssl3

ZerBea, 2023-01-19:
If Debian developers decide not to backport libssl3 to older Debian releases, hcxdumptool/hcxtools 6.2.7 will be the last version working for you.
I think we can close this report, because it is not a bug and OpenSSL 3.0 is available.