ZerBea/hcxdumptool #251: Errors (qcacld-3.0 driver)

BobNich, 2023-02-12:
**Device**:  Android with qcacld-3.0 driver

Before running the tool, _no operations were performed to change the mode_ of the internal wlan interface.

> hcxdumptool -I
                                               
```
wlan interfaces:
phy0 d0497ca2bfb8 wlan0 (driver:cnss_pci)
phy0 d2497c24bfb8 p2p0 (driver:cnss_pci)
phy0 d2497c25bfb8 wifi-aware0 (driver:cnss_pci)
```
> hcxdumptool -i wlan0 --check_driver

```
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...
warning: interface wlan0 (phy0) is shared
hcxdumptool may not work as expected on shared physical devices

starting driver test...
failed to detect wlan interface: interface possible not comected
driver doesn't support CFG80211
kernel possible complied without Wireless Extensions (expected: CONFIG_CFG80211_WEXT=y and CONFIG_CFG80211_WEXT_EXPORT=y): Operation not supported
warning: failed to init socket

terminating...
1 driver error encountered
usually this error is related to pselect() after SIGTERM has been received
ERRORs < 10 are related to a slow initialization and can be ignored
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device
```

> hcxdumptool -i wlan0

```
initialization of hcxdumptool 6.2.6 (depending on the capabilities of the device, this may take some time)...

warning: interface wlan0 (phy0) is shared
hcxdumptool may not work as expected on shared physical devices

failed to set monitor mode, ioctl(SIOCSIWMODE) not supported by driver: Operation not supported
warning: failed to init socket

terminating...
```

---
Initially, the necessary configs _(CONFIG_CFG80211_WEXT=y and CONFIG_CFG80211_WEXT_EXPORT=y)_ were not included in the system kernel:

After including them in **/proc/config.gz** we have:

> \# CONFIG_CFG80211_WEXT is not set

Therefore, suspicions fell on the adapter driver.
I looked at the information on the **qcacld-3.0** driver and found this issue: https://github.com/aircrack-ng/aircrack-ng/issues/2179

There was a discussion that the **SIOCSIWMODE** call was not implemented.

In file https://github.com/EssentialOpenSource/qcacld-3.0/blob/master/core/hdd/src/wlan_hdd_wext.c we can see: 

```
static const iw_handler we_handler[] = {
                ...
  NULL, /* SIOCSIWMODE */
                ...
}
```

What can you advise to do in this case?

ZerBea, 2023-02-12:
hcxdumptool use WIRELESS EXTENSIONS to control the interface via ioctl() system calls.
Unfortunately WIRELESS EXTENSIONS are either completely deactivated or not used by the driver.
In that case hcxdumptool will not work.

To get rid of it, I'm working on a version that will use NETLINK as an alternative to WIRELESS EXTENSIONS:
https://github.com/ZerBea/hcxdumptool/issues/250#issuecomment-1426658244

To figure out if your interface support NETLINK instead of WIRELES EXTENSIONS, please unzip the attachment and run make (ignore the 2 warnings, because this is a bleeding test version).
Than run
$ hcxdumptool -I -C
and comment the output (if there is an output).

[hcxlabtool_test.zip](https://github.com/ZerBea/hcxdumptool/files/10716102/hcxlabtool_test.zip)

BobNich, 2023-02-12:
Here is the output with disabled wi-fi:

```
wlan interfaces:
phy0	d0497ca2bfb8	(spoofed MAC:1ae7d1db04f9 detected)	wlan0	(driver:cnss_pci)
phy0	d2497c24bfb8	p2p0	(driver:cnss_pci)
phy0	d2497c25bfb8	wifi-aware0	(driver:cnss_pci)
```

The output with enabled wi-fi:

```
wlan interfaces:
phy0	d2497c25bfb8	wifi-aware0	(driver:cnss_pci)
```

BobNich, 2023-02-12:
I also ran this one:

`$ ./hcxlabtool -I -C`
```

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0  14 d0497ca2bfb8 1ae7d1db04f9 - wlan0            cnss_pci (WIRELESS EXTENSIONS & NETLINK)
  0  15 d2497c24bfb8 d2497c24bfb8 - p2p0             cnss_pci (WIRELESS EXTENSIONS & NETLINK)
  0  16 d2497c25bfb8 d2497c25bfb8 + wifi-aware0      cnss_pci (WIRELESS EXTENSIONS & NETLINK)

+ monitor mode available
- no monitor mode available

wlan0 default scan list: frequency [channel] tx-power

p2p0 default scan list: frequency [channel] tx-power

wifi-aware0 default scan list: frequency [channel] tx-power
  2412 [1] 30.0 dBm	  2417 [2] 30.0 dBm	  2422 [3] 30.0 dBm	  2427 [4] 30.0 dBm
  2432 [5] 30.0 dBm	  2437 [6] 30.0 dBm	  2442 [7] 30.0 dBm	  2447 [8] 30.0 dBm
  2452 [9] 30.0 dBm	  2457 [10] 30.0 dBm	  2462 [11] 30.0 dBm	  2467 [12] 30.0 dBm
  2472 [13] 30.0 dBm	  5180 [36] 30.0 dBm	  5200 [40] 30.0 dBm	  5220 [44] 30.0 dBm
  5240 [48] 30.0 dBm	  5260 [52] 30.0 dBm	  5280 [56] 30.0 dBm	  5300 [60] 30.0 dBm
  5320 [64] 30.0 dBm	  5500 [100] 30.0 dBm	  5520 [104] 30.0 dBm	  5540 [108] 30.0 dBm
  5560 [112] 30.0 dBm	  5580 [116] 30.0 dBm	  5600 [120] 30.0 dBm	  5620 [124] 30.0 dBm
  5640 [128] 30.0 dBm	  5660 [132] 30.0 dBm	  5680 [136] 30.0 dBm	  5700 [140] 30.0 dBm
  5720 [144] 30.0 dBm	  5745 [149] 30.0 dBm	  5765 [153] 30.0 dBm	  5785 [157] 30.0 dBm
  5805 [161] 30.0 dBm	  5825 [165] 30.0 dBm
```

ZerBea, 2023-02-12:
Thanks.
The result is very interesting:

wlan0  = no monitor mode
p2p = no monitor mode
wifi-aware0 = monitor mode

wifi-aware0 reported that it support WIRELESS EXTENSONS (but we know it will not support all EXTENSIONS)

Pleas no run:
$ hcxlabtool -i wifi-aware0

You should see some information about RSN elements and that the channel is successfully changed.
If set worked, we can guess that NETLINK is working to change channels.

ZerBea, 2023-02-12:
BTW:
The interface phy0 is split into three virtual interfaces. That will slow down all attacks.

BobNich, 2023-02-12:
```
$ ./hcxlabtool -i wifi-aware0
failed to set monitor mode

1 errors during runtime

bye-bye
```

ZerBea, 2023-02-12:
Thanks.
Now we know:
driver report that it support WIRELESS EXTENSIONS - but it doesn't!
driver report that it support NETLINK - but it doesn't!

I'm running out of ideas and I don't see a chance that hcxlabtool/hcxdumptool will work on this driver.

BobNich, 2023-02-12:
Thanks a lot, @ZerBea! I think we can close this issue.

ZerBea, 2023-02-12:
No, please leave it open.

hcxlabtool set monitor mode via NETLINK if the driver support NETLINK.
hcxlabtool set monitor mode via WIRELESS extensions if the driver support WIRELESS EXTENSIONS.

Let's find out what happens if we set monitor mode via
echo 4 > /sys/module/wlan/parameters/con_mode
and skip the ways mentioned above.

ZerBea, 2023-02-12:
Let me think about it.....

BobNich, 2023-02-12:
```
$ echo 4 > /sys/module/wlan/parameters/con_mode
$ ./hcxlabtool -I -C
```
```
available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0  24 d0497ca2bfb8 d0497ca2bfb8 + wlan0            cnss_pci (NETLINK)

+ monitor mode available
- no monitor mode available

wlan0 default scan list: frequency [channel] tx-power
  2412 [1] 30.0 dBm	  2417 [2] 30.0 dBm	  2422 [3] 30.0 dBm	  2427 [4] 30.0 dBm
  2432 [5] 30.0 dBm	  2437 [6] 30.0 dBm	  2442 [7] 30.0 dBm	  2447 [8] 30.0 dBm
  2452 [9] 30.0 dBm	  2457 [10] 30.0 dBm	  2462 [11] 30.0 dBm	  2467 [12] 30.0 dBm
  2472 [13] 30.0 dBm	  5180 [36] 30.0 dBm	  5200 [40] 30.0 dBm	  5220 [44] 30.0 dBm
  5240 [48] 30.0 dBm	  5260 [52] 30.0 dBm	  5280 [56] 30.0 dBm	  5300 [60] 30.0 dBm
  5320 [64] 30.0 dBm	  5500 [100] 30.0 dBm	  5520 [104] 30.0 dBm	  5540 [108] 30.0 dBm
  5560 [112] 30.0 dBm	  5580 [116] 30.0 dBm	  5600 [120] 30.0 dBm	  5620 [124] 30.0 dBm
  5640 [128] 30.0 dBm	  5660 [132] 30.0 dBm	  5680 [136] 30.0 dBm	  5700 [140] 30.0 dBm
  5720 [144] 30.0 dBm	  5745 [149] 30.0 dBm	  5765 [153] 30.0 dBm	  5785 [157] 30.0 dBm
  5805 [161] 30.0 dBm	  5825 [165] 30.0 dBm

bye-by
```
---
`$ ./hcxlabtool -i wlan0`
```
failed to open rx socket

1 errors during runtime

bye-bye
```

ZerBea, 2023-02-12:
Thae atteeched version will not ry to set monitor mode vie NETLINK or WIRELESS EXTENSIONS:

[hcxlabtool_test.zip](https://github.com/ZerBea/hcxdumptool/files/10716367/hcxlabtool_test.zip)

Please set monitor mode via
echo 4 > /sys/module/wlan/parameters/con_mode
and try again:
./hcxlabtool -i wlan0

ZerBea, 2023-02-12:
Ok, that is the next big problem: It looks like raw packet sockets ore not supported by the driver.

$ ./hcxlabtool -i wlan0

failed to open rx socket

BobNich, 2023-02-12:
After using the new hcxlabtool_test I see the same output:

`$ ./hcxlabtool -i wlan0`
```
failed to open rx socket

1 errors during runtime

bye-bye
```

ZerBea, 2023-02-12:
Ok, thanks.
Now we can close this report, because I don't see a chance here.

closed 2023-02-12 by ZerBea