ZerBea/hcxdumptool #253: error: unknown type name ‘EVP_MAC’

adrastee, 2023-02-18:
```
root@raspberrypi:~/hcxdumptool $ openssl version
OpenSSL 3.0.8 7 Feb 2023 (Library: OpenSSL 3.0.8 7 Feb 2023)
```
```
root@raspberrypi:~/hcxdumptool $ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto  -DVERSION_TAG=\"6.2.7-59-g0a12c7b\" -DVERSION_YEAR=\"2023\"
hcxdumptool.c:61:8: error: unknown type name ‘EVP_MAC’
 static EVP_MAC *hmac;
        ^~~~~~~
hcxdumptool.c:62:8: error: unknown type name ‘EVP_MAC’
 static EVP_MAC *cmac;
        ^~~~~~~
hcxdumptool.c:63:8: error: unknown type name ‘EVP_MAC_CTX’
 static EVP_MAC_CTX *ctxhmac;
        ^~~~~~~~~~~
hcxdumptool.c:64:8: error: unknown type name ‘EVP_MAC_CTX’

```

ZerBea, 2023-02-18:
hcxtools / hcxdumptool moved to OpenSSL 3.0 API as mentioned in README.md section "Requirements":
`
* libopenssl (>= 3.0) and openssl-dev installed

and changelog:
```
12.11.2022
==========
start moving to OpenSSL 3.0 EVP API
this is a huge step forward an will break backward compatibility
```
https://wiki.openssl.org/index.php/OpenSSL_3.0

As well as in several bug reports, e.g.:
https://github.com/ZerBea/hcxdumptool/issues/247#issuecomment-1397049676

Now OpenSSL 3.0 is default crypto lib on Arch Linux
https://archlinux.org/packages/core/x86_64/openssl/
and several other Distributions, e.g. Debian.
https://packages.debian.org/bookworm/openssl

It is mandatory to update to OpenSSL >= 3.0.

closed 2023-02-18 by ZerBea

adrastee, 2023-02-18:
@ZerBea 

OpenSSL 3.0.8 is >= 3.0, no ?!

```
root@raspberrypi:~/hcxdumptool $ openssl version
OpenSSL 3.0.8 7 Feb 2023 (Library: OpenSSL 3.0.8 7 Feb 2023)

```

ZerBea, 2023-02-19:
I feel sorry, my fault. I meant OpenSSL development package >= 3.0.
EVP headers are missing:
`unknown type name ‘EVP_MAC’`

adrastee, 2023-02-19:
OK, thanks.