ZerBea/hcxdumptool #254: Android build fails - use of undeclared identifier 'VERSION_YEAR'/'VERSION_TAG'

Skyller360, 2023-02-26:
Hello,

I'm trying to build for Android but ran into multiple errors.

I have no errors when I launch the `make` command:
```
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7-61-gef5376e\" -DVERSION_YEAR=\"2023\"
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-61-gef5376e\" -DVERSION_YEAR=\"2023\"
```

First time running `ndk-build`, I had the error that it didn't found `openssl/conf.h`:
```
./hcxdumptool.c:44:10: fatal error: 'openssl/conf.h' file not found          
#include <openssl/conf.h>
         ^~~~~~~~~~~~~~~~
```

I'm supposing that it is correctly installed and with the correct headers since the `make` command didn't produce any errors.
I copied the `/usr/include/openssl` folder to the root of `hcxdumptool` so it have a direct access to it.

The problem I now have is related to `VERSION_YEAR` and `VERSION_TAG`:

```
$ ndk-build            
[arm64-v8a] Compile        : hcxdumptool <= hcxdumptool.c
In file included from ./hcxdumptool.c:33:
./include/android-ifaddrs/ifaddrs.c:89:9: warning: unused variable 'l_result' [-Wunused-variable]
    int l_result;
        ^
./include/android-ifaddrs/ifaddrs.c:291:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
In file included from ./hcxdumptool.c:51:                                    
./include/rpigpio.h:4:9: warning: 'PAGE_SIZE' macro redefined [-Wmacro-redefined]
#define PAGE_SIZE       (4*1024)
        ^
/home/kali/Downloads/android-ndk-r25c/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/sys/user.h:38:9: note: previous definition is here
#define PAGE_SIZE 4096
        ^
In file included from ./hcxdumptool.c:55:                                    
./include/pcap.c:248:37: error: use of undeclared identifier 'VERSION_TAG'
        sprintf(sysinfo, "hcxdumptool %s", VERSION_TAG);
                                           ^
./hcxdumptool.c:8425:55: error: use of undeclared identifier 'VERSION_TAG'   
fprintf(stdout, "%s %s (C) %s ZeroBeat\n", eigenname, VERSION_TAG, VERSION_YEAR);
                                                      ^
./hcxdumptool.c:8425:68: error: use of undeclared identifier 'VERSION_YEAR'  
fprintf(stdout, "%s %s (C) %s ZeroBeat\n", eigenname, VERSION_TAG, VERSION_YEAR);
                                                                   ^
./hcxdumptool.c:8710:13: error: use of undeclared identifier 'VERSION_TAG'   
        eigenname, VERSION_TAG, VERSION_YEAR, eigenname,
                   ^
./hcxdumptool.c:8710:26: error: use of undeclared identifier 'VERSION_YEAR'  
        eigenname, VERSION_TAG, VERSION_YEAR, eigenname,
                                ^
./hcxdumptool.c:8748:13: error: use of undeclared identifier 'VERSION_TAG'   
        eigenname, VERSION_TAG, VERSION_YEAR,
                   ^
./hcxdumptool.c:8748:26: error: use of undeclared identifier 'VERSION_YEAR'  
        eigenname, VERSION_TAG, VERSION_YEAR,
                                ^
./hcxdumptool.c:8757:40: error: use of undeclared identifier 'VERSION_TAG'   
        "usage: %s -h for help\n", eigenname, VERSION_TAG, VERSION_YEAR, eigenname);
                                              ^
./hcxdumptool.c:8757:53: error: use of undeclared identifier 'VERSION_YEAR'  
        "usage: %s -h for help\n", eigenname, VERSION_TAG, VERSION_YEAR, eigenname);
                                                           ^
./hcxdumptool.c:9476:139: error: use of undeclared identifier 'VERSION_TAG'  
fprintf(stdout, "initialization of %s %s (depending on the capabilities of the device, this may take some time)...\n", basename(argv[0]), VERSION_TAG);
                                                                                                                                          ^               
4 warnings and 10 errors generated.                                          
make: *** [/home/kali/Downloads/android-ndk-r25c/build/core/build-binary.mk:422: obj/local/arm64-v8a/objs/hcxdumptool/hcxdumptool.o] Error 1
```

I tried with Ubuntu 22.04, Kali-2022.4 and also Manjaro-xfce-22.0.4-230222, the latter being based on Arch.
Every time, I'm stuck to this error.

Am I wrong assuming that the deps are correctly installed ?

Thank you !

Regards,

ZerBea, 2023-02-26:
This values are defined in the Makefile:
PRODUCTION		:= 0
PRODUCTION_VERSION	:= 6.2.7
PRODUCTION_YEAR		:= 2023

ZerBea, 2023-02-26:
Set PRODUCTION := 1 than this values should be used instead of reading them from git.

BTW:
I don't think that hcxdumptool will work as expected, because it depend an WIRELESS EXTENSIONS.
Unfortunately most Android kernels and a lot of drivers (mostly Realtek) do not support this.

Please try hcxlabtool which do not have this dependencies:
https://github.com/ZerBea/wifi_laboratory

ZerBea, 2023-02-26:
hcxlabtool:
```
$ make
cc -O3 -Wall -Wextra  -o hcxlabtool hcxlabtool.c -DSTATUSOUT
$ sudo make install
cc -O3 -Wall -Wextra  -o hcxlabtool hcxlabtool.c -DSTATUSOUT
install -D -m 0755 hcxlabtool /usr/local/bin/hcxlabtool
rm -f hcxlabtool
rm -f *.o *~
```

hcxdumptool PRODUCTION := 1
```
$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7\" -DVERSION_YEAR=\"2023\"
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7\" -DVERSION_YEAR=\"2023\"

$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
install -D -m 0755 hcxpioff /usr/bin/hcxpioff
```

hcxdumptool PRODUCTION := 0
```
$ make
mkdir -p .deps
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxdumptool.d -o hcxdumptool hcxdumptool.c -lssl -lcrypto   -DVERSION_TAG=\"6.2.7-61-gef5376e\" -DVERSION_YEAR=\"2023\"
cc -O3 -Wall -Wextra -std=gnu99   -MMD -MF .deps/hcxpioff.d -o hcxpioff hcxpioff.c   -DVERSION_TAG=\"6.2.7-61-gef5376e\" -DVERSION_YEAR=\"2023\"

$ sudo make install
install -D -m 0755 hcxdumptool /usr/bin/hcxdumptool
install -D -m 0755 hcxpioff /usr/bin/hcxpioff
```

This warning
```
In file included from ./hcxdumptool.c:51:                                    
./include/rpigpio.h:4:9: warning: 'PAGE_SIZE' macro redefined [-Wmacro-redefined]
#define PAGE_SIZE       (4*1024)
```
is fixed by latest commit.

This warning
```
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
```
Is out of my scope, because it is an Android header file.

ZerBea, 2023-03-17:
Closed, due to missing response.

closed 2023-03-17 by ZerBea

ZerBea, 2023-04-12:
Please test latest git head.
There are various fixes (see README.md).

Skyller360, 2023-04-17:
It work without any issue, thank you !