ZerBea/hcxdumptool #256: hcxdumptool works with alfa awus036ac on version 6.2.5-50-g75ddcb4 but not on the latest version

DeDe43ISR, 2023-03-07:
I have used Alfa awus036ac a year ago and had no issue running hcxdumptool with this driver :
https://github.com/aircrack-ng/rtl8812au
Chipset : RTL8812AU

I have tried to use it today with the latest version and got error saying that the driver doesnt support WIRELESS EXTENSION.
Reveting hcxdumptool to verison 6.2.5-50-g75ddcb4 solved this issue.

I'm using Ubuntu 22.04 with 5.19.0-35-generic kernel.

ZerBea, 2023-03-07:
That driver depend on NETLINK and does not provide WIRELESS EXTENSIONS.
The aircrack-ng driver may work, but others do not.
```
$ hcxlabtool -L

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
 14  18 74da380623e7 74da380623e7 + wlp39s0f3u1u1u1  rtl88XXau (NETLINK)

+ monitor mode available
- no monitor mode available
```

In contrast an interface that support NETLINK and WIRELESS EXTENSIONS
```
$ hcxlabtool -L

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
 15  19 74da38eb19fc 74da38eb19fc + wlp39s0f3u1u1u2  mt7601u (NETLINK & WIRELESS EXTENSIONS)

+ monitor mode available
- no monitor mode available
```

By latest commit, I changed the restrictions.
https://github.com/ZerBea/hcxdumptool/commit/bfcb216bee91e74ec2c3a01a3efe36226663e01f
Now you get a warning, but hcxdumptool will not terminate.

closed 2023-03-07 by ZerBea

ZerBea, 2023-03-07:
Please try hcxlabtool. It take full advantage of NL80211:
```
$ hcxlabtool -I wlp39s0f3u1u1u2

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
 16  20 74da380623e7 74da380623e7 + wlp39s0f3u1u1u2  rtl88XXau (NETLINK)

available frequencies: frequency [channel] tx-power

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] 20.0 dBm	  5075 [ 15] 30.0 dBm	  5080 [ 16] 30.0 dBm
  5085 [ 17] 30.0 dBm	  5090 [ 18] 30.0 dBm	  5100 [ 20] 30.0 dBm	  5120 [ 24] 30.0 dBm
  5140 [ 28] 30.0 dBm	  5160 [ 32] 23.0 dBm	  5180 [ 36] 23.0 dBm	  5200 [ 40] 23.0 dBm
  5220 [ 44] 23.0 dBm	  5240 [ 48] 23.0 dBm	  5260 [ 52] 20.0 dBm	  5280 [ 56] 20.0 dBm
  5300 [ 60] 20.0 dBm	  5320 [ 64] 20.0 dBm	  5340 [ 68] 20.0 dBm	  5360 [ 72] 30.0 dBm
  5380 [ 76] 30.0 dBm	  5400 [ 80] 30.0 dBm	  5420 [ 84] 30.0 dBm	  5440 [ 88] 30.0 dBm
  5460 [ 92] 30.0 dBm	  5480 [ 96] 26.0 dBm	  5500 [100] 26.0 dBm	  5520 [104] 26.0 dBm
  5540 [108] 26.0 dBm	  5560 [112] 26.0 dBm	  5580 [116] 26.0 dBm	  5600 [120] 26.0 dBm
  5620 [124] 26.0 dBm	  5640 [128] 26.0 dBm	  5660 [132] 26.0 dBm	  5680 [136] 26.0 dBm
  5700 [140] 26.0 dBm	  5720 [144] 13.0 dBm	  5745 [149] 13.0 dBm	  5765 [153] 13.0 dBm
  5785 [157] 13.0 dBm	  5805 [161] 13.0 dBm	  5825 [165] 13.0 dBm	  5845 [169] 13.0 dBm
  5865 [173] 13.0 dBm	  5885 [177] 30.0 dBm

bye-bye
```
and provide several new featues.
Just stop wpa_supplicant and NetworkManager and run
$ sudo hcxlabtool