ZerBea/hcxdumptool #257: [Question / Improvement]: "TX power Entropy Pool": Minimizing Detection/Triangulation Rate

adde88, 2023-03-17:
Hi there @ZerBea!
Forgive my ignorance on this question! I just had an idea suddenly while playing with some lab. equipment and **hcxdumptool**.
I've found it 'relatively easy' to kinda trianglate/detect where many of these packets from **hcxdumptool** are originating from based on the "BSSID RX power" on the chipset reciever packets. (*To a few meters radius*)
So my question for improvement is;

1. Would it be possible to make **hcxdumptool** either voluntarily, by default, or by force, somehow have a pool of entropy dedicated to making the TX power of a wireless device using **hcxdumptool** fluctuate. To such a point where triangulating the device becomes improbable/unlikely/impossible?
2. Such a change would likely bare with it some unforeseen consequences, i assume. Such as "*RX power*" fluctuation, but hopefully at a correct entropy, a stable signal should still be be able to be established.
3. Could i be totally off-course? Could making such a change possibly lead it to making it even easier to detect the device?
4. Or would such a change need to be made outside userland, within the wireless drivers?
If so, i assume alot more changes could me made to hide from detection!

This is probably just a baloon, soon going flat as some guy posts the obvious error in my logic, but i'm still giving it a try!

ZerBea, 2023-03-17:
BTW:
I close this here, because it is more a discussion than an issue.
https://github.com/ZerBea/hcxdumptool/discussions

closed 2023-03-17 by ZerBea

ZerBea, 2023-03-17:
I made some tests. Under lab environment the RX RSSI is the nearly same (~ -47 dBm) regardless if hcxlabtool is running 20 dBm or 16 dBm. RX RSSI fluctuate if I walk through the lab. It also fluctuate if I touch the antenna or if I change direction of the TX DF-antenna. But all this is expected.

ZerBea, 2023-03-18:
Your idea is good but it is not feasible. I spent the last night with several tests and noticed that non of the tested interfaces support dynamic tx power management.

I'm able to set tx power limit by wireless regulatory domain
```
$ sudo iw reg set IN
$ iw reg get
global
country IN: DFS-UNSET
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 30), (N/A)
	(5250 - 5350 @ 80), (N/A, 24), (N/A)
	(5470 - 5725 @ 160), (N/A, 24), (N/A)
	(5725 - 5875 @ 80), (N/A, 30), (N/A)

$ hcxlabtool -I wlp39s0f3u1u1u1
available frequencies: frequency [channel] tx-power
  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm
```

from e.g. 20dBm to 30dBm
```
$ sudo iw reg set US
$ iw reg get
global
country US: DFS-FCC
	(902 - 904 @ 2), (N/A, 30), (N/A)
	(904 - 920 @ 16), (N/A, 30), (N/A)
	(920 - 928 @ 8), (N/A, 30), (N/A)
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)

$ hcxlabtool -I wlp39s0f3u1u1u1
available frequencies: frequency [channel] tx-power
  2412 [  1] 30.0 dBm	  2417 [  2] 30.0 dBm	  2422 [  3] 30.0 dBm	  2427 [  4] 30.0 dBm
  2432 [  5] 30.0 dBm	  2437 [  6] 30.0 dBm	  2442 [  7] 30.0 dBm	  2447 [  8] 30.0 dBm
  2452 [  9] 30.0 dBm	  2457 [ 10] 30.0 dBm	  2462 [ 11] 30.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled
```

but if I try to set dynamic tx power it failed:
```
$ sudo iw phy0 set txpower fixed 2000
$ hcxlabtool -I wlp39s0f3u1u1u1
available frequencies: frequency [channel] tx-power
  2412 [  1] 30.0 dBm	  2417 [  2] 30.0 dBm	  2422 [  3] 30.0 dBm	  2427 [  4] 30.0 dBm
  2432 [  5] 30.0 dBm	  2437 [  6] 30.0 dBm	  2442 [  7] 30.0 dBm	  2447 [  8] 30.0 dBm
  2452 [  9] 30.0 dBm	  2457 [ 10] 30.0 dBm	  2462 [ 11] 30.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled
```