ZerBea/hcxdumptool #259: Can I use this in conjunction with airodump-ng

ChillVibesMushroom, 2023-03-25:
Hey man I was wondering if I would be able to use this tool to capture the handshake alongside airodump-ng in order to see decrypted packets in wireshark flow through the traffic section?

ZerBea, 2023-03-25:
You can use Wireshark/tshark.
You can't use aitodump-ng.

From help:
```
usage: hcxdumptool <options>
       press ctrl+c to terminate hcxdumptool
       press GPIO button to terminate hcxdumptool
       hardware modification is necessary, read more:
       https://github.com/ZerBea/hcxdumptool/tree/master/docs
       do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
       do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
       do not run hcxdumptool on virtual machines or emulators
       do not run hcxdumptool in combination with tools (channel hopper), that take access to the interface (except: tshark, wireshark, tcpdump)
       do not use tools like macchanger, because hcxdumptool runs its own MAC space and will ignore this changes
       stop all services (e.g.: wpa_supplicant.service, NetworkManager.service) that take access to the interface
```

Explanation:
hcxdumptool/hcxlabtool are interactive tools that use its own channel management. You can run all tools that do not change channels in parallel.

ChillVibesMushroom, 2023-03-31:
I used airodump-ng and it worked correctly ?

ChillVibesMushroom, 2023-03-31:
https://hashcat.net/forum/thread-7791.html

but why do you not want us to use it with other tools :/

ZerBea, 2023-03-31:
For sure, you can other tools. But you have to know that hcxdumptool/hcxlabtool
run its own monitor mode
use its own MAC address pool
have an own channel management

It is possible that it interfere with other tools that have e.g. its own channel management.

An example:
hcxdumptool is interactive. It transmit and it expect a response from the target.
You run hcxdumptool with channel option -c 6
I'm sure you can imagine if you run airodump-ng
"By default, airodump-ng hop on 2.4GHz channels."

hcxdumptool transmit on channel 6 and expect a response on channel 6, but airodump-ng set the interface to channel e.g. 8
In that case hcxdumptool never get a response and jam channel 8.

ZerBea, 2023-03-31:
I'm not a friend of shared interfaces.
There is only one physical interface. If you share it, everything is halved.
 You can't run the interface on 2 different frequencies at the same time.

ZerBea, 2023-04-05:
Closed, because hcxdumptool is not part of the aircrack-ng suite and it should not be used in combination with aircrack-ng tools.

closed 2023-04-05 by ZerBea

ChillVibesMushroom, 2023-04-06:
> You can use Wireshark/tshark. You can't use aitodump-ng.
>
> From help:
>
> ```
> usage: hcxdumptool <options>
>        press ctrl+c to terminate hcxdumptool
>        press GPIO button to terminate [...]

I think I just understood what you mean thanks.

ZerBea, 2023-04-06:
Glad to hear this.
It is really a big problem, because hcxdumptool tranmit a BEACON and announce the operating channel as IE_TAG inside this BEACON.
If another tool change the channel, hcxdumptool become a spammer.

ChillVibesMushroom, 2023-04-17:
Zera I have a question, how would I Deauthenticate a client from an ap using hcxdumptool for a little while and then switch to capturing traffic ? 

I know we can use --disable_deauthentication that means hcxdumptool will just capture traffic which is what I need but I also need a param that will set a time for Deauthenticate a then switch back to capturing traffic

Do I just use the program twice using the same card just different terminals? Thnx

ZerBea, 2023-04-17:
hcxdumptool stops attacks on APs if it got either a PMKID or an EAPOL M1M2M3.
It also stops if attemptapmax is reached.

hcxdumptool is not(!) designed to capture traffic, because it reduce bit rate and bandwidth (to increase RANGE and to make sure to get a PMKID, an EAPOL M1M2M3 or an EAPOL M1M2ROGUE)
Work flow:
first get PMKID or EAPOL M1M2M3 (or an EAPOL M1M2ROGUE from a CLIENT)
than recover the PSK/PMK by hashcat/JtR
Now you can record and decrypt the the traffic by tshark / Wireshark.
Once the PSK or the PMK is added to tshark  / Wireshark you can decrypt the entire traffic (always) as long as the PSK/PMK is not changed by the admin.

BTW:
`--disable_deauthentication
`
will only disable this old school kind of an attack.
hcxdumptool run multiple attacks to  make sure to get a PMKID / EAPOL M1M2M3 or M1M2ROGUE.
All other attack modes are still active.

ChillVibesMushroom, 2023-04-17:
> hcxdumptool stops attacks on APs if it got either a PMKID or an EAPOL M1M2M3.
> It also stops if attemptapmax is reached.
>
> hcxdumptool is not(!) designed to capture traffic, because it reduce bit rate [...]

What do you recommend to capture traffic with?

ZerBea, 2023-04-18:
As mentioned above: tshark or Wireshark
Once the PSK or the PMK is added to tshark / Wireshark you can decrypt the entire traffic (always) as long as the PSK/PMK is not changed by the admin.

ChillVibesMushroom, 2023-04-18:
How would I add it Wireshark would I just save the file and reopen it when I capture with hcx dumptool running parameter --donot_deathenticate

ChillVibesMushroom, 2023-04-18:
These are all going to be over the air attacks I won't be connected to the network I need to know what to do there I'm doing this at a home lab

ZerBea, 2023-04-18:
Than you don't need hcxdumptool.

Set interface to monitor mode
Add the PSK or the PMK to Wireshark as described here:
https://wiki.wireshark.org/HowToDecrypt802.11

Now use Wireshark to capture traffic, decode it on the fly and save it to pcapng file