ZerBea/hcxdumptool #287: Few notes on minor issues

kimocoder, 2023-04-08:
Just listing a few things that need to be adressed below;

hcxdumptool.c:1760 process80211eapol_m1() error: memcmp() 'pmkid->pmkid' too small (16 vs 32)
hcxdumptool.c:3583 set_interface() error: double free of 'ufld'
hcxdumptool.c:3622 get_interfacelist() warn: ignoring unreachable code.
hcxdumptool.c:4185 get_rpigpiobasemem() warn: inconsistent indenting

ZerBea, 2023-04-08:
I'll check them.

kimocoder, 2023-04-08:
Fixed. Thanks. closing.

closed 2023-04-08 by kimocoder

ZerBea, 2023-04-08:
Should be fixed by:
https://github.com/ZerBea/hcxdumptool/commit/ade502f75500882e6c54b78cf5b309d575ff5020
https://github.com/ZerBea/hcxdumptool/commit/8aff15ac869f040bb8932fe77e5b5191aa4e7909
https://github.com/ZerBea/hcxdumptool/commit/251bcc55f98111a729e0d64cfaadf11d012980f9
https://github.com/ZerBea/hcxdumptool/commit/87fcb83b6f0bca1d7e6155a3c3735c89cae9efa1

ZerBea, 2023-04-08:
I still expect some more warnings, because the code changes (from old hcxdumptool code to new hcxlabtool code) are huge. Mostly this are ugly "copy and past errors" errors.
BTW:
My favorite editor is Geany (and Geany plugins).