ZerBea/hcxdumptool #301: not an issue just a comment

ukscone, 2023-04-13:
just upgraded from a version from last year and whilst the new version works I do miss the old style --enable_status=XXX stuff as it was much easier to keep an eye on while doing other stuff

ZerBea, 2023-04-14:
The old status function is very CPU cycle intensive. It show status when an event occurs.
The new version collect events and show them timer controlled. That is much (much) faster.
The entire engine is now based on epoll() and timerfd() instead of pselect() and counters, which is ultra fast.
If you take a look at Makefile, you*ll see that the entire status display and the GPS handling can be commented out, e.g.:
`#DEFS		+= -DSTATUSOUT -DNMEAOUT`
This is for headless operation and speed up hcxdumptool again a lot. 

Compared to the old WIRELESS EXTENSION version, the NL80211 version is trimmed for effectiveness and speed.
Everything that cost CPU cycles is removed. To increase range, bandwidth is reduced to 20MHz and bit rate is reduced to 1MB/s 2.4GHz  o 6MB/s. above 5GHz.

But I leave this request open.
If the entire  transformation from WIRELESS EXTENSIONS to NL80211 is finished, I try to add a real time waterfall status display (show EAPOL only) that can be selected by option.

Active RCASCAN is still missing, too as well as requesting EAP-IDs. I have to add this before the next release.

BTW:
At anytime (even though if hcxdumptool is still running), you can do
```
$ hcxpcapngtool -o /tmp/current.22000 dump.pcapng
$ hcxhashtool -i /tmp/current.22000 --info=stdout
$ rm /tmp/current.22000
```

To simplify it, you can add this commands to a bash script, e.g.: showeapol
```
#!/bin/bash

hcxpcapngtool -o /tmp/current.22000 dump.pcapng
hcxhashtool -i /tmp/current.22000 --info=stdout
rm /tmp/current.22000
```

ZerBea, 2023-04-14:
By last commit I added a new option:
```
--rds=<digit>                  : sort real time display
                                  default: sort by time (last seen on top)
                                  1 = sort by status (last PMKID/EAPOL on top)
```

on rds=1 last last last AP of which a PMKID/EAPOL M1M2M3 was received is on top

ukscone, 2023-04-14:
thanks and now you've explained why it makes sense

ZerBea, 2023-04-15:
Compared to the old WIRELESS EXTENSION version, NL80211 version is a thousand times more effective.

b4cker, 2023-04-26:
where can I contact you ZerBea?

ZerBea, 2023-04-26:
You can get my email address via git API:
https://api.github.com/users/ZerBea/events/public

b4cker, 2023-04-26:
wow, I didnt know that API, thanks!

YuryKomar, 2023-06-01:
can someone share an Android version of this tool for capturing PMKID to file

ZerBea, 2023-06-01:
hcxdumptool should run fine on Android:
https://github.com/ZerBea/hcxdumptool/issues/313#issuecomment-1532600605

Unfortunately I don't have an Android environment to compile it.  maybe some Android users can help.

YuryKomar, 2023-06-01:
> hcxdumptool should run fine on Android: [#313 (comment)](https://github.com/ZerBea/hcxdumptool/issues/313#issuecomment-1532600605)
> 
> Unfortunately I don't have an Android environment to compile it. maybe some Android users can help.

Ok. thanks.

deeeblack, 2023-06-09:
With the introduction of 6.3.0 I can't seem to get it to run in the background with '&' at the end of the command line. Or maybe it is running, but It does disappear from htop. Thus I assume its not running.... But I could be wrong.

ZerBea, 2023-06-09:
I'll check it. Could be related to the new real time display.
For headless opearation this can be completely disabled by uncomment the feature in Makefile:
move:
`DEFS		+= -DSTATUSOUT -DNMEAOUT`
to
`DEFS		+= -DNMEAOUT`

ZerBea, 2023-06-09:
Ok, tested it.
Terminal 1:
`$ sudo hcxdumptool -i wlp5s0f3u3 &
`

Terminal 2:
```
$ ps -All | grep hcxdumptool
4 S     0    4262    4261  0  80   0 -   769 -      pts/1    00:00:00 hcxdumptool
```

Doing exactly what expected.

Terminating in terminal 2:
```
$ sudo killall hcxdumptool
$ ps -All | grep hcxdumptool
$
```

Terminal 1 ctrl+c:
```
exit on sigterm

bye-bye
^C
[1]+  Fertig                  sudo hcxdumptool -i wlp5s0f3u3
```

And now the funny part:
top doesn't show hcxdumptool task!

deeeblack, 2023-06-09:
Awwwww I have a lot to learn....yea what confused me is the fact you can see the process in top until you close the session..... I'll give it another few runs. I just checked out the timestamps in the .pcapng and it does seem to still be running. I guess I was just after something visual to ensure it was up.

ZerBea, 2023-06-09:
That is correct. If you close the terminal, hcxdumptool will stop because it will not detach from the terminal (there is no code inside to run hcxdumptool as a deamon).

There are several ways to detach a process from a terminal. One of them is setsid:
$ su
$ setsid hcxdumptool -i interface
close terminal.

More information is here:
$ man setsid

a tutorial and some wore ways is here:
https://www.baeldung.com/linux/detach-process-from-terminal

deeeblack, 2023-06-10:
SETSID worked a treat! I'll have a play with all those methods as well now.

# ELEVATE PRIVS
`sudo -i`

# SHUTDOWN INTERFERING SERVICES
```
sudo systemctl stop NetworkManager.service
sudo systemctl stop wpa_supplicant.service
```

#START HCXDUMPTOOL ON SELECTED INTERFACE
`setsid hcxdumptool -i wlan1`

#TURN PI OFF TO STOP.

ZerBea, 2023-06-10:
I prefer a different method - hcx dumptool is compiled without -DSTATUS
and it is started by bash_profile if the RPI is powerd on:
```
#!/bin/bash

sleep 10
chmod 777 -R /home
cd /home
iw reg set IN
#hcxdumptool --gpio_button=4 --gpio_statusled=17 --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --onwatchdog=reboot --bpf=own.bpfc --essidlist=standard.essidliste --beacontx=5
hcxdumptool --gpio_button=4 --gpio_statusled=17 --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --onwatchdog=reboot --bpf=own.bpfc --essidlist=standard.essidliste --beacontx=5 -c 1a,6a,11a,2a,1a,6a,11a,13a,1a,6a,11a,3a,1a,6a,11a,12a,1a,6a,11a,4a,1a,6a,11a,10a,1a,6a,11a,5a,1a,6a,11a,9a,1a,6a,11a,7a,1a,6a,11a,8a
#hcxdumptool --gpio_button=4 --gpio_statusled=17 --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --onwatchdog=reboot --bpf=own_ext.bpfc --essidlist=standard.essidliste --beacontx=5 -c 1a,6a,11a,2a,1a,6a,11a,13a,1a,6a,11a,3a,1a,6a,11a,12a,1a,6a,11a,4a,1a,6a,11a,10a,1a,6a,11a,5a,1a,6a,11a,9a,1a,6a,11a,7a,1a,6a,11a,8a,1a,6a,11a,36b,1a,6a,11a,40b,1a,6a,11a,44b,1a,6a,11a,48b
#hcxdumptool --gpio_button=4 --gpio_statusled=17 --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --onwatchdog=reboot --bpf=own_ext.bpfc --essidlist=standard.essidliste --tot=8640 --attemptclientmax=2 --beacontx=5 -t 120 -c 1a,8a,3a,6a,4a,7a,2a,5a --disable_deauthentication --disable_proberequest --disable_association --disable_reassociation
#hcxdumptool --gpio_button=4 --gpio_statusled=17 --onsigterm=exit --ongpiobutton=poweroff --ontot=reboot --onerror=reboot --onwatchdog=reboot --bpf=own.bpfc --essidlist=standard.essidliste --attemptapmax=1000 --attemptclientmax=20 --beacontx=5 -c 1a,6a,11a,2a,1a,6a,11a,13a,1a,6a,11a,3a,1a,6a,11a,12a,1a,6a,11a,4a,1a,6a,11a,10a,1a,6a,11a,5a,1a,6a,11a,9a,1a,6a,11a,7a,1a,6a,11a,8a
#systemctl start dhcpcd.service
#systemctl start ssh.service
poweroff
```

deeeblack, 2023-06-10:
Perhaps a little bit beyond my skillset. However, I'll use that s a benchmark and have a go at it. Appreciate it.

PavloM7, 2023-06-11:
Hi Zerbea, can you help pls? When I try to convert "file.pcapng" to "file.22000" with hcxpcapngtool I get:
"Information: no hashes written to hash files"
I think my problem is writing the .pcapng file, because the .pcapng file appears right after I press enter, instantly, which I think is not normal. What I am doing:
**sudo systemctl stop NetworkManager.service
sudo systemctl stop wpa_supplicant.service**

**hcxdumptool -i wlan0 -w Desktop/file_f.pcapng**
*The file appears immediately as soon as I press enter*
**hcxpcapngtool -o /home/kali/Desktop/f_hash.22000 /home/kali/Desktop/file_f.pcapng**
And l get: "Information: no hashes written to hash files"

ZerBea, 2023-06-12:
Immediately after starting hgcxdumptool, it create a dump file and it store an initial packet. This packet contain some information about the used MAC addresses and hashes. Hcxpcapngtool can work in this information.

You have to run hcxdumptool until you get an information that your attack is successful and the mandatory EAPOL messages are received or retrieved.
hcxdumptool will show you that (I recommend to use -rds=1 so that they are always on top of the status display)
On AP display (above) take a look at  the columns P and 3.
On CLIENT display take a look at the column 2.
An explanation if in --help:
```
real time display:
 R = + AP display:     AP is in TX range or under attack
 S = + AP display:     AUTHENTICATION KEY MANAGEMENT PSK
 P = + AP display:     got PMKID
 1 = + AP display:     got EAPOL M1 (CHALLENGE)
 3 = + AP display:     got EAPOL M1M2M3 (AUTHORIZATION)
 E = + CLIENT display: got EAP-START MESSAGE
 2 = + CLIENT display: got EAPOL M1M2 (ROGUE CHALLENGE)
```

Once you got this information, you can stop hcxdumptool and convert the content of the pcapng file to a hc22000 file.

ZerBea, 2023-06-13:
Some APs do not transmit a PMKID. Looks like your AP is among them. 
In that case, you need a CLIENT connected to the AP (or to hcxdumptool) to get a handshake.

ZerBea, 2023-06-13:
The screenshot who the split screen.
Above = results of attacks on APs.
column R = AP is in RANGE and respond to hcxdumptool
column 1 = AP respond an EAPOL frame on hcxdumptool request
column 3 = empty = no handshake, because no CLIENT is connected = nothing hashcat / JtR can work on
column P = empty = AP doesn't use PMKIDs = nothing hashcat / JtR can work on
column S = AP use crypto systems supported by hashcat / JtR

Below = results of attacks on CLIENTs.
column 2 = CLIENT connected to hcxdumptool and respond an EAPOL M2 hashcat / JtR can work on

ZerBea, 2023-06-15:
The Linux kernel team decided to drop WIRELESS EXTENSIONS (WEXT) on new WiFi hardware starting with kernel 6.3.0.
From my point of view it was a good decision that has led to the redesign of the hcxdumptool as well.

Version 6.2.9 is the latest "old style" version. 

Starting with version 6.3.0 nearly everything changed.
Filter lists in user space are slow and completely replaced by BPF. 

The real time display is now always on. On headless operation it can be disabled completely.

Goal of all changes is to save CPU cycles.

The changelog gives an overview what changed and when it changed.

The simplest command to do nearly the same is:
`$ hcxdumptool -i INTERFACE -F --rds=1 --bpf=filter.bpfc`

You can add e.g. tcpdump to airgeddon to create the BPF code as suggested by this user here:
https://github.com/ZerBea/hcxdumptool/issues/116#issuecomment-1585045127
It doesn't matter which tool is used (there is more than one good tool to do this) to create BPF code, because hcxdumptool take every valid BPF code that is in decimal count.
The simplest way to create the filter code is to use tcpdump and its high level language.

BTW:
Some more information is here:
https://github.com/kimocoder/wifite2/commit/eb2f2bbd348eb217753c0c1c0a200379f851e3e4

ZerBea, 2023-06-16:
I forgot to mention:
Since WiFi6, channel numbers are not longer unique. My solution is either to use the frequency or to add the band to the channel number. Internal hcxdumptool use frequencies (channel numbers are converted to frequencies).

OscarAkaElvis, 2023-06-16:
omg... this changed so much... I can see the `-o` option was deprecated as well... I guess now `-w` should be used and I hope the format of the output file is the same.

Understood about the versions... now I have 3 ranges:
<6.0.0
>=6.0.0 and <6.3.0
>=6.3.0

The conditional about the versions is not a problem. I sorted it out. Now I need to figure out how to do the same using new stuff. Please, help me out checking this:

[image]

The bottom highlighted in red part is the command. As you can see all are sharing only the `-i ${interface}` part. The rest needs to be in a var which will change depending on the version.

I don't have defined the filter bpfc file. Not sure yet what is that. Can it be static? I mean, can you give me one to test? and can I use the same always? I hope so. If you tell me that is something that changes each time depending on the target that is going to be a very big problem for me. What format does it have?

ZerBea, 2023-06-16:
Let's say the target MAC is 11:22:33:44:55:66 and you would like to get a PMKID of this target:

First create as much as filters as you want or combine several MACs in a single filter
```
$ hcxdumptool -m INTERFACE
$ tcpdump -i INTERFACE wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66  > attack112233445566.bpf
```

Now run
`$ hcxdumptool -F -i interface --bpf=attack112233445566.bpf` -w 112233445566.pcapng

The same filter, but it allow to retrieve undirected PROBEREQUEST frames, too:
```
$ hcxdumptool -m INTERFACE
$ tcpdump -i INTERFACE wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66 or wlan add3 ff:ff:ff:ff:ff:ff > attack112233445566_pr.bpf
```

That sounds complex, but it is ultra fast, because we're doing filtering in kernel space.

Please notice that (if running this kind of filter) nearly 75% of the real capabilities of hcxdumptool are unused and you will loose a lot of useful information. It does not make sense to capture a PMKID only or a 4way handshake only and to disregard the rest of the traffic:
https://github.com/evilsocket/pwnagotchi/issues/835#issuecomment-598597214

Please answer this question yourself:
How much time would aircrack-ng, hashcat or JtR take to recover this complex PSK without the help of hcxdumptool/hcxtools?
`v+#gqu5g9pqnp%+7jbf%2uex+8hlo3ms`

With the help of hcxdumptool/hcxtools ~ 2 seconds
```
$ hashcat -m 22000 --nonce-error-corrections=8 eapol.22000 wordlist
f6b32d9fc030aa42a250f1e93785bc2d:a0f3c1503e62:b0c090467cab:hascat:v+#gqu5g9pqnp%+7jbf%2uex+8hlo3ms
                                                          
Session..........: hashcat
Status...........: Cracked
Hash.Mode........: 22000 (WPA-PBKDF2-PMKID+EAPOL)
Hash.Target......: eapol.22000
Time.Started.....: Fri Jun 16 14:59:57 2023 (0 secs)
Time.Estimated...: Fri Jun 16 14:59:57 2023 (0 secs)
Kernel.Feature...: Pure Kernel
Guess.Base.......: File (wordlist)
Guess.Queue......: 1/1 (100.00%)
Speed.#1.........:       89 H/s (0.60ms) @ Accel:64 Loops:256 Thr:32 Vec:1
Recovered........: 1/1 (100.00%) Digests (total), 1/1 (100.00%) Digests (new)
Progress.........: 2/2 (100.00%)
Rejected.........: 1/2 (50.00%)
Restore.Point....: 0/2 (0.00%)
Restore.Sub.#1...: Salt:0 Amplifier:0-1 Iteration:0-1
Candidate.Engine.: Device Generator
Candidates.#1....: v+#gqu5g9pqnp%+7jbf%2uex+8hlo3ms -> v+#gqu5g9pqnp%+7jbf%2uex+8hlo3ms
Hardware.Mon.#1..: Temp: 48c Util:  7% Core:1965MHz Mem:4001MHz Bus:8

Started: Fri Jun 16 14:59:56 2023
Stopped: Fri Jun 16 14:59:58 2023
```

And I'm not running a big GPU. A GTX 1650 (mobile) and an ASUS TUF notebook - that's all:
```
OpenCL Platform ID #1
  Vendor..: NVIDIA Corporation
  Name....: NVIDIA CUDA
  Version.: OpenCL 3.0 CUDA 12.1.98

  Backend Device ID #2 (Alias: #1)
    Type...........: GPU
    Vendor.ID......: 32
    Vendor.........: NVIDIA Corporation
    Name...........: NVIDIA GeForce GTX 1650
    Version........: OpenCL 3.0 CUDA
    Processor(s)...: 16
    Clock..........: 1560
    Memory.Total...: 3903 MB (limited to 975 MB allocatable in one block)
    Memory.Free....: 3776 MB
    Local.Memory...: 48 KB
    OpenCL.Version.: OpenCL C 1.2 
    Driver.Version.: 530.41.03
    PCI.Addr.BDF...: 01:00.0
```

Now please answer the second question yourself
Running airgeddon script, will the user get the same information?

BTW:
hcxdumptool provide ASSOCIATION and REASSOCIATION attacks instead of running stupid DEAUTHENTICATIONs.
This kind of an attack is working if MANAGEMENT FRAME PROTECTION (MFP) is activated. It also force the downgrade of aWPA3 CLIENT to WPA2.
Also hcxdumptool is able to request an EAPOL M2 of a CLIENT. That is useful if the CLIENT is in attack range, but the AP not.

OscarAkaElvis, 2023-06-16:
Thanks for the info.... I can see I'll need to add a new dependency (tcpdump)... hmmm ok, that is doable.

Yeah, sure that the resulting file will be better for a later offline cracking, you are right about that.

Ok I still have some questions....

The first doubt is about the first command `hcxdumptool -m INTERFACE` what is it for exactly. I launched it and I can see that is just checking available frequencies and it sets monitor mode for the card. I guess I can avoid that because I already have validated that the user have monitor mode set in an earlier stage. Just want to confirm.

The second. If I just want to target one BSSID 11:22:33:44:55:66 using my monitor wlan0mon card, I guess the tcpdump comand should be `tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 > attack112233445566.bpf` . I tested it targeting a test AP I have which is vulnerable to PMKID but the file is generated is empty. I tested old versions of hcxdumptool so many times over this AP and always worked. There must be something I'm missing. I left it even for a couple of minutes running but no nothing happened. I think I'm not understanding the point of this command. I understood that is a filter to be used later by hcxdumptool in order to get PMKID faster and create a cleaner and easier to crack file... but what's happening here? why is not working for me? I need to understand the commands and the process to later integrate it in the script.

Thanks.

ZerBea, 2023-06-16:
tcpdump is only a simple example to create such kinds of filters. There are several other ways to do this, but all of them are more complicated.

That was a copy and paste error in my comment above
wrong command:
`tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 > attack112233445566.bpf`
correct command:
`tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 -ddd > attack112233445566.bpf`

```
       -d     Dump  the  compiled  packet-matching  code in a human readable
              form to standard output and stop.

              Please mind that although code compilation is always  DLT-spe‐
              cific, typically it is impossible (and unnecessary) to specify
              which DLT to use for the dump because tcpdump uses either  the
              DLT  of  the input pcap file specified with -r, or the default
              DLT of the network interface specified with -i, or the partic‐
              ular DLT of the network interface specified with -y and -i re‐
              spectively. In these cases the dump shows the same exact  code
              that  would  filter  the  input  file or the network interface
              without -d.

              However, when neither -r nor -i is  specified,  specifying  -d
              prevents  tcpdump  from  guessing a suitable network interface
              (see -i).  In this case the DLT defaults to EN10MB and can  be
              set to another valid value manually with -y.

       -dd    Dump packet-matching code as a C program fragment.

       -ddd   Dump  packet-matching code as decimal numbers (preceded with a
              count).
```
ddd create decimal numbers, accepted by hcxdumptool.

ZerBea, 2023-06-16:
An example:
```
$ sudo hcxdumptool -m wlp5s0f3u2
$ sudo tcpdump -i wlp5s0f3u2 wlan addr1 11:22:33:44:55:66 -ddd > attack112233445566.bpf
$ cat attack112233445566.bpf
13
48 0 0 3
100 0 0 8
7 0 0 0
48 0 0 2
76 0 0 0
2 0 0 0
7 0 0 0
64 0 0 6
21 0 3 860116326
72 0 0 4
21 0 1 4386
6 0 0 262144
6 0 0 0
```

You can create this filters and store them e.g. in the folder filters.
Now you can tell hcxdumpool via a variable whic filter ist should use.
E.g.:
```
export FILTER1="filters/attack112233445566.bpf"
$ hcxdumptool --bpf=$FILTER1
```

ZerBea, 2023-06-16:
Now the complex explanation:
The filter code is a peace of assembler code:
```
$ sudo tcpdump -i wlp5s0f3u2 wlan addr1 11:22:33:44:55:66 -d
(000) ldb      [3]
(001) lsh      #8
(002) tax      
(003) ldb      [2]
(004) or       x
(005) st       M[0]
(006) tax      
(007) ld       [x + 6]
(008) jeq      #0x33445566      jt 9	jf 12
(009) ldh      [x + 4]
(010) jeq      #0x1122          jt 11	jf 12
(011) ret      #262144
(012) ret      #0
```
or c code:
```
$ sudo tcpdump -i wlp5s0f3u2 wlan addr1 11:22:33:44:55:66 -dd
{ 0x30, 0, 0, 0x00000003 },
{ 0x64, 0, 0, 0x00000008 },
{ 0x7, 0, 0, 0x00000000 },
{ 0x30, 0, 0, 0x00000002 },
{ 0x4c, 0, 0, 0x00000000 },
{ 0x2, 0, 0, 0x00000000 },
{ 0x7, 0, 0, 0x00000000 },
{ 0x40, 0, 0, 0x00000006 },
{ 0x15, 0, 3, 0x33445566 },
{ 0x48, 0, 0, 0x00000004 },
{ 0x15, 0, 1, 0x00001122 },
{ 0x6, 0, 0, 0x00040000 },
{ 0x6, 0, 0, 0x00000000 },
```
If the code isv converted to decimals, hcxdumptool read this code peace
```
$ sudo tcpdump -i wlp5s0f3u2 wlan addr1 11:22:33:44:55:66 -ddd
13
48 0 0 3
100 0 0 8
7 0 0 0
48 0 0 2
76 0 0 0
2 0 0 0
7 0 0 0
64 0 0 6
21 0 3 860116326
72 0 0 4
21 0 1 4386
6 0 0 262144
6 0 0 0
```
and add it to the kernel.

From no on, the code snipped is active and does the entire filtering for hcxdumptool.

ZerBea, 2023-06-16:
That is the Linux kernel way how to filter traffic.

OscarAkaElvis, 2023-06-16:
Ok, let me some time to digest all of this info and to do some tests. I hope the resulting file can be parsed in the same way by hcxpcapngtool without any changes. My conditional command to trigger a successful capture is still this:

`if hcxpcapngtool -o "${tmpdir}${standardpmkid_filename}" "${tmpdir}pmkid.pcapng" | grep -Eq "PMKID(\(s\))? written" 2> /dev/null; then`

Thanks.

ZerBea, 2023-06-16:
Conclusion:
using BPF save CPU cycles, because filtering is done inside the kernel and not by hcxdumptool
using active monitor mode save CPU cycles, because requesting missing responses is done by the interface and not by hcxdumptool.

ZerBea, 2023-06-16:
Another powerful tool is tshark. It can be used inside scripts and hcxdumptool pcapng output is 100% compatible to it.
Additional tshark (or Wireshark if you prefer a GUI) can run in parallel with hcxdumptool (on the same interface).

You can filter everything, directly on the fly or later on on the dump file, e.g. to get all PMKIDs from a pcapng dump file:
`$ tshark -r /dump.pcapng -Y "wlan.rsn.ie.pmkid" -T fields -e wlan.rsn.ie.pmkid`

An overview of the filter options is here:
https://tshark.dev/capture/capture_filters/

More information is here:
https://www.wireshark.org/docs/dfref/w/wlan.html

OscarAkaElvis, 2023-06-16:
I already have the tshark part implemented for later (after getting the pmkid capture) as an option to transform the file from hashcat format to aircrack format. I'm developing right now the base (dependencies, checks etc...), and later I'll do the testing to implement the new way. I'll tell you if I have any issue. Thanks again.

OscarAkaElvis, 2023-06-16:
hmmm.... weird. I finished implementing it and I have no errors but also is not working... nothing is captured. Let's troubleshoot this:

After implementing all the validations, tcpdump dependency if >=6.30 version is used and bla, bla, bla, here is the code which constructs the command:

[image]

The tcpdump is constructing well the .bpf file in the temp dir. Then the command is launched and this appears first:

[image]

Then this other start after some seconds:

[image]

But nothing happens until reach the timeout. I tested it using 20, 30 and 40 seconds. Same result.

I double checked after this test using hcxdumptool old v6.0.0 that is working and PMKID is captured correctly from my AP.

As you can see, the channel is 6 from 2.4ghz band, so I'm adding to the command the `-c 6a` which I think is correct. Not sure what can be happening here but is not capturing the PMKID 🙄 

Could it be related with the fact of airgeddon is setting monitor mode before and I'm not launching the `hcxdumptool -m INTERFACE` command? hmm.... I'll do some tests about this. Any other suggestion or anything weird that you can spot in the command? Thanks.

ZerBea, 2023-06-16:
Is the target MAC in addr1, addr2 and addr3 of the filter?

BTW:
I sent you a PM.

OscarAkaElvis, 2023-06-16:
The tcpdump command I'm launching was this:

`tcpdump -i "${interface}" wlan addr1 "${bssid}" -ddd > "${tmpdir}pmkid.bpf"`

That part I think is ok because the file is well generated and contains the right content.

after some testing trial and error... I decided to generate traffic to the AP just authenticating a real user to the PMKID vulnerable AP and something was shown, but I think that doesn't mean a PMKID was captured yet

[image]

I need to do more tests because my first thoughts about this is that it is not very reliable. At least for the purpose in airgeddon which is to get just a single pmkid from a single target. It is supposed that airgeddon is a pentest tool to be launched over a specific target. That is one of the main reasons to not having "massive or random" attacks to surrounding APs and it's focused on a single target. The reason is that I want to be used mainly in legal and tailored wifi penetration tests. Let's see if this version finally is able to do what airgeddon needs. Need to do more testing to confirm but for now, I had bad results.

I have no more time today to keep testing. I'll keep trying other day.

ZerBea, 2023-06-16:
The BPF is merciless.
wlan addr1 means that all frames are filtered out that are not addressed to the target MAC.

This frames are filtered out:
BEACON (target must be in wlan addr2)
PROBERESPONSE (target must be in wlan addr2)
AUTHENTICATION RESPONSE  (target must be in wlan addr2)
ASSOCIATION RESPONSE  (target must be in wlan addr2)
EAPOL M1, that contain the PMKID  (target must be in wlan addr2)

Explanation is here:
https://networkengineering.stackexchange.com/questions/25100/four-layer-2-addresses-in-802-11-frame-header

ZerBea, 2023-06-17:
An example to reproduce wrong BPF techniques.

To reproduce get the PMKID example hash from:
https://hashcat.net/wiki/doku.php?id=example_hashes

store it to hc22000 file:
`$ echo "WPA*01*4d4fe7aac3a2cecab195321ceb99a7d0*fc690c158264*f4747f87f9f4*686173686361742d6573736964***" > example.hc22000
`
connvert it to cap file:
`$ hcxhash2cap --pmkid-eapol=example.hc22000
`

```
MAC_AP: fc690c158264
MAC_CLIENT: f4747f87f9f4
ESSID: 686173686361742d6573736964
PMKID: 4d4fe7aac3a2cecab195321ceb99a7d0
```

```
$ whoismac -x 686173686361742d6573736964
hashcat-essid
```

```
$ ls
f4747f87f9f4.cap
```

```
tshark filter syntax	tcpdump filter syntax
wlan.ta			wlan addr1
wlan.sa			wlan addr2
wlan.bssid		wlan.addr3
```

```
$ tshark -r f4747f87f9f4.cap
    1 06:16:33,207219 fc:69:0c:15:82:64 → ff:ff:ff:ff:ff:ff 802.11 111 Beacon frame, SN=996, FN=0, Flags=........, BI=100, SSID="hashcat-essid" 
    2 06:16:33,207220 fc:69:0c:15:82:64 → f4:74:7f:87:f9:f4 EAPOL 155 Key (Message 1 of 4)
```
Correct filtering:
```
$ tshark -r f4747f87f9f4.cap -Y "wlan.bssid==fc:69:0c:15:82:64"
    1 06:16:33,207219 fc:69:0c:15:82:64 → ff:ff:ff:ff:ff:ff 802.11 111 Beacon frame, SN=996, FN=0, Flags=........, BI=100, SSID="hashcat-essid" 
    2 06:16:33,207220 fc:69:0c:15:82:64 → f4:74:7f:87:f9:f4 EAPOL 155 Key (Message 1 of 4)
```

Now the impact of wrong filtering - we set the BSSID as wlan.ta (tcpdump syntax = wlan addr1):
```
$ tshark -r f4747f87f9f4.cap -Y "wlan.ra==fc:69:0c:15:82:64"
$
```

All wanted frames are filtered out!

OscarAkaElvis, 2023-06-17:
So you think I'm not getting the PMKID because the filter is not well prepared? ok, maybe... now let's create the right bpf file using tcpdump. Data and stuff the user have in airgeddon on PMKID menu:

BSSID of the target
channel of the target
ESSID of the target
A card set into monitor mode used previously to get the target's data

Let's suppose they are:
Card: wlan0mon
BSSID: 11:22:33:44:55:66
Channel: 6
ESSID: demoPMKID

How could be the tcpdump right command to be used later on hcxdumptool to get a PMKID?

Right now I have this:
`tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 -ddd > "/tmp/pmkid.bpf"`

And then my hcxdumptool looks like this:
`hcxdumptool -i wlan0mon -c 6a -F --rds=1 --bpf=/tmp/pmkid.bpf -w /tmp/pmkid.pcapng`

Can you help me to generate a better or more precise bpf file to be able to get a PMKID of a target?

I'm guessing based on your comments that a better bpf file could be this way:
`tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 wlan addr2 11:22:33:44:55:66 wlan addr3 11:22:33:44:55:66 -ddd > "/tmp/pmkid.bpf`

Configuring it and testing it in some minutes...

OscarAkaElvis, 2023-06-17:
Ok, I was missing the "or" ... I added them and it worked!!!!

[image]

Finally the tcpdump filter creation command was:
`tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66 -ddd > "/tmp/pmkid.bpf`

And my tshark transformation over the right file worked flawlessy. So I think I have all I need. Thanks for the help and the patience 😄

ZerBea, 2023-06-17:
You're welcome. Would be nice if all this is helpful to improve airgeddon.

Correct, this is the "all-in-one" solution that covers the target MAC in addr1 (ra) addrs 2 (ta) and addr 3 (bssid):
`$ sudo tcpdump -i wlan0mon wlan addr1 11:22:33:44:55:66 or wlan addr2 11:22:33:44:55:66 or wlan addr3 11:22:33:44:55:66 -ddd > "/tmp/pmkid.bpf
`

OscarAkaElvis, 2023-06-17:
great, it works like a charm now. And I must admit it is faster now.

I needed to perform so many changes, but it was worth in the end I think, you were right. The hardest part was to prepare the "structure" to keep it robust validating all and adding the new dependency to be checked only if version >=6.3.0 , etc etc... but now it is done, commit here: https://github.com/v1s1t0r1sh3r3/airgeddon/commit/ef71d9670ba7e955fe188eff5c5663fe5cd85dbe

Regards.

ZerBea, 2023-06-17:
I fully agree.

BTW:
You can play with BPF, e.g. remove unwanted ctrl frames:
`$ sudo tcpdump -i  wlp5s0f3u2 "(wlan addr3 11:22:33:44:55:66) and (not type ctl subtype ack and not type ctl subtype rts and not type ctl subtype cts)" -ddd > /tmp/pmkid.bpf
`
Cheers
Mike

OscarAkaElvis, 2023-06-17:
not sure if it worths to tune in more. Do you think could be interesting? as you know, we want only to get the needed stuff to get a valid PMKID and then it is supposed that is going to be cracked offline. That changes that you are suggesting could improve this in any way?

ZerBea, 2023-06-17:
Le't say your hunting for three PMKIDs of three different NETWORKs:
```
00:00:00:00:00:01
00:00:00:00:00:02
00:00:00:00:00:03
```

create on filter for the three NETWORKs:
`$ sudo tcpdump -i INTERFACE wlan addr3 00:00:00:00:00:01 or wlan addr3 00:00:00:00:00:02 or wlan addr3 00:00:00:00:00:03 -ddd > /tmp/pmkid.bpf
`

and let hcxdumptool hunt for them:
`$ sudo hcxdumptool -F --bpf=/tmp/pmkid.bpf`

OscarAkaElvis, 2023-06-19:
Ok but as I said, in airgeddon everything is done intentionally only for a concrete target. It is designed to perform professional wifi audits.

I removed the wlan addr1 and the wlan addr2 leaving only wlan addr3 in the filter whichs seems to be the only needed to get the PMKID.

ZerBea, 2023-06-19:
Exactly that is the purpose of the BPF.

OscarAkaElvis, 2023-06-20:
yeah, don't worry. We know it's not very reliable and sometimes may cause problems in the .cap file. It is warned always to the user before using it